emsesp/EMS-ESP #2: 3 way valve

danimaciasperea, 2018-05-28:
Hi Paul,

Nice work here adapting EMS Bus code to ESP.

I have seen that you are looking for a way to control the 3-way valve to switch the domestic hot water on and off. Maybe you can try using [UBAFunktionstest ](https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#ubafunktionstest_write_only) datagrams (0X1D) and perform in the ColdShot method a test mode setting the valve to heating circuit (0x00) during some minutes.

In my house I also have "problems" with people taking too long showers...

proddy, 2018-05-28:
Thanks Daniel. I was thinking of using the 0x1D too but it's experimental
and worried that I'd mess it up. But worth a try I guess. Did you get my
code working?

danimaciasperea, 2018-05-28:
No, I was just having a look. I am still using Arduino Mega and a Wifly RN-XV module. In the last weeks I have developed and [documented ](https://domoticproject.com/calduino-connecting-arduino-ems-bus) a [library for Arduino](https://github.com/danimaciasperea/Calduino) to talk with the EMS Bus.
I would like to learn more about ESP modules this summer, so maybe I will have try.

proddy, 2018-05-28:
ah you built Calduino! I actually used some of your ideas too in my code design

danimaciasperea, 2018-05-28:
Yeah, I already saw that ;)

proddy, 2018-06-02:
My plan is to find a Nefit Easy thermostat (that supports the shower-timer) or a user with one and then sniff the traffic to determine which commands it is issuing to the boiler and reverse engineer into it my code.

proddy, 2018-12-14:
@danimaciasperea - I got my hands on an Easy and we were both right. They use the UBAFunktionstest datagrams (0X1D) to switch off the 3-way valve to temporary stop the hot tap water. I've adopted the same approach. Just wanted to let you know!

closed 2018-12-15 by proddy