emsesp/EMS-ESP #4: VSC - PlatformIO building error

blacksocket, 2018-09-28:
Hi Paul,

I am trying to build d1_mini firmware on PlatformIO according to your procedure and platformio.ini-example but unfortunately I am getting an error message below. Do you know what could be the issue ? "class "Ticker" has no member "attach"
--------------------------------------------------
> Executing task: C:\Users\tomas\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing tool-mkspiffs @ ~1.200.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
PackageManager: Installing tool-espotapy @ ~1.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting boiler.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Time> 1.5
|-- <PubSubClient> 2.6
|-- <ArduinoJson> 5.13.2
|-- <CRC32> 2.0.0
|-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 3.1.4
Compiling .pioenvs\nodemcuv2\src\boiler.ino.cpp.o
Compiling .pioenvs\nodemcuv2\libef2\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs\nodemcuv2\libef2\ESP8266WiFi\ESP8266WiFiMulti.cpp.oC:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:65:8: error: no matching function for call to 'Ticker::Ticker()'

^C
ompiling .pioenvs\nodemcuv2\libef2\ESP8266WiFi\ESP8266WiFiSTA.cpp.oC
:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:65:8: note: candidates are:Compiling .pioenvs\nodemcuv2\libef2\ESP8266WiFi\ESP8266WiFiScan.cpp.o

In file included from C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:37:0:
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note: Ticker::Ticker(fptr, uint32_t, uint16_t, resolution_t)
Ticker(fptr callback, uint32_t timer, uint16_t repeat = 0, resolution_t resolution = MICROS);
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note:   candidate expects 4 arguments, 0 provided
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note: constexpr Ticker::Ticker(const Ticker&)
class Ticker {
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note:   candidate expects 1 argument, 0 provided
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:67:8: error: no matching function for call to 'Ticker::Ticker()'
#define HOSTNAME "raspberrypi"
^
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:67:8: note: candidates are:
In file included from C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:37:0:
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note: Ticker::Ticker(fptr, uint32_t, uint16_t, resolution_t)
Ticker(fptr callback, uint32_t timer, uint16_t repeat = 0, resolution_t resolution = MICROS);
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note:   candidate expects 4 arguments, 0 provided
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note: constexpr Ticker::Ticker(const Ticker&)
class Ticker {
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note:   candidate expects 1 argument, 0 provided
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:69:8: error: no matching function for call to 'Ticker::Ticker()'
// app specific - do not change
^
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:69:8: note: candidates are:
In file included from C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:37:0:
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note: Ticker::Ticker(fptr, uint32_t, uint16_t, resolution_t)
Ticker(fptr callback, uint32_t timer, uint16_t repeat = 0, resolution_t resolution = MICROS);
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note:   candidate expects 4 arguments, 0 provided
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note: constexpr Ticker::Ticker(const Ticker&)
class Ticker {
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note:   candidate expects 1 argument, 0 provided
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:71:8: error: no matching function for call to 'Ticker::Ticker()'
#define TOPIC_START MQTT_BOILER MQTT_TOPIC_START
^
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:71:8: note: candidates are:
In file included from C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:37:0:
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note: Ticker::Ticker(fptr, uint32_t, uint16_t, resolution_t)
Ticker(fptr callback, uint32_t timer, uint16_t repeat = 0, resolution_t resolution = MICROS);
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note:   candidate expects 4 arguments, 0 provided
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note: constexpr Ticker::Ticker(const Ticker&)
class Ticker {
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note:   candidate expects 1 argument, 0 provided
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:73:8: error: no matching function for call to 'Ticker::Ticker()'
#define TOPIC_THERMOSTAT_TEMP MQTT_BOILER "thermostat_temp"                      // for received thermostat temp changes
^
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:73:8: note: candidates are:
In file included from C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:37:0:
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note: Ticker::Ticker(fptr, uint32_t, uint16_t, resolution_t)
Ticker(fptr callback, uint32_t timer, uint16_t repeat = 0, resolution_t resolution = MICROS);
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:61:2: note:   candidate expects 4 arguments, 0 provided
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note: constexpr Ticker::Ticker(const Ticker&)
class Ticker {
^
C:\Users\tomas\.platformio\lib\Ticker_ID1586/Ticker.h:49:7: note:   candidate expects 1 argument, 0 provided
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:87:5: error: expected unqualified-id before string constant
// default values
^
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino: In function 'void setup()':
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:1345:20: error: 'class Ticker' has no member named 'attach'
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:1353:24: error: 'class Ticker' has no member named 'attach'
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:1355:22: error: 'class Ticker' has no member named 'attach'
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:1357:25: error: 'class Ticker' has no member named 'attach'
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:1391:50: error: no matching function for call to 'ESPHelper::consoleSetHelpProjectsCmds()'
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:1391:50: note: candidate is:
In file included from C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:19:0:
src\ESPHelper.h:80:12: note: void ESPHelper::consoleSetHelpProjectsCmds(String)
void   consoleSetHelpProjectsCmds(String help);
^
src\ESPHelper.h:80:12: note:   candidate expects 1 argument, 0 provided
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino: In function 'void _showerColdShotStart()':
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:1527:29: error: 'class Ticker' has no member named 'attach'
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino: In function 'void _showerColdShotStop()':
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:1545:33: error: 'class Ticker' has no member named 'detach'
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino: In function 'void loop()':
C:/Users/tomas/EMS-ESP-Boiler/src/boiler.ino:1725:98: error: 'SHOWER_PAUSE_TIME' was not declared in this scope
*** [.pioenvs\nodemcuv2\src\boiler.ino.cpp.o] Error 1
=============================================================================================== [ERROR] Took 7.20 seconds ===============================================================================================

======================================================================================================= [SUMMARY] =======================================================================================================
Environment nodemcuv2   [ERROR]
Environment d1_mini     [SKIP]
=============================================================================================== [ERROR] Took 7.21 seconds ===============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

proddy, 2018-09-28:
Ticker is a standard library is standard in the ESP8266/Arduno (see https://github.com/esp8266/Arduino/tree/master/libraries/Ticker). It looks like your PlatformIO has gone and grabbed https://github.com/sstaub/Ticker. Check your platformio.ini, scrub your .piolibdeps and try again.

blacksocket, 2018-09-28:
Hi Paul,
Thank you for the quick answer! ;-) I removed all additional libs from PIO and leave only these described in platformio.ini provided by you namely:
lib_deps =
  Time
  PubSubClient
  ArduinoJson
and additional CRC2 because I have got a message about that.
Now I am getting the errors below:
----------------------------------------
> Executing task: C:\Users\tomas.WINDOWS-9P5QTO1\.platformio\penv\Scripts\platformio.exe run <

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting boiler.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Time> 1.5
|-- <PubSubClient> 2.6
|-- <ArduinoJson> 5.13.2
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
|-- <ESP8266WiFi> 1.0
|-- <CRC32> 2.0.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs\nodemcuv2\src\boiler.ino.cpp.o
Compiling .pioenvs\nodemcuv2\libef2\ESP8266WiFi\WiFiServer.cpp.o
Compiling .pioenvs\nodemcuv2\libef2\ESP8266WiFi\WiFiServerSecureAxTLS.cpp.o
Compiling .pioenvs\nodemcuv2\libef2\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
C:/Users/tomas.WINDOWS-9P5QTO1/EMS-ESP-Boiler/src/boiler.ino:87:5: error: expected unqualified-id before string constant
// default values
^
C:/Users/tomas.WINDOWS-9P5QTO1/EMS-ESP-Boiler/src/boiler.ino: In function 'void setup()':
C:/Users/tomas.WINDOWS-9P5QTO1/EMS-ESP-Boiler/src/boiler.ino:1391:50: error: no matching function for call to 'ESPHelper::consoleSetHelpProjectsCmds()'
C:/Users/tomas.WINDOWS-9P5QTO1/EMS-ESP-Boiler/src/boiler.ino:1391:50: note: candidate is:
In file included from C:/Users/tomas.WINDOWS-9P5QTO1/EMS-ESP-Boiler/src/boiler.ino:19:0:
src\ESPHelper.h:80:12: note: void ESPHelper::consoleSetHelpProjectsCmds(String)
void   consoleSetHelpProjectsCmds(String help);
^
src\ESPHelper.h:80:12: note:   candidate expects 1 argument, 0 provided
C:/Users/tomas.WINDOWS-9P5QTO1/EMS-ESP-Boiler/src/boiler.ino: In function 'void loop()':
C:/Users/tomas.WINDOWS-9P5QTO1/EMS-ESP-Boiler/src/boiler.ino:1725:98: error: 'SHOWER_PAUSE_TIME' was not declared in this scope
*** [.pioenvs\nodemcuv2\src\boiler.ino.cpp.o] Error 1
=============================================================================================== [ERROR] Took 4.96 seconds ===============================================================================================

======================================================================================================= [SUMMARY] =======================================================================================================
Environment nodemcuv2   [ERROR]
Environment d1_mini     [SKIP]
=============================================================================================== [ERROR] Took 4.97 seconds ===============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

proddy, 2018-09-28:
Ah, I've seen this problem before on non-US locales. It may be the backslashes tripping up the compiler in the PROJECT_CMDS definition. There's a few open issues on this topic. Try making it a single line and see if that compiles correctly. For example in boiler.ino
`#define PROJECT_CMDS " test " `

blacksocket, 2018-09-28:
I removed all backslashes from the #define PROJECT_CMDS definition and nothing changed. The same error in:
"C:/Users/tomas.WINDOWS-9P5QTO1/EMS-ESP-Boiler/src/boiler.ino:87:5: error: expected unqualified-id before string constant"

Line 87 and next:

// default values
// thermostat support, shower timing and shower alert all enabled (1) (disabled = 0)
#define BOILER_THERMOSTAT_ENABLED 1
#define BOILER_SHOWER_TIMER 1
#define BOILER_SHOWER_ALERT 0

// logging - EMS_SYS_LOGGING_VERBOSE, EMS_SYS_LOGGING_NONE, EMS_SYS_LOGGING_BASIC (see ems.h)
#define BOILER_DEFAULT_LOGGING EMS_SYS_LOGGING_NONE

// shower settings
#ifndef SHOWER_TEST
const unsigned long SHOWER_PAUSE_TIME   = 15000;  // 15 seconds, max time if water is switched off & on during a shower
const unsigned long SHOWER_MIN_DURATION = 180000; // 3 minutes, before recognizing its a shower
const unsigned long SHOWER_MAX_DURATION = 420000; // 7 minutes, before trigger a shot of cold water
const unsigned long SHOWER_COLDSHOT_DURATION = 5; // 5 seconds for cold water - note, must be in seconds
const unsigned long SHOWER_OFFSET_TIME       = 8000; // 8 seconds grace time, to calibrate actual time under the shower
#else
// for DEBUGGING only
const unsigned long SHOWER_PAUSE_TIME   = 15000;  // 15 seconds, max time if water is switched off & on during a shower
const unsigned long SHOWER_MIN_DURATION = 20000;  // 20 secs, before recognizing its a shower
const unsigned long SHOWER_MAX_DURATION = 25000;  // 25 secs, before trigger a shot of cold water
const unsigned long SHOWER_COLDSHOT_DURATION = 5; // in seconds! how long for cold water shot
const unsigned long SHOWER_OFFSET_TIME       = 0; // 0 seconds grace time, to calibrate actual time under the shower
#endif

blacksocket, 2018-09-28:
I will try to fix it and report about the status when find something.. Thank you Paul for your time and help! ;-)

blacksocket, 2018-09-29:
Hi Paul, I have found that there is something wrong with the // public libraries and defined in platformio.ini. lib_deps. Could you please reupload all the files used in platformio project building especially platformio.ini ?

I added the library defined in boiler.ino code below
// public libraries
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson
#include <CRC32.h>       // https://github.com/bakercp/CRC32
**#include <PubSubClient>**

Now the errors are gone but have new one below:
--------------------------
> Executing task in folder EMS-ESP-Boiler: C:\Users\tomas.WINDOWS-9P5QTO1\.platformio\penv\Scripts\platformio.exe run <

Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting boiler.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Time> 1.5
|-- <PubSubClient> 2.6
|-- <ArduinoJson> 5.13.2
|-- <ESP8266WiFi> 1.0
|-- <CRC32> 2.0.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 3.1.4
Compiling .pioenvs\d1_mini\src\ESPHelper.cpp.o
Compiling .pioenvs\d1_mini\src\ems.cpp.o
Compiling .pioenvs\d1_mini\src\emsuart.cpp.o
Generating LD script .pioenvs\d1_mini\ld\eagle.app.v6.common.ld
Compiling .pioenvs\d1_mini\lib784\Time_ID44\DateStrings.cpp.o
Compiling .pioenvs\d1_mini\lib784\Time_ID44\Time.cpp.o
Compiling .pioenvs\d1_mini\libd19\PubSubClient_ID89\PubSubClient.cpp.o
Archiving .pioenvs\d1_mini\lib35b\libArduinoJson_ID64.a
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\BearSSLHelpers.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\CertStoreBearSSL.cpp.o
Archiving .pioenvs\d1_mini\lib784\libTime_ID44.a
Archiving .pioenvs\d1_mini\libd19\libPubSubClient_ID89.a
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiClient.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiClientSecureAxTLS.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiServer.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiServerSecureAxTLS.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiUdp.cpp.o
Compiling .pioenvs\d1_mini\lib56c\CRC32\CRC32.cpp.o
Compiling .pioenvs\d1_mini\lib402\ESP8266mDNS\ESP8266mDNS.cpp.o
Compiling .pioenvs\d1_mini\libbe9\ArduinoOTA\ArduinoOTA.cpp.o
Archiving .pioenvs\d1_mini\libef2\libESP8266WiFi.a
Compiling .pioenvs\d1_mini\lib15a\Ticker_ID1586\Ticker.cpp.o
Archiving .pioenvs\d1_mini\lib56c\libCRC32.a
Archiving .pioenvs\d1_mini\libFrameworkArduinoVariant.a
Compiling .pioenvs\d1_mini\FrameworkArduino\Esp-version.cpp.o
Archiving .pioenvs\d1_mini\lib15a\libTicker_ID1586.a
Compiling .pioenvs\d1_mini\FrameworkArduino\Esp.cpp.o
[... 49 lines cut ...]
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs\spiffs_hydrogen.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs\spiffs_nucleus.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs_api.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs_hal.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\time.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\uart.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\umm_malloc\umm_malloc.c.o
Archiving .pioenvs\d1_mini\libFrameworkArduino.a
Linking .pioenvs\d1_mini\firmware.elf
.pioenvs\d1_mini\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
.pioenvs\d1_mini\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
**.pioenvs\d1_mini\libFrameworkArduino.a(core_esp8266_main.cpp.o): In function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'**
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\d1_mini\firmware.elf] Error 1
============================================================================================== [ERROR] Took 30.81 seconds ==============================================================================================

======================================================================================================= [SUMMARY] =======================================================================================================
Environment nodemcuv2   [SKIP]
Environment d1_mini     [ERROR]
============================================================================================== [ERROR] Took 30.85 seconds ==============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

proddy, 2018-09-29:
I'm not sure why its not working. I took a new laptop with windows10,
installed VSCode and Platformio, downloaded the files from my GitHub and it
builds correctly. PlatforrmIO should be able to find the correct libraries
and download them. Which OS are you running on?

blacksocket, 2018-09-30:
I am using also win 10. I will try to switch to English locale and reinstall VSCode and Platformio and buid it again and I will let you know if this helps. Thanks Paul!

blacksocket, 2018-10-04:
1. Installing VSCode and Restart the PC (if using Windows) to apply the new PATH settings.
2. Installing Gitlens
3. Installing C/C++ InteliSense (Platformio IDE depend on this)
4. Installing PlatformIO IDE
5. Create New Project via Wizard and select board: "Wemos D1 R2 & Mini (wemos)" and Framework: "Arduino". I tried also without this step -just git clone the repo.
6. Clon EMS-ESP-Boiler repo into the Main project folder.
7. Then I am getting a message that the Arduino extension is recommended but I ignore it.
8. Configure platformio.ini like this below:
---------------------------
[platformio]
env_default = d1_mini

[common]
platform = espressif8266
build_flags = -DMQTT_MAX_PACKET_SIZE=300
build_flags_custom = '-DWIFI_SSID="myWifi"' '-DWIFI_PASSWORD="password"' '-DMQTT_IP="myIP"' '-DMQTT_USER=""' '-DMQTT_PASS=""' 
lib_deps = Time, PubSubClient, ArduinoJson

[env:d1_mini]
board = d1_mini
platform = ${common.platform}
framework = arduino
lib_deps = ${common.lib_deps}
build_flags = ${common.build_flags} ${common.build_flags_custom}
upload_speed = 921600
upload_port = COM4
--------------------------------------------------------------------
9. Then build and getting an error below. 
10. The question - did you install any additional libraries via Platformio libraries or any additional extensions ? Are you using only these libraries that are defined in platformio.ini (They have been automatically installed) ?
in c:\Users\tomas.WINDOWS-9P5QTO1\.platformio\platforms\ i have only **espressif8266** folder
in c:\Users\tomas.WINDOWS-9P5QTO1\Documents\PlatformIO\Projects\EMS-ESP-Boiler\.piolibdeps\ automatically have been installed these libs:
**ArduinoJson_ID64
CRC32_ID1202
PubSubClient_ID89
Time_ID44**
in c:\Users\tomas.WINDOWS-9P5QTO1\Documents\PlatformIO\Projects\EMS-ESP-Boiler\lib\ **is empty**
Can you see any differences or additional packages or different steps you did ? I tried this installation from scratch on Win 10, Win7 many times and always getting similar errors. 

Error message:
------------------------------------------------------------------------------
> Executing task in folder EMS-ESP-Boiler: C:\Users\tomas.WINDOWS-9P5QTO1\.platformio\penv\Scripts\platformio.exe run <

Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting boiler.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Time> 1.5
|-- <PubSubClient> 2.6
|-- <ArduinoJson> 5.13.2
|-- <Ticker> 1.0
|-- <ESP8266WiFi> 1.0
|-- <CRC32> 2.0.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs\d1_mini\src\ESPHelper.cpp.o
Compiling .pioenvs\d1_mini\src\boiler.ino.cpp.o
Compiling .pioenvs\d1_mini\src\ems.cpp.o
Compiling .pioenvs\d1_mini\src\emsuart.cpp.o
C:/Users/tomas.WINDOWS-9P5QTO1/Documents/PlatformIO/Projects/EMS-ESP-Boiler/src/boiler.ino:87:5: error: expected unqualified-id before string constant
// default values
^
C:/Users/tomas.WINDOWS-9P5QTO1/Documents/PlatformIO/Projects/EMS-ESP-Boiler/src/boiler.ino: In function 'void setup()':
C:/Users/tomas.WINDOWS-9P5QTO1/Documents/PlatformIO/Projects/EMS-ESP-Boiler/src/boiler.ino:1391:50: error: no matching function for call to 'ESPHelper::consoleSetHelpProjectsCmds()'
C:/Users/tomas.WINDOWS-9P5QTO1/Documents/PlatformIO/Projects/EMS-ESP-Boiler/src/boiler.ino:1391:50: note: candidate is:
In file included from C:/Users/tomas.WINDOWS-9P5QTO1/Documents/PlatformIO/Projects/EMS-ESP-Boiler/src/boiler.ino:19:0:
src\ESPHelper.h:80:12: note: void ESPHelper::consoleSetHelpProjectsCmds(String)
void   consoleSetHelpProjectsCmds(String help);
^
src\ESPHelper.h:80:12: note:   candidate expects 1 argument, 0 provided
C:/Users/tomas.WINDOWS-9P5QTO1/Documents/PlatformIO/Projects/EMS-ESP-Boiler/src/boiler.ino: In function 'void loop()':
C:/Users/tomas.WINDOWS-9P5QTO1/Documents/PlatformIO/Projects/EMS-ESP-Boiler/src/boiler.ino:1725:98: error: 'SHOWER_PAUSE_TIME' was not declared in this scope
Generating LD script .pioenvs\d1_mini\ld\eagle.app.v6.common.ld
*** [.pioenvs\d1_mini\src\boiler.ino.cpp.o] Error 1
=============================================================================================== [ERROR] Took 4.43 seconds ===============================================================================================

======================================================================================================= [SUMMARY] =======================================================================================================
Environment nodemcuv2   [SKIP]
Environment d1_mini     [ERROR]
=============================================================================================== [ERROR] Took 4.44 seconds ===============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

blacksocket, 2018-10-04:
[image]

blacksocket, 2018-10-04:
[image]

blacksocket, 2018-10-04:
[image]

blacksocket, 2018-10-04:
[image]

blacksocket, 2018-10-04:
[image]

blacksocket, 2018-10-04:
Hi Paul,

I tested this code on Win 10 and Win7 and it works fine. The code compiled successfully:

> Executing task in folder EMS-ESP-Boiler: C:\Users\tomas.WINDOWS-9P5QTO1\.platformio\penv\Scripts\platformio.exe run <

Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > WeMos D1 R2 & mini
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting boiler.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Time> 1.5
|-- <PubSubClient> 2.6
|-- <ArduinoJson> 5.13.2
|-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
|-- <CRC32> 2.0.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs\d1_mini\src\boiler.ino.cpp.o
Compiling .pioenvs\d1_mini\lib13a\Time_ID44\DateStrings.cpp.o
Compiling .pioenvs\d1_mini\lib13a\Time_ID44\Time.cpp.o
Compiling .pioenvs\d1_mini\lib80b\PubSubClient_ID89\PubSubClient.cpp.o
Archiving .pioenvs\d1_mini\libd89\libArduinoJson_ID64.a
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\BearSSLHelpers.cpp.o
Archiving .pioenvs\d1_mini\lib13a\libTime_ID44.a
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\CertStoreBearSSL.cpp.o
Archiving .pioenvs\d1_mini\lib80b\libPubSubClient_ID89.a
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiClient.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiClientSecureAxTLS.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiServer.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiServerSecureAxTLS.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
Compiling .pioenvs\d1_mini\libef2\ESP8266WiFi\WiFiUdp.cpp.o
Compiling .pioenvs\d1_mini\lib0bd\Ticker\Ticker.cpp.o
Compiling .pioenvs\d1_mini\lib9de\CRC32_ID1202\CRC32.cpp.o
Compiling .pioenvs\d1_mini\lib402\ESP8266mDNS\ESP8266mDNS.cpp.o
Archiving .pioenvs\d1_mini\lib0bd\libTicker.a
Archiving .pioenvs\d1_mini\libef2\libESP8266WiFi.a
Compiling .pioenvs\d1_mini\libbe9\ArduinoOTA\ArduinoOTA.cpp.o
Archiving .pioenvs\d1_mini\libFrameworkArduinoVariant.a
Archiving .pioenvs\d1_mini\lib9de\libCRC32_ID1202.a
Compiling .pioenvs\d1_mini\FrameworkArduino\Esp-version.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\Esp.cpp.o
Archiving .pioenvs\d1_mini\lib402\libESP8266mDNS.a
Compiling .pioenvs\d1_mini\FrameworkArduino\FS.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\FunctionalInterrupt.cpp.o
Archiving .pioenvs\d1_mini\libbe9\libArduinoOTA.a
Compiling .pioenvs\d1_mini\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\IPAddress.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\MD5Builder.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\Print.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\Schedule.cpp.o
[... 38 lines cut ...]
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs\spiffs_check.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs\spiffs_gc.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs\spiffs_hydrogen.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs\spiffs_nucleus.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs_api.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\spiffs_hal.cpp.o
Compiling .pioenvs\d1_mini\FrameworkArduino\time.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\uart.c.o
Compiling .pioenvs\d1_mini\FrameworkArduino\umm_malloc\umm_malloc.c.o
Archiving .pioenvs\d1_mini\libFrameworkArduino.a
Linking .pioenvs\d1_mini\firmware.elf
Building .pioenvs\d1_mini\firmware.bin
Retrieving maximum program size .pioenvs\d1_mini\firmware.elf
Checking size .pioenvs\d1_mini\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  45.4% (used 37224 bytes from 81920 bytes)
PROGRAM: [===       ]  34.4% (used 359496 bytes from 1044464 bytes)
============================================================================================= [SUCCESS] Took 26.49 seconds =============================================================================================

======================================================================================================= [SUMMARY] =======================================================================================================
Environment nodemcuv2   [SKIP]
Environment d1_mini     [SUCCESS]
============================================================================================= [SUCCESS] Took 26.50 seconds =============================================================================================

Terminal will be reused by tasks, press any key to close it.

proddy, 2018-10-04:
next add all the #DEFINE's after the "int connectionStatus;" line and
compile again...

blacksocket, 2018-10-04:
I added all #define from the previous boiler.ino code and now getting this:
[image]

proddy, 2018-10-04:
Start deleting the #define lines block by block until it compiles. Starting
with project_cmds

blacksocket, 2018-10-04:
After deleting PROJECT_CMDS code built successfully:
#define PROJECT_CMDS                                                                                                   \
    "*  s show statistics"                                                                                         \
    "*  P publish stats to MQTT"                                                                                   \
    "*  v [n] set logging (0=none, 1=basic, 2=verbose)"                                                            \
    "*  p poll response on/off"                                                                                    \
    "*  T thermostat Support on/off"                                                                               \
    "*  S shower timer on/off"                                                                                     \
    "*  A shower alert on/off"                                                                                     \
    "*  r [n] request for data from EMS. Examples:"                                                                \
    "*     from Boiler: 33=UBAParameterWW, 18=UBAMonitorFast, 19=UBAMonitorSlow, 34=UBAMonitorWWMessage"           \
    "*     from Thermostat: 91=RC20StatusMessage, A8=RC20Temperature, 6=RC20Time, 2=Version"                       \
    "*  t [n] set thermostat temperature to n"                                                                     \
    "*  m [n] set thermostat mode (0=low, 1=manual, 2=clock)"                                                      \
    "*  w [n] set boiler warm water temperature to n (min 30)"                                                     \
    "*  a [n] boiler warm water on (n=1) or off (n=0)"                                                             \
    "*  x [n] experimental (warning: for debugging only!)"

[image]

blacksocket, 2018-10-04:
Yes, I was trying to delete  backslashes "\n\r" from the PROJECT_CMDS without success:

[image]

mrfixit1, 2018-10-05:
blacksocket: try removing the \ backslash continuation characters at the end of lines 45 - 58 and concatenating the commands on one long line. This worked for me. Keep the backslashes (\n\r) in each of the individual command strings.

proddy, 2018-10-05:
thanks @mrfixit1 I think I'll re-write that piece since this is causing problems on some environments

blacksocket, 2018-10-05:
Hi mrfixit1, thanks for your help! Do you mean concatenate them including all of the quotation marks like this ?
#define PROJECT_CMDS                                                                                                   
"*  s show statistics\n\r""*  P publish stats to MQTT\n\r""*  v [n] set logging (0=none, 1=basic, 2=verbose)\n\r""*  p poll response on/off\n\r""*  T thermostat Support on/off\n\r""*  S shower timer on/off\n\r""*  A shower alert on/off\n\r""*  r [n] request for data from EMS. Examples:\n\r""*     from Boiler: 33=UBAParameterWW, 18=UBAMonitorFast, 19=UBAMonitorSlow, 34=UBAMonitorWWMessage\n\r""*     from Thermostat: 91=RC20StatusMessage, A8=RC20Temperature, 6=RC20Time, 2=Version\n\r""*  t [n] set thermostat temperature to n\n\r""*  m [n] set thermostat mode (0=low, 1=manual, 2=clock)\n\r""*  w [n] set boiler warm water temperature to n (min 30)\n\r""*  a [n] boiler warm water on (n=1) or off (n=0)\n\r""*  x [n] experimental (warning: for debugging only!)"

mrfixit1, 2018-10-05:
Yes

blacksocket, 2018-10-05:
For me this is also not working:
[image]

mrfixit1, 2018-10-05:
Looks like you deleted the #define PROJECT_CMDS at the start of line 45, try putting it back

blacksocket, 2018-10-05:
#define PROJECT_CMDS is still there i just show the message that cover the define in line 44..

blacksocket, 2018-10-05:
Indeed it works for me too!!! ;-)) Thanks mrfixit1! Thank you ALL for your help!

mrfixit1, 2018-10-05:
New line = new statement, unless there is a continuation \ to join them, which in our cases does not work....

proddy, 2018-10-06:
@mrfixit1 @blacksocket I improved the code to make the command list dynamic and avoid the compile errors. I also added a new function 'h' to list which commands are available.

blacksocket, 2018-10-07:
Hi Paul, I  have just one question. What kind of address are you using to simulate Buderus Controler in this program ? Is this Buderus Service Key with the Address 0x0B ? I am asking because when I connected Wemos to the EMS circuit I am getting a message "(00:02:08) Didn't receive acknowledgement from the 0xa8, so resending (attempt #1/4)...". I am not sure if this is caused by the electronic circuit itself or by the software. I am using RC 300 in my house.

proddy, 2018-10-08:
Yes x0B for our device. I explain this in detail in my readme. When you
have everything connected turn on verbose messaging with ‘v 2’ to see the
incoming telegrams. It’ll tell you what is broadcasting on the ems bus. A8
is probably not compatible with your thermostat (not boiler). Just comment
out that call.

blacksocket, 2018-10-09:
Paul, "v 2" works for me:
[image]

proddy, 2018-10-12:
great, looks like its working. Did you get messages from your thermostat
too?

blacksocket, 2018-10-14:
yes it is working, however I did not get messages from my thermostat..

proddy, 2018-10-14:
That's strange, I double checked the code it should print out anything that lands on the bus. Your thermostat should be sending out signals all the time. Check how your connecting to the bus and that your thermostat is still working correctly. It could also be the circuit (not sure which one you're using)

mareksugar, 2018-10-28:
Hi proddy,
first of all, thank you for a lot of work which you did with this project.
Secondly, sorry that I didn’t open new tread, but this one is related to my problem - receiving response messages.
Reading boilers broadcast messages is working without problem.
After week of trying, I have no idea where the problem is.
After every poll message I receive last 2 bytes of it (like echo), but without answer from boiler.
Something what you describe here:
https://github.com/susisstrolch/EMS-ESP12/issues/2

I use susisstrolch board, without powering circuit (I use stand-alone 5V power supply, tried 3.3V too).
On EMS bus I have boiler only, without thermostat (I want to control boiler power with UBASollwerte/UBASetpoints -1a message).
Do you have any idea what could be a problem? I check few times my hardware connection.
[image]

proddy, 2018-10-28:
> susisstrolch/EMS-ESP12#2

It looks like sending is not working. I remember Jeurgen's boards wern't
designed to also transmit. How are you powering the ESP? The ESP8266 takes
a lot of current when doing a Tx (>300mA).

mareksugar, 2018-10-28:
His schematic is same as your , only LM393 has compactator.
For power supply I use this 3W module, but I dont see any voltage drops with oscloscope
https://www.aliexpress.com/item/5PCS-TSP-05-replace-HLK-PM01-AC-DC-220V-to-5V-mini-power-supply-module-intelligent/32705954189.html?spm=a2g0s.9042311.0.0.27424c4datbokx

proddy, 2018-10-28:
I really don't know what the problem is then if you say you're not using the buck converter to drive the ESP8266 like in Juergen's board. There are others that have used my library with this board so I'm assuming it works. I can only state it works fine with esp8266 dev boards as mentioned in my doc.

As this issue is related to platformio compile errors, please open a new issue where we can track status.

closed 2018-10-28 by proddy

lobocobra, 2019-05-27:
Had the same problem and it occured when I put into platformio.ini "ticker" with a version
=> then it gets (as described by proddy) another version.

For me it solved to delete the complete folder .`piolibdeps`

Saultes45, 2020-12-02:
I had the same problem. Did 2 things:

- deleted the folder `.piolibdeps`
- deleted the "ticker" line in the `platformio.ini`

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
lib_deps = 
	adafruit/Adafruit FXAS21002C@^1.3.0
        [ticker line deleted here] <----------------------
```