emsesp/EMS-ESP #5: Sonoff DEV

Chaot1, 2018-10-07:
Hello,
ist it possible to use a Sonoff SV or a Sonoff DEV board.

proddy, 2018-10-08:
Not sure why not. I’ll need to look at the schematics.

Chaot1, 2018-10-08:
https://www.itead.cc/wiki/Sonoff_DEV
https://www.itead.cc/wiki/Sonoff_SV

Both are also working with Tasmota Firmware and Espurna:

https://github.com/xoseperez/espurna
https://github.com/arendst/Sonoff-Tasmota

The DEV Version is a perfect Board because you have all things on it that you need for the Project.
- Micro USB Port for Programming and Power supply
- 5V Output (2x)
- 3.3V Output (4x)
- 7 Input Channels
- 1 Analog input Channel
So you don't need some more extra Boards (except the EMS Board from bbqkees) and there is also the Chance to connect some more Sensors for addition Heating control.

proddy, 2018-10-12:
a wemos d1 or any nodemcu dev board would work equally well and has more inputs/outputs. But its all really the same. Just make sure with the Sonoff you're able to switch the RX/TX to different pins.

closed 2018-10-28 by proddy