emsesp/EMS-ESP #6: Pooling is not working

mareksugar, 2018-10-28:
I got problem with receiving messages from pooling boiler.
It is possible to show in log all message - including boiler poll request ?

proddy, 2018-10-28:
if you set logging to verbose you'll see all messages on the EMS bus.

The poll requests are sent very quickly and you'll get a lot of data if you want to capture them. You could add a line after #342 in ems.cpp like

```c
myDebug("received poll 0x%02x", value);
```

mareksugar, 2018-11-08:
so is full of poll requests:
0x09received poll 0x8breceived poll 0x89received poll 0x09received poll 0x89received poll 0x09received poll 0x89received poll 0x09received poll 0x8creceived poll 0x89received poll 0x09received poll 0x8dreceived poll 0x89received poll 0x09received poll 0x89received poll 0x09received poll 0x89received poll 0x09received poll 0x89received poll 0x09received poll 0x89received poll 0x09received poll 0x89received poll 0x09received poll 0x89received poll 0x09received poll 0x8ereceived poll 0x89received poll 0x09received poll 0x8freceived poll 0x89received poll 0x09received poll 0x89received poll 0x09received poll 0x89received poll..

what are 0b 89 messages?
should it be 0b 88?

[image]

proddy, 2018-11-08:
this looks fine. It means on the ems bus you have a device with id 0x09. I'm not sure what it is. The boiler (master) sends out a poll with a single byte 0x89 and the 0x09 device responds back with sending its own echo. They are only single bytes, not messages.

What are you trying to do exactly?

proddy, 2018-11-08:
it should work without the poll. It's actually disabled by default. I see there's a lot of noise on the line too which could be preventing the sending/Tx from working, which could be caused if you're using a different circuit that doesn't handle a steady ESP8266.

proddy, 2018-11-08:
looking at https://emswiki.thefischer.net/doku.php?id=wiki:ems:ems-telegramme and https://www.buderus.dk/files/200708172223530.Buderus%20Logamatic%20EMS.pdf it does seem the EM10 sits on the EMS bus with an ID 0x09. Worth checking.

mareksugar, 2018-11-08:
but now I dont have EM10 connected right now
ok BC10 has address 09. it is front display, I have no idea that front display has address too
http://img0064.popscreencdn.com/104384294_buderus-regelung-logamatic-mc10-mit-rc35-brm10-u-bc10-.jpg

proddy, 2018-11-09:
Ok, so Polling is working and I'm not sure what the actual problem is - most probably the circuit. I'm closing this issue and please open a new one if you need help with another issue.

closed 2018-11-09 by proddy