emsesp/EMS-ESP #7: cannot write a command to the boiler

pascalbuijs, 2018-11-28:
Hi,

First of all, great work on this project!

i have you're prebuild firmware working on a nodemcu with a prebuild converter board from bbqkees. Polling/Reading from the boiler is no problem. I have everything integrated in HomeAssistant like you're example. Except when i try to write a command nothing happens. When i try typing v 2 or r in a telnet session i get a "-ERROR unknown command" message.
From the debug window in the webinterface i get "[583061] [WEBSOCKET] Requested action: dbgcmd" but nothing happens. In the boiler config tab i have everything enabled. Any idee?

Pascal Buijs

proddy, 2018-11-28:
Hi Pascal

glad you got it partially working with the pre-built firmware using bbqkees' board.

The firmware is based of espurna and to be honest, i got kinda bored porting all the functionality over so left out the telnet commands and send instructions :-) I'm not sure how good your software skills are but I do recommend you compile and upload the code to your nodemcu yourself. It should only take about 10 mins to setup platformio and then commiting changes over the air (OTA) takes a few seconds after that. It's a lot of fun too! And if you need help setting up the environment I can help.

pascalbuijs, 2018-11-29:
Hi Proddy,

thanks for you're reply. I was hoping something like this was the issue after all. I allready gave platformio a try but was not able to compile and upload the code. I will try again. I should learn it anyway sometimes :-) . I hope i'll manage and will report my progress. Thanks again.

proddy, 2018-11-29:
Just download the open source visual studio code (its 43MB), install the PlatformIDE extension, open the folder with my code in it which you already have, modify a few params in platformio.ini and off you go. It's amazingly simple and you'll never look back. if you run into problems let me know.

pascalbuijs, 2018-11-29:
Uploaded the code with platformio and Got it working! :-).  Just after the upload i received a push notification from HA "boiler has booted". Having still some issues like the hot water temp is not changing after changing it's value from within HA. From a telnet session it is ok so probably something in the HA config. I was wondering how the updates of the sensors where handled because sometimes it took a while and after some digging in the log i missed the mqtt string like how it is showing in espurna.  after i found the followingin boiler,ino i knew is was how it was designed

// only send values if something has changed, to save unnecessary wifi traffic
    if (previousPublishCRC != checksum) {
        previousPublishCRC = checksum;
        if (ems_getLogging() != EMS_SYS_LOGGING_NONE) {
            myDebug("Publishing data to MQTT topics\n");
        }

        // send values via MQTT
        myESP.publish(TOPIC_BOILER_DATA, data);
    }

All seems ok now. Thank you again for the great work on this! I will test some more and will let you know.

pascalbuijs, 2018-11-29:
[image]

proddy, 2018-11-30:
awesome! nicely done. I've updated my home assitant since then. Try adding this to your ui-lovelace.yaml

`          - type: thermostat
            entity: climate.thermostat`

and include a new climate.yaml file from configuration.yaml with contents:

```  - platform: mqtt
    name: Thermostat
    modes:
      - low
      - manual
      - auto
    mode_state_topic: "home/boiler/thermostat_mode"
    temperature_command_topic: "home/boiler/thermostat_temp"
    temperature_state_topic: "home/boiler/thermostat_seltemp"
    current_temperature_topic: "home/boiler/thermostat_currtemp"
    temp_step: 0.5

pascalbuijs, 2018-12-04:
very nice! 

[image]

The thermostat is nog showing the mode in my config. Did you change something in the esp code what is relevant? If so i will update it.

pascalbuijs, 2018-12-05:
Ok, will you do that? My mqtt skills are not that good (yet) to fully explain the problem. i've just been playing with HA and mqtt for about a month now...

scheric, 2018-12-05:
Just to be sure
You are only discribig that it does not work in Lovelace. does it work in the old frontend?
Did anyone look in the states tab under developer tools?
It is this icon `<>` it should give the thermostat status there.

pascalbuijs, 2018-12-05:
in de states tab the values are correct. In the old frontend i don't have it programmed.

[image]

proddy, 2018-12-06:
I checked the code and thermostat messages and its working fine. No changes needed in lovelace as it should display either Manual or Auto. As RC30/35 units only sends low, manual or auto and low is not used or supported we can ignore it. I'll remove it from the code to avoid confusion.

closed 2018-12-06 by proddy