emsesp/EMS-ESP #9: Nefit Easy support

balk77, 2018-12-07:
Hi, I just installed a "bbq Kees" board and loaded the connected Wemos D1 Mini with your code. Connection with the boiler seems to work but the connection with my Nefit Easy does not:
```
(00:07:25) EMS-ESP-Boiler system stats:
(00:07:25)   System Logging is set to None
(00:07:25)   # EMS type handlers: 13
(00:07:25)   Thermostat is enabled, Poll is disabled, Shower timer is enabled, Shower alert is disabled
(00:07:25)   EMS Bus Stats: Connected=yes, # Rx telegrams=5, # Tx telegrams=36, # Crc Errors=7, Rx Status=idle, Tx Status=idle, Last Tx Action=none
(00:07:25) 
(00:07:25) Boiler stats:
(00:07:25)   Warm Water activated: on
(00:07:25)   Warm Water circulation pump available: off
(00:07:26)   Warm Water selected temperature: 60 C
(00:07:26)   Warm Water desired temperature: 70 C
(00:07:26)   Warm Water current temperature: 42.5 C
(00:07:26)   Warm Water # starts: 214 times
(00:07:26)   Warm Water active time: 12 days 4 hours 0 minutes
(00:07:26)   Warm Water 3-way valve: off
(00:07:26)   Selected flow temperature: 80 C
(00:07:26)   Current flow temperature: 47.9 C
(00:07:26)   Return temperature: 46.9 C
(00:07:26)   Gas: off
(00:07:26)   Boiler pump: on
(00:07:26)   Fan: off
(00:07:26)   Ignition: off
(00:07:26)   Circulation pump: off
(00:07:26)   Burner selected max power: 16 %
(00:07:26)   Burner current power: 0 %
(00:07:26)   Flame current: 0.0 uA
(00:07:26)   System pressure: 1.9 bar
(00:07:26)   Outside temperature: ? C
(00:07:26)   Boiler temperature: 55.0 C
(00:07:26)   Pump modulation: 30 %
(00:07:26)   Burner # restarts: 157 times
(00:07:26)   Total burner operating time: 12 days 8 hours 50 minutes
(00:07:26)   Total heat operating time: 0 days 4 hours 50 minutes
(00:07:26) 
(00:07:26) Thermostat stats:
(00:07:26)   Thermostat time is 00:00:00 0/0/2000
(00:07:26)   Setpoint room temperature: 0.0 C
(00:07:26)   Current room temperature: 0.0 C
(00:07:26)   Mode is set to <255>
(00:07:26) 
(00:07:26) Shower stats:
(00:07:26)   Hot water is stopped
(00:07:26)   Shower is off
(00:07:26) 
```

```
(00:08:57) System Logging is set to Verbose
(00:09:00) Requesting type UBAParameterWW(0x33) from dest 0x08
(00:09:01) Sending read telegram: len=6, telegram: 0b 88 33 00 63 3b 
(00:09:01) Corrupt telegram: len=5, telegram: 88 33 00 63 3b 
(00:09:01) Boiler telegram for us, type 0x33 len=18, telegram: 08 0b 33 00 08 ff 3c fb 00 28 00 00 46 ec ff ff 00 7b 
(00:09:02) <--- UBAParameterWW(0x33) received
(00:09:02) Publishing boiler data via MQTT
(00:09:02) Boiler broadcast, type 0x07 len=18, telegram: 08 00 07 00 0b 00 01 00 00 00 00 00 00 00 00 00 00 5a 
(00:09:03) Boiler broadcast, type 0x07 len=18, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 76 
(00:09:04) Boiler broadcast, type 0x18 len=30, telegram: 08 00 18 00 50 01 d6 0f 00 01 01 20 40 01 a0 80 00 01 d4 00 00 13 30 45 01 09 00 00 00 8f 
(00:09:04) <--- UBAMonitorFast(0x18) received
(00:09:04) Boiler broadcast, type 0x2a len=26, telegram: 08 00 2a 00 00 00 00 00 00 00 00 00 aa 00 00 80 00 00 80 00 80 00 80 00 00 c5 
(00:09:04) Boiler broadcast, type 0x34 len=24, telegram: 08 00 34 00 00 01 9f 80 00 20 00 00 01 00 00 44 70 00 39 d6 00 80 00 9b 
(00:09:04) <--- UBAMonitorWWMessage(0x34) received
(00:09:13) Boiler broadcast, type 0x18 len=30, telegram: 08 00 18 00 50 01 d5 0f 00 01 01 20 40 01 9f 80 00 01 d3 00 00 13 30 45 01 09 00 00 00 8b 
(00:09:13) <--- UBAMonitorFast(0x18) received
(00:09:14) Boiler broadcast, type 0x18 len=30, telegram: 08 00 18 00 50 01 d5 0f 00 01 01 20 40 01 9f 80 00 01 d3 00 00 13 30 45 01 09 00 00 00 8b 
(00:09:14) <--- UBAMonitorFast(0x18) received
(00:09:14) Boiler broadcast, type 0x2a len=26, telegram: 08 00 2a 00 00 00 00 00 00 00 00 00 aa 00 00 80 00 00 80 00 80 00 80 00 00 c5 
(00:09:14) Boiler broadcast, type 0x34 len=24, telegram: 08 00 34 00 00 01 9f 80 00 20 00 00 01 00 00 44 70 00 39 d6 00 80 00 9b 
(00:09:14) <--- UBAMonitorWWMessage(0x34) received
(00:09:24) Boiler broadcast, type 0x34 len=24, telegram: 08 00 34 00 00 01 9e 80 00 20 00 00 01 00 00 44 70 00 39 d6 00 80 00 c3 
(00:09:24) <--- UBAMonitorWWMessage(0x34) received
(00:09:24) Boiler broadcast, type 0x18 len=30, telegram: 08 00 18 00 50 01 d5 0f 00 01 01 20 40 01 9e 80 00 01 d0 00 00 13 30 45 01 09 00 00 00 31 
(00:09:24) <--- UBAMonitorFast(0x18) received
(00:09:24) Boiler broadcast, type 0x2a len=26, telegram: 08 00 2a 00 00 00 00 00 00 00 00 00 aa 00 00 80 00 00 80 00 80 00 80 00 00 c5 
(00:09:24) <unknown> (not for us), type 0x18 len=6, telegram: 18 88 18 00 18 c5 
(00:09:25) Boiler (not for us), type 0x18 len=29, telegram: 08 18 18 00 50 01 d5 0f 00 01 01 20 40 01 9e 80 00 01 d0 00 00 13 30 45 01 09 00 00 ba 
(00:09:25) <--- UBAMonitorFast(0x18) received
(00:09:30) Requesting type RC20Temperature(0xa8) from dest 0x17
(00:09:30) Sending read telegram: len=6, telegram: 0b 97 a8 00 63 9d 
(00:09:30) Noisy data: len=3, telegram: a8 00 63 
(00:09:33) <unknown> (not for us), type 0x1a len=7, telegram: 18 08 1a 00 50 27 fc 
```
It seems that my thermostat is not recognized. At the "not for us" mark, I have changed the temperature on my Nefit Easy.
Is there any way I can help you to get the Easy supported?

proddy, 2018-12-07:
I'm working on the Netfit Easy support. For now you can use 0x18 as the THERMOSTAT_ID and this will give you some basic results like temperature

proddy, 2018-12-08:
I'm adding support for the Nefit Easy. They have completely different messages types to the RC20 and RC3x series. I only just received the Easy and it'll be quite some work to reverse engineer the packets. So be patient!

However what is strange is that your ESP resets itself. I'll do some stress testing on bbqkees board and a Wemos D1 mini later this weekend. Are you compiling the code with the standard settings?

proddy, 2018-12-08:
Also the documentation clearly states "The code is designed for a Moduline300 (RC20) thermostat. "

balk77, 2018-12-15:
👍 Goed bezig! Contact me if you need help (testing etc)

pipovanpimperzele, 2018-12-17:
I have the following setup: Buderus Ketel type GB172i-24W with a Bosch TC100 thermostat.
I am wondering if the EMS protocol is similar as Nefit Easy.
Anyhow I am planning to buy a bbq Kees board and to use a D1 mini.

balk77, 2018-12-17:
I am trying to move away from their web based interface because it is not reliable. Many times, the thermostat does not respond to these requests.

I will do some work on this topic in the Christmas break.

pipovanpimperzele, 2018-12-17:
Can you elaborate more about the 'web service'? I am using home assistant. 
Is it like home assistant trying to emulate the 'EasyMode' app ? 
I am looking for an easy way to read the temperatures, settings,... off my TC100 thermostat using home assistant .

proddy, 2018-12-17:
This code will read the temp settings from the Easy. It just won't be able to write ems messages to set them since this is disabled on an Easy/TC100. It's, unfortunately, a very closed system. I'm making final changes to my code and then I'll upload with the Easy read-only support. To write you will need to install a service on your device that runs HA and have HA call it. See https://github.com/robertklep/nefit-easy-core

proddy, 2018-12-19:
> My problem with the *connected* Nefit Easy is that it in practice is not
> always *connected*. It many times, the app cannot connect and similarly
> the robertklep software is not able to reliably set the thermostat. Maybe
> I'll follow a hybrid approach: set temperature via the robertklep software
> and check the actual setpoint via EMS-ESP-Boiler. If not the same: repeat.
> Alternatively I will install the Moduline 300 that is still somewhere in a
> So, the *webservice* is the Nefit server which is able to set the
> temperature. The robertklep software basically mimics the telephone app and
> sets the temperature on the Nefit server. The Nefit server on its turn
> talks to the thermostat.

The Easy should always be connected - it sounds like a wifi problem in the
house! The Easy runs its own webserver so no comms needed to the
internet/cloud service. You can send XMPP type messages directly to the
Easy if you know the serial number, access key and password. If I have time
I'll see if I can implement that into my code after xmas.

proddy, 2018-12-22:
support for read added. Writing may come later

balk77, 2018-12-22:
I am still struggling with a continuously restarting wemos d1. I have flashed a new esp but get the same result. Do you have any clue how I can resolve my issue? Or at least debug?

proddy, 2018-12-22:
Look at the doc, getting started step 7. The serial output will tell you
what is going on. Are you familiar with platformio and coding or new to
most of this ? Which is ok and can help you better understand how it works

proddy, 2018-12-22:
Well, I may have messed something up. I hooked up a brand new wemos d1 mini pro via USB and uploaded the firmware. I see lots of issues connecting to the WiFi which is causing the crash. Perhaps the libraries changed. I'll look into it...

balk77, 2018-12-22:
My board is a normal d1 mini, version 3.1.0, not the pro version. It is running stable now for several minutes. 
I've downloaded a fresh zip and compiled it through Atom/Platformio. I have previously added the `-DUSE_SERIAL` flag but the serial output was not very informative.

proddy, 2018-12-22:
I think I’ll rewrite my WiFi library. It’s very impatient and retries are
too quick on slower networks

proddy, 2018-12-27:
completely re-wrote my mqtt and wifi libs. Is part of release 1.1.1

closed 2018-12-27 by proddy

balk77, 2019-01-02:
Proddy, thanks a lot for the 1.2.1 update. The WiFi connection is a lot more stable now! I have not seen a disconnect in 20 minutes. 👍

proddy, 2019-01-03:
good to hear. I've been running a few prototypes boards non stop for a few months now without disconnections so do let me know if you get connection loses or resets. It may be related to out-of-memory os watch-dog errors. My code is still not 100% primed to make best use of the flash and SRAM

Peterpc58, 2019-02-02:
Hi, I installed a "bbq Kees" board and am new to this.
When I do autodetect, I get this:
<--- Version(0x02) received
Unrecognized device found. Product ID 115, Version 03.06
<--- Version(0x02) received
Unrecognized device found. Product ID 114, Version 01.03
I have a Nefit Topline Compact HRC25 HR
How can I read my Nefit?