emsesp/EMS-ESP #11: Req: Plain EMS Messaging via MQTT

sysrun, 2018-12-13:
Hi,

your code works quite good on my Buderus (BC10/RC35)

I am trying to implement more functionality (Like Party-Mode etc) into the code. For that, it would be awesome to have a "raw" functionality:

- Send a raw HEX String (like "0B103F910101") without CRC via a MQTT Topic
   The Code is the adding the CRC32 and send it to the Bus
- Get raw Telegrams: Publish the raw incoming telegrams as HEX to a specific MQTT Topic

Is it possible to add this functionality?

Cheers, Frederik

sysrun, 2018-12-13:
Awesome :)

Exact format would be "[src] [dest] [type] [offset] [data]" as HEX - like "0B103F910101" - the ESP (Your code) would then add "[crc] <BRK>" and push it to the Bus. I think the ESP do no't have to be aware about the data, its just for trying different commands. So IMHO we don't need a re-send logic here. Just fire and forget.

On the other hand, if the "debug mode" is enabled, the ESP should just pipe the incoming BUS data to a special MQTT Topic - no check, no decoding, just the plain data as HEX

My point for that is: Its easier for me to try thing on the bus - instead of Uploading new code every time (whichs sometimes does not work over OTA so i have to down into my cellar...)

Sidenote: I am experimenting with the EMS since several years now and build the HW and a NodeJS decoder myself :) Now i got BBQKEES ready-to-go circuit  and i want to use your code instead of a Raspberry ;)

sysrun, 2018-12-13:
Idea 2 would be some kind of Telnet <-> Bus bridge-mode

proddy, 2018-12-13:
ok, probably using telnet is easy then and not bother with mqtt.

So for sending how about a command from telnet that takes a parameter and sends this as a valid telegram ([data] [crc]) to the ems bus. Fire and forget. If anything comes back it'll be echo'd in the 'verbose' logging mode. What that work?

sysrun, 2018-12-13:
Yeah, i think that would work :) CRC is calculated by the ESP?

How would the reply be logged? Is it possible to get a "raw" log-mode? No colors, no text, just data (with a newline at the end)

proddy, 2018-12-13:
yes CRC is calculated by the code. Raw log mode is easy to add too - what needs to be shown, all telegrams from all devices on the bus?

sysrun, 2018-12-13:
Nice :)

Yeah just the dirty raw data coming in: "11 22 33 44 55 66 .. ..\n" - thats it :)

sysrun, 2018-12-13:
What i imagine is something like this:

- Telnet to the stick
- Send "br" - Bridge-Mode will be activated:
  - No Logoutput anymore, just raw HEX-Strings of Telegrams coming in
  - To Send data: "send 001122334455" (not sure if a send prefix would be needed)
- Send another "br" to exit the Bridge-Mode - resume normal ops

proddy, 2018-12-13:
I've added raw mode. I'll work on the send piece later after I've figured out this darn Nefit Easy

sysrun, 2018-12-13:
Awesome :) Do you have a dedicated branch for testing?

proddy, 2018-12-14:
no, when I've added the send logic from telnet I'll upload the new version. I see you've dabbled in coding, why not contribute yourself and do a pull request?

sysrun, 2018-12-14:
Happy to contribute new functionalities later - i tried to implement the above logic myself - well, C is really not my home-turf :(

proddy, 2018-12-22:
Raw mode supported. Let me know if it works for you. To use, set the logging to Raw and then use the 'r' command

sysrun, 2018-12-22:
Awesome! :) Thank you!

I already tried to implement the RC35 - it's a bit complicated because we need to issue two queries: One for the mode and one for the state (temperatures)

proddy, 2018-12-22:
you always need 2 queries, same with RC20 and RC30. I can do the coding for you , but no idea if it'll work.

proddy, 2018-12-22:
I'll close this issue as it related to the Raw mode I implemented. Can you open a new issue for RC35 support and then'll see how I can help you remotely.

closed 2018-12-22 by proddy