emsesp/EMS-ESP #13: Outside temperature wrong when under 0°C

Analogiker, 2018-12-18:
If the outside temperature is under 0°C, then a wrong number is detected:
Console output:  Outside temperature: 6552.8 C

proddy, 2018-12-18:
good catch. I never tested for negative numbers. I checked the code it negative floats are rendered correctly so the problem is interpreting the values from the telegram. Would you mind sending me the raw data? In Verbose logging mode copy&paste a 0x19 (UBAMonitorSlow) message. Thanks

Analogiker, 2018-12-18:
Thanks for your quick reply.
-------------------------
<--- UBAMonitorSlow(0x19) received
(00:01:57) Publishing data to MQTT topics
(00:01:58) Boiler broadcast, type 0x1c len=16, data: 08 00 1c 00 00 3d b9 7b 00 00 00 00 00 00 00 74  
-------------------------
shows 0x8000 (not available ?)

proddy, 2018-12-18:
could you send the data from type 0x19? This has the outdoor temp in it.

Analogiker, 2018-12-18:
oh sorry 

(00:28:50) Boiler broadcast, type 0x19 len=32, data: 08 00 19 00 ff f3 01 6e 80 00 00 00 00 64 02 ff f1 12 bf 14 00 00 00 0c a4 5e 02 d1 ff 00 00 e8 (OK)

proddy, 2018-12-18:
ok got it. and whats the real temp outside roughly so I can double check?

Analogiker, 2018-12-18:
It's somewhat under zero °C. Depending on my position -0,5 to -0,1 
(00:57:41) Boiler broadcast, type 0x19 len=32, data: 08 00 19 00 ff f2 01 7f 80 00 00 00 00 37 02 ff f3 12 bf 20 00 00 00 0c a4 6a 02 d2 01 00 00 a8 (OK)

proddy, 2018-12-18:
is it -1.2 ?

Analogiker, 2018-12-18:
Year. That could be possible. I have no other outside thermostat.

proddy, 2018-12-18:
ok, I think I fixed it. It's hard to tell because my boiler doesn't support it. The fixed code is part of version 1.1 which I'd upload soon to github

Analogiker, 2018-12-19:
Proddy, great job. Thank you very much!! 

I wish you some nice holidays

proddy, 2018-12-19:
I'm still not certain if they use a 1's-complement or 2's-complement for representing negative values so the results may be off by 0.1. I'll leave this open so someone can double check and confirm.

proddy, 2018-12-22:
fixed in latest build.

closed 2018-12-22 by proddy