emsesp/EMS-ESP #14: Request: please add RC30/RC35 specific Data Types

SpaceTeddy, 2018-12-19:
Dear proddy,
Thanks a lot for your amazing EMS Code for the ESP. This may be exactly the part i need to continue the idea to connect the Buderus / Sieger / Nefit via MQTT to my fhem Home server. Basically to connect it to fhem, it will need some time, but first, I would like to Get and Set as much as possible data from the boiler. Currently I use a Buderus KM50 at my Sieger BK15 to get some UBA messages, but to set Temperature values etc is not possible, because the Sieger es73 is not fully SW compatible to the Buderus RC30 type.

Anyhow, I tried to understand your code and I saw that same RC30/RC35 related data types, i.E. ( HK1Betriebsart (0x3D); HK1MonitorMessage (0x3E);HK1Schaltzeiten (0x3F) ), are not implemented in your code, which are very interesting for RC3x users.

May you show me a way how to implement these or other data types in your code?

thanks a lot and again, congrats to the project.

Chris

proddy, 2018-12-19:
Thanks. When I started I only had an RC20 older Moduline 300 which was which the code was first based on. Then I borrowed an Easy/TC100 so adapted that which I'll checkin this week, and since yesterday bbqkees lent me his Moduline 400 so I can finally add full support for the RC30/35. It shouldn't be too difficult, just making the time is hard these days.

If you're proficient in C/C++ coding and don't mind getting your hands dirty I could do with some help with either coding or testing some of my early dev builds.

SpaceTeddy, 2018-12-19:
Well, to tell the truth, I've ordered the EMS Adapter board from bbqkees yesterday and hope that will arrive soon. In the meantime I'm going to prepare everything I can do to start developing. I've already installed the Toolchain and study you implementation. Means, don't worry and no need to harry up :)
I'm not professional in writing code, but I'm quite optimistic to implement RC3x related stuff and change the MQTT stuff.
BTW: If you need a additional RC30, I have a spare part here and can send it over.

regards,
chris

proddy, 2018-12-22:
I got my hands on a Moduline 400 with I expect is RC35. Although interestingly non of the EMS types match what is stated in the wiki for an RC35. The 4 heater circuits in my thermostat listen to types 3F/40/41 49/4A/4B 53/54/55 and 5D/5E/5F. I've implemented this in the code and will update shortly so you can test.

To be honest I am confused with RC20, RC30, RC35 and how they map to the Nefit Moduline series.

SpaceTeddy, 2018-12-22:
yes, what I heared is, that codes and authentification to the UBA is sometimes different from brand to brand. All types are developed by Bosch Thermodynamic and sold under brand Buderus, Junkers, Sieger and diffent don't german types like your Netfit. My Sieger es73 looks almost the same like the RC30, but has a different FW flashed, which has different codes and UBA auth. Sieger i.E. sold six different UBA types... so, the situation is not as simple. 
If only the message type code is different and the payload fits, https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme should be added with your investigations.
For me it would be perfect, if you could add the RC3x thermostat code that it could be easy adjustable to change the type codes, probably for my Sieger es73 thermostat.

thanks a lot

Chris

proddy, 2018-12-22:
RC30 added.

closed 2018-12-22 by proddy