emsesp/EMS-ESP #15: boilTemp gives very high value (sometimes)

balk77, 2018-12-19:
Hi, boilTemp sometimes gives a very high value, in this case 2073.6 degrees. See two mqtt topics below. 

{"wWSelTemp":"60","wWActivated":"on","wWCurTmp":"42.2","wWHeat":"off","curFlowTemp":"44.5","retTemp":"44.0","burnGas":"off","heatPmp":"on","fanWork":"off","ignWork":"off","wWCirc":"off","selBurnPow":"0","curBurnPow":"0","sysPress":"1.6","boilTemp":"45.4","pumpMod":"35"}
{"wWSelTemp":"60","wWActivated":"on","wWCurTmp":"42.2","wWHeat":"off","curFlowTemp":"44.5","retTemp":"44.0","burnGas":"off","heatPmp":"on","fanWork":"off","ignWork":"off","wWCirc":"off","selBurnPow":"0","curBurnPow":"0","sysPress":"1.6","boilTemp":**"2073.6"**,"pumpMod":"0"}

proddy, 2018-12-19:
strange. Could you set the logging to verbose and send me the telegram of type 0x19 (UBAMonitorSlow). It should broadcast every minute.

carcass, 2018-12-22:
I believe this is Nefit Easy specific. It sends requests for outside temperature(?) and current code doesn't handle partial or non-zero offset responses.

(01:35:28) Thermostat -> Boiler, type 0x19 telegram: 18 88 19 00 02 DB (len 6)
(01:35:28) Boiler -> Thermostat, type 0x19 telegram: 08 18 19 00 80 00 51 (len 7)

proddy, 2018-12-22:
I need to see the 0x19 broadcasts (Boiler -> all) and not the ones the Thermostat requests. Can you capture those and reply back.

carcass, 2018-12-22:
please check my comment
Thermostat sends a request for 2 bytes with 0 offset -- outside temperature
Boiler responds with 8000 (NOT SET) and CRC is 51
your code does:
EMS_Boiler.boilTemp = _toFloat(2, data); // 0x8000 if not available
byte 2 is 51 
byte 3 is not set
hex 5100 = 20736

proddy, 2018-12-22:
that's correct. But the boilTemp is never set at this point, only when a full 0x19 broadcast is received.

carcass, 2018-12-22:
I don't see anything that would prevent this code from running upon receiving 0x19 telegram with length >=5 and correct crc

proddy, 2018-12-22:
fixed in latest build.

closed 2018-12-22 by proddy