emsesp/EMS-ESP #18: Gather EMS Ids & Versions

sysrun, 2018-12-26:
I am trying to implement my RC35 (Buderus) controller i found out that the current implementation is not able to gather the thermostat information correctly as mu RC35 is using HK1 for heating.

I'd like to gather some more informations about the different setups.

I issued the following commands to get some more infos:

BC10:
r 0b 88 02 00 08 -> Product ID 123. Version 04.05

Original RC35:
r 0b 90 02 00 08 -> Product ID 86. Version 01.15

Maybe we can compile a list of Product IDs & Versions somehow

proddy, 2018-12-26:
Awesome! I was waiting for this. How are your programming skills? I can do
the first version and push it my dev branch.

sysrun, 2018-12-26:
Well, skills are quite good - i can understand 90% of the C code ;) Binary ops etc are no problem. So extending of existing code is not really a problem (adding new ems telegrams etc.)

SpaceTeddy, 2018-12-28:
dear sysrun,
my Sieger BK15 with es73 (RC30 like) controller:
`
Sending raw telegram: 0B 88 02 00 08 (CRC=94, #data=1)
(01:19:52) Corrupt telegram: telegram: 00 08 (CRC=94)
(01:19:52) Boiler -> me, type 0x02 telegram: 08 0B 02 00 40 03 06 00 00 00 00 00 (CRC=DE, #data=8)
(01:19:52) <--- Version(0x02) received
(01:19:52) Product ID 64. Version 03.06
`

`
(01:20:27) Sending raw telegram: 0B 90 02 00 08 (CRC=54, #data=1)
(01:20:27) Corrupt telegram: telegram: 00 08 (CRC=54)
(01:20:27) Thermostat -> me, type 0x02 telegram: 10 0B 02 00 4C 02 08 (CRC=18, #data=3)
(01:20:27) <--- Version(0x02) received
(01:20:27) Product ID 76. Version 02.08
`

proddy, 2018-12-28:
I'm working on some new code that auto detects the EMS devices. I've
included the Sieger BK15 and ES73 but will need your help to fine tune the
code. I'm testing now and hope to push to GitHub this weekend.

SpaceTeddy, 2018-12-28:
yes for sure, I can test it.

sysrun, 2019-01-02:
Just tried the 1.2.2 - works for me :) Awesome!
Found a Boiler compatible device, model MC10/UBA3 Boiler with TypeID 0x08, Product ID 123, Version 04.05
Found a Thermostat. Model RC35 with TypeID 0x10, Product ID 86, Version 01.15

One thing: I don't have any room sensor/thermostat (RC35 is build into the heater).
**_Current room temperature: 3200.0 C_**

proddy, 2019-01-03:
can you do a 't 3e' which will fetch the status message telegram from built-in thermostat, and post the results back. The room temperature should be byte 3 and 4 of the data package. If there is no sensor it should be 0x80 and 0x00.

sysrun, 2019-01-03:
Requesting type RC35StatusMessage(0x3E) from dest 0x10
(09:52:35) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(09:52:35) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 80 02 28 7D 00 00 00 00 00 00 00 00 64 11 3A 00 (CRC=4E), #data=16

proddy, 2019-01-03:
Ok, think I fixed this in my latest 1.2.3

sysrun, 2019-01-03:
Looks good now (Current room temperature: ? C)

SpaceTeddy, 2019-01-05:
- two additional Product ID'd and SW Versions, because I've installed an RFM20 + RC20RF as well.

`- RFM20 -> ID = 0x09 -> raw telegram: 08 89 02 00 63 (CRC=C7), #data=1 -> Unrecognized device found. Product ID 68, Version 02.03`

`- RC20RF -> ID = 0x18 -> raw telegram: 0B 98 02 00 63 (CRC=7F), #data=1 -> Unrecognized device found. Product ID 93, Version 02.00`

proddy, 2019-01-05:
thanks @SpaceTeddy. These will be included in the next build.

Bonusbartus, 2019-01-15:
The autodetect is not working for me either, since the last update (comming from 1.2.0) I cannot get any boiler information. I am using a Nefit Topline Compact HRC30 CW5 with a moduline 300 thermostat. 
In the old versions, all data seemed to work perfectly.
the discovered deviceIDs are:
0x08: Unrecognized device found. Product ID 115, Version 03.06
0x09: Unrecognized device found. Product ID 114, Version 01.03

proddy, 2019-01-16:
That's because these devices haven't been added to the list. You'll need to manually add to ems_devices.h 

For now to get it working, hard code these in my_devices.h like

#define MY_BOILER_MODELID EMS_MODEL_UBA
#define MY_THERMOSTAT_MODELID EMS_MODEL_RC20

proddy, 2019-01-23:
@Bonusbartus I've added these to version 1.3.2

lobocobra, 2019-03-04:
Luckily I have no boiler attached, because the water would be quite cold :-)
=> Warm Water current temperature: -3200.0 C

Not a real isse for me, but maybe it makes sense to eliminate values <0° and higher as 110°?

proddy, 2019-03-05:
Good idea. But can you create a new github bug for this request and I’ll make the change. Thanks

proddy, 2019-03-06:
@lobocobra for the minus -3200C warm water temperature problem could do a `log v` `boiler read 34` and post back the telegram. The code should be able to detect n/a values.

lobocobra, 2019-04-07:
Thanks!  with your latest fix the 3200C warm water temperature problem is solved.

closed 2019-05-03 by proddy