emsesp/EMS-ESP #22: EMS msg buffer limitation of 32 bytes.

SpaceTeddy, 2019-01-01:
Hi,
i'm still playing around with different msg types of the thermostat and some of them have a longer return message byte count of 32bytes. For example to read out ID 0x37 (user defined switching settings). 
`(16:25:14) 10 0B 3F 00 01 1F 00 39 01 57 00 87 21 1F 20 39 21 57 20 87 41 1F 40 39 41 57 40 87 61 1F 60 C5 `
I already modified the EMS_MAXBUFFERSIZE to 256, but even with 256byte buffer, the system does not recognizes more then 27 data bytes. Side effect is that the system seems to be less stable.  
Is there a possibility to increase the received message lenght?

many thanks,
chris

proddy, 2019-01-01:
what does ''b 37" give you? It should fetch all the data.

Then compare to "r 10 0b 37 00' which is essentially the same. The 00 value
means give me everything. It works on my RC20 but may not work for other
devices as its dependent on the firmware versions I believe.

Look for the #data" at the end the log message when in verbose mode. It
shows the amount of data fetched.

But, if I look at the wiki it says on an RC35 the 0x37 telegram should be
only 15 bytes long?

SpaceTeddy, 2019-01-01:
Hi proddy,

I edited my question some minutes after initial posting. I mean 0x3F instead of 0x37. Sry for that.

https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme
**Quelle| Ziel | Typ   | Name                    | Länge | Bemerkung**
RC35 |        | 0x3F | HK1Schaltzeiten1 | 99     | Schaltprogramm HK1 Eigen1 siehe Schaltzeiten

max. packet size is 99 for this type, but this is probably not fully needed.
For me more important is type 0x3D, which has 42 bytes len (38 data) if I follow the wiki. All data of this type could be needed if the RC35 should be implemented.
**RC35 |   | 0x3D | HK1Betriebsart | 42**

thanks again, and happy new year!!!

proddy, 2019-01-01:
can do the same test though. I'd like to see the output when you request
all the data (with 0) or with a number like 40. This will determine how to
implement this correctly. Perhaps in several separate requests.

SpaceTeddy, 2019-01-01:
--- Boiler request 0x3F
b 3f
(02:36:26) Requesting type (0x3F) from dest 0x08
(02:36:26) Boiler -> all, type 0x18 telegram: 08 00 18 00 23 01 71 64 19 09 01 25 60 80 00 02 2B 01 3C 00 3C 0B 2D 48 00 C8 00 02 00 (CRC=9B, #data=25)
(02:36:26) <--- UBAMonitorFast(0x18) received
(02:36:26) Sending read of type 0x3F to 0x08: telegram: 0B 88 3F 00 63 (CRC=0B, #data=1)
(02:36:26) Corrupt telegram: telegram: 63 (CRC=0B)
(02:36:26) Boiler -> me, type 0x3F telegram: 08 0B 3F 00 (CRC=12)
---------

--> thermostat request 0x3F
t 3f
(02:36:45) Requesting type (0x3F) from dest 0x10
(02:36:46) Sending read of type 0x3F to 0x10: telegram: 0B 90 3F 00 63 (CRC=CB, #data=1)
(02:36:46) Corrupt telegram: telegram: 00 63 (CRC=CB)
(02:36:46) Thermostat -> me, type 0x3F telegram: 10 0B 3F 00 01 1F 00 39 01 57 00 87 21 1F 20 39 21 57 20 87 41 1F 40 39 41 57 40 87 61 1F 60 (CRC=C5, #data=27)
------------

--> raw request -> gives no feedback to request 
r 10 0b 3F 00
(02:43:51) Sending raw telegram: 10 0B 3F 00 (CRC=D2)
(02:43:51) Corrupt telegram: telegram: 3F 00 (CRC=D2)
(02:43:54) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 01 10 01 39 0A 01 00 (CRC=48, #data=8)
(02:43:54) <--- RCTime(0x06) received

--> raw request -> gives no feedback to request 
r 0b 10 3F 00
....

--> raw request -> gives no feedback to request 
r 0b 90 3F 00

proddy, 2019-01-01:
try:

r 0B 90 3F 00 63

this should give you up the 99 bytes back.

Then try

r 0b 90 3f 00 23 which should give you 35 bytes back

proddy, 2019-01-01:
> Hi, it seems max bytes are #27 data.
> I sent also r 0b 90 3f 00 10 and then data bytes are #16
>
> r 0B 90 3F 00 63
> (03:05:19) Sending raw telegram: 0B 90 3F 00 63 (CRC=CB, #data=1)
> (03:05:19) Corrupt telegram: telegram: 63 (CRC=CB)
> (03:05:19) Thermostat -> me, type 0x3F telegram: 10 0B 3F 00 01 1F 00 39
> 01 57 00 87 21 1F 20 39 21 57 20 87 41 1F 40 39 41 57 40 87 61 1F 60
> (CRC=C5, #data=27)
>
> r 0b 90 3f 00 23
> (03:06:41) Sending raw telegram: 0B 90 3F 00 23 (CRC=8B, #data=1)
> (03:06:41) Corrupt telegram: telegram: 00 23 (CRC=8B)
> (03:06:41) Thermostat -> me, type 0x3F telegram: 10 0B 3F 00 01 1F 00 39
> 01 57 00 87 21 1F 20 39 21 57 20 87 41 1F 40 39 41 57 40 87 61 1F 60
> (CRC=C5, #data=27)
>
> r 0b 90 3f 00 10
> (03:08:24) Sending raw telegram: 0B 90 3F 00 10 (CRC=B8, #data=1)
> (03:08:24) Corrupt telegram: telegram: 00 10 (CRC=B8)
> (03:08:24) Thermostat -> me, type 0x3F telegram: 10 0B 3F 00 01 1F 00 39
> 01 57 00 87 21 1F 20 39 21 57 20 87 (CRC=F5, #data=16)

Ok looks like 27 is the max size of the block, but you can set the offset
(3rd byte) to request the remainder. But none of this should matter as in
the code I fetch specific values. I’ll add RC35 support within the next few
days for you to try.

SpaceTeddy, 2019-01-01:
understood, to get the full message I have to do several readings. 
As I already mentioned, I try to compare the es73 with the Buderus RCxx types and I would like to gather as much as information as possible. Didn't know that I can read the 32 byte msg also with offset.
thank you so much

closed 2019-01-01 by SpaceTeddy