emsesp/EMS-ESP #23: Wrong and random values from Boiler

darookee, 2019-01-02:
**Bug description**
I get unrealistic and seemingly random values from the boiler. 

Here a few examples:
```
(16:08:29)   Return temperature: 67.0 C
...
(16:08:29)   Flame current: 5.8 uA
(16:08:29)   System pressure: 1.1 bar
(16:08:29)   Current System Service Code: =H
(16:08:29)   Outside temperature: 4.4 C
(16:08:29)   Boiler temperature: ? C
(16:08:29)   Pump modulation: 85 %
---
(16:08:46)   Return temperature: 12.8 C
...
(16:08:46)   Flame current: 12.8 uA
(16:08:46)   System pressure: 0.0 bar
(16:08:46)   Current System Service Code:
(16:08:46)   Outside temperature: 4.4 C
(16:08:46)   Boiler temperature: ? C
(16:08:46)   Pump modulation: 85 %
---
(16:10:13)   Return temperature: 1.7 C
...
(16:10:13)   Flame current: -1664.0 uA
(16:10:13)   System pressure: 12.8 bar
(16:10:13)   Current System Service Code:
(16:10:13)   Outside temperature: 4.4 C
(16:10:13)   Boiler temperature: ? C
(16:10:13)   Pump modulation: 85 %
```

**Device information**
I use a Buderus Logamax plus GB192-25 and the integrated thermostat. Here are a few 0x19 and 0x18 I got:

```
Boiler -> all, type 0x18 telegram: 08 00 18 00 54 03 1F 64 00 00 01 64 40 80 00 02 5F 02 DC 00 00 0B 30 41 01 31 00 01 00 00 00 (CRC=7F, #data=27)
Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF, #data=11)
Boiler -> all, type 0x18 telegram: 08 00 18 00 54 03 1F 64 00 00 01 44 40 80 00 02 5F 02 DB 00 00 0B 30 41 01 31 00 01 00 00 00 (CRC=BC, #data=27)
Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF, #data=11)
Boiler -> all, type 0x19 telegram: 08 00 19 00 00 2C 80 00 80 00 00 00 00 00 00 1E 8E 01 14 E1 00 00 00 00 B5 81 00 0C CF 80 00 (CRC=AC, #data=27)
Boiler -> all, type 0x1C telegram: 08 00 1C 00 92 06 0F 05 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=1B, #data=25)
```

proddy, 2019-01-02:
Looks like with your boiler it broadcasts sometimes with an offset. The 4th byte is the offset which I assumed is 0 always for broadcast but I also see '1B' too. I'll modify the code. Please see if that helped.

darookee, 2019-01-02:
After flashing 1.2.1 I don't get any values at all as the boiler is not detected. Can I force it somehow? I tried pressing `D` but this gives the following:
```
l 4
System Logging set to Verbose
(00:06:18) 0x10 -> 0x08, type 0x23 telegram: 10 08 23 00 38 64 64 (CRC=33), #data=3
D
(00:06:19) 0x08 -> all, type 0x18 telegram: 08 00 18 00 38 02 41 64 0B 09 01 25 40 80 00 02 73 01 A1 00 30 0D 2D 48 00 C8 00 02 18 00 00 (CRC=65), #data=27

Thermostat set to disabled
Boiler set to disabled
Scanning EMS bus for devices. This may take a few seconds.
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x08
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x08
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x09
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x21
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x11
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x10
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x17
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x10
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x10
Boiler not initialized, queuing the read request.
Requesting type Version(0x02) from dest 0x18
(00:06:19) 0x08 -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
(00:06:19) 0x10 -> all, type 0x06 telegram: 10 00 06 00 13 01 16 02 08 29 02 00 10 FF 00 (CRC=C2), #data=11
(00:06:19) 0x08 -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 93 00 93 80 00 00 80 00 80 00 80 00 00 (CRC=82), #data=21
(00:06:20) 0x08 -> all, type 0x34 telegram: 08 00 34 00 3C 02 73 02 73 A1 00 01 03 00 00 5F 88 00 11 C7 00 80 00 (CRC=0A), #data=19
(00:06:20) 0x08 -> all, type 0x18 telegram: 08 00 18 00 38 02 42 64 0B 09 01 25 40 80 00 02 73 01 A1 00 2E 0D 2D 48 00 C8 00 02 18 00 00 (CRC=C0), #data=27
(00:06:20) 0x08 -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
(00:06:22) 0x08 -> all, type 0x18 telegram: 08 00 18 00 38 02 42 64 0B 09 01 25 40 80 00 02 73 01 A1 00 2F 0E 2D 48 00 C8 00 02 18 00 00 (CRC=F2), #data=27
(00:06:22) 0x08 -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
(00:06:22) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(00:06:22) 0x08 -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(00:06:23) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(00:06:23) 0x10 -> 0x08, type 0x1A telegram: 10 08 1A 00 38 64 64 (CRC=88), #data=3
(00:06:23) 0x10 -> all, type 0xFF telegram: 10 00 FF 00 01 67 00 00 (CRC=AB), #data=4
(00:06:23) 0x10 -> all, type 0xFF telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02), #data=6
(00:06:23) 0x08 -> all, type 0x18 telegram: 08 00 18 00 38 02 43 64 0B 09 01 25 40 80 00 02 73 01 A1 00 2F 0E 2D 48 00 C8 00 02 18 00 00 (CRC=A5), #data=27
(00:06:23) 0x08 -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
(00:06:24) 0x08 -> all, type 0x18 telegram: 08 00 18 00 38 02 43 64 0B 09 01 25 40 80 00 02 73 01 A1 00 30 0E 2D 48 00 C8 00 02 18 00 00 (CRC=9D), #data=27
(00:06:24) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(00:06:24) 0x08 -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
```

proddy, 2019-01-02:
That is strange. I'll modify the logging slightly and check it in tonight. Then please send this 'r 0b 88 02 00 63' which will return the version from the boiler.

darookee, 2019-01-03:
After updating I sent the raw telegram, here is the log output:

```
r 0b 88 02 00 63
(00:01:08) Sending read of type 0x18 to 0x00: telegram: 0B 80 18 00 63 (CRC=D7), #data=1
(00:01:08) Sending read of type 0x18 to 0x00: telegram: 0B 80 18 00 63 (CRC=D7), #data=1
(00:01:08) 0x08 -> all, type 0x18 telegram: 08 00 18 00 54 03 3E 64 34 0A 11 65 40 80 00 02 51 02 D4 00 FF 0B 3D 48 00 C9 00 03 18 00 00 (CRC=FC), #data=27
(00:01:09) 0x08 -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
(00:01:09) 0x08 -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 57 01 58 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(00:01:09) Sending read of type 0x19 to 0x00: telegram: 0B 80 19 00 63 (CRC=D3), #data=1
(00:01:09) 0x08 -> all, type 0x34 telegram: 08 00 34 00 3C 02 51 02 51 89 00 01 03 00 00 5F C2 00 11 D2 00 80 00 (CRC=35), #data=19
(00:01:10) 0x08 -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(00:01:10) 0x08 -> all, type 0x18 telegram: 08 00 18 00 54 03 3F 64 34 0A 11 65 40 80 00 02 52 02 D5 01 00 0B 3D 48 00 C9 00 03 18 00 00 (CRC=2C), #data=27
(00:01:10) 0x08 -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
(00:01:10) Sending read of type 0x19 to 0x00: telegram: 0B 80 19 00 63 (CRC=D3), #data=1
(00:01:11) Sending read of type 0x33 to 0x00: telegram: 0B 80 33 00 63 (CRC=7B), #data=1
(00:01:11) Sending read of type 0x33 to 0x00: telegram: 0B 80 33 00 63 (CRC=7B), #data=1
(00:01:12) Sending read of type 0x14 to 0x00: telegram: 0B 80 14 00 63 (CRC=E7), #data=1
(00:01:12) Sending read of type 0x14 to 0x00: telegram: 0B 80 14 00 63 (CRC=E7), #data=1
(00:01:13) Sending raw telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(00:01:14) 0x08 -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
(00:01:17) 0x08 -> all, type 0x18 telegram: 08 00 18 00 54 03 48 64 34 0A 11 65 40 80 00 02 52 02 DC 00 FD 0B 3D 48 00 C9 00 03 18 00 00 (CRC=EA), #data=27
(00:01:17) 0x08 -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
(00:01:18) 0x08 -> all, type 0x18 telegram: 08 00 18 00 54 03 4A 64 34 0A 11 65 40 80 00 02 53 02 DC 00 FF 0B 3D 48 00 C9 00 03 18 00 00 (CRC=9A), #data=27
(00:01:18) 0x08 -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
(00:01:18) 0x08 -> all, type 0x19 telegram: 08 00 19 00 00 19 80 00 80 00 00 00 00 55 00 1E D9 01 18 D2 00 00 00 00 B9 10 00 0D 07 80 00 (CRC=FA), #data=27
(00:01:19) 0x08 -> all, type 0x1C telegram: 08 00 1C 00 92 06 0F 05 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=1B), #data=25
(00:01:19) 0x08 -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 57 01 56 80 00 00 80 00 80 00 80 00 00 (CRC=D9), #data=21
(00:01:19) 0x08 -> all, type 0x34 telegram: 08 00 34 00 3C 02 53 02 53 89 00 01 03 00 00 5F C2 00 11 D2 00 80 00 (CRC=A9), #data=19
(00:01:20) 0x08 -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
```

I was not sure which lines were in response of the version request... :-|

proddy, 2019-01-03:
Can you try again with my latest version, 1.2.3

darookee, 2019-01-03:
I just saw that I can define the boiler type in `my_config.h` and tried that. Now I get the correct values :-)

```
  Selected flow temperature: 57 C
  Current flow temperature: 40.9 C
  Return temperature: 29.6 C
```

Just the thermostat seems to have troubles, the time is correct, but it does not display any of the other values. I'm not sure if I have an RC35, though. I just 'assumed' by reading the EMS wiki and knowing that I can set 'holiday mode', which seems to be a feature available only on the RC35...

```
Thermostat stats:
  Thermostat type: RC35 (or compatible [TypeID 0x10] Product ID:86
  Thermostat time is 15:38:07 3/1/2019
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Mode is set to ?
```

proddy, 2019-01-03:
great. I added that feature for you, and others that will have the same problem until I've built up a library of EMS devices. I'm sure it must be published somewhere on the internet, just haven't found it yet so its all trial and error. So looks like you have an RC35 compatible, but they do differ between Budrus, Nefit, Sieger, Bosch etc. Can you do a 't 2' which will tell you the firmware version? Your thermostat may respond to different heating circuits. Then do a 't 41' (for an RC30) and a 't 3e' (for a RC35)

darookee, 2019-01-03:
`t 2`:
```
t 2
Requesting type Version(0x02) from dest 0x10
(01:50:29) Thermostat -> all, type 0xFF telegram: 10 00 FF 0F 01 A5 02 D8 (CRC=A4), #data=4
(01:50:29) Thermostat -> all, type 0xFF telegram: 10 00 FF 0D 01 A5 01 6F (CRC=35), #data=4
(01:50:31) Sending read of type 0x02 to 0x10: telegram: 0B 90 02 00 63 (CRC=3F), #data=1
(01:50:31) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(01:50:31) Sending read of type 0x02 to 0x10: telegram: 0B 90 02 00 63 (CRC=3F), #data=1
(01:50:33) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
(01:50:33) Thermostat -> Boiler, type 0x23 telegram: 10 08 23 00 39 64 64 (CRC=37), #data=3
```
`t 3e`:
```
Requesting type RC35StatusMessage(0x3E) from dest 0x10
(01:51:06) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
Calling schedule fetch of values from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(01:51:06) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(01:51:06) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(01:51:07) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 39 64 64 (CRC=8C), #data=3
<--- UBASetPoints(0x1A) received
 SetPoint=57, hk_power=100 ww_power=100
(01:51:07) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 67 00 00 (CRC=AB), #data=4
(01:51:07) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02), #data=6
(01:51:07) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(01:51:08) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
(01:51:08) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
(01:51:09) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 63 (CRC=2F), #data=1
(01:51:09) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 63 (CRC=2F), #data=1
(01:51:10) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(01:51:10) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(01:51:10) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(01:51:11) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 01 11 03 26 2F 03 00 10 FF 00 (CRC=BE), #data=11
<--- RCTime(0x06) received
(01:51:11) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(01:51:12) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(01:51:12) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(01:51:12) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 63 (CRC=A7), #data=1
(01:51:13) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 63 (CRC=A7), #data=1
(01:51:14) Corrupt telegram: telegram: 0B 08 00 18 00 39 02 4E 64 0B 09 01 25 40 80 00 02 7C 01 97 00 31 0D 2D 48 00 C8 00 02 18 00 00 (CRC=55), #data=28
(01:51:14) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
```

`t 41`:

```
Requesting type RC30StatusMessage(0x41) from dest 0x10
(01:51:55) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 93 00 93 80 00 00 80 00 80 00 80 00 00 (CRC=82), #data=21
(01:51:55) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 7B 02 7C A1 00 01 03 00 00 5F D7 00 11 D6 00 80 00 (CRC=A4), #data=19
<--- UBAMonitorWWMessage(0x34) received
(01:51:55) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
(01:51:56) Sending read of type 0x41 to 0x10: telegram: 0B 90 41 00 63 (CRC=2A), #data=1
(01:51:57) Sending read of type 0x41 to 0x10: telegram: 0B 90 41 00 63 (CRC=2A), #data=1
(01:51:57) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(01:51:58) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
(01:52:01) 0x09 -> Boiler, type 0x1A telegram: 09 08 1A 00 00 00 00 (CRC=D2), #data=3
(01:52:01) 0x09 -> Boiler, type 0x23 telegram: 09 08 23 00 00 00 00 (CRC=69), #data=3
(01:52:02) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 11 (CRC=30), #data=2
(01:52:04) Boiler -> all, type 0x18 telegram: 08 00 18 00 39 02 57 64 0B 09 01 25 40 80 00 02 7B 01 97 00 2F 0E 2D 48 00 C8 00 02 18 00 00 (CRC=ED), #data=27
<--- UBAMonitorFast(0x18) received
(01:52:04) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(01:52:04) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 93 00 92 80 00 00 80 00 80 00 80 00 00 (CRC=E6), #data=21
(01:52:04) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 7B 02 7B A1 00 01 03 00 00 5F D7 00 11 D6 00 80 00 (CRC=C6), #data=19
<--- UBAMonitorWWMessage(0x34) received
Publishing boiler data via MQTT
Calling schedule fetch of values from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
[... cut ...]
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(01:52:06) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(01:52:06) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(01:52:07) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(01:52:07) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
l (01:52:08) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
```

darookee, 2019-01-03:
I just noticed that tap water and central heating show up as `off`, which they are not. Or would this just be `on` when it is heating up?

```
  Hot tap water is off
  Central Heating is off
```

darookee, 2019-01-03:
The results of `U 0`

```
System Logging set to Thermostat only
Doing a deep scan on all message types to the thermometer start at 0x%. Reboot ESP when finished.
> Scanning thermostat message type #0x00..
> Scanning thermostat message type #0x01..
> Scanning thermostat message type #0x02..
> Scanning thermostat message type #0x03..
> Scanning thermostat message type #0x04..
> Scanning thermostat message type #0x05..
> Scanning thermostat message type #0x06..
> Scanning thermostat message type #0x07..
> Scanning thermostat message type #0x08..
> Scanning thermostat message type #0x09..
(00:00:49) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 39 64 64 (CRC=8C), #data=3
 SetPoint=57, hk_power=100 ww_power=100
(00:00:49) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 67 00 00 (CRC=AB), #data=4
(00:00:49) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02), #data=6
> Scanning thermostat message type #0x0A..
> Scanning thermostat message type #0x0B..
> Scanning thermostat message type #0x0C..
> Scanning thermostat message type #0x0D..
> Scanning thermostat message type #0x0E..
> Scanning thermostat message type #0x0F..
> Scanning thermostat message type #0x10..
> Scanning thermostat message type #0x11..
> Scanning thermostat message type #0x12..
> Scanning thermostat message type #0x13..
> Scanning thermostat message type #0x14..
> Scanning thermostat message type #0x15..
> Scanning thermostat message type #0x16..
> Scanning thermostat message type #0x17..
> Scanning thermostat message type #0x18..
> Scanning thermostat message type #0x19..
> Scanning thermostat message type #0x1A..
> Scanning thermostat message type #0x1B..
> Scanning thermostat message type #0x1C..
(00:01:08) Thermostat -> all, type 0xFF telegram: 10 00 FF 08 01 A5 01 18 03 03 01 01 18 03 2F (CRC=20), #data=11
> Scanning thermostat message type #0x1D..
> Scanning thermostat message type #0x1E..
> Scanning thermostat message type #0x1F..
> Scanning thermostat message type #0x20..
(00:01:11) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 01 13 03 05 0B 03 00 10 FF 00 (CRC=60), #data=11
> Scanning thermostat message type #0x21..
> Scanning thermostat message type #0x22..
> Scanning thermostat message type #0x23..
> Scanning thermostat message type #0x24..
> Scanning thermostat message type #0x25..
> Scanning thermostat message type #0x26..
> Scanning thermostat message type #0x27..
> Scanning thermostat message type #0x28..
> Scanning thermostat message type #0x29..
> Scanning thermostat message type #0x2A..
> Scanning thermostat message type #0x2B..
> Scanning thermostat message type #0x2C..
> Scanning thermostat message type #0x2D..
> Scanning thermostat message type #0x2E..
> Scanning thermostat message type #0x2F..
> Scanning thermostat message type #0x30..
> Scanning thermostat message type #0x31..
> Scanning thermostat message type #0x32..
> Scanning thermostat message type #0x33..
> Scanning thermostat message type #0x34..
> Scanning thermostat message type #0x35..
(00:01:33) Thermostat -> Boiler, type 0x23 telegram: 10 08 23 00 39 64 64 (CRC=37), #data=3
> Scanning thermostat message type #0x36..
> Scanning thermostat message type #0x37..
> Scanning thermostat message type #0x38..
> Scanning thermostat message type #0x39..
[... 213 lines cut ...]
> Scanning thermostat message type #0xF3..
> Scanning thermostat message type #0xF4..
> Scanning thermostat message type #0xF5..
> Scanning thermostat message type #0xF6..
> Scanning thermostat message type #0xF7..
(00:04:47) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 39 64 00 (CRC=E8), #data=3
 SetPoint=57, hk_power=100 ww_power=0
(00:04:47) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 67 00 00 (CRC=AB), #data=4
(00:04:47) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 80 00 02 2A 39 00 2A 1E 01 15 03 03 01 01 15 03 32 00 00 11 01 03 FF FF 00 (CRC=94), #data=27
> Scanning thermostat message type #0xF8..
(00:04:47) Thermostat -> all, type 0xFF telegram: 10 00 FF 19 01 A5 06 04 00 00 00 00 FF 64 4B 00 3C 01 FF 01 02 (CRC=93), #data=17
(00:04:48) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02), #data=6
> Scanning thermostat message type #0xF9..
> Scanning thermostat message type #0xFA..
> Scanning thermostat message type #0xFB..
> Scanning thermostat message type #0xFC..
> Scanning thermostat message type #0xFD..
> Scanning thermostat message type #0xFE..
> Scanning thermostat message type #0xFF..
> Scanning thermostat message type #0x00..
```

I didn't look at the code to see how the on and off state of tapwater and heating active is determined, sorry. It makes sense now... :-)

The boiler is the [Buderus Logamax plus GB192-25 iW](https://www.buderus.de/de/produkte/catalogue/alle-produkte/8087_logamax-plus-gb192i) and the thermostat is the one that comes integrated in the front lid. I found a sticker in there that said 'BC30', maybe that is relevant...?

proddy, 2019-01-03:
> Looking at the Buderus page it seems the thermostat is the RC310
> <https://www.buderus.de/de/produkte/catalogue/alle-produkte/102256_logamatic-rc310>
> ...

ok, we're getting closer. From the U dump it shows that the thermostat is
not listening to any of the Tx messages we sent. I need to think about
this....

proddy, 2019-01-03:
ok, we're getting closer. From the U dump it shows that the thermostat is not listening to any of the Tx messages we sent. Just did a quick look at the boiler and I have a feeling it uses the EMS Plus type of telegram packages, which my code doesn't support - but should be fun to decipher. I'll think about some tests you can run.

proddy, 2019-01-04:
Can you set logging to basic 'l 2' and do a 't 2'. I'd like to see the firmware version.

I can see that your thermostat is using EMS Plus (https://emswiki.thefischer.net/doku.php?id=wiki:ems:plus-telegramme) since the 3rd byte is FF.

So to support EMS+ we would need to test for the 3rd byte being >F0 and then taking the type from bytes 4&5. Everything shifts across by two bytes. It'll be a lot of trial and error and not something I can code without having a physical device.

darookee, 2019-01-04:
`t 2` seems to do nothing

```
l 2
System Logging set to Basic
t 2
Requesting type Version(0x02) from dest 0x10
<--- UBAMonitorFast(0x18) received
<--- UBAMonitorFast(0x18) received
<--- UBAMonitorFast(0x18) received
<--- UBAMonitorFast(0x18) received
<--- UBAMonitorWWMessage(0x34) received
<--- UBAMonitorFast(0x18) received
<--- UBAMonitorFast(0x18) received
```

When using verbose logging I get this. It looks like the thermostat doesn't respond at all

```
l 4
System Logging set to Verbose
t 2
Requesting type Version(0x02) from dest 0x10
(00:06:05) Sending read of type 0x02 to 0x10: telegram: 0B 90 02 00 63 (CRC=3F), #data=1
(00:06:05) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(00:06:06) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
Publishing boiler data via MQTT
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:06:06) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 02 D9 64 64 0A 11 65 40 80 00 02 18 02 55 01 55 0B 3D 48 00 C9 00 03 18 00 00 (CRC=57), #data=27
<--- UBAMonitorFast(0x18) received
(00:06:06) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:06:07) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 02 39 02 3A 80 00 00 80 00 80 00 80 00 00 (CRC=8B), #data=21
(00:06:07) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 18 02 18 89 00 01 03 00 00 60 4F 00 11 E9 00 80 00 (CRC=34), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:06:07) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 02 DC 64 64 0A 11 65 40 80 00 02 18 02 57 01 53 0B 3D 48 00 C9 00 03 18 00 00 (CRC=0F), #data=27
<--- UBAMonitorFast(0x18) received
(00:06:07) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:06:09) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(00:06:09) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(00:06:09) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(00:06:10) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
(00:06:10) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
(00:06:10) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 02 E6 64 64 0A 11 65 40 80 00 02 18 02 5F 01 52 0B 3D 48 00 C9 00 03 18 00 00 (CRC=74), #data=27
<--- UBAMonitorFast(0x18) received
(00:06:11) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:06:11) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 63 (CRC=2F), #data=1
(00:06:11) 0x09 -> Boiler, type 0x2A telegram: 09 88 2A 14 01 (CRC=35), #data=1
(00:06:11) Boiler -> 0x09, type 0x2A telegram: 08 09 2A 14 00 (CRC=48), #data=1
(00:06:11) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 63 (CRC=2F), #data=1
(00:06:12) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(00:06:12) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(00:06:13) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(00:06:13) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 02 EE 64 64 0A 11 65 40 80 00 02 18 02 65 01 54 0C 3D 48 00 C9 00 03 18 00 00 (CRC=EE), #data=27
<--- UBAMonitorFast(0x18) received
```

For now the information I get from the boiler is probably all I need, as the thermostat is basically just the input device for the boiler, directly attached to it. But if I can help in any way just tell me what I can do :-)
If it helps, I have a spare 'internet-connection-gateway'-thing that came with the boiler, not sure if that uses EMS+, though...

proddy, 2019-01-05:
did you build the circuit yourself or using a pre-fab one like one from bbqkees? Since it recognizes your boiler you should be able to send it Tx packages, but again this is the most complex piece of the code (and circuit) and hard to calibrate the timings. So if you do a 'b 2' and in verbose logging what does it come back with?

darookee, 2019-01-06:
I bought a board from bbqkees, but I had to remove the pin headers to fit it in the enclosure I wanted to use, so I might have messed something up there. I'm going to check that tomorrow. I tried `b 2` with verbose logging:

```
(04:01:30) 0x09 -> Boiler, type 0x1A telegram: 09 08 1A 00 00 00 00 (CRC=D2), #data=3
(04:01:30) 0x09 -> Boiler, type 0x23 telegram: 09 08 23 00 00 00 00 (CRC=69), #data=3
(04:01:31) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 01 14 06 01 24 06 00 10 FF 00 (CRC=49), #data=11
<--- RCTime(0x06) received
(04:01:32) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 03 64 64 0B 0A 11 E5 40 80 00 02 31 02 F8 00 41 0B 3D 48 00 C9 00 03 18 00 00 (CRC=AF), #data=27
<--- UBAMonitorFast(0x18) received
(04:01:33) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
b 2
Requesting type Version(0x02) from dest 0x08
(04:01:35) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 03 61 64 0B 0A 11 E5 40 80 00 02 32 02 F7 00 40 0B 3D 48 00 C9 00 03 18 00 00 (CRC=5B), #data=27
<--- UBAMonitorFast(0x18) received
(04:01:36) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(04:01:36) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 93 00 92 80 00 00 80 00 80 00 80 00 00 (CRC=E6), #data=21
(04:01:36) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 31 02 32 89 00 05 03 00 00 61 57 00 12 17 00 80 00 (CRC=41), #data=19
<--- UBAMonitorWWMessage(0x34) received
(04:01:37) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(04:01:37) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(04:01:38) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(04:01:39) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
(04:01:40) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 03 5A 64 0B 0A 11 E5 40 80 00 02 33 02 F5 00 3E 0B 3D 48 00 C9 00 03 18 00 00 (CRC=0A), #data=27
<--- UBAMonitorFast(0x18) received
(04:01:41) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
```

It does not seem to be responding... I'll check the Tx line and test again... :-/

proddy, 2019-01-06:
strange indeed. Double check that the Tx on kees' board is on D8 on the
ESP8266. Also that you don't have a Serial connected. Better still, don't
power it via the USB on a PC

darookee, 2019-01-10:
I finally got around to checking the board with an arduino uno and it works (I used the sketch from bbqkees to request 0x02 from the boiler and it got a correct response). I'm now going to check the ESP when I have a bit more time.

proddy, 2019-01-10:
ok, thats good. It must be something in my code then, perhaps the timing
which I set in the emsuart.h. I can help you debug this

darookee, 2019-01-11:
I changed my ESP from a d1_mini to a witty cloud and the problem still persists. I used [this](https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/blob/master/Arduino-Code/EMSBusSimpleRegisterReadRequest.ino) sketch from bbqkees to check the EMS converter is working. I noticed that you are using ` #define EMSUART_BAUD 9600` and `#define EMS_TX_BRK_WAIT 2070`, wheres the sketch I tried uses 9700 baud, and `delay(2)`. I already changed the baud rate, but that didn't make it work, then I changed `2070` to `2000` and it didn't change anything either

```
(00:01:40) Sending raw telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(00:01:41) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 38 64 64 (CRC=88), #data=3
<--- UBASetPoints(0x1A) received
 SetPoint=56, hk_power=100 ww_power=100
(00:01:41) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 67 00 00 (CRC=AB), #data=4
(00:01:41) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02), #data=6
(00:01:42) Boiler -> all, type 0x18 telegram: 08 00 18 00 38 02 30 64 12 09 01 25 40 80 00 02 7B 01 D3 00 70 0E 2D 48 00 C8 00 02 18 00 00 (CRC=01), #data=27
<--- UBAMonitorFast(0x18) received
(00:01:42) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:01:44) Boiler -> all, type 0x18 telegram: 08 00 18 00 38 02 30 64 12 09 01 25 40 80 00 02 7B 01 D3 00 70 0E 2D 48 00 C8 00 02 18 00 00 (CRC=01), #data=27
<--- UBAMonitorFast(0x18) received
(00:01:44) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:01:44) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B5 00 B5 80 00 00 80 00 80 00 80 00 00 (CRC=42), #data=21
(00:01:44) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 7B 02 7B A1 00 01 03 00 00 63 A1 00 12 78 00 80 00 (CRC=50), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:01:48) Boiler -> all, type 0x18 telegram: 08 00 18 00 38 02 28 64 12 09 01 25 40 80 00 02 7B 01 D3 00 71 0E 2D 48 00 C8 00 02 18 00 00 (CRC=E2), #data=27
<--- UBAMonitorFast(0x18) received
(00:01:48) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:01:49) Boiler -> all, type 0x18 telegram: 08 00 18 00 38 02 28 64 12 09 01 25 40 80 00 02 7B 01 D3 00 71 0D 2D 48 00 C8 00 02 18 00 00 (CRC=B4), #data=27
<--- UBAMonitorFast(0x18) received
(00:01:49) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:01:53) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 01 0A 0B 28 05 04 00 10 FF 00 (CRC=56), #data=11
<--- RCTime(0x06) received
(00:01:54) Boiler -> all, type 0x18 telegram: 08 00 18 00 38 02 28 64 12 09 01 25 40 80 00 02 7B 01 D3 00 71 0D 2D 48 00 C8 00 02 18 00 00 (CRC=B4), #data=27
<--- UBAMonitorFast(0x18) received
(00:01:54) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:01:54) Thermostat -> all, type 0xFF telegram: 10 00 FF 08 01 A5 03 11 03 03 01 03 11 01 36 (CRC=DF), #data=11
(00:01:54) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B6 00 B5 80 00 00 80 00 80 00 80 00 00 (CRC=C0), #data=21
(00:01:54) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 7B 02 7B A1 00 01 03 00 00 63 A1 00 12 78 00 80 00 (CRC=50), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:01:56) Boiler -> all, type 0x18 telegram: 08 00 18 00 38 02 30 64 12 09 01 25 40 80 00 02 7B 01 D3 00 71 0D 2D 48 00 C8 00 02 18 00 00 (CRC=33), #data=27
```

What should I do now? Further reduce the delay value?

muanton, 2019-01-11:
Logamax plus GB192-25 is the latest generation boiler and it uses EMS Plus. I have the older one which is EMS. Just recently I needed to add a new thermostat to the system and Bosch engineer told me that new thermostats are not compatible with the old bus, so needed to buy old RC10.

darookee, 2019-01-11:
It is using EMS Plus, but it seems that it is largely compatible with 'old' EMS. At least I get readable data, thanks to @proddy, and also when using the test sketch from bbqkees, which also did send a request and got a response from the boiler. I would not be surprised if the actual Bosch thermostats would refuse to talk to 'old' devices so they can sell new stuff... ;-)

proddy, 2019-01-11:
yes, as I [said ](https://github.com/proddy/EMS-ESP/issues/23#issuecomment-451410053) your thermostat uses EMS Plus which isn't implemented so you will not be able to read or write to the built in RC310. But you should be able to request commands from the boiler, like "boiler read 18" should come back with something. Please try that to verify the circuit & ESP is working.

Adding EMS Plus is possible if more people want this and I get some help. It's uncharted territory.

darookee, 2019-01-15:
I received my Arduino Mega today and tried again with the example sketch from bbqkees. I just modified the requested data from `0x18` to `0x02` and here is the result.

Boiler
```
EMS bus request started.
Do request
Received a dataframe!
Correct frametype received.
-----Datagram----- BYTES=17
HEX: 0x8 0xB 0x2 0x0 0xD0 0x1 0x4 0x0 0x0 0x0 0x0 0x0 0x0 0x3 0x28 0x0 0xF4
-----End of Datagram-----
Do request
Received a dataframe!
Correct frametype received.
-----Datagram----- BYTES=17
HEX: 0x8 0xB 0x2 0x0 0xD0 0x1 0x4 0x0 0x0 0x0 0x0 0x0 0x0 0x3 0x28 0x0 0xF4
-----End of Datagram-----
```

Thermostat
```
EMS bus request started.
Do request
Received a dataframe!
Correct frametype received.
-----Datagram----- BYTES=16
HEX: 0x10 0xB 0x2 0x0 0x9E 0x12 0x3 0x0 0xFF 0x0 0x0 0x0 0x0 0x9A 0x0 0x65
-----End of Datagram-----
```

So the circut is working, I should be able to send and receive from both, the boiler and the thermostat, and it should respond, at least to the `0x02` request, which it unfortunately does not using EMS-ESP. What can I do now? :-)

proddy, 2019-01-15:
ok so the circuit is fine. Are you're 100% certain you set the Tx to the correct pin on ESP? (note, these are not the standard pins but D7 and D8). The baud is 9600 (9700 is a mistake by the original owner. bbqkees didn't write the code). What we'll need to do is uncomment some lines which will show all the messages coming in after a write/Tx operation. We then can see if anything is coming back and just being ignored because of the noise filter or because of CRC failures. I'm away this week so won't be able to guide you until later in the month when I'm back home. If you're comfortable with coding then look in ems.cpp for the 'noisey' line and comment it out.

darookee, 2019-01-16:
I connected the ESP board to my witty cloud again, checking that Rx is on D7 and Tx is on D8 (using [this pinout diagram](https://cdn.instructables.com/FCP/LLA4/IYOQCC3V/FCPLLA4IYOQCC3V.LARGE.jpg)). I then uncommented the `Noisy data:...`line, uploaded it to the ESP and tried to request `0x02` from the boiler:

```
boiler read 02
Requesting type Version(0x02) from dest 0x08
(00:00:46) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:00:46) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 03 73 64 00 02 11 64 40 80 00 02 19 02 DB 00 00 0C 30 59 00 CC 00 01 00 00 00 (CRC=D7), #data=27
<--- UBAMonitorFast(0x18) received
(00:00:46) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:00:48) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(00:00:48) Noisy data: telegram: 63 (CRC=FF)
(00:00:48) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(00:00:48) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
(00:00:50) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 03 79 64 00 02 11 64 40 80 00 02 19 02 E1 00 00 0C 30 59 00 CC 00 01 00 00 00 (CRC=3C), #data=27
<--- UBAMonitorFast(0x18) received
(00:00:50) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:00:51) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 2C 01 50 80 00 00 80 00 80 00 80 00 00 (CRC=A7), #data=21
(00:00:51) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 19 02 19 89 00 01 03 00 00 66 58 00 12 E9 00 80 00 (CRC=DC), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:00:51) Boiler -> all, type 0x18 telegram: 08 00 18 00 54 03 7A 64 00 02 11 64 40 80 00 02 19 02 E3 00 00 0C 30 59 00 CC 00 01 00 00 00 (CRC=CE), #data=27
<--- UBAMonitorFast(0x18) received
(00:00:52) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
```
Here are a few other `Noisy data...` lines I got:
```
(00:01:13) 0x09 -> Boiler, type 0x23 telegram: 09 08 23 00 00 00 00 (CRC=69), #data=3
(00:01:13) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(00:01:13) Noisy data: telegram: 63 (CRC=93)
(00:01:13) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(00:01:13) Noisy data: telegram: 63 (CRC=93)
(00:01:14) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(00:01:14) Noisy data: telegram: 63 (CRC=3B)
(00:01:14) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(00:01:14) Noisy data: telegram: 63 (CRC=3B)
(00:01:15) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 63 (CRC=A7), #data=1
(00:01:15) Noisy data: telegram: 63 (CRC=A7)
(00:01:15) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 63 (CRC=A7), #data=1
(00:01:15) Noisy data: telegram: 63 (CRC=A7)
```

proddy, 2019-01-16:
Looks like the Tx is dead, nothing is being sent out. Either faulty wiring or something not supported on your ESP board. Try connect to the Rx and Tx to the real Rx/Tx and uncomment line #138 in emsuart.cpp. This will use the serial ports. Do make sure DEBUG_SUPPORT is *not* enabled and also do not connect the ESP board to a PC using the USB port.

proddy, 2019-01-16:
> I'm powering the ESP with a USB PSU, the wittycloud does not have it's USB
> 'shield' connected, so there should be no conflict with the USB serial
> communication, DEBUG_SUPPORT is not enabled, I removed the
> system_uart_swap() and connected Rx/Tx on the ESP to Rx/Tx of the EMS
> boiler read 2
> (00:10:30) Boiler -> all, type 0x18 telegram: 08 00 18 00 30 01 F2 64 0B 09 01 25 40 80 00 02 6C 01 B8 00 33 0D 2D 48 00 C8 00 02 18 00 00 (CRC=E8), #data=27
> (00:10:31) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
> (00:10:32) Boiler -> all, type 0x18 telegram: 08 00 18 00 30 01 F1 64 0B 09 01 25 40 80 00 02 6C 01 B8 00 30 0D 2D 48 00 C8 00 02 18 00 00 (CRC=BD), #data=27
> (00:10:32) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
> (00:10:33) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 93 00 94 80 00 00 80 00 80 00 80 00 00 (CRC=A7), #data=21
> (00:10:33) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 6C 02 6C A1 00 01 03 00 00 66 5D 00 12 EA 00 80 00 (CRC=64), #data=19
> (00:10:33) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
> (00:10:33) Noisy data: telegram: 63 (CRC=FF)
> (00:10:34) Noisy data: telegram: 63 (CRC=FF)
> (00:10:34) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
> (00:10:35) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
> (00:10:35) Boiler -> all, type 0x18 telegram: 08 00 18 00 30 01 F9 64 0B 09 01 25 40 80 00 02 6C 01 B8 00 30 0D 2D 48 00 C8 00 02 18 00 00 (CRC=37), #data=27
> (00:10:36) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11

I’m kinda out of ideas. Do you have a scope or a signal analyzer by chance?
You said same thing happens with a wemos? You changed the board definition
in platformio too right?

Yosh01, 2019-01-16:
> I'm powering the ESP with a USB PSU, the wittycloud does not have it's USB 'shield' connected, so there should be no conflict with the USB serial communication, `DEBUG_SUPPORT` is not enabled, I [...]

I would suggest you to replace the board with a wemos one.
The Witty Cloud Module has the RGB LED connected to PIN8 - a noisy communication is a result. If you take look on the datasheet, there are only a few outputs not occupied by LDR, LED or other things.
Only D0, D1 and D5 are free to use and will probably not work with a faked serial.
Take a look on the pinout:
https://cdn.instructables.com/F3I/RGHJ/JBMJVA03/F3IRGHJJBMJVA03.LARGE.jpg?auto=webp&width=697

proddy, 2019-01-17:
I suspect the board definition is incorrect in platformio

darookee, 2019-01-17:
Unfortunately I don't have a signal analyzer, but I'll look around if I can get one.

I also swapped the witty cloud with the d1 mini and have it currently running with the Tx/Rx not swapped, still the same problem:

```
<--- UBAMonitorWWMessage(0x34) received
(00:01:56) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(00:01:56) Noisy data: telegram: 63 (CRC=FF)
(00:01:56) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(00:01:56) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(00:01:56) Noisy data: telegram: 63 (CRC=FF)
(00:01:57) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
(00:01:59) Boiler -> all, type 0x18 telegram: 08 00 18 00 36 02 14 64 0B 09 01 25 40 80 00 02 64 01 B8 00 33 0E 2D 48 00 C8 00 02 18 00 00 (CRC=CF), #data=27
<--- UBAMonitorFast(0x18) received
(00:01:59) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
(00:02:01) Boiler -> all, type 0x18 telegram: 08 00 18 00 36 02 13 64 0B 09 01 25 40 80 00 02 64 01 B8 00 34 0D 2D 48 00 C8 00 02 18 00 00 (CRC=00), #data=27
<--- UBAMonitorFast(0x18) received
(00:02:01) Boiler -> all, type 0x18 telegram: 08 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF), #data=11
<--- UBAMonitorFast(0x18) received
```

I don't think that it is anything with the boards, they are working without problems when using it for other things; or the wiring, as it is exactly the same when I tried it with the Mega, which worked. I also tried updating platformio (3.6.0).

tibi03, 2019-01-21:
I have similar issue , no communication direct to me , all parameters are from broadcasts ,my board is build by me.

darookee, 2019-01-22:
@tibi03 That's good, kind of.

- What boiler and thermostat are you using?
- What ESP board are you using?
- Do you have an Arduino Mega on hand and could you test the [this](https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/blob/master/Arduino-Code/EMSBusSimpleRegisterReadRequest.ino) from bbqkees to confirm that it's not your EMS board that prevents the sending?

tibi03, 2019-01-22:
RC310 https://www.buderus.de/de/produkte/catalogue/alle-produkte/102256_logamatic-rc310 ,  (model: LogamaxPlus GB172-30 iKW H https://www.buderus.com/ro/ro/ocs/produse/logamax-plus-gb172i-757615-p/    , i dont have an arduino mega . i am using nodemcuv2 board with the adapter from the scheme provided by proddy.

(00:01:17) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
(00:01:18) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 00 ED 80 00 21 00 00 01 00 00 2C 57 00 27 55 00 80 00 (CRC=67), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:01:18) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 77 00 00 00 00 40 40 00 ED 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=10), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:23) Thermostat -> all, type 0xA2 telegram: 10 00 A2 00 00 00 00 00 00 (CRC=CA), #data=5
(00:01:24) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4), #data=3
<--- UBASetPoints(0x1A) received
 SetPoint=0, hk_power=0 ww_power=0
(00:01:24) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 67 00 00 (CRC=AB), #data=4
(00:01:25) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02), #data=6
(00:01:27) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 77 00 00 00 00 40 40 00 EB 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=64), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:28) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
(00:01:28) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 00 EB 80 00 21 00 00 01 00 00 2C 57 00 27 55 00 80 00 (CRC=AE), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:01:37) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 00 EB 80 00 21 00 00 01 00 00 2C 57 00 27 55 00 80 00 (CRC=AE), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:01:38) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 76 00 00 00 00 40 40 00 EB 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=BB), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:38) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
(00:01:40) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 01 0D 16 29 37 01 00 10 FF 00 (CRC=01), #data=11
<--- RCTime(0x06) received
(00:01:47) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
(00:01:48) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 00 EB 80 00 21 00 00 01 00 00 2C 57 00 27 55 00 80 00 (CRC=AE), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:01:48) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 76 00 00 00 00 40 40 00 EB 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=BB), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:50) Thermostat -> all, type 0xFF telegram: 10 00 FF 0D 01 A5 01 6B 02 0A (CRC=CA), #data=6
(00:01:54) Thermostat -> Boiler, type 0x23 telegram: 10 08 23 00 00 00 00 (CRC=7F), #data=3
(00:01:57) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 76 00 00 00 00 40 40 00 EB 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=BB), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:58) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
(00:01:58) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 00 EB 80 00 21 00 00 01 00 00 2C 57 00 27 55 00 80 00 (CRC=AE), #data=19
<--- UBAMonitorWWMessage(0x34) received
Publishing boiler data via MQTT
Publishing hot water and heating state via MQTT
Publishing thermostat data via MQTT
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:02:00) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 01 (CRC=20), #data=2
(00:02:03) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(00:02:06) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(00:02:07) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 00 EB 80 00 21 00 00 01 00 00 2C 57 00 27 55 00 80 00 (CRC=AE), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:02:07) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
(00:02:08) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 76 00 00 00 00 40 40 00 EB 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=BB), #data=25
<--- UBAMonitorFast(0x18) received
(00:02:08) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 04 80 00 80 00 00 00 00 00 00 32 44 00 D9 B7 00 00 00 00 AD 60 00 0A EF 80 00 (CRC=ED), #data=27
<--- UBAMonitorSlow(0x19) received
(00:02:08) Boiler -> all, type 0x1C telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4B), #data=25
(00:02:09) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
(00:02:11) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
[... cut ...]
(00:02:27) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 76 00 00 00 00 40 40 00 EB 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=BB), #data=25
<--- UBAMonitorFast(0x18) received
(00:02:27) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
(00:02:28) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 00 EB 80 00 21 00 00 01 00 00 2C 57 00 27 55 00 80 00 (CRC=AE), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:02:29) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1

tibi03, 2019-01-23:
hello , after i changed rx and tx to the standard pins and explicit disable debug , now i can send and receive telegrams correctly 

> (01:08:13) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
(01:08:13) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(01:08:13) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 49 80 00 80 00 00 00 00 00 00 32 B7 00 DC 2B 00 00 00 00 AF C2 00 0B 28 80 00 (CRC=38), #data=27
<--- UBAMonitorSlow(0x19) received

proddy, 2019-01-23:
@tibi03 glad you got it working. I tested with a nodemcuv2 with the pins switched and it worked fine so interested to understand why it didn't on your dev board. the system_set_os_print is not necessary as this is done in MyEsp with the `SerialAndTelnet.setDebugOutput(false)` command

so now only to fix the Tx issue with @darookee

proddy, 2019-01-23:
@darookee I just pushed v1.3.2 with a few modifications. There's a check called 'txCapable' which shows if the code is able to transmit. Use info to see the status. If it says no then it means the bus is not sending poll requests to initiate a send. If that is the case then I will need to investigate how to trigger a handshake between our device and modern boilers like yours. let me know how you get on.

darookee, 2019-01-23:
I just uploaded it to the d1_mini and get this
```
EMS Bus Stats:
  Bus Connected=yes, Tx Capable=yes, # Rx telegrams=0, # Tx telegrams=16, # Crc Errors=0
```
I still wonder why sending works with the bbqkees example code but not with this one :-|

proddy, 2019-01-23:
@darookee I really don't know anymore. So far you seem the only one that has this issue. You're using a normal Wemos D1, you tried switching Tx/Rx, the Poll is working, the circuit works with bbqkees's arduino code so obviously its something nasty in my code - I just don't know what. What I would typically do is hook up a cheap [logical analyzer](https://www.aliexpress.com/item/USB-Logic-SCM-24MHz-8-Channel-24M-seconds-Logic-Analyzer-Debugger-for-ARM-FPGA-Logic-Analyzer/32873970588.html) see what's on the Tx line.  If this is really important for you we can make a small debug program to sniff it out

darookee, 2019-01-24:
@proddy I just ordered the logic analyzer, which should be here by the end of the week.

darookee, 2019-01-26:
@proddy I got the analyzer today and tried capturing things. I connected this between the ESP and the EMS convertert. I really don't know what this is telling me, but it looks like it at least is trying to send things (D0 is Tx, D1 is Rx). This is the ESP doing the automatic data request, I was too slow to start the capture after manually trying to read something... :-| 
[image]

proddy, 2019-01-27:
The esp is sending on the Tx which means it is receiving a poll request and the UART is working. Which 
 is good and bad because I still don't know why the EMS is not responding to the send requests. What you need to do now is also put some probes on the EMS bus (UREF) to see if all the bytes we send on the Tx end up on the serial bus in the correct order. The destination device should then respond immediately. Note if you're taking the 8V from the service jack or the 12V from the EMS then it may blow your logic analyzer so you need to tap into the signal after the comparator. If you built a breadboard that would be easy, if using one of bbqkees' board you may want to check with him first where to add the probe.

darookee, 2019-02-06:
@proddy sorry for taking so long, I didn't have much time the last weeks to tinker with this.
I can confirm that all of your assumptions are correct.

- I have an RC310, it is using EMS+
- I tested with wemos D1 and a witty cloud
- The jumper is set to 100E, J5 is empty, the board gets power from the ESP
- Yes, tested with the default Tx pin and the 'swapped' Tx pin
- arduino mega and the bbqkees board with the [example code](https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/blob/master/Arduino-Code/EMSBusSimpleRegisterReadRequest.ino) get responses to read requests

higgers, 2019-03-06:
> hello , after i changed rx and tx to the standard pins and explicit disable debug , now i can send and receive telegrams correctly

How do you explicitly disable debug?  Is this a build option?  I'd like to try swapping over to the RX/TX pins to see if it solves my problem.

higgers, 2019-03-06:
I've swapped over to the RX/TX pins but still can't seem to transmit.

I've tried hooking up both a cheapo Chinese USB logic analyser and a cheapo USB Chinese oscilloscope but can't get the things to work with my laptop.  Looks like I'll have to wait until the weekend before I can borrow a proper scope to do some more analysis.

higgers, 2019-03-06:
I just looked at the example code on bbqkees' project that you linked to in one of your responses above.  I'm pretty sure I've got an Arduino Mega in a box somewhere.  If I can get the transmit code working on that before the weekend then I could do a scope capture of a transmit from the Wemos and the Mega and compare.  That should give us a good idea of what's causing the problem.

If you have any suggestions of tests I could do then let me know.  I'll only have the scope for an hour or two so would need to get all tests done in a fairly small time window.

proddy, 2019-03-06:
@higgers to be honest bbqkees is the best person to give advice. I cheap logic analyzer should suffice. Place it on the Rx and Tx to make sure the Tx is actually transmitting when doing a 'boiler read' for example. If that looks ok, grab the big boy oscciliscope and hook it up to the incoming EMS lines (either the 2 wire or capture it from the jack on bbqkees board). Look for something like in this [picture](https://emswiki.thefischer.net/lib/exe/fetch.php?cache=&media=wiki:ems:ems-read.jpg). If you're in NL and around the Eindhoven area perhaps we can assist.

higgers, 2019-03-07:
I'm based in the UK so can't take you up on your very kind offer.  Thanks though, very kind!

Checking the TX signal along the path from the Wemos to the boiler isn't a problem, I was asking more about what commands would be useful to send to the boiler while I've got the scope.  I'll look at how the 'boiler read' command works.  I have a wireless thermostat in the system too so maybe a 'thermostat scan' would be useful too?

proddy, 2019-03-07:
Well, I'm thinking I should build in a new command which puts the board into a kind-of test mode, so basically disabling all automatic fetches and sends. That way you can connect your cheapo logic analyzer (no need for a scope) to the UEMS lines (which is the circuit after the bridge rectifier), do a 'read boiler' command and see if the Tx calls end up in the right order on the EMS line. This is basically how I built and tested my circuit.

proddy, 2019-03-07:
Also I started collecting the info on EMS+ and putting it in a GitHub project (https://github.com/proddy/EMS-ESP/projects/1).

higgers, 2019-03-07:
I like the idea of the test mode, that would make it easier to analyse data on the scope since there'd be less "noise" on the TX line.

kwertie01, 2019-03-10:
I think I have the same problem that darookee experienced. I only receive broadcasts because my TX is not functioning. I ordered the EMS interface board including the Wemos from bbqkees. To test if the interface board is functioning and my boiler/thermostat (Nefit Smartline/Moduline 400) are able to supporting writes, I connected my ardruino mega and used the sketch from bbqkees [EMSBusSimpleRegisterReadRequest.ino](https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz/blob/master/Arduino-Code/EMSBusSimpleRegisterReadRequest.ino)
With this setup the TX is working fine! 
I haven't tried an other ESP8266 yet. I think I have one (used for an other project) I will try it later. I also will do the test using the normal RX/TX lines instead of the swapped ones (it helped for tibi03)
I have no logic analyzer yet but I ordered one today.

proddy, 2019-03-11:
@kwertie01 yes there are others reporting similar issues with the Tx using newer models of boilers and thermostats. It's something that can be easily fixed with a $4 logic analyzer and some trial & error in the UART code to determine the root cause. Hope you're able to help!

higgers, 2019-03-11:
I managed to get some time to look into this yesterday though our efforts were hampered by the fact that the oscilloscope seemed to die during the car journey to my house.  Sad times.  Anyway, after some trial and error I managed to get a USB oscilloscope working with my laptop and could see that the TX line on the Wemos seems to be permanently at 3.3v.  Does this mean it's constantly sending BRK frames?  At no point did I see any data on the TX line though I'd caveat that by saying I couldn't get constant capture running on my scope so I couldn't see what happens when the Wemos first boots.  I could only start a ten second capture on the scope and then alt-tab to the Wemos telnet session and issue a command (I tried 'send' and 'autodetect').  I couldn't see any output on the TX line after these commands.

Typing 'queue' in the telnet session showed that commands were arriving on the queue but not leaving.  The queue got progressively longer.  I then started sprinkling a few myDebug calls through likely looking functions but ran out of time before I could make any headway.

What conditions are required for a command to be sent from the queue?  I guess there could be two possibilities that would result in an ever expanding queue:

1. Commands are added to the queue, sent out via the TX line but not removed from the queue
2. Commands are added to the queue but not sent out via the TX line

If we assume that I am seeing option 2 (because I've not seen any activity on the TX line) is it possible to remove/reduce any checks on send commands and test if that results in activity on the TX line?

proddy, 2019-03-11:
The code will always try and send any Tx messages that are on the queue when its status is idle (`EMS_Sys_Status.emsTxStatus = EMS_TX_STATUS_IDLE`). Then you should be seeing 'Sending ..." commands in verbose mode - can you confirm you see these messages.

To bypass all tests use the raw mode, so `log r` to reduce the chatter and then send commands using `send XX XX XX XX....` . See the README documentation for the format which is basically `[src] [dest] [type] [offset] [data]` with src `0x0B` and dest the id of your ems device like `0x10` for an RC30 thermostat. You should then see something on the Tx line. If not, something is very wrong!

higgers, 2019-03-11:
Here's the first 60 seconds or so of messages directly after a reboot with the log set to verbose:

`log v
System Logging set to Verbose
(00:00:16) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 02 1B 4C 00 01 02 20 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=C9), #data=25
<--- UBAMonitorFast(0x18) received
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(00:00:18) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 02 16 4C 00 01 02 20 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=EF), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:21) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:00:22) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 02 0A 4C 00 01 02 20 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=53), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:22) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 00 F8 80 00 A0 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=6B), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:00:23) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 02 07 4C 00 01 02 20 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=75), #data=25
<--- UBAMonitorFast(0x18) received
Publishing boiler data via MQTT
(00:00:28) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 F8 4C 00 01 02 20 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=DC), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:31) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:00:32) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 F0 4C 00 01 02 20 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=72), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:32) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 00 F9 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=25), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:00:34) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 E9 4C 00 01 02 20 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=46), #data=25
<--- UBAMonitorFast(0x18) received
Publishing boiler data via MQTT
(00:00:41) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:00:42) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 DE 4C 00 01 02 20 C0 00 F9 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=B0), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:42) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 00 F8 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=33), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:00:43) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 DA 4C 00 01 02 20 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=F1), #data=25
<--- UBAMonitorFast(0x18) received
Publishing boiler data via MQTT
(00:00:51) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:00:52) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 D0 4C 00 01 02 20 C0 00 F8 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=F8), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:52) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 00 F9 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=25), #data=17
<--- UBAMonitorWWMessage(0x34) received
Publishing boiler data via MQTT
(00:00:57) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 CA 4C 00 01 02 20 C0 00 F9 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=A2), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:01) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:01:02) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 C8 4C 00 01 02 20 C0 00 F9 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=05), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:02) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 00 F9 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=25), #data=17
<--- UBAMonitorWWMessage(0x34) received
Publishing boiler data via MQTT
Calling scheduled data refresh from EMS devices..
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:01:11) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:01:12) Boiler -> all, type 0x07 telegram: 88 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=E6), #data=15
(00:01:12) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 C2 4C 00 01 02 20 C0 00 F9 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=0C), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:12) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 00 F9 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=25), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:01:12) Boiler -> all, type 0x19 telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 02 64 00 03 7D 08 F0 C5 00 00 00 07 3C FA 00 03 62 80 00 (CRC=8B), #data=27
<--- UBAMonitorSlow(0x19) received
Publishing boiler data via MQTT`

I can't see any 'Sending...' commands there.  Could you post a similar log from your device so I can see what you mean?

proddy, 2019-03-11:
hmm, that is strange. When I'm at home I'll send you an example. In the meantime try my earlier suggestion 'send 0b 18 18 00' and see what comes back, in verbose mode

higgers, 2019-03-11:
Here's an extract from the log in verbose mode:

```
send 0b 08 18 20
(00:15:15) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 E9 4C 00 01 02 20 C0 00 FF 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=24), #data=25
<--- UBAMonitorFast(0x18) received
Publishing boiler data via MQTT
(00:15:20) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:15:21) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 E0 4C 00 01 02 20 C0 00 FF 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=55), #data=25
<--- UBAMonitorFast(0x18) received
(00:15:21) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 00 FF 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=51), #data=17
<--- UBAMonitorWWMessage(0x34) received
```

proddy, 2019-03-11:
ok, then sending is not working at all. Its probably stuck on the queue. You should see the message 'Sending raw' and a message come back "Boiler -> me'. Look at the _ems_sendTelegram() function in ems.cpp.

The problem you have, like others, is that you're not receiveing a Poll request from the bus master (UBA). This is the trigger for us to allow to send anything. Look at the first lines of `ems_parseTelegram()` where its explained. Perhaps you can print out the Poll requests to verify this? I reckon it has something to do with the handshake we need to initialize to allow our device to be recognized by the Bus master. A few of us are already looking into this and some experiments have been reported in earlier issues

higgers, 2019-03-11:
I added a debug statement at the top of 'ems_parseTelegram' like this:

```
            EMS_Sys_Status.emsPollTimestamp = millis(); // store when we received a last poll
            EMS_Sys_Status.emsTxCapable     = true;

            // SH++
            myDebug("Telegram received with length 1 (Poll): %02X", value);

            // do we have something to send thats waiting in the Tx queue? if so send it if the Queue is not in a wait state
```

It doesn't seem to be being called.  The log after uploading the new firmware looks like this:

```
log v
System Logging set to Verbose
(00:00:15) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 B9 4C 00 01 02 20 C0 01 13 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=44), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:16) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:00:16) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 13 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=79), #data=17
<--- UBAMonitorWWMessage(0x34) received
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(00:00:25) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:00:26) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 B7 4C 00 01 02 20 C0 01 12 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=0C), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:26) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 12 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=6F), #data=17
<--- UBAMonitorWWMessage(0x34) received
Publishing boiler data via MQTT
(00:00:35) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:00:36) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 B5 4C 00 01 02 20 C0 01 12 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=AB), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:36) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 12 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=6F), #data=17
<--- UBAMonitorWWMessage(0x34) received
Publishing boiler data via MQTT
(00:00:45) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:00:46) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 B5 4C 00 01 02 20 C0 01 12 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=AB), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:46) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 12 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=6F), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:00:55) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:00:56) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 B3 4C 00 01 02 20 C0 01 12 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=5B), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:56) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 12 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=6F), #data=17
<--- UBAMonitorWWMessage(0x34) received
Publishing boiler data via MQTT
(00:01:05) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:01:06) Boiler -> all, type 0x07 telegram: 88 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=E6), #data=15
(00:01:06) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 B3 4C 00 01 02 20 C0 01 11 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=61), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:06) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 11 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=55), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:01:06) Boiler -> all, type 0x19 telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 01 64 00 03 8E 08 F0 E3 00 00 00 07 3D 18 00 03 73 80 00 (CRC=14), #data=27
<--- UBAMonitorSlow(0x19) received
Publishing boiler data via MQTT
Calling scheduled data refresh from EMS devices..
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:01:15) Boiler -> all, type 0x18 telegram: 88 00 18 00 3A 01 B2 4C 00 01 02 20 C0 01 12 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=84), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:15) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:01:16) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 12 80 00 80 00 00 01 00 01 B3 CB 00 00 1B 00 (CRC=6F), #data=17
<--- UBAMonitorWWMessage(0x34) received
Publishing boiler data via MQTT
```

That seems to confirm your theory that the code is never reaching the state required to kick off sending a message to the boiler.

[50 more replies]

closed 2019-07-26 by proddy