emsesp/EMS-ESP #24: CRC errors since b527949

darookee, 2019-01-02:
**Bug description**
I get a log of CRC errors since b527949, with 13603d6 it works without the errors. I don't think it is related to #23, as #23 happens with 13603d6 as well.

```
EMS Bus Stats: Connected=yes, # Rx telegrams=0, # Tx telegrams=34441, # Crc Errors=34485
```
and a few log lines

```
(16:17:45) Sending read of type 0x41 to 0x10: telegram: 0B 90 41 00 63 (CRC=2A, #data=1)
(16:17:45) Corrupt telegram: telegram: 63 (CRC=2A)
(16:17:45) Thermostat -> Boiler, type 0x23 telegram: 10 08 23 00 36 64 64 (CRC=0B, #data=3)
(16:17:45) Sending read of type 0x41 to 0x10: telegram: 0B 90 41 00 63 (CRC=2A, #data=1)
(16:17:45) Corrupt telegram: telegram: 63 (CRC=2A)
(16:17:46) Sending read of type 0x41 to 0x10: telegram: 0B 90 41 00 63 (CRC=2A, #data=1)
(16:17:46) Corrupt telegram: telegram: 63 (CRC=2A)
```

**Device information**
Buderus Logamax plus GB192-25 and the integrated thermostat.

proddy, 2019-01-02:
this is normal as there will always be noise on the line. I improved the code slightly to distinguish between noise and just poorly formatted telegrams. See if the latest change set in version 1.2.1 helps

darookee, 2019-01-02:
Ah, OK. I was thinking it was an error because it didn't happen in earlier versions and it looked like it was resetting the ESP after a while, which, in fact, it did not.
I just flashed 1.2.1 and the CRC errors are down to 1 after ~10 minutes. Thanks! :-)

closed 2019-01-02 by darookee