emsesp/EMS-ESP #25: findings V 1.2.1

SpaceTeddy, 2019-01-02:
dear Paul,

Following items I've found in 1.2.1:

info: 
- my boiler and Thermostat is automatically recognized.
- OTA update is working
- no corrupt telegram messages

minor:
- copy paste issue for RC35 boiler commands. RC30 defines are used instead RC35 defines
`EMS_MODEL_RC3, EMS_TYPE_RC35Set, "RC35Set", _process_RC35Set},
    {EMS_MODEL_RC30, EMS_TYPE_RC35StatusMessage, "RC35StatusMessage", _process_RC35StatusMessage},}`

modifications:
**ems.cpp**
- activated ES73 thermostat for read/write
`{EMS_MODEL_ES73, EMS_THERMOSTAT_READ_YES, EMS_THERMOSTAT_WRITE_YES}`
- added same ES73 commands like RC35
`// ES73
    {EMS_MODEL_ES73, EMS_TYPE_RCOutdoorTempMessage, "RCOutdoorTempMessage", _process_RCOutdoorTempMessage},
    {EMS_MODEL_ES73, EMS_TYPE_RCTime, "RCTime", _process_RCTime},
    {EMS_MODEL_ES73, EMS_TYPE_RC35Set, "RC35Set", _process_RC35Set},
    {EMS_MODEL_ES73, EMS_TYPE_RC35StatusMessage, "RC35StatusMessage", _process_RC35StatusMessage},
    {EMS_MODEL_ES73, EMS_TYPE_UBASetPoints, "UBASetPoints", _process_SetPoints},`
- added ES73 ID in ems_getThermostatValues()
`else if (model_id == EMS_MODEL_ES73) {
        ems_doReadCommand(EMS_TYPE_RC35StatusMessage, type); // to get the setpoint temp
        ems_doReadCommand(EMS_TYPE_RC35Set, type);           // to get the mode
    }`
- added ES73 ID in ems_setThermostatTemp(float temperature)
`else if (model_id == EMS_MODEL_ES73) {
        EMS_TxTelegram.type               = EMS_TYPE_RC35Set;
        EMS_TxTelegram.offset             = EMS_OFFSET_RC35Set_temp;
        EMS_TxTelegram.comparisonPostRead = EMS_TYPE_RC35StatusMessage; // call a different type to refresh temperature value
    }`
- added ES73 ID in ems_setThermostatMode(uint8_t mode)
`else if (model_id == EMS_MODEL_ES73) {
        EMS_TxTelegram.type   = EMS_TYPE_RC35Set;
        EMS_TxTelegram.offset = EMS_OFFSET_RC35Set_mode;
    }`

**boiler.ino**
- added selFlowTemp, extTemp in mqtt boiler_data
`rootBoiler["selFlowTemp"] = _float_to_char(s, EMS_Boiler.selFlowTemp);
  rootBoiler["outdoorTemp"] = _float_to_char(s, EMS_Boiler.extTemp);`

question:
- I use the MQTT message for my home server software (fhem). Unfortunately I really don't know how to set the thermostat temperature in type 0x3D if the thermostat is set in "Auto" Mode. In 0x3D you can set the night/day/holiday temperature with offset 0x01/0x02/0x03. Other method could be to write the setpoint temperature to 0X3E Offset 0x02, but I didn't try this till now. Therefore it could be a possibility to read out the "day mode" in 0x3E and send it via mqtt
- get day mode: 
#define EMS_OFFSET_RC35Get_mode_day 1   // position of thermostat day mode
add 
uint8_t mode_day;          // 0=night, 1=day
in struct _EMS_Thermostat 

void _process_RC35StatusMessage(uint8_t * data, uint8_t length)
    uint8_t v = data[EMS_OFFSET_RC35Get_mode_day];
    EMS_Thermostat.mode_day      = bitRead(v, 1);         //get day mode flag
    EMS_Thermostat.setpoint_roomTemp = ((float)data[EMS_Thermostat.mode_day]) / (float)EMS_Thermostat.mode_day;
... write to setpoint temperature offset 

👍

proddy, 2019-01-02:
Good finds! thanks. I've updated the code. I will dig into the RC35 messaging. There's a lot already published on these telegrams so it should be quite easy. Only thing is I can't really test it. I may start updating my dev branch in GitHub with some experimental code.

proddy, 2019-01-03:
So for the RC35 modes, you can read them using 0x3E (EMS_TYPE_RC35StatusMessage). The Holiday mode is `bitRead(data[0], 5)`, Summer Mode is `bitRead(data[1],0)` and Day Mode is `bitRead(data[1],1)`. 

To set the temperature values you would use 0x3D (EMS_TYPE_RC35Set) with `data[3]` for Summer Temp, `data[2]` for Day Temp and `data[1]` for Night Temp.

This is all pretty straight forward to implement. Thing is I can't really test it. Would you mind closing this issue (since its a bug) and open a feature request for the mode support?

closed 2019-01-04 by SpaceTeddy