emsesp/EMS-ESP #27: MQTT receive command topic not working?

SpaceTeddy, 2019-01-03:
dear proddy,

sry to issue a topic again, but since 1.2.x I've got trouble to "sent" MQTT messages to the ESP8266. receiving messages is not an issue at all.
With V. 1.1.x is was working well. As I see in the changelog, you have modified the MQTT client, therefore I would like to kindly as you if I have to do something different in V1.2.x to V1.1.x.

My test setup:
ESP8266 <-> mosquitto (rPi) <-> MQTT.fx (Mac)
User and PW is set on ESP8266 & MQTT.fx

any idea?

many thanks

chris

[image]

[image]

SpaceTeddy, 2019-01-03:
update: 
ok, if I activate the debug output in void MyESP::_mqttOnMessage
`myDebug_P(PSTR("[MQTT] Received %s => %s"), topic, message);`
I can clearly see that the thermostat command messages are received by the ESP, but the system does not change the mode or temperature.
Strange is, that I receive also a lot of heating_active and tapwater_active messages, but I do not send there messages 🧐
home/boiler/tapwater_active => 0
home/boiler/heating_active => 0`

[image]

BTW: I don't use HA

SpaceTeddy, 2019-01-03:
well,
strcmp of the command topic is the problem, because the received topic value is home/boiler/thermostat_cmd_mode and the compare is on TOPIC_THERMOSTAT_CMD_MODE only.
so, i guess that home/boiler/ must be cut first, or compare to ...(dirty hack)...
`if (strcmp(topic, "home/boiler/"TOPIC_THERMOSTAT_CMD_MODE) == 0)`

It is working then...
[055277] [MQTT] Received home/boiler/thermostat_cmd_mode => manual
MQTT topic event: home/boiler/thermostat_cmd_mode
MQTT message: manual
MQTT topic: thermostat mode value manual
Setting thermostat mode to 1

For tapwater_active and heating_active I deactivated for testing
        //myESP.mqttSubscribe(TOPIC_BOILER_TAPWATER_ACTIVE);
        //myESP.mqttSubscribe(TOPIC_BOILER_HEATING_ACTIVE);
in void MQTTcallback(unsigned int type, const char * topic, const char * message)

I should also drink something ... because I really like your project. 👍

proddy, 2019-01-03:
yes, I had already fixed the subscribe problem. I'll fix the CMD_MODE too and do a new build soon

SpaceTeddy, 2019-01-03:
works in 1.2.3

closed 2019-01-03 by SpaceTeddy