emsesp/EMS-ESP #29: findings V 1.2.3

SpaceTeddy, 2019-01-03:
only related to ES73 aka RC35 compatible types:

- night/day/auto mode -> offset is **7** .... 6 should be Room-offset setting -> verified with my boiler
`#define EMS_OFFSET_RC35Set_mode 7             // position of thermostat mode`

- void _process_RC35StatusMessage(uint8_t * data, uint8_t length) , we have to read the Thermostat current_temp from here (0x3E) as well! Offset is 3.
`#define EMS_TYPE_RC35StatusMessage_curr 3     // current temp`
`EMS_Thermostat.curr_roomTemp = _toFloat(EMS_TYPE_RC35StatusMessage_curr, data);`

proddy, 2019-01-03:
oops, The offset 7 vs 6 is a bug my side. I can't count. Fixed.

What I really need is someone with an official RC35 that can confirm the 0x3E has the current temp at offset 3.

proddy, 2019-01-03:
buffer increased from 500 to 600

proddy, 2019-01-03:
and made your changes in 1.2.4

SpaceTeddy, 2019-01-03:
please increase also StaticJsonBuffer<512> jsonBuffer;
without increasing this JsonBuffer, the thermostat_data message is not complete

thx and have a good sleep.

proddy, 2019-01-03:
done

closed 2019-01-03 by proddy