emsesp/EMS-ESP #31: BC25 (0x09) detected as MM10

sysrun, 2019-01-05:
The BC25 Module (0x09) is detected as MM10 (0x21)

(00:00:09) Sending read of type 0x02 to 0x09: telegram: 0B 89 02 00 63 (CRC=F7), #data=1
(00:00:09) 0x09 -> me, type 0x02 telegram: 09 0B 02 00 7D 01 05 (CRC=C1), #data=3
<--- Version(0x02) received
Boiler recognized. Model MM10 Mixer Module with TypeID 0x21, Product ID 125, Version 01.05

proddy, 2019-01-05:
This is because i guessed the MM10 would have a product id of 125. I've added BC25 to release 1.2.5. Would be good if you could check when its checked in

proddy, 2019-01-09:
ok, try again with 1.3.0

sysrun, 2019-01-10:
Looks good now :)

I've been a bit puzzled first. Had to connect to the esp wifi first and set the wifi params. 

Thought it would be used from the platformio.ini ?!