emsesp/EMS-ESP #35: telnet set command uses wrong characters

SpaceTeddy, 2019-01-12:
to setup a new wifi connection via telnet I used the set wifi_ssid command.
Result: the set command ignores uppercases and modifies it to lowercases.
Currently I see no way to change from AP to Client mode. I had to flash back to 1.2.4.

[image]

kind regards,
chris

proddy, 2019-01-12:
that is a silly bug, I lowercase all the telent commands. I'll fix that now, but you can still hardcode the wifi/mqtt settings in my_config.h

SpaceTeddy, 2019-01-12:
Well, I tried to hardcode the new WiFi said, but for ever reason, the system tried to connect to the old ssid. Did you store the old settings anywhere? If yes, do you use these settings only?

proddy, 2019-01-12:
they are stored in SPIFFS, the internal ESP's RAM which is like a file system. When you start type 'set erase' and this will reset all your settings and take the ones from my_config.h

SpaceTeddy, 2019-01-12:
Alright. 😎

SpaceTeddy, 2019-01-13:
fixed!

closed 2019-01-13 by SpaceTeddy