emsesp/EMS-ESP #36: Debug for wifi connect

muzixiaosanshi, 2019-01-12:
Hi proddy:

I use linux enviroment to build code,when I finished build and upload firmware to ESP8266 board and press reset button on the board.I search IP on my router but I cann't found it connect to my router.I found board return to AP mode.I found in your wiki:
If the WiFi, MQTT, MDNS or something else fails to connect, re-build the firmware using the `-DDEBUG_SUPPORT` option.
So I use this command "sudo platformio --DDEBUG_SUPPORT run -t upload",but it cann't work.Does this command is right?

muzixiaosanshi, 2019-01-12:
I config my wifi SSID and passwd in /src/myconfig.h

muzixiaosanshi, 2019-01-12:
Hi proddy:
I forgot to tell you I have already configured wifi ssid and passwd in /src/my_config.h.I want to know why my board can't connect to my router.Thanks.

proddy, 2019-01-12:
I don't think you did anything wrong. Default settings should start up in AP mode and should be visible as ems-esp on your router. I will need to do some more testing here to check that it works correctly on a non Windows environment.  If the wifi fails then you can't get telnet which is used for logging information which is why I in extreme situations you can use DEBUG_SUPPORT and then see the wifi connect details on the Serial USB. I've updated the doc to make this clearer

muzixiaosanshi, 2019-01-12:
Hi proddy:
Thanks for your reply. If default setting is AP,how can 8266 connect to my router? And you mentioned above "should be visible as ems-esp on your router."I can't understand this,could you explain this clearly for me,Thanks.

muzixiaosanshi, 2019-01-12:
Now,esp8266 is invisible on my router.So i want to rebuild firmware use debug_support to see log use usb serial.Thanks.

muzixiaosanshi, 2019-01-13:
Hi proddy:
Thanks for your help,Now my ESP board can connect to my router.I found one

muzixiaosanshi, 2019-01-13:
Hi Proddy:
Thanks for your help,my ESP can connect to my router.But I found one problem.I wonder you have ever met before.I found "set wifi_ssid [value]" value - must be lowercase letters.if I input capital ,I saw log print  corresponding lowercase letters.For example:My router name is "TP-LINK-E9EA",when I input "set wifi_ssid TP-LINK-E9EA".Log print "set wifi_ssid  tp-link-e9ea".So I change my router name from capital to lowercase letters,then it can connect sucessfully.

proddy, 2019-01-13:
Yes, the lower case bug was fixed in the latest release

proddy, 2019-01-13:
> @proddy <https://github.com/proddy>,Do you have instruction for build
> environment for windows?now I use linux environment,I want to build use
> windows environment.This is my first time to develop base ESP8266 use
> visual studio code.

Yes it’s described step by step in my readme

muzixiaosanshi, 2019-01-13:
Ok,I will read your readme carefully,Thanks.

closed 2019-01-13 by muzixiaosanshi