emsesp/EMS-ESP #37: Buderus RC300 support ?

S-Przybylski, 2019-01-15:
**Describe the solution you'd like**
Does somebody know if Buderus RC300 (see http://documents.buderus.com/download/pdf/file/6720804905.pdf) is also supported by the project? It seems that this module uses EMS plus.

proddy, 2019-01-16:
EMS Plus isn't supported, but if there are more people then it's not too difficult to implement. So far I think you're the second that has shown interest

tibi03, 2019-01-20:
i also have a RC310 https://www.buderus.de/de/produkte/catalogue/alle-produkte/102256_logamatic-rc310 , from my first tests only the date and time are read from the thermostat and for the boiler (model: LogamaxPlus GB172-30 iKW H https://www.buderus.com/ro/ro/ocs/produse/logamax-plus-gb172i-757615-p/ ) not all the parameters are available , maybe some new types are used . thanks

tibi03, 2019-01-20:
with a version from October 2018 :

(00:15:33) EMS-ESP-Boiler system stats:
(00:15:33)   System Logging is set to None
(00:15:33)   Thermostat is enabled, Poll is disabled, Shower timer is enabled, Shower alert is disabled
(00:15:33)   EMS Bus Stats: Connected=yes, # Rx telegrams=0, # Tx telegrams=87, # Crc Errors=1701, Rx Status=idle, Tx Status=idle, Last Tx Action=read
(00:15:33)
(00:15:33) Boiler stats:
(00:15:33)   Warm Water activated: ?
(00:15:33)   Warm Water circulation pump available: ?
(00:15:33)   Warm Water selected temperature: ? C
(00:15:33)   Warm Water desired temperature: ? C
(00:15:33)   Warm Water current temperature: 36.4 C
(00:15:33)   Warm Water # starts: 207 times
(00:15:33)   Warm Water active time: 7 days 18 hours 15 minutes
(00:15:33)   Warm Water 3-way valve: on
(00:15:33)   Selected flow temperature: 0 C
(00:15:33)   Current flow temperature: 39.5 C
(00:15:33)   Return temperature: ? C
(00:15:33)   Gas: on
(00:15:33)   Boiler pump: on
(00:15:33)   Fan: on
(00:15:33)   Ignition: off
(00:15:33)   Circulation pump: off
(00:15:33)   Burner selected max power: 67 %
(00:15:33)   Burner current power: 46 %
(00:15:33)   Flame current: 24.7 uA
(00:15:33)   System pressure: 0.0 bar
(00:15:33)   Outside temperature: 2.5 C
(00:15:33)   Boiler temperature: ? C
(00:15:33)   Pump modulation: 100 %
(00:15:33)   Burner # restarts: 114 times
(00:15:33)   Total burner operating time: 37 days 19 hours 54 minutes
(00:15:33)   Total heat operating time: 30 days 1 hours 39 minutes
(00:15:33)
(00:15:33) Thermostat stats:
(00:15:33)   Thermostat time is 19:05:00 20/1/2019
(00:15:33)   Setpoint room temperature: 0.0 C
(00:15:33)   Current room temperature: 0.0 C
(00:15:33)   Mode is set to ?
(00:15:33)
(00:15:33) Shower stats:
(00:15:33)   Hot water is stopped
(00:15:33)   Shower is off

tibi03, 2019-01-21:
i want to thank you for the great work at this project. i was able to read some parameters in latest version after forced the boiler and thermostat models : 

Boiler stats:
  Boiler type: MC10/UBA3 Boiler [TypeID 0x08] Product ID:123
  Hot tap water is off
  Central Heating is off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water is set to Comfort
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 38.7 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 9995 times
  Warm Water active time: 7 days 19 hours 20 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 52 C
  Current flow temperature: 49.0 C
  Return temperature: ? C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 50 %
  Burner current power: 17 %
  Flame current: 14.1 uA
  System pressure: 0.0 bar
  Current System Service Code: -H
  Outside temperature: 2.5 C
  Boiler temperature: ? C
  Pump modulation: 25 %
  Burner # restarts: 12743 times
  Total burner operating time: 38 days 3 hours 51 minutes
  Total heat operating time: 30 days 8 hours 31 minutes

Thermostat stats:
  Thermostat type: RC35 (or compatible [TypeID 0x10] Product ID:86
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 10:22:50 21/1/2019
  Mode is set to ?

 Shower stats:
  Shower Timer is off

i can help with with tests on RC310 and boiler

tibi03, 2019-01-23:
i assume this is the prod id for RC310:

(01:15:26) Thermostat -> me, type 0x02 telegram: 10 0B 02 00 9E 12 03 00 FF 00 00 00 00 (CRC=9A), #data=9
<--- Version(0x02) received
Unrecognized device found. Product ID 158, Version 18.03

> (01:15:21) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 01 96 80 00 31 00 00 01 00 00 2C 6A 00 27 90 00 80 00 (CRC=05), #data=19
<--- UBAMonitorWWMessage(0x34) received
(01:15:23) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 63 (CRC=FF), #data=1
(01:15:23) Boiler -> me, type 0x02 telegram: 08 0B 02 00 7B 06 08 00 00 00 00 00 00 03 (CRC=A0), #data=10
<--- Version(0x02) received
Boiler recognized. Model MC10/UBA3 Boiler with TypeID 0x08, Product ID 123, Version 06.08
(01:15:23) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
(01:15:23) Sending read of type 0x02 to 0x09: telegram: 0B 89 02 00 63 (CRC=F7), #data=1
(01:15:23) 0x09 -> me, type 0x02 telegram: 09 0B 02 00 7D 03 03 (CRC=C3), #data=3
<--- Version(0x02) received
Boiler recognized. Model BC25 Base Controller with TypeID 0x09, Product ID 125, Version 03.03
(01:15:25) Sending read of type 0x02 to 0x0B: telegram: 0B 8B 02 00 63 (CRC=E7), #data=1
(01:15:25) Sending read of type 0x02 to 0x0B: telegram: 0B 8B 02 00 63 (CRC=E7), #data=1
(01:15:26) Sending read of type 0x02 to 0x10: telegram: 0B 90 02 00 63 (CRC=3F), #data=1
(01:15:26) Thermostat -> me, type 0x02 telegram: 10 0B 02 00 9E 12 03 00 FF 00 00 00 00 (CRC=9A), #data=9
<--- Version(0x02) received
Unrecognized device found. Product ID 158, Version 18.03
(01:15:26) Sending read of type 0x02 to 0x11: telegram: 0B 91 02 00 63 (CRC=37), #data=1
(01:15:27) Sending read of type 0x02 to 0x11: telegram: 0B 91 02 00 63 (CRC=37), #data=1
(01:15:27) Sending read of type 0x02 to 0x17: telegram: 0B 97 02 00 63 (CRC=07), #data=1
(01:15:28) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
(01:15:28) Thermostat -> Boiler, type 0x23 telegram: 10 08 23 00 00 00 00 (CRC=7F), #data=3
l(01:15:29) Sending read of type 0x02 to 0x17: telegram: 0B 97 02 00 63 (CRC=07), #data=1
(01:15:29) Sending read of type 0x02 to 0x18: telegram: 0B 98 02 00 63 (CRC=7F), #data=1

proddy, 2019-01-23:
Thanks, I've included it in the known devices. To implement EMS Plus I need some more information and message types and format. I can't seem to find anything out there. Any help in searching...?

viimD, 2019-01-23:
Hello proddy. Thanks to bbqkees I have your code running for my Neft Enviline warmtepomp. The attached thermostat is the 'Nefit thermostaat moduline 1010'. The heatpump has an EMS+ bus and it would be very nice to be able to include these messages. 
When I enter 'info' I get no hits. When I enter 'log v' I see a lot of commands. A few I've already figured out: Room temperature, modulation and pump speed. Does this help your research?

Now I'm trying to get these values into Domoticz, but I have no clue :)

tibi03, 2019-01-23:
yes, i did a scan on all type IDs on the thermostat : 

below you can find a selection with all the responses : 

> Thermostat -> me, type 0x02 telegram: 10 0B 02 00 9E 12 03 00 FF 00 00 00 00 (CRC=9A), #data=9
	 Thermostat -> me, type 0x01 telegram: 10 0B 01 00 32 35 33 30 37 35 39 32 30 30 34 32 33 37 37 33 38 31 31 32 33 31 39 00 FF FF FF (CRC=0B), #data=27
	 Thermostat -> me, type 0x02 telegram: 10 0B 02 00 9E 12 03 00 FF 00 00 00 00 (CRC=9A), #data=9
	 Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 01 10 17 2E 08 02 00 10 FF 00 (CRC=BE), #data=11
	 Thermostat -> me, type 0x0A telegram: 10 0B 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 00 (CRC=03), #data=15
	 Thermostat -> me, type 0x33 telegram: 10 0B 33 00 00 FF 3C FB 00 19 00 03 46 00 FF 00 00 (CRC=31), #data=13
	 Thermostat -> me, type 0x35 telegram: 10 0B 35 00 11 01 00 3C (CRC=7C), #data=4
	 Thermostat -> me, type 0x36 telegram: 10 0B 36 06 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=96), #data=12
	 Thermostat -> me, type 0xA0 telegram: 10 0B A0 00 00 00 01 00 (CRC=D5), #data=4
	 Thermostat -> me, type 0xA2 telegram: 10 0B A2 00 00 00 00 00 00 (CRC=37), #data=5
	 Thermostat -> me, type 0xC2 telegram: 10 0B C2 00 10 9E 00 0E 11 41 31 31 0B F5 92 09 07 1D 18 92 09 07 1D 1E 10 9E 00 0C 31 41 31 (CRC=40), #data=27
	 Thermostat -> me, type 0xC4 telegram: 10 0B C4 00 00 (CRC=7A), #data=1

tibi03, 2019-01-23:
RC310
an unknown type ID :

(04:07:45) Thermostat -> Boiler, type 0x23 telegram: 10 08 23 00 37 64 64 (CRC=0F), #data=3

proddy, 2019-01-23:
> Hello proddy. Thanks to bbqkees I have your code running for my Neft Enviline warmtepomp. The attached thermostat is the 'Nefit thermostaat moduline 1010 (code 0x38)'. The heatpump has an EMS+ bus and it would be very nice to be able to include these messages.
> When I enter 'info' I get no hits. When I enter 'log v' I see a lot of commands. A few I've already figured out: Room temperature, modulation and pump speed. Does this help your research?
> 
> Now I'm trying to get these values into Domoticz, but I have no clue :)

@viimD could you use the latest build (1.3.2) and let me know the product ID of the Moduline 1010 so I can add to the library? Unfortunately I haven't built in support for EMS+ (or EMS2) as there is very little information on the message protocols. As for Domoticz I believe there is a special format for receiving MQTT messages. It's not really an open system so I can't help you there.

proddy, 2019-01-23:
@tibi03 looks like the RC310 can talk both EMS and EMS+. With the info you sent me I could make a start to decypher the telegrams. Could you do another run sometime and tell me the values of current temp and selected temp which will help me in reverse engineering the data.

tibi03, 2019-01-24:
i did a cap today on the ems bus will i modified some parameters on RC310 , below you can find the scenarios and in the attach you can find all the telegrams : 

[ems-cap.txt](https://github.com/proddy/EMS-ESP/files/2791042/ems-cap.txt)
> 
> 
>  auto and off  21.5 degrees in room
> 
> (00:40:32) Thermostat -> all, type 0xFF telegram: 10 00 FF 08 01 B9 00 (CRC=3C), #data=3
> (00:40:32) Thermostat -> all, type 0xFF telegram: 10 00 FF 15 01 A5 00 (CRC=EC), #data=3
> (00:40:32) Thermostat -> all, type 0xFF telegram: 10 00 FF 19 01 A5 05 (CRC=89), #data=3
> (00:40:33) Thermostat -> all, type 0xFF telegram: 10 00 FF 1B 01 A5 FF EC (CRC=2A), #data=4
> (00:40:34) Thermostat -> all, type 0xFF telegram: 10 00 FF 04 01 A5 00 00 00 30 02 2C 01 (CRC=90), #data=9
> (00:40:34) Thermostat -> all, type 0xFF telegram: 10 00 FF 20 01 A5 00 (CRC=5D), #data=3
> (00:40:36) Thermostat -> Boiler, type 0x23 telegram: 10 08 23 00 00 00 64 (CRC=1B), #data=3
> (00:40:36) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 64 (CRC=A0), #data=3
> (00:40:36) Thermostat -> all, type 0xFF telegram: 10 00 FF 03 01 A5 00 00 00 00 (CRC=D2), #data=6
> (00:40:37) Thermostat -> all, type 0xFF telegram: 10 00 FF 1A 01 A5 06 (CRC=92), #data=3
> (00:40:40) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 D8 (CRC=50), #data=4
> 
> (00:40:43) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 01 (CRC=20), #data=2
> (00:40:47) Thermostat -> all, type 0xA2 telegram: 10 00 A2 00 00 00 00 00 00 (CRC=CA), #data=5
> 
> 
> 
> manual and off 21.5 degrees in room
> 
> (00:46:29) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
> 
> (00:46:38) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
> 
> (00:46:43) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 01 (CRC=20), #data=2
> (00:46:46) Thermostat -> all, type 0xF7 telegram: 10 00 F7 00 FF 01 B9 A1 0D (CRC=81), #data=5
> (00:46:46) Thermostat -> all, type 0xFF telegram: 10 00 FF 19 01 A5 01 (CRC=8D), #data=3
> (00:46:46) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 B9 00 (CRC=7C), #data=3
> (00:46:47) Thermostat -> all, type 0xFF telegram: 10 00 FF 08 01 A5 05 A0 (CRC=A2), #data=4
> (00:46:47) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
> (00:46:48) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 67 00 00 (CRC=AB), #data=4
> (00:46:48) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02), #data=6
> (00:46:49) Boiler -> all, type 0x1C telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4B), #data=25
> (00:46:49) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
> (00:46:49) Thermostat -> all, type 0xFF telegram: 10 00 FF 07 01 A5 00 05 A0 00 (CRC=86), #data=6
> 
> 
> 
> manual off 10 degrees  21.5 degrees in room
> 
> (00:50:12) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 B9 00 (CRC=7C), #data=3
> (00:50:13) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6), #data=13
> (00:50:14) Thermostat -> all, type 0xFF telegram: 10 00 FF 0D 01 A5 02 22 01 53 (CRC=B0), #data=6
> (00:50:16) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
> (00:50:16) Thermostat -> all, type 0xFF telegram: 10 00 FF 03 01 A5 14 00 00 14 14 (CRC=D8), #data=7
> (00:50:16) Thermostat -> all, type 0xFF telegram: 10 00 FF 0A 01 B9 14 (CRC=38), #data=3
> (00:50:17) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 D9 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=D6), #data=21
> (00:50:17) Thermostat -> all, type 0xFF telegram: 10 00 FF 15 01 A5 02 (CRC=EE), #data=3
> 
> 
> 
> manual off 20 degrees  21.5 degrees in room
> 
> 
> (00:54:26) Thermostat -> all, type 0xFF telegram: 10 00 FF 03 01 A5 28 11 00 28 28 (CRC=FF), #data=7
> (00:54:26) Thermostat -> all, type 0xFF telegram: 10 00 FF 0A 01 B9 28 (CRC=04), #data=3
> (00:54:27) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 01 0A 18 2B 18 03 00 10 FF 00 (CRC=3C), #data=11
> <--- RCTime(0x06) received
> (00:54:28) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 B9 00 (CRC=7C), #data=3
> (00:54:29) Thermostat -> all, type 0xFF telegram: 10 00 FF 0A 01 A5 02 (CRC=16), #data=3
> 
> 
> 
> manual on 25 degrees  22 degrees in room
[... 126 lines cut ...]
> (01:21:09) Thermostat -> all, type 0xFF telegram: 10 00 FF 03 01 A5 2C 39 00 2C (CRC=63), #data=6
> (01:21:13) Thermostat -> all, type 0xFF telegram: 10 00 FF 1B 01 A5 FF F4 (CRC=32), #data=4
> (01:21:14) Thermostat -> all, type 0xFF telegram: 10 00 FF 04 01 A5 29 (CRC=4D), #data=3
> 
> 
> (01:22:10) Thermostat -> all, type 0xFF telegram: 10 00 FF 0D 01 A5 02 02 01 73 (CRC=10), #data=6
> 
> (01:22:25) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 29 64 64 (CRC=CC), #data=3
>  SetPoint=41, hk_power=100 ww_power=100
> (01:22:25) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 67 00 00 (CRC=AB), #data=4
> (01:22:25) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02), #data=6
> (01:22:44) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 01 (CRC=20), #data=2
> (01:22:54) Thermostat -> all, type 0xFF telegram: 10 00 FF 1B 01 A5 FF F6 (CRC=30), #data=4
[... cut ...]
>  SetPoint=41, hk_power=100 ww_power=100
> (01:23:41) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 67 00 00 (CRC=AB), #data=4
> (01:23:42) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 E1 21 2C 29 00 2C 30 02 01 05 01 03 02 01 01 74 00 00 11 01 03 08 CF 00 (CRC=AD), #data=27
> (01:23:42) Thermostat -> all, type 0xFF telegram: 10 00 FF 19 01 A5 08 04 FF F6 00 00 FF 64 3C 00 3C 01 FF 01 02 (CRC=4F), #data=17
> (01:23:42) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02), #data=6
> 
>

tibi03, 2019-01-24:
room temp not sure if it is in this messages  10 00 FF 00 01 A5 00 XX  only last byte changed over time as my window was opened and after that was closed   : 

lowest value 0xB2 : 17.8 C ???    0xC8 : 20 C ???      0xCC: 20.4 C ????

on RC310 i can see only values 0.5 degree not 0.1 degree 

> (06:42:31) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CE 21 00 00 00 00 30 00 C2 01 01 03 00 C2 02 B3 00 00 11 01 00 08 0A 00 (CRC=86), #data                                                           =27
	Line 188: (06:42:49) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CD (CRC=45), #data=4
	Line 219: (06:44:31) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CC (CRC=44), #data=4
	Line 227: (06:45:08) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CC 21 00 00 00 00 30 00 BF 01 01 03 00 BF 02 B6 00 00 11 01 00 07 F6 00 (CRC=FC), #data                                                           =27
	Line 242: (06:45:56) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CB (CRC=43), #data=4
	Line 268: (06:47:24) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CA (CRC=42), #data=4
	Line 272: (06:47:41) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CA 21 00 00 00 00 30 00 BD 01 01 03 00 BD 02 B8 00 00 11 01 00 07 E2 00 (CRC=23), #data                                                           =27
	Line 288: (06:48:38) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C9 (CRC=41), #data=4
	Line 372: (06:50:14) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C9 21 00 00 00 00 30 00 BA 01 01 03 00 BA 02 BB 00 00 11 01 00 07 D7 00 (CRC=73), #data                                                           =27
	Line 375: (06:50:45) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C8 (CRC=40), #data=4
	Line 407: (06:52:26) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C7 (CRC=4F), #data=4
	Line 411: (06:52:49) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C7 21 00 00 00 00 30 00 B8 01 01 03 00 B8 02 BD 00 00 11 01 00 07 C8 00 (CRC=37), #data                                                           =27
	Line 444: (06:54:18) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C6 (CRC=4E), #data=4
	Line 462: (06:55:23) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C6 21 00 00 00 00 30 00 B5 01 01 03 00 B5 02 C0 00 00 11 01 00 07 B8 00 (CRC=52), #data                                                           =27
	Line 471: (06:56:11) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C5 (CRC=4D), #data=4
	Line 498: (06:57:19) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C4 (CRC=4C), #data=4
	Line 504: (06:57:58) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C4 21 00 00 00 00 30 00 B3 01 01 03 00 B3 02 C2 00 00 11 01 00 07 AD 00 (CRC=B4), #data                                                           =27
	Line 537: (06:59:36) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C3 (CRC=4B), #data=4
	Line 557: (07:00:32) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C3 20 00 00 00 00 30 00 B0 01 01 03 00 B0 02 C5 00 00 11 01 00 07 A2 00 (CRC=BC), #data                                                           =27
	Line 566: (07:01:15) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C2 (CRC=4A), #data=4
	Line 599: (07:03:05) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C2 21 00 00 00 00 30 00 AD 01 01 03 00 AD 02 C8 00 00 11 01 00 07 90 00 (CRC=F2), #data                                                           =27
	Line 603: (07:03:07) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C1 (CRC=49), #data=4
	Line 629: (07:05:08) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C0 (CRC=48), #data=4
	Line 642: (07:05:41) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 C0 21 00 00 00 00 30 00 AB 01 01 03 00 AB 02 CA 00 00 11 01 00 07 84 00 (CRC=16), #data                                                           =27
	Line 670: (07:07:25) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 BF (CRC=37), #data=4
	Line 684: (07:08:18) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 BF 21 00 00 00 00 30 00 A8 01 01 03 00 A8 02 CD 00 00 11 01 00 07 79 00 (CRC=E4), #data                                                           =27
	Line 718: (07:10:14) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 BE (CRC=36), #data=4
	Line 730: (07:10:49) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 BE 21 00 00 00 00 30 00 A6 01 01 03 00 A6 02 CF 00 00 11 01 00 07 6F 00 (CRC=86), #data                                                           =27
	Line 776: (07:13:22) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 BE 21 00 00 00 00 30 00 A3 01 01 03 00 A3 02 D2 00 00 11 01 00 07 68 00 (CRC=C5), #data                                                           =27
	Line 780: (07:13:24) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 BD (CRC=35), #data=4
	Line 816: (07:15:55) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 BD 21 00 00 00 00 30 00 A1 01 01 03 00 A1 02 D4 00 00 11 01 00 07 62 00 (CRC=E7), #data                                                           =27
	Line 832: (07:16:26) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 BC (CRC=34), #data=4
[... cut ...]
	Line 2464: (08:51:35) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CA 21 31 3A 00 31 30 00 41 05 01 03 00 41 03 34 00 00 11 01 03 07 E8 00 (CRC=DB), #data=27
	Line 2467: (08:51:55) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CB (CRC=43), #data=4
	Line 2517: (08:54:13) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CB 21 31 3A 00 31 30 00 3E 05 01 03 00 3E 03 37 00 00 11 01 03 07 F1 00 (CRC=78), #data=27
	Line 2531: (08:54:59) Thermostat -> all, type 0xFF telegram: 10 00 FF 00 01 A5 00 CC (CRC=44), #data=4

viimD, 2019-01-25:
> @viimD could you use the latest build (1.3.2) and let me know the product ID of the Moduline 1010 so I can add to the library? Unfortunately I haven't built in support for EMS+ (or EMS2) as there is [...]

Sorry it took me some time to figure out how to upload the new build (last one bbqkees did for me) but I succeeded! How can I see the Product ID of the Moduline 1010? It sends signals with 0x38, does that help?

@tibi03 at my heatpump the thermostat and the screen of the heatpump also only show temperature values of 0.5 degrees. However, the app that I can use from Nefit does show the room temperature in 0,1 degrees!

tibi03, 2019-01-25:
hello, i did some testes and i can see the temp 0.1 degrees in the menu of RC310 not on the main display, so i figured it out that the telegram with this pattern should be the room temp : 

10 00 FF 00 01 A5 0z xx    z= 1 and xx = 00  25.6 C

10 00 FF 00 01 A5 0z xx    z= 0 and xx = D6  21.4 C

@all who have RC300/RC310 (maybe RC200 e dont know ):  please can you try to do a capture and compare the readings ?

it is also included in a longer one probably with more parameters : 

10 00 FF 00 01 A5 00 D7 (CRC=5F), #data=4
10 00 FF 00 01 A5 00 D7 21 00 00 00 00 30 01 84 01 01 03 01 84 01 F1 00 00 11 01 00 08 63 00 (CRC=CC), #data=27
10 00 FF 00 01 A5 00 D7 21 32 3C 00 32 32 05 A0 02 01 03 01 81 01 F4 00 00 11 01 02 08 64 00 (CRC=0A), #data=27
10 00 FF 00 01 A5 00 D7 21 30 38 00 30 30 05 A0 02 01 03 01 7F 01 F6 00 00 11 01 02 08 64 00 (CRC=2B), #data=27
10 00 FF 00 01 A5 00 D7 21 31 3B 00 31 31 05 A0 02 01 03 01 7C 01 F9 00 00 11 01 02 08 64 00 (CRC=34), #data=27
10 00 FF 00 01 A5 00 D7 21 2E 33 00 2E 2E 05 A0 02 01 03 01 79 01 FC 00 00 11 01 02 08 64 00 (CRC=0E), #data=27
10 00 FF 00 01 A5 00 D7 21 2E 33 00 2E 2E 05 A0 02 01 03 01 76 01 FF 00 00 11 01 02 08 66 00 (CRC=0F), #data=27

tibi03, 2019-01-26:
@viimD great , i will capture more data and try to find more parameters , it is not so hard to understand EMS+ .

proddy, 2019-01-27:
> Sorry it took me some time to figure out how to upload the new build (last one bbqkees did for me) but I succeeded! How can I see the Product ID of the Moduline 1010? It sends signals with 0x38, does that help?

@viimD set logging to basic 'log b' and do an 'autodetect'. Hopefully it will discover your ems devices and show the product id and version. Then we can add to `ems_devices.h`

viimD, 2019-01-27:
@proddy Does not dicover any device...
autodetect
Scanning EMS bus for devices. This may take a few seconds...
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
Requesting type Version(0x02) from dest 0x10
Requesting type Version(0x02) from dest 0x11
Requesting type Version(0x02) from dest 0x17
Requesting type Version(0x02) from dest 0x18
Requesting type Version(0x02) from dest 0x21

I'm sure (from the bbqkees arduino script) that my thermostat is 0x38. I guess this destination is not included in the 'autodetect'?

proddy, 2019-01-27:
@viimD yes, sorry, its not included yet. Can you do a 'thermostat read 2' instead?

suncarver, 2019-01-27:
Hi, I'm using proddy's excellent ESP code on a Wemos D1 mini with my Buderus GB172 and RC310. bbqkees helped me to get the gateway up and running.

APP_VERSION "1.3.1" autodetect shows:

**EMS Bus nodes   found: | TypeID | Product ID | Version**
Boiler found. Model Nefit Trendline | 0x08 | 123 | 06.08    ??????? I do not have a Nefit component afaik ??           
Boiler found. Model BC25 Base Controller | 0x09 | 125 | 03.03
Thermostat found. Model RC310 | 0x10 | 158 | 18.03

Unclear: I have BC25 and have telegrams from and to 0x08, but to telegrams from and to 0x09.

I have a GB172-14 boiler (BC25 inside acc. spec) and an RC310 thermostat. RC310 seems to talk EMS as well as EMS+.
All boiler messages are provided to MQTT as far as I can see. For RC310, I tried to interprete the EMS+ telegrams - those with type 0xFF.
When I filter all the 0xFF type telegrams, I find those with 0x01 and 0xA5 (EMS+ telegrams seem to have 2 Bytes):
[image]

See [https://www.mikrocontroller.net/topic/309075?goto=3928825#3928825](url) which gave the hint.

In my case, it seems to work. The code must be extended to deal with 0xFF and reading 2 Byte telegrams types. A first step is to have uin16_t instead of uint8_t for the telegram type and adjust headerlength from 4 to 6 in case of EMS+ telegrams. This leads to change of data start byte. 

See also comments in [http://www.domoticz.com/forum/viewtopic.php?f=38&t=14132&p=202036&hilit=RC310#p202036](url)

Good to see interest in RC310 and EMS+ telegrams here!

proddy, 2019-01-27:
@suncarver I thought the product id's were manufacturer unique, but it seems those dutch Nefit trendlines are just a rebrand of the German Buderus Logamax GB172. It's all trial and error so I've never really sure.

I've been following your ems+ progress on the Domoticz forum, nice work. I think we're close to coding the messages.

suncarver, 2019-01-27:
Hi proddy,  "It's all trial and error so I've never really sure."   Of course, thats clear to me and I really appreciate to have a forum here to collect all the information to learn and improve. I'd like to thank you for this community work across various products and brands. So if I can provide more info, let me know..

viimD, 2019-01-27:
@proddy I did a 'thermostat read 2' but only see this:
log b
System Logging set to Basic
thermostat read 2
Requesting type Version(0x02) from dest 0x00

any other options?

suncarver, 2019-01-27:
@tibi03 : as you have also an RC310, I'd like to comment on your message with my finding

> 10 00 FF 00 01 A5 00 D7 21 00 00 00 00 30 01 84 01 01 03 01 84 01 F1 00 00 11 01 00 08 63 00 (CRC=CC), #data=27

10 - sent from RC310
00 to all = broadcast
FF indicator for EMS+ telegram -> hint to interprete 01 A5 later
00 offset (important for me to react only to messages which have offset 0x00 
               -there are more messages 01 A5 which have an offset.
01 - high Byte of the EMS+ telegram
A5 - low Byte of the EMS+ telegram so 0x01 << 8 + 0xA5 ==> 0x01A5 into an uint16_t (instead uint8_t)
              01 could be HK heating circle 1 - I read RC310 can now support many components. to 
              be verified - I have no 0x02A5 messages
0x00 and 0xD7 -> room temp (from 2 Bytes). 
              EMS_Thermostat.curr_roomTemp = _toFloat(EMS_TYPE_RC310StatusMessage_curr, data);
0x21  -> assumed night temp set point
0x00  -> assumed set point day (fits for me - 0x00 is not plausible for my assumtion)

proddy, 2019-01-27:
> @proddy I did a 'thermostat read 2' but only see this:
> log b
> System Logging set to Basic
> thermostat read 2
> Requesting type Version(0x02) from dest 0x00
> 
> any other options?

type 'send 0b b8 02 63'

note: as your thermostat 0x38 is not recognized, you could have added it yourself to my_config.h

viimD, 2019-01-27:
Hmm, no result either. 

Sorry for my n00b questions, but I used your prebuilt firmware. Then adding my thermostat is no option i guess..?

proddy, 2019-01-27:
> Hmm, no result either.
> 
> Sorry for my n00b questions, but I used your prebuilt firmware. Then adding my thermostat is no option i guess..?

that's strange. @tibi03 does ''send 0b b8 02 63' give anything back for you? I think you both have a RC310 right?

viimD, 2019-01-27:
I have the nefit moduline 1010h https://www.warmteservice.nl/Verwarming/Warmtepompen/Warmtepomp-toebehoren/Nefit-Moduline-1010H-warmtepomp-kamerthermostaat/p/03553075

That is a really basic heatpump thermostat

tibi03, 2019-01-28:
@proddy RC310 has address 0x10, i dont have a device with 0x38 address

S-Przybylski, 2019-02-09:
I set up my device today. I have a Buderus GB172-20 with RC300 and a solar thermie (module SM100).
Login:
...
Boiler type device found. Model Buderus GB172/Nefit Trendline with TypeID 0x08,                                                                                                                                                              Product ID 123, Version 04.09
* Setting Boiler type to Model Buderus GB172/Nefit Trendline, TypeID 0x08, Produ                                                                                                                                                             ct ID 123, Version 04.09

* Connected to: EMS-ESP version 1.5.2
* ESP8266 is in AP mode with SSID ems-esp
* Free RAM:17 KB, Load:0%

Info:
info
EMS-ESP System stats:
  System logging set to None
  LED is off
  # connected Dallas temperature sensors=0
  Thermostat is enabled, Boiler is enabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus Stats:
  Bus Connected=yes, # Rx telegrams=59, # Tx telegrams=0, # Crc Errors=10

Boiler stats:
  Boiler type: Buderus GB172/Nefit Trendline [Type ID: 0x08] Product ID:123 Version:04.09
  Hot tap water is off
  Central Heating is active
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water is set to Comfort
  Warm Water selected temperature: 51 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 44.59 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 59453 times
  Warm Water active time: 244 days 19 hours 4 minutes
  Warm Water 3-way valve: on
  Selected flow temperature: 81 C
  Current flow temperature: 64.19 C
  Return temperature: ? C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 123 %
  Burner current power: 25 %
  Flame current: 17.79 uA
  System pressure: 0.0 bar
  Current System Service Code: =H
  Heating temperature setting on the boiler: 70 C
  Boiler circuit pump modulation max. power: 100 %
  Boiler circuit pump modulation min. power: 10 %
  Outside temperature: 9.50 C
  Boiler temperature: ? C
  Pump modulation: 100 %
  Burner # restarts: 201944 times
  Total burner operating time: 536 days 5 hours 9 minutes
  Total heat operating time: 291 days 10 hours 5 minutes
  Total UBA working time: 1570 days 4 hours 46 minutes

Thermostat stats:
  Thermostat type: RC310 [Type ID: 0x10] Product ID:158 Version:11.07
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 16:40:45 9/2/2019
  Mode is set to ?

Shower stats:
  Shower Timer is off
Read operations not yet supported for this model thermostat
------------------
[... 377 lines cut ...]
<--- UBAMonitorWWMessage(0x34) received
(01:06:33) Boiler -> all, type 0x18 telegram: 08 00 18 00 29 01 13 64 00 01 01 A0 40 80 00 01 EB 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=46), #data=25
<--- UBAMonitorFast(0x18) received
(01:06:43) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DC 00 00 80 00 00 80 00 80 00 80 00 04 (CRC=4D), #data=21
(01:06:43) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 01 EA 01 EA 21 00 05 03 00 05 61 04 00 E8 3D 00 80 00 (CRC=C3), #data=19
<--- UBAMonitorWWMessage(0x34) received
(01:06:43) Boiler -> all, type 0x18 telegram: 08 00 18 00 29 01 0E 64 00 01 01 A0 40 80 00 01 EA 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=AC), #data=25
<--- UBAMonitorFast(0x18) received
(01:06:43) Thermostat -> Boiler, type 0x07 telegram: 10 88 07 00 0E (CRC=2F), #data=1
(01:06:44) Boiler -> Thermostat, type 0x07 telegram: 08 10 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=A8), #data=13
(01:06:46) Boiler -> all, type 0x18 telegram: 08 00 18 00 29 01 0D 64 00 01 01 A0 40 80 00 01 EA 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=D4), #data=25
<--- UBAMonitorFast(0x18) received
(01:06:49) 0x30 -> all, type 0xFF telegram: 30 00 FF 00 02 66 01 62 00 12 (CRC=73), #data=6
(01:06:52) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 11 (CRC=30), #data=2
(01:06:53) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DC 00 00 80 00 00 80 00 80 00 80 00 04 (CRC=4D), #data=21
(01:06:53) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 01 EA 01 EA 21 00 05 03 00 05 61 04 00 E8 3D 00 80 00 (CRC=C3), #data=19
<--- UBAMonitorWWMessage(0x34) received
(01:06:53) Boiler -> all, type 0x18 telegram: 08 00 18 00 29 01 0C 64 00 01 01 A0 40 80 00 01 EA 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=0B), #data=25
<--- UBAMonitorFast(0x18) received
-----------------------------------
The RC300 uses only the outside temperature. I do not have any room temperature device.

It seems that some parts of the RC300 are not displayed and also the SM100 isn't recognized.
Can you please support me to get the maximum out of it?
I can provide Logs if you advice me

proddy, 2019-02-09:
Hi and welcome. The RC300 range are all EMS+ and the code doesn't currently support that yet, but we're planning to work on it soon.

proddy, 2019-04-19:
I added the basics for recognizing the RC300/RC310 and generally supporting all EMS+ types but need your help in testing and refining. Use the 1.7 build from the dev branch, thanks,

suncarver, 2019-04-28:
@proddy, @gl3nni3 and all, thanks for the RC310 extension!
I used the dev branche and can confirm 

```
Boiler found. Model Buderus GB172/Nefit Trendline (DeviceID:0x08 ProductID:123 Version:06.08)
* Setting Boiler to model Buderus GB172/Nefit Trendline (DeviceID:0x08 ProductID:123 Version:06.08)
Device found. Model BC25 Base Controller with DeviceID 0x09, ProductID 125, Version 03.03
* Setting Thermostat model to RC300/RC310 (DeviceID:0x10 ProductID:158 Version:1 
```
I get some corrupt messages:

> (00:06:03.014) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 20 78 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 8A 05 00 68 ED FF 3F 0
> (00:06:03.057) Corrupt telegram: telegram: 0B 33 00 20 (CRC=78)
> 
> (00:06:03.811) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 20 E4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 23 8D 05 00 68 ED FF 3F 0
> (00:06:03.854) Corrupt telegram: telegram: 0B 14 00 20 (CRC=E4)
> (00:06:04.045) Corrupt telegram: telegram: 0B 08 00 18 00 2C 01 B6 47 17 09 01 25 40 80 00 02 09 80 00 00 55 FF 2D 48 00 C8 00 02 00 (CRC=84), #data=26
> 
> (00:07:03.014) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 20 D4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 66 74 06 00 68 ED FF 3F 0
> (00:07:03.287) Corrupt telegram: telegram: 0B 88 18 00 20 (CRC=90)
> 
> (00:07:05.155) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 20 EC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 C3 7C 06 00 68 ED FF 3F 0
> (00:07:05.198) Corrupt telegram: telegram: 0B 16 00 20 (CRC=EC)
> 
> (00:07:05.580) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 20 E4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 6C 7E 06 00 68 ED FF 3F 0
> (00:07:05.659) Corrupt telegram: telegram: 0B 14 00 20 (CRC=E4)

Room temp, mode and hc are correct. SelTemp is shown as 20°C in the RC310 Display:

[image]

proddy, 2019-04-28:
Wonderful! A few corrupt messages here and there are ok and quite normal.
They are mainly residual noise left on the line. Can you confirm that
sending works and there’s nothing stuck in the Tx queue? (‘Queue’)

suncarver, 2019-04-29:
"Tx queue is empty".
OK, I can ignore the corrupt messages.
Problem found: The Wemos sends MQTT Messages, but stops sending after a while.

[image]

 Then telnet session can still be startet, info shows ? values. "Reboot" in telnet session and "info" shows values again.
"Publish" shows MQTT telegram in NodeRed again:
[image]

closed 2019-07-27 by proddy