emsesp/EMS-ESP #38: MQTT messaging: jsons variables not correct

Yosh01, 2019-01-16:
Hi, 
thanks for the awesome code :-)
Everything works fine excepts the mqtt processing.
I have changed ems_devices.h to fit to my needs (Using MC10/BC10 with RC35@boiler).
[image]

ems_devices.h:
`    // various boilers and buderus type devices
    {EMS_MODEL_UBA, **72**, 0x08, "MC10/UBA3 Boiler"},// MODIFIED
    {EMS_MODEL_BC10, **68**, 0x09, "BC10 Base Controller"},//MODIFIED
`  
Boiler and thermostats are read now correctly:
`Boiler stats:
  Boiler type: MC10/UBA3 Boiler [TypeID 0x08] Product ID:72
  Hot tap water is off
  Central Heating is off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water is set to Comfort
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 57.5 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 52299 times
  Warm Water active time: 225 days 19 hours 15 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 51 C
  Current flow temperature: 52.2 C
  Return temperature: ? C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 100 %
  Flame current: 52.4 uA
  System pressure: 0.0 bar
  Current System Service Code: -H
  Outside temperature: 8.1 C
  Boiler temperature: 47.0 C
  Pump modulation: 100 %
  Burner # restarts: 221937 times
  Total burner operating time: 587 days 16 hours 14 minutes
  Total heat operating time: 361 days 21 hours 49 minutes
  Total UBA working time: 3681 days 7 hours 42 minutes`

`Thermostat stats:
  Thermostat type: RC35 (or compatible [TypeID 0x10] Product ID:86
  Setpoint room temperature: 21.0 C
  Current room temperature: 3200.0 C
  Thermostat time is 13:27:12 16/1/2019
  Mode is set to auto
`
The mqtt message payload of the RC35 seems to be ok:
`{"thermostat_currtemp":"21.0","thermostat_seltemp":"21.0","thermostat_mode":"auto"}`

But the mqtt message payload of tapwater_active/boilder_data seems to be incorrect:
`{"wWSelTemp":"**100**","selFlowTemp":"**100**","outdoorTemp":"**100**","wWActivated":"**100**","wWComfort":"Comfort","wWCurTmp":"100","wWCurFlow":"100","wWHeat":"100","curFlowTemp":"100","retTemp":"100","burnGas":"100","heatPmp":"100","fanWork":"100","ignWork":"100","wWCirc":"100","selBurnPow":"100","curBurnPow":"100","sysPress":"100","boilTemp":"100","pumpMod":"100","ServiceCode":"0Y"}` 
Has anybody an idea about this issue?
Thanks a lot :-)

proddy, 2019-01-16:
Thanks for the ems_devices.h updates. I'll include them in the next build.

I see a number of issues
 1) printing out current room temp is incorrectly rendered for an RC35. @SpaceTeddy may be able to help
 2) the MQTT message for boiler_data has all value of 100? Not sure why here. I'll need to look into the code again. 
 3) same with tapwater_active. Can you show the output of the MQTT topic. Is it also '100' ?

Yosh01, 2019-01-16:
Hi, 
that was a fast response :-) Thanks!

1. this is not a real issue for me, because the RC35 is mounted on the boiler and the outdoor temperature is used. I read about the 3200 degrees issue before. 
2. Yes - both telegramms (boilder_data and tapwater_active) contains 100 as values except the Service Code (in my case it´s "0Y")
3. Enclosed some screenshots:
[image]

SpaceTeddy, 2019-01-16:
@Yosh01, may you send the telegram feedback of type 0x3d please?
`thermostat read 3d`

Yosh01, 2019-01-16:
@SpaceTeddy,
got this output:
thermostat read 3d
Requesting type RC35Set(0x3D) from dest 0x10
(07:57:50) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
(07:57:50) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 01 22 2A 0A 00 28 02 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 14 05 05 01 02 (CRC=25), #data=27
<--- RC35Set(0x3D) received

SpaceTeddy, 2019-01-16:
may you give 3E as well please?

Yosh01, 2019-01-16:
@proddy 
> same with tapwater_active. Can you show the output of the MQTT topic. Is it also '100' ?

I double checked this with another mqtt tool and tapwater_active is not delivered with the '100' in the values.
tapwater_active messages '0' and looks fine. 
The problem is only exisiting in boilder_data.

proddy, 2019-01-16:
I think the JSON is being serialized by your MQTT broker. Maybe its an option you can disabled. Or we can put the option into the code too. If you add after line 433 in ems-esp.ino `myDebug(data);` what does it show you?

Yosh01, 2019-01-16:
@SpaceTeddy 
Requesting type RC35StatusMessage(0x3E) from dest 0x10
(08:13:00) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(08:13:00) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 2A 7D 00 00 00 00 00 00 00 00 64 11 34 00 (CRC=2B), #data=16
<--- RC35StatusMessage(0x3E) received

Yosh01, 2019-01-16:
@proddy,
Probably - I use mosquitto as broker on a debian server. 
I will check it with myDebug(data); 
Reflashing is in progress ;)

SpaceTeddy, 2019-01-16:
ok, "7D 00" means, that you don't have an external temperature controller mounted. 0xFD00 / 10 is your 3200°C. Currently I don't understand why you get a real value (21°C) via mqtt.

Yosh01, 2019-01-16:
Seems to be in the code...

`Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
<--- UBAMonitorFast(0x18) received
<--- UBAMonitorFast(0x18) received
<--- RC35StatusMessage(0x3E) received
{"wWSelTemp":"100","selFlowTemp":"100","outdoorTemp":"100","wWActivated":"100","wWComfort":"Comfort","wWCurTmp":"100","wWCurFlow":"100","wWHeat":"100","curFlowTemp":"100","retTemp":"100","burnGas":"100","heatPmp":"100","fanWork":"100","ignWork":"100","wWCirc":"100","selBurnPow":"100","curBurnPow":"100","sysPress":"100","boilTemp":"100","pumpMod":"100","ServiceCode":"0H"}
Publishing boiler data via MQTT
Publishing hot water and heating state via MQTT
Publishing thermostat data via `MQTT`

proddy, 2019-01-16:
yup, something in the code. need to think about why this is happening.

Yosh01, 2019-01-16:
> ok, "7D 00" means, that you don't have an external temperature controller mounted. 0xFD00 / 10 is your 3200°C. Currently I don't understand why you get a real value (21°C) via mqtt.

Me too - looks strange...  
If I take a look into the 'info' section via telnet, actualy it shows:
`Thermostat stats:
  Thermostat type: RC35 (or compatible [TypeID 0x10] Product ID:86
  Setpoint room temperature: 17.0 C
  Current room temperature: 3200.0 C
  Thermostat time is 22:03:42 16/1/2019
  Mode is set to auto`

...but via mqtt the following is transmitted:
`{"thermostat_currtemp":"21.0","thermostat_seltemp":"21.0","thermostat_mode":"auto"}`
 
Looks like, that the info section shows the correct readings, but via mqtt omething else is delivered (both values are wrong) ...

SpaceTeddy, 2019-01-16:
I use mosquitto on raspberry pi and Mqtt.fx. in the past i had soe trouble with the mqtt "retain" flag on some messages. Are you sure that retain was not sent with this massage?

Yosh01, 2019-01-16:
> I use mosquitto on raspberry pi and Mqtt.fx. in the past i had soe trouble with the mqtt "retain" flag on some messages. Are you sure that retain was not sent with this massage?
You´re right. The messages were retained. After deleting the messages I got a payload with 17° (but for both values). I was also using Mqtt.fx (osx). In node-red the messages where overwritten and not retained...

proddy, 2019-01-16:
Ahh. That’s a mistake on my side. my_config.h needs to be changed too

#define MQTT_RETAIN false

Yosh01, 2019-01-22:
@proddy Why is this issue closed?
The retain Problem is closed (and was not the reported issue)- but the issue with the repeating 100 / 0 values in MQTT is still existing.

proddy, 2019-01-22:
my bad, sorry. I forgot about the mqtt string issue

proddy, 2019-01-22:
@Yosh01 are you using platformio or Arduino IDE to build? Can you check which version of ArduinoJson you are using? (check the library.* files)

Yosh01, 2019-01-23:
@proddy - Thanks for the hint ;) I have checked, updated, downgraded  and tried several librarie- versions in Arduino IDE without success. After I have moved over to Plattformio it works like a charm...

closed 2019-01-23 by Yosh01

proddy, 2019-01-23:
@Yosh01 great! You'll fine PlatformIO with Visual Studio Code so much better. The reason for the error is because Arduino IDE downloads the latest beta version of ArduinoJson (version 6) which is still a little buggy. PlatformIO uses the stable version 5 by default. See https://arduinojson.org/v5/doc/

Gert05, 2019-01-26:
I had the same problem (using Arduino IDE). After moving from ArduinoJson 5.11.2 to 5.13.4 it seems to have disappeared.

proddy, 2019-01-26:
@Gert05 good to know. I'll add it to the doc