emsesp/EMS-ESP #39: some findings with bosch condens 2500 boiler and Bosch CW100 thermostat

m1588, 2019-01-18:
First of all I'd like to thank you for the excellent work done!
 
I do have Bosch condens 2500 boiler (based on Heatronic 3 PCB) working with Bosch CW100 thermostat (https://www.bosch-thermotechnology.com/tr/en/ocs/null/cw-100-754594-p/). 

The boiler is broadcasting EMS telegrams from address 0x08 and the thermostat broadcasts both EMS and EMS+ telegrams from address 0x18. I have added new Boiler TypeID 0x08, Product ID:95 to ems_devices.h and it is recognised by the EMS-ESP during thermostat-boiler version exchange when thermostat is powered on: 
> (00:00:39) 0x18 -> 0x08, type 0x02 telegram: 98 88 02 00 0A (CRC=77), #data=1
> (00:00:39) 0x08 -> 0x18, type 0x02 telegram: 88 18 02 00 5F 2B 0A 00 00 00 00 00 00 00 (CRC=FD), #data=10

I would say that the version telegram from thermostat is too short, does not contain FW version and not parsed by EMS-ESP. I have attached the [initial-exchange1.txt](https://github.com/proddy/EMS-ESP/files/2772177/initial-exchange1.txt) output for your reference. It is also interesting that no version exchange is done during normal operation, so in order for boiler to be recognised after EMS-ESP reset thermostat needs to be reloaded.

After that the SW can decode following messages:
- UBAMonitorFast 
- UBAMonitorSlow
- UBATotalUptimeMessage
- UBAParameterWW

That made possible to monitor the system using home assistant which is great!

Nevertheless EMS-ESP can decode boiler broadcasts now, it keep saying that it can't access EMS bus.
I have added polling messages debug and discovered strange thing  - the boiler is polling only 0x18 address (telegram = 0x98) and no other destination are polled. 

Have you seen such situations before? Any thoughts on that?

proddy, 2019-01-18:
Thanks @m1588 for adding to our EMS database. I'll include these devices in the net version. Regarding your points
- The Thermostat is indeed EMS Plus so I'm thinking maybe I should just start adding ems+ support since more people are asking.
- The Thermostat is responding with a single 0A data byte when asked for a version, which means 'unsupported'. EMS plus has a different type range for the version details. I don't know exactly which yet.
- The logic to see if the EMS bus is connected is by checking if a poll request is sent from the boiler to our device (service key). See line #577 in `ems.cpp`. The boiler should cycle through the complete range sending out these 1 byte poll requests. What we need to do is put some extra debug print statements in your code to see what is happening, something like in ems.cpp line #575:
``
myDebug("Poll sent to 0x%02d",  (value | 0x80));
``

m1588, 2019-01-18:
Couple days ago I added following line to ems.cpp:
577:        if (value > 0x80)  { myDebug("poll request, value=0x%02X",value); };
however, as I said earlier, the only requests seen were 0x98 (thermostat polling):
poll request, value=0x98
poll request, value=0x98
poll request, value=0x98
poll request, value=0x98

E.g. no service key, no other EMS devices are expected to be polled by the bus master.

proddy, 2019-01-23:
I'm starting to see a pattern here. A few people with other boilers reported that sending (Tx) is not working. The code only sends messages when it receives a poll request to the service key (0xB8) and in your case that never arrives. I'm thinking perhaps our device has to register on the bus first, not sure. I need to do some more digging..

m1588, 2019-01-23:
I have enabled full bus dump 

`(ems.cpp#577         myDebug("**1-byte received, value=0x%02X**",value);)`

During dump start thermostat was off, after I have connected it there is kinda handshake between thermostat and boiler:

```
(00:07:02) 0x18 -> Boiler, type 0x07 telegram: 98 88 07 00 0E (CRC=67), #data=1
(00:07:02) Boiler -> 0x18, type 0x07 telegram: 88 18 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=7C), #data=14
(00:07:02) 0x18 -> Boiler, type 0xEF telegram: 98 88 EF 00 01 (CRC=E3), #data=1
```
And then version exchange follows. After that boiler start thermostat polling.

I have attached full thermostat startup dump, hope it will add some clarity to what is going on on a bus.
[initial-exchange-1byte.txt](https://github.com/proddy/EMS-ESP/files/2788445/initial-exchange-1byte.txt)

proddy, 2019-01-25:
thanks, this is really useful. Looks like type 0x07 is the handshake. So when a EMS device boots up it sends a single byte data 0x0E to this message, and when it receives sends a type 0x07 acknowledges with a 0x01.

Could you try a 'send B8 88 07 00 0E' to see if this activates the polls to 0xB8 ? If not I will need to implement a handler for the 0x07 message type.

m1588, 2019-01-25:
@proddy, unfortunately I am on vacation now, will do that in a week.
BTW, will the send work when the firmware thinks it has no access to EMS BUS? (Error! Unable to connect to EMS bus. Check connection and make sure you're not in DEBUG_SUPPORT mode. Retrying in 10 seconds...)
If I am not mistaken, the bus write requests are never taken out of the queue untill bus master successfully poll us.

proddy, 2019-01-25:
that's correct, if the code doesn't receive any successful telegrams (i.e. packages with correct CRC) then it assumes the bus has no connection and sending will be suspended. I've improved this logic in v1.3.2. The sends will only work when the boiler/master sends a poll to our device and it will take a message a time out of the queue.

proddy, 2019-01-26:
also try a 'send b8 88 02 00 01' which sends the version of out device to the boiler

m1588, 2019-02-03:
I have updated firmware to latest 1.4.1 version. After I have set boiler and thermostat types to 8 and 18 some datagrams started to be recognised (e.g. UBAMonitorFast, UBAMonitorWWMessage, RCTime from thermostat).
However, as we are not polled by the boiler, bus status is:
```
EMS Bus Stats:
  Bus Connected=no, Tx Capable=yes, # Rx telegrams=0, # Tx telegrams=0, # Crc Errors=0

```
So, after I am issuing 'send b8 88 02 00 01' it gets stuck in tx queue and seems to be never sent.
Any suggestions on how to forcibly sent telegram even when core thinks "Bus Connected=no"?

m1588, 2019-02-03:
Started to look into code, and it seems there is a typo in ems.cpp line 279 (ems_getTxCapable value displayed as Bus Connected and vice versa):
```
    myDebug("  Bus Connected=%s, Tx Capable=%s, # Rx telegrams=%d, # Tx telegrams=%d, # Crc Errors=%d",
            (ems_getTxCapable() ? "yes" : "no"),
            (ems_getBusConnected() ? "yes" : "no"),
            EMS_Sys_Status.emsRxPgks,
            EMS_Sys_Status.emsTxPkgs,
            EMS_Sys_Status.emxCrcErr);

```

proddy, 2019-02-03:
Could you try the same commands with the latest 1.5.0 release?

m1588, 2019-02-03:
It seems that something is broken on 1.5.0 version. The board is reloading some time after starts and telmet console saying:

> Unknown command. Use ? for help.
> info
> Unknown command. Use ? for help.
> info
> Unknown command. Use ? for help.
> info
> Unknown command. Use ? for help.
> ?Unknown command. Use ? for help.
>

m1588, 2019-02-03:
got that fixed with set serial off.
However, the messages from queue are not sent (probably because we are not polled):

> Tx queue (14/50)
>  [1] action=read dest=0x08 type=0x02 offset=0 length=6 dataValue=99 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:00:02)
>  [2] action=read dest=0x18 type=0x02 offset=0 length=6 dataValue=99 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:00:02)
>  [3] action=read dest=0x08 type=0x18 offset=0 length=6 dataValue=99 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:01:00)
>  [4] action=read dest=0x08 type=0x19 offset=0 length=6 dataValue=99 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:01:00)
>  [5] action=read dest=0x08 type=0x33 offset=0 length=6 dataValue=99 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:01:00)
>  [6] action=read dest=0x08 type=0x16 offset=0 length=6 dataValue=99 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:01:00)
>  [7] action=read dest=0x08 type=0x14 offset=0 length=6 dataValue=99 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:01:00)
>  [8] action=? dest=0x08 type=0x02 offset=0 length=6 dataValue=0 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:01:13)
>

proddy, 2019-02-03:
yes I understand the problem, just not sure why. The polling is important since its a single half-duplex bus used by all devices so all comms have to wait their turn.

can you try sending messages to the boiler using type 0x07 ?
`send  0b 88 07 00 01` 

and if that doesn't work, type 0x02 to tell the boiler our own version number
`send 0b 88 02 00 01`

the line you need to modify is #566 in ems.cpp

I'll start using the 'dev' branch of my github repo for these EMS+ changes.

m1588, 2019-02-04:
I managed to make FW sent telegrams you refered to, however the boiler does not react to them.

I even tried to simulate thermostat handshake - but again, no luck.
I have also tried to change 

>         if (value == (EMS_ID_ME | 0x80)) {
to
>         if (value == EMS_ID_ME) {

I have enabled 1-byte message debug, and captured thermostat startup exchange.
The thermostat start write after 0x18 1-byte request appears on bus. I have tried to simulate that behaviour with thermostat turned off, but boiler does not reply to my sends:
```
send 98 88 07 00 0E
send 18 88 07 00 0E
```
I have attached thermostat startup exchange dump, hope that will help.
[boiler-thermostat-startup.txt](https://github.com/proddy/EMS-ESP/files/2829247/boiler-thermostat-startup.txt)

proddy, 2019-02-04:
one more try, this time using Write commands to the boiler from our device:

`send 8b 88 07 00 01`

or if that didn't work:

`send 8b 88 02 00 01`

m1588, 2019-02-04:
tried different variants - no luck:

```
(00:00:33) Sending raw telegram: 8B 88 07 00 01 (CRC=41), #data=1
(00:00:35) Sending raw telegram: 8B 88 02 00 01 (CRC=55), #data=1
(00:00:36) Boiler -> all, type 0x18 telegram: 88 00 18 00 23 01 5F 46 2A 09 33 25 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (CRC=AB), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:36) Boiler -> all, type 0x34 telegram: 88 00 34 00 32 01 D0 01 D0 81 00 00 03 00 00 0E 66 00 01 2E 00 (CRC=B8), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:00:37) Sending raw telegram: 0B 88 07 00 01 (CRC=89), #data=1
(00:00:38) Sending raw telegram: 0B 88 02 00 01 (CRC=9D), #data=1
(00:00:40) Sending raw telegram: 8B 08 07 00 01 (CRC=25), #data=1
(00:00:41) Sending raw telegram: 8B 08 02 00 01 (CRC=31), #data=1
(00:00:43) Sending raw telegram: 0B 08 07 00 01 (CRC=ED), #data=1
(00:00:44) Sending raw telegram: 0B 08 02 00 01 (CRC=F9), #data=1
(00:00:46) Boiler -> all, type 0x07 telegram: 88 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=6F), #data=15
(00:00:46) Boiler -> all, type 0x18 telegram: 88 00 18 00 23 01 5D 46 29 09 33 25 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (CRC=12), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:46) Boiler -> all, type 0x34 telegram: 88 00 34 00 32 01 CF 01 CF 81 00 00 03 00 00 0E 66 00 01 2E 00 (CRC=F3), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:00:46) Boiler -> all, type 0x19 telegram: 88 00 19 00 00 06 80 00 80 00 FF FF 00 64 00 06 CF 00 70 E9 00 00 00 00 62 82 00 05 A1 80 00 (CRC=45), #data=27
<--- UBAMonitorSlow(0x19) received
(00:00:47) Sending raw telegram: 0B 08 07 00 01 (CRC=ED), #data=1
(00:00:48) Sending raw telegram: 0B 08 02 00 01 (CRC=F9), #data=1
(00:00:56) Boiler -> all, type 0x18 telegram: 88 00 18 00 23 01 5F 46 29 09 33 25 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 02 18 (CRC=B5), #data=25
<--- UBAMonitorFast(0x18) received
(00:00:56) Boiler -> all, type 0x34 telegram: 88 00 34 00 32 01 CF 01 CF 81 00 00 03 00 00 0E 66 00 01 2E 00 (CRC=F3), #data=17
<--- UBAMonitorWWMessage(0x34) received

```

proddy, 2019-02-04:
hmm, kinda running out of ideas. if you power the thermostat off and back on, can you see capture the very first telegrams? I think you did this before but please try again. We need to figure out this handshake

curantil, 2019-02-26:
I'm not sure this is the exact same issue, but I'll report my findings.

For me the first message the thermostat (CT200) sends out at startup is `18 88 EF 00 01` which is immediately responded to by the Bosch Easy Control Adapter with `08 18 EF 00`. (CRC's not included for consitancy)

When the thermostat is disconnected and I manually do a `send 18 88 EF 00 01`no response is given. However, if I change `EMS_ID_ME` to 0x18 (that of the thermostat), and I send `18 88 EF 00 01` I get the expected reply.

With the ID on 0B I can send `0B 88 EF 00 01` and I get the expected reply of `08 0B EF 00`.

I'll try to emulate the rest of the initialization sequence.

philrich, 2019-05-01:
hi, i see you tried the same things i did - with same success: none.. see here #103. did you get any further? @curantil how did you manage to get a reply with src ID 0x0B (as mentioned in your last sentence)?

proddy, 2019-07-15:
can you guys try again with the latest 1.8 firmware and with different tx_mode settings?

m1588, 2019-07-19:
For my bosch condens 2500 the only working mode is 3, and it seems that is works fine.
I will monitor and update the thread in case tha crashes will be noticed.

proddy, 2019-07-27:
thanks. closing and we can re-open if 1.8.1 has problems with the Bosch 2500

closed 2019-07-27 by proddy

mellbo, 2019-10-29:
@m1588 , do you have logs from thermostat with start heating sequence ?
I don`t have thermostat, and I can`t see what code sending for starting heating.

Thanks in advance.

m1588, 2019-10-29:
My thermostsat is Bosch cw100 and boiler bosch condens 2500 (which uses HT3 protocol)
I have captured Set message, and it seems it has very basic format:

> #Set 16
(00:09:22.218) Thermostat -> all, type 0x01A5, telegram: 98 00 FF 03 01 A5 20 (CRC=45) #data=1
<--- RCPLUSStatusMessage_HC1(0x1A5)
#Set 20
(00:08:22.765) Thermostat -> all, type 0x01B9, telegram: 98 00 FF 0A 01 B9 28 (CRC=3D) #data=1
<--- RCPLUSSetMessage(0x1B9)
#Set 22
(00:07:34.706) Thermostat -> all, type 0x01B9, telegram: 98 00 FF 0A 01 B9 2C (CRC=39) #data=1
<--- RCPLUSSetMessage(0x1B9)
#Set 23
(00:09:50.654) Thermostat -> all, type 0x01B9, telegram: 98 00 FF 0A 01 B9 2E (CRC=3B) #data=1
<--- RCPLUSSetMessage(0x1B9)

The last byte is the desired temperature with 0.5C resolution coded with DEC2HEX(temp_in_C*2).

However trying to replicate RCPLUSSetMessage with "send 98 00 FF 0A 01 B9 32" seems does not have an effect.

mellbo, 2019-10-29:
thanks for your post this information.
do you have log for stop heating too?

i try send start/stop command to boiler without ems thermostat.

m1588, 2019-10-29:
@mellbo nope, but I will try to catch that

mellbo, 2019-10-29:
thanks in advance.

mellbo, 2019-10-29:
>ever trying to replicate RCPLUSSetMessage with "send 98 00 FF 0A 01 B9 32" seems does not have an effect.

With this datagram, I  think that, thermostat send in network just his current settings for other clients information. but i am looking for the code of starting heating.

mellbo, 2019-11-02:
hello my friend. some newz for me ?

mellbo, 2019-11-04:
@m1588  i want to ask you, because you have Bosch Condens 2500W

 with version 1.9.3, what selection you made for `tx_mode` ?  For me work only with 1 or 2.  Never with tx_mode = 3. But some people tell me that for me is normal to be tx_mode = 3 (Junkers HT3)

I try with tx_mode = 3 but TX appear with no signal.

```
EMS Bus stats:
  Bus is connected, protocol: Junkers HT3
  Rx: # successful read requests=0, # CRC errors=2
  Tx: no signal
```

mellbo, 2019-11-09:
> @mellbo nope, but I will try to catch that

can you made for one log (r) when central heating working ? I want to understand, what thermostat send to boiler when this starting heating. 

thanks.

mellbo, 2019-12-27:
I fixed that. with 8B 08 1A 00 [temp] [power] [circulation pump:0xFF] 00

proddy, 2019-12-30:
@mellbo that's correct. this function has always been in EMS-ESP since version 1.6