emsesp/EMS-ESP #43: Web interface on the esp

tibi03, 2019-01-25:
It will be nice if all the parameters can be displayed in a web page hosted on microcontroller  and set temp and modes for thermostat .

Thank you.

proddy, 2019-01-26:
I had thought of that originally, and it is possible but decided on the telnet route. Reason being the circuit & code was intended to be a ems message decoder and pass on its data values to a more intelligent home automation system (like Home assistant, openhab, domitcz, node-red) and not intended to be a closed system. The telnet server I created was purely for debugging since I couldn't use the serial/uart port.

Also adding http server would increase the firmware size and it would be slow (unless its on an esp32).

Can you describe the use case you are after? Do you want continuous monitoring? Is there something you can't control with telnet or a mqtt subsystem?

tibi03, 2019-01-26:
I want only a web page to dispaly all the known parameters as is on telnet with auto refresh at X seconds , set temp and mode for thermostat and maybe last 5 showers duration and time date, not the telegrams . For now i am not using any home automation hub and this is the reason why a web interface its a great thing :) thanks

proddy, 2019-01-27:
to be honest I still don't understand the use case on why you would want to monitor the ESP devices real-time on a web page. Home automation systems are intended to do this and have the logic to trigger events based on incoming data, and also retain history. If you have an old PC, Synology or Raspberry PI I would strongly suggest installing hass.io. My other concern with having a web server on the ESP8266 is that it'll use up valuable processing power which is needed to process the huge amount of EMS telegrams coming in. Even in the code now, which is quite optimized I still see some packages being skipped.

proddy, 2019-07-03:
had a change of mind, I think moving all the config to the web is a marvelous idea so have started working on a new version to replace telnet. Watch out for v1.9 soon

tibi03, 2019-07-04:
Hello. This is great :)

proddy, 2019-08-17:
first version in 1.9

closed 2019-08-17 by proddy