emsesp/EMS-ESP #45: Bosch Easy Control Adapter support

curantil, 2019-01-31:
I'm trying to connect to my Bosch Easy Control Adapter with the EMS-ESP code.

It seems to work partially (I.e. water temperature seems to have sane values), but other fields or commands don't work. (I.e Gas and pump are always off, even when heating. The thermostat gives a flame symbol when the gas is on, so it should be communicated somehow. And for instance the "thermostat temp" command doesn't seem to work.)

Is there a way I can contribute to make supporting this product easier?

curantil, 2019-01-31:
Thanks.

I think the autodetect also runs at startup? I have a log here saying:
```
Scanning EMS bus for devices...
Unrecognized device found. Product ID 206, Version 02.01
Unrecognized device found. Product ID 206, Version 02.01
Unrecognized device found. Product ID 203, Version 02.00
Unrecognized device found. Product ID 206, Version 02.01
Unrecognized device found. Product ID 203, Version 02.00
```

curantil, 2019-01-31:
I assume the first command has influence on the second command, but when I execute `set boiler_type 8` it says `Invalid parameter for set command.`
When I do `boiler read 2` it returns `Unrecognized device found. Product ID 206, Version 02.01`

(Running EMS-ESP version 1.4.0 btw)

proddy, 2019-01-31:
you need 1.4.1

curantil, 2019-01-31:
That doesn't seem to make a difference:
```
set boiler_type 8        
boiler_type changed to 8
Note, some changes will only have effect after the ESP is restarted (use ! command)
boiler read 2
Unrecognized device found. Product ID 206, Version 02.01
```
Even after restarting

proddy, 2019-01-31:
what boiler do you have? what does 'log v' show you?

curantil, 2019-01-31:
I have a Buderus Logamax U122.

I don't know what I should look for in the log, but here is a part of the verbose log:
```
log v
System Logging set to Verbose
(00:01:39) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 01 EA 00 43 09 01 00 00 80 00 80 00 01 E0 00 00 FF 2D 48 00 00 FF 00 00 (CRC=62), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:40) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 01 C2 80 00 20 00 00 01 19 00 00 00 00 00 00 00 (CRC=39), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:01:49) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 01 C2 80 00 20 00 00 01 19 00 00 00 00 00 00 00 (CRC=39), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:01:50) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 01 EA 00 43 09 01 00 00 80 00 80 00 01 E0 00 00 FF 2D 48 00 00 FF 00 00 (CRC=62), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:59) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 01 EA 00 43 09 01 00 00 80 00 80 00 01 E0 00 00 FF 2D 48 00 00 FF 00 00 (CRC=62), #data=25
<--- UBAMonitorFast(0x18) received
Publishing boiler data via MQTT
Calling scheduled data refresh from EMS devices..
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:02:00) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 01 C2 80 00 20 00 00 01 19 00 00 00 00 00 00 00 (CRC=39), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:02:00) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(00:02:00) Boiler -> me, type 0x18 telegram: 08 0B 18 00 00 01 EA 00 43 09 01 00 00 80 00 80 00 01 E0 00 00 FF 2D 48 00 00 FF 00 00 (CRC=D1), #data=25
<--- UBAMonitorFast(0x18) received
(00:02:00) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(00:02:00) Boiler -> me, type 0x19 telegram: 08 0B 19 00 80 00 80 00 80 00 00 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=36), #data=27
<--- UBAMonitorSlow(0x19) received
(00:02:00) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=71), #data=15
(00:02:01) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(00:02:01) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 3C FB FF 28 FF 02 46 00 FF (CRC=44), #data=11
<--- UBAParameterWW(0x33) received
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(00:02:01) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 63 (CRC=AF), #data=1
(00:02:01) Boiler -> me, type 0x16 telegram: 08 0B 16 00 FF 4C 43 00 06 FA 0A 01 03 64 64 00 (CRC=9F), #data=12
<--- UBAParametersMessage(0x16) received
(00:02:01) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 63 (CRC=A7), #data=1
(00:02:01) Boiler -> me, type 0x14 telegram: 08 0B 14 00 00 00 00 (CRC=12), #data=3
<--- UBATotalUptimeMessage(0x14) received
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(00:02:02) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=C1), #data=15
(00:02:02) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 01 F4 00 43 09 01 00 00 80 00 80 00 01 E0 00 00 FF 2D 48 00 00 FF 00 00 (CRC=79), #data=25
<--- UBAMonitorFast(0x18) received
(00:02:06) 0x18 -> Boiler, type 0x1A telegram: 18 08 1A 00 20 (CRC=91), #data=1
(00:02:07) 0x18 -> all, type 0x06 telegram: 18 00 06 00 13 01 15 1F 24 35 03 A5 (CRC=B2), #data=8
(00:02:09) 0x18 -> Boiler, type 0x35 telegram: 18 08 35 00 11 (CRC=1C), #data=1
(00:02:09) 0x18 -> Boiler, type 0x33 telegram: 18 88 33 01 09 (CRC=7A), #data=1
(00:02:09) Boiler -> 0x18, type 0x33 telegram: 08 18 33 01 FF 3C FB FF 28 FF 02 46 00 (CRC=29), #data=9
<--- UBAParameterWW(0x33) received
(00:02:09) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 01 F4 00 43 09 01 00 00 80 00 80 00 01 D6 00 00 FF 2D 48 00 00 FF 00 00 (CRC=02), #data=25
<--- UBAMonitorFast(0x18) received
(00:02:10) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 01 C2 80 00 20 00 00 01 19 00 00 00 00 00 00 00 (CRC=39), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:02:19) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 01 C2 80 00 20 00 00 01 19 00 00 00 00 00 00 00 (CRC=39), #data=17
```

curantil, 2019-02-07:
I have verbose log files when starting the Adapter both connected and not connected to the thermostat. Would that help? And is there anything I should look for?

proddy, 2019-02-07:
Can you remind me what the issue exactly is? Please use the latest firmware and confirm
 - your boiler is correctly recognized when booting up
 - the values from the boiler are automatically read and displayed after 'info' command
 - the temp values are read correctly from the Easy thermostat
 - you're able to read values from the boiler e.g. `boiler read 2`

note changing values on the Easy (like thermostat temp ...) will not work as the Easy doesn't have Write support yet.

curantil, 2019-02-07:
Using the latest version:
- At boot or autodetect it gets detected as `Unrecognized device found. TypeID 0x02, Product ID 206, Version 02.01` Should the device be recognized or is this expected behavior?
- Gas and boiler pump don't seem to be read from the boiler. Probably some other values too, but those I could verify. The thermostat displays when the boiler is heating water or heating the water for the radiators, so I would expect that those could be read.
- Where should I see the temp values from the Easy thermostat? The water temperature settings on the boiler are displayed, but I cannot see the set point of the thermostat anywhere.
- boiler read 2 works. Is there a list of possible IDs that can be read? (I assume the EMS wiki, but which table should I look at?)

proddy, 2019-02-07:
- Your boiler (Buderus Logamax U122) isn't in ems_devices.h yet because I didn't know the type. Can you tell me what 'boiler read 2' says and I'll add it
- The type 0x02, product id 206 is the Bosch Easy. I'll add that too. For now do a `set thermostat_type 2` and then another `autodetect`. You probably are not seeing the values because it hasn't recognized it yet. Look for 'thermostat enabled/disabled' at the top when you do a `info` command.
- the gas and boiler pump, what values do you see with 'info' ? These values should come from type 0x18. Can you do a `boiler read 18` in verbose logging and let me know what data it returns?
- use `types` to list all the commands you can issue that are parsed. Of course you can do any read command and decipher the byte sequence yourself using the wiki but its not 100% accurate.

curantil, 2019-02-08:
- `boiler read 2`returns `Unrecognized device found. TypeID 0x02, Product ID 206, Version 02.01` (Info shows it as `Boiler type: Generic Type [Type ID: 0x08] Product ID:0 Version:not set`)
- `info` says `Thermostat is enabled'
Set thermostat_type 2 returns:
```
set thermostat_type 2
thermostat_type changed.
autodetect
Scanning EMS bus for devices.
Unrecognized device found. TypeID 0x02, Product ID 206, Version 02.01
Unrecognized device found. TypeID 0x02, Product ID 206, Version 02.01
Unrecognized device found. TypeID 0x02, Product ID 203, Version 02.00
```
- `boiler read 18` gives:
```
boiler read 18
Requesting type UBAMonitorFast(0x18) from dest 0x08
(00:04:58) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(00:04:58) Boiler -> me, type 0x18 telegram: 08 0B 18 00 00 02 E4 00 43 09 01 00 00 80 00 80 00 01 FE 00 00 FF 2D 48 00 00 FF 00 00 (CRC=23), #data=25
<--- UBAMonitorFast(0x18) received
(00:04:59) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=71), #data=15
(00:04:59) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=C1), #data=15
Calling scheduled data refresh from EMS devices..
Read operations not yet supported for this model thermostat
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:05:00) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(00:05:00) Boiler -> me, type 0x18 telegram: 08 0B 18 00 00 02 E4 00 43 09 01 00 00 80 00 80 00 01 FE 00 00 FF 2D 48 00 00 FF 00 00 (CRC=23), #data=25
<--- UBAMonitorFast(0x18) received
(00:05:00) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(00:05:00) Boiler -> me, type 0x19 telegram: 08 0B 19 00 80 00 80 00 80 00 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=27
<--- UBAMonitorSlow(0x19) received
(00:05:00) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=71), #data=15
(00:05:00) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(00:05:00) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 3C FB FF 28 FF 02 46 00 FF (CRC=44), #data=11
<--- UBAParameterWW(0x33) received
Publishing boiler data via MQTT
(00:05:01) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 63 (CRC=AF), #data=1
(00:05:01) Boiler -> me, type 0x16 telegram: 08 0B 16 00 FF 4C 43 00 06 FA 0A 01 03 64 64 00 (CRC=9F), #data=12
<--- UBAParametersMessage(0x16) received
(00:05:01) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 63 (CRC=A7), #data=1
(00:05:01) Boiler -> me, type 0x14 telegram: 08 0B 14 00 00 00 00 (CRC=12), #data=3
<--- UBATotalUptimeMessage(0x14) received
(00:05:02) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=C1), #data=15
(00:05:02) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=C1), #data=15
(00:05:02) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 02 E4 00 43 09 01 00 00 80 00 80 00 01 FE 00 00 FF 2D 48 00 00 FF 00 00 (CRC=90), #data=25
<--- UBAMonitorFast(0x18) received
(00:05:03) Boiler -> all, type 0x19 telegram: 08 00 19 00 80 00 80 00 80 00 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=24), #data=27
<--- UBAMonitorSlow(0x19) received
(00:05:03) Boiler -> all, type 0x1C telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4B), #data=25
(00:05:03) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 3A 80 00 20 00 00 01 19 00 00 00 00 00 00 00 (CRC=68), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:05:09) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 02 EE 00 43 09 01 00 00 80 00 80 00 01 FE 00 00 FF 2D 48 00 00 FF 00 00 (CRC=99), #data=25
<--- UBAMonitorFast(0x18) received
(00:05:12) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 02 EE 00 43 09 01 00 00 80 00 80 00 01 FE 00 00 FF 2D 48 00 00 FF 00 00 (CRC=99), #data=25
<--- UBAMonitorFast(0x18) received
(00:05:13) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 3A 80 00 20 00 00 01 19 00 00 00 00 00 00 00 (CRC=68), #data=17
<--- UBAMonitorWWMessage(0x34) received
```

proddy, 2019-02-08:
everything looks fine (except we need to add your thermostat to the list). You said the boiler flags like gas on/off an pump on/off are not showing? Can you do an 'info' and tell me what you see.

curantil, 2019-02-08:
This is a snapshot when requesting higher temperature for the heating.
The burner is at 67% power, so I would assume gas and boiler/circulation pump should show something.
```
info
EMS-ESP System stats:
  System logging set to None
  LED is off
  # connected Dallas temperature sensors=0
  Thermostat is enabled, Boiler is enabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus Stats:
  Bus Connected=yes, # Rx telegrams=325, # Tx telegrams=0, # Crc Errors=0

Boiler stats:
  Boiler type: Generic Type [Type ID: 0x08] Product ID:0 Version:not set
  Hot tap water is off
  Central Heating is off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water is set to Comfort
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 43.0 C
  Warm Water current tap water flow: 2.5 l/min
  Warm Water # starts: 0 times
  Warm Water active time: 0 days 0 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 0 C
  Current flow temperature: 61.0 C
  Return temperature: 39.0 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 67 %
  Flame current: 0.0 uA
  System pressure: 0.0 bar
  Current System Service Code: -H
  Heating temperature setting on the boiler: 76 C
  Boiler circuit pump modulation max. power: 100 %
  Boiler circuit pump modulation min. power: 100 %
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # restarts: 0 times
  Total burner operating time: 0 days 0 hours 0 minutes
  Total heat operating time: 0 days 0 hours 0 minutes
  Total UBA working time: 0 days 0 hours 0 minutes

Thermostat stats:
  Thermostat type: Generic Type [Type ID: 0x02] Product ID:0 Version:not set
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 00:00:00 0/0/2000
  Mode is set to ?

Shower stats:
  Shower Timer is off
```

proddy, 2019-02-08:
if you look at the 8th bit of the 0x18 message its always zero in the message dump you sent us, which contains all the flags for gas, fan, heat pump etc. So either your boiler has a different EMS message format or the gas wasn't on when you took the snapshot. Can you check yourself? Also look at the code around line 933 in ems.cpp. It's an open source project after all. 😄

curantil, 2019-02-08:
In the `Boiler -> all, type 0x18 telegram: ` there indeed never seems to be a non-zero value for the 8th byte. So all the values for that block are false. 

What would be the best way to try to figure out where those messages are hidden? My first idea would be to filter out all the messages that can be matched with a known meaning. And then see if relation can be found. Although I don't know how many messages would remain.

proddy, 2019-02-12:
turn the hot water on, make sure the gas is burning and the pump is on, capture the 0x18 telegram and let's see what it says. If the 8th byte is not zero we're ok.

curantil, 2019-02-20:
The 8th byte is always 0 as far as I can see. Here is a log of only the 0x18 telegrams. The first two lines are from before the hot water is turned on. The last two from when the hot water is running. As you can see, the 8th byte is always 00.
```
(00:14:32) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 02 08 00 00 00 00 00 00 80 00 80 00 02 1C 00 00 FF 30 48 00 00 FF 00 00 (CRC=B9), #data=25
(00:14:38) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 02 08 00 00 00 00 00 00 80 00 80 00 02 1C 00 00 FF 30 48 00 00 FF 00 00 (CRC=B9), #data=25
(00:14:48) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 02 08 00 00 00 00 00 00 80 00 80 00 02 4E 00 00 FF 30 43 00 00 FF 00 00 (CRC=46), #data=25
(00:14:58) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 02 08 00 00 00 00 00 00 80 00 80 00 02 4E 00 00 FF 3D 48 00 00 FF 00 00 (CRC=54), #data=25
```
I'll try to see if I can decipher the rest of the telegram based on your code.

proddy, 2019-02-20:
those messages coming back don't look correct at all. I did a quick google and the Buderus Logamax U122 was produced in 2004 so perhaps it uses additional EMS messages. I'm just not sure at this point.

curantil, 2019-02-21:
As it is the Easy Control Adapter that translates betweent iRT and EMS I could expect that it is the Easy Control Adapter that causes the "dialect".

curantil, 2019-02-22:
I'm trying to recreate the initialization procedure of the thermostat, but it seems the Adapter doesn't respond to the raw telegrams I send.

In your docs, there it is mentioned that you have 200ms seconds before the bus master times out. I tried to find anything related to the timing back in your code, but I don't seem to find it. Is that correct or am I overlooking something?

proddy, 2019-02-22:
I also spent a lot of time capturing the EMS messages when the Easy was booting, to recreate the handshake with the boiler but unfortunately, it didn't help. There's a GitHub issue somewhere where I posted the ideas I had using messages 0x07 and 0x02. I hope you have better luck than I did.

As for the 200ms, since we immediately Send on receiving a poll I decided not to build in the timeout. But I did have to hack a pause into the Tx line to allow the bus to drop. See line 17 in EMS-ESP/src/emsuart.h

curantil, 2019-02-22:
Could you point me to the issue you mention? I cannot find it.

I am still figuring out the exact flow. Are we only sending after a poll of our own ID? Wouldn't that make the bus master ignore it as it doesn't come at the time slice of the right device?

proddy, 2019-02-23:
remind me, what issue are we trying to solve here? It jumped from Bosch Easy to the boiler not showing gas/pump settings correctly?

proddy, 2019-02-23:
The other issue I was referring too is [here](https://github.com/proddy/EMS-ESP/issues/39)

curantil, 2019-02-24:
> remind me, what issue are we trying to solve here? It jumped from Bosch Easy to the boiler not showing gas/pump settings correctly?

Well, I see this issue as getting the Bosch Easy Control Adapter to work as much as possible. So getting the gas/pump settings is part of that in my view. 

> The other issue I was referring too is [here](https://github.com/proddy/EMS-ESP/issues/39)

Ok, thanks. I'll see if I can contribute.

curantil, 2019-02-28:
Ok, this registration procedure seems to work if I send is as raw messages:
```
send 0B 88 EF 00 01 
send 0B 00 02 00 CB 02 00 00 00 00 00 00 00 01 00 
send 0B 00 01 1B 52 52 43 32 00 
send 0B 88 02 00 03 
send 0B 88 01 00 1B 
send 0B 88 04 00 03 
send 0B 88 02 09 01 
send 0B 88 16 01 02 
send 0B 88 33 00 0A 
send 0B 88 19 00 02 
send 0B 88 04 04 01 
send 0B 88 1C 05 03 
send 0B 88 18 00 18 
send 0B 88 04 0F 01 
send 0B 00 06 00 13 02 0E 16 2D 10 04 A5 
send 0B 08 1A 00 00 
send 0B 08 35 00 01 
send 0B 88 34 08 01 
```
If the thermostat is not connected this will make the "blue" (searching thermostat) light on the Easy Control Adapter go green (found thermostat). There is some kind of heart beat detection which makes the Easy Control Adapter forget the device after a certain time, but I have not figured out the exact rules for that.

Executing: `send 0B 08 1A 00 4C` successfully starts the heating flow at 67%. `send 0B 08 1A 00 00` stops the heating successfully. Our device doesn't need to be registered for that.

I'll now try to figure out where the values went that do not correspond to the EMS documentation.

proddy, 2019-03-04:
The `send 0B 00 02 00 CB 02 00 00 00 00 00 00 00 01 00` is interesting. Basically, it's our device broadcasting to everyone our version number and product id. Maybe that is the secret sauce.

proddy, 2019-03-11:
@curantil the work that Glenn did at https://github.com/proddy/EMS-ESP/issues/44 should also work for your thermostat. Perhaps worth trying

proddy, 2019-05-28:
re-opening to have another look at supporting Bosch Easy devices

closed 2020-05-16 by proddy