emsesp/EMS-ESP #46: Nefit Topline Compact HRC25 HR

Peterpc58, 2019-02-02:
Hi, I installed a "bbq Kees" board and am new to this.
When I do autodetect, I get this:
<--- Version(0x02) received
Unrecognized device found. Product ID 115, Version 03.06
<--- Version(0x02) received
Unrecognized device found. Product ID 114, Version 01.03
I have a Nefit Topline Compact HRC25 HR
How can I read my Nefit?*Before creating a new issue please check that you have:*

proddy, 2019-02-02:
Hi Peter,  which version of the firmware are you using? It should detect the Topline Compact and set the boiler type. If you type 'info' or 'set' you should be able to see this.

The product ID 114 is interesting. Do you have any other devices attached, like a thermostat?

Bonusbartus, 2019-02-02:
Proddy, those are the same numbers I found on my topline compact hrc30cw5 with the older version:

"the discovered deviceIDs are:
0x08: Unrecognized device found. Product ID 115, Version 03.06
0x09: Unrecognized device found. Product ID 114, Version 01.03"

proddy, 2019-02-02:
yes, the Topline Compact should be recognized. The 0x09 in the BC10 base controller which is built into these units.

Peterpc58, 2019-02-03:
The bbq kees board came with version 1.3.1
I have no thermostat connected to the boiler.
I use my computer to switch the heating with a relay. (using Homeseer)
When I enter 'set boiler_type 9' and then 'info', I get this:
'Thermostat is disabled, Boiler is disabled, Poll is disabled, Tx is enabled, Shower Timer is enabled, Shower Alert is disabled
  EMS Bus Stats: Connected=yes, # Rx telegrams=0, # Tx telegrams=0, # Crc Errors=2'
When I set 'log v' I see this:
'(15:24:19) 0x08 -> all, type 0x19 telegram: 08 00 19 00 80 00 01 D5 80 00 00 00 00 5A 02 AF 70 05 40 38 00 00 00 03 5A 7E 00 EC 72 00 00 (CRC=8F), #data=27
(15:24:20) 0x08 -> all, type 0x1C telegram: 08 00 1C 00 00 4E 5B C4 00 00 00 00 00 00 00 (CRC=88), #data=11
(15:24:20) 0x08 -> all, type 0x34 telegram: 08 00 34 00 3C 02 2C 80 00 21 00 00 03 00 01 E5 BA 01 C2 FE 00 (CRC=FB), #data=17
(15:24:20) 0x08 -> all, type 0x07 telegram: 08 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=12), #data=13
(15:24:20) 0x08 -> all, type 0x18 telegram: 08 00 18 00 50 01 DD 64 00 01 02 20 E2 02 2C 80 00 01 D9 00 00 10 30 41 00 CA 00 00 00 (CRC=5C), #data=25'
and that keeps on going....

proddy, 2019-02-03:
I recommend you upgrade to the latest firmware which supports your boiler type.

Peterpc58, 2019-02-03:
Trying to upload version 1.4.1 with Arduino 1.8.8, but I get this error in ems.cpp:

Arduino: 1.8.8 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Users\Bulldog\AppData\Local\Temp\arduino_build_15361\sketch\ems.cpp: In function 'void _process_Version(uint8_t*, uint8_t)':

ems.cpp:1163:15: error: 'class MyESP' has no member named 'fs_saveConfig'

         myESP.fs_saveConfig();

               ^

ems-esp:104:1: error: initializer-string for array of chars is too long [-fpermissive]

 };

 ^
exit status 1
'class MyESP' has no member named 'fs_saveConfig'

proddy, 2019-02-03:
I can't you help with the Arduino IDE I'm afraid. I use Platformio as stated in the doc. Perhaps someone else on the forums can help you. You need to enable SPIFFs for your board.

Peterpc58, 2019-02-03:
Installed Platformio, it was all new for me. But I managed to update the board.
The Boiler is recognised now and it shows all the readings!
Now figuring out how to get the readings in Homeseer with mqtt....
Thanks for your help!

Peterpc58, 2019-02-04:
Installed Mosquito and EMS-ESP directly connected to it.
Only had to fill in the ip address of my server without any port reference.
Then installed the mscMQTT plugin for Homeseer.
It came with some examples to configure, but after filling in the basics and stop/start the plugin, it found the Boiler_data.
I now have all the data in 19 Homeseer devices.
With the devices I can also publish commands to the EMS-ESP, but what is the command trough MQTT to switch on/off the boiler_data:wWActivated, so I can set it to off when I am not at home?

proddy, 2019-02-04:
that isn't implemented. The only ones I added are to set the thermostat mode and temp (see TOPIC_THERMOSTAT_CMD_TEMP and TOPIC_THERMOSTAT_CMD_MODE) in my_config.h

its easy to add though if you really need it.

proddy, 2019-02-04:
ok, I just added it and also updated the doc. It'll be in 1.5.2 which I'm still debugging

Peterpc58, 2019-02-04:
Hi Proddy,
That is quick! Thank you. 
Is it possible to update the 1.5.2 OTA, or do I have to connect it to my PC?

proddy, 2019-02-04:
OTA is preferred. Use `upload_port = ems-esp.local` in your platformio.ini file.

Peterpc58, 2019-02-04:
Updated to 1.5.2
When I send a MQTT to the EMS-ESP, will it show in the log then?

proddy, 2019-02-05:
yes, says "Setting boiler warm water off" or "Setting boiler warm water on". You'll also see the setting when you type 'info'

Peterpc58, 2019-02-05:
I am a little bit further.
When I start mosquitto, I have to add -v to see all the traffic.
When I send 1 to wwactivated, I see this in mosquitto:
1549397609: Received PUBLISH from mcsMQTT on doberman2 (d0, q0, r0, m0, 'wwactivated', ... (1 bytes))
1549397609: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'wwactivated', ... (1 bytes))
But noting in the log on the EMS-ESP

I also see this commands from the EMS-ESP
1549397740: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'home/ems-esp/tapwater_active', ... (1 bytes))
1549397740: Received PUBLISH from ems-esp (d0, q1, r0, m168, 'home/ems-esp/heating_active', ... (1 bytes))
1549397741: Sending PUBACK to ems-esp (Mid: 168)
1549397741: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'home/ems-esp/heating_active', ... (1 bytes))
But what does it mean?

Peterpc58, 2019-02-05:
When I do that, it is working, but the ems-esp keeps on sending this messages then:
1549398657: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'home/ems-esp/tapwater_active', ... (1 bytes))
1549398657: Received PUBLISH from ems-esp (d0, q1, r0, m1371, 'home/ems-esp/heating_active', ... (1 bytes))
1549398657: Sending PUBACK to ems-esp (Mid: 1371)
1549398657: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'home/ems-esp/heating_active', ... (1 bytes))
1549398658: Received PUBLISH from ems-esp (d0, q1, r0, m1372, 'home/ems-esp/boiler_data', ... (372 bytes))
1549398658: Sending PUBACK to ems-esp (Mid: 1372)
1549398658: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'home/ems-esp/boiler_data', ... (372 bytes))
1549398658: Received PUBLISH from ems-esp (d0, q1, r0, m1373, 'home/ems-esp/tapwater_active', ... (1 bytes))
1549398658: Sending PUBACK to ems-esp (Mid: 1373)
1549398658: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'home/ems-esp/tapwater_active', ... (1 bytes))
1549398658: Received PUBLISH from ems-esp (d0, q1, r0, m1374, 'home/ems-esp/heating_active', ... (1 bytes))
1549398658: Sending PUBACK to ems-esp (Mid: 1374)
1549398658: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'home/ems-esp/heating_active', ... (1 bytes))
1549398658: Received PUBLISH from ems-esp (d0, q1, r0, m1375, 'home/ems-esp/boiler_data', ... (372 bytes))
1549398658: Sending PUBACK to ems-esp (Mid: 1375)
1549398658: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'home/ems-esp/boiler_data', ... (372 bytes))
1549398658: Received PUBLISH from ems-esp (d0, q1, r0, m1376, 'home/ems-esp/tapwater_active', ... (1 bytes))
1549398658: Sending PUBACK to ems-esp (Mid: 1376)
1549398658: Sending PUBLISH to mcsMQTT on doberman2 (d0, q0, r0, m0, 'home/ems-esp/tapwater_active', ... (1 bytes))
1549398658: Received PUBLISH from ems-esp (d0, q1, r0, m1377, 'home/ems-esp/heating_active', ... (1 bytes))
1549398658: Sending PUBACK to ems-esp (Mid: 1377)
And does not stop. Is that ok?

proddy, 2019-02-05:
no! that is not ok. reboot the circuit or it may heat up if you leave on too long.

Can you re-flash with the latest build from github and try again?

Peterpc58, 2019-02-05:
I was already using 1.5.2
When I reboot, all is ok.
But as soon as I send a command from HS, it keeps on sending.

proddy, 2019-02-05:
just to be sure, pull from github, recompile and flash again. I'll try and reproduce the problem from this side too. Do you have mqtt qos also set to 1 on the HS?

Peterpc58, 2019-02-05:
In the general settings of mcsMQTT I only see "Default QOS" and can select At Most, At Least or Exactly.
It is on At Most

Peterpc58, 2019-02-05:
Also found at the device.
Have set them both on at least.
Same problem.
Did a reboot on the ems-esp and have to stop for today.
Tomorrow I don't have time, but on Thursday I will flash the ems-esp again and test again.
In the meantime I will not switch the wwactivated.
Thanks for all your help!

Peterpc58, 2019-02-07:
I have downloaded version 1.5.2 again and OTA it to the board.
Now all is well.
Don't know if the qos was the problem or something in the upload went wrong.
But it is working now.
In the menu of the boiler you can set some more options, like:
wWConfort set to off, Eco or Comfort
boilTemp set to 30 - 85 C
Is it possible to also add these settings to the boiler settings?

proddy, 2019-02-07:
glad it works now! yes, it's possible to add all that functionality. How good are you programming skills? :-) It would be great for others to contribute and expand the features, otherwise I spend all my evenings doing change requests for others ;-) And I really need to fix this EMS Plus support...

Peterpc58, 2019-02-07:
My programming skills are not very good, but with some hints I can always try and learn.

proddy, 2019-02-07:
no problem, I'll do the first one and show you how. Do you want to be able to set the settings from the telnet command line or via MQTT or both?

Peterpc58, 2019-02-07:
MQTT

proddy, 2019-04-06:
setting mode was added in 1.6

closed 2019-04-06 by proddy