emsesp/EMS-ESP #47: RC10 Moduline 100 support

dutchrazor, 2019-02-03:
Hello,

First of all thanks for this great project. I was wondering if you could help me with my Moduline 100. It is connected to a Nefit Smartline Basic HR. 

I see the following devices on the bus:

```
Scanning EMS bus for devices...
Boiler type device found. Model Sieger BK15 Boiler with TypeID 0x08, Product ID 64, Version 03.05
* Setting Boiler type to Model Sieger BK15 Boiler, TypeID 0x08, Product ID 64, Version 03.05
Boiler type device found. Model RFM20 Receiver with TypeID 0x09, Product ID 68, Version 02.03
Unrecognized device found. Product ID 79, Version 01.02
```

I'm not sure what the RFM20 receiver is , but I think ID 79 is the RC10 thermostat with bus ID 0x17.

From this [topic](https://www.domoticz.com/forum/viewtopic.php?f=38&t=14132&start=100#p177153) I found and verified the following commands:

Read thermostat currrent setpoint

```
(00:40:57) Sending raw telegram: 0B 97 B0 04 01 (CRC=97), #data=1
(00:40:57) 0x17 -> me, type 0xB0 telegram: 17 0B B0 04 2C (CRC=E7), #data=1 
```

Thermostat current temperature is broadcast:

`(00:40:06) 0x17 -> all, type 0xB1 telegram: 17 00 B1 00 04 2C 00 D5 00 00 00 D5 (CRC=F0), #data=8`

The thermostat at this moment said it was 21.0 and later jumped to 21.5 so I assume D5 (=213) would be 21.3 degrees.

Setting temperature 

> And I'm able to write a new setting (0x29 = 41 = 20,5 degrees) with the following command: 

(00:35:22) Sending raw telegram: 0B 17 B0 04 29 (CRC=DB), #data=1
(00:35:23) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 80 00 00 00 00 00 00 00 00 00 00 00 (CRC=47), #data=13
(00:35:23) 0x17 -> Boiler, type 0x1A telegram: 17 08 1A 00 00 00 00 00 (CRC=3A), #data=4

Two questions:
-Do you know why in the telegram 97 sets the new temp? I thought that that position was the target ID and should be 17?
-Are these all the required telegrams for implementation?

Thanks.

proddy, 2019-02-03:
Thanks for the info. It shouldn't be too difficult to add RC10 support. I'll include it in the new release 1.5.0 and we'll finetune as we go along.

dutchrazor, 2019-02-03:
Cool :)

proddy, 2019-02-03:
i added the stubs for your RC10 in v1.5.0. Can you test it?

dutchrazor, 2019-02-03:
That's quick! 

For some reason, I now get a response "unknown command" when sending autodetect. I'll try to erase flash in case it's a coincidence:

`telnet 192.168.4.1
Trying 192.168.4.1...
Connected to 192.168.4.1.
Escape character is '^]'.
[000172] [MQTT] disabled
[000253] [WIFI] MODE AP --------------------------------------
[000254] [WIFI] SSID  ems-esp
[000254] [WIFI] IP    192.168.4.1
[000255] [WIFI] MAC   86:0D:8E:8A:7F:98
[UART] Opened Rx/Tx connection

* Connected to: EMS-ESP version 1.5.0
* ESP8266 is in AP mode with SSID ems-esp
* Free RAM:22 KB, Load:0%
*
* Commands:
*  ?=help, CTRL-D=quit, !=reboot
*  set
*  set <wifi_ssid | wifi_password | mqtt_host | mqtt_username | mqtt_password> [value]
*  set erase
*  set serial
*  set led <on | off>                    toggle status LED on/off
*  set led_gpio <pin>                    set the LED pin (onboard=2)
*  set dallas_gpio <pin>                 set the pin for the external Dallas temperature sensor (D5=14)
*  set thermostat_type <hex type ID>     set the thermostat type id (e.g. 10 for 0x10)
*  set boiler_type <hex type ID>         set the boiler type id (e.g. 8 for 0x08)
*  info                                  show the values
*  log <n | b | t | r | v>               set logging mode to none, basic, thermostat only, raw or verbose
*  publish                               publish values to MQTT
*  types                                 list supported EMS telegram type IDs
*  queue                                 list Tx queue
*  autodetect                            discover EMS devices and set boiler and thermostat automatically
*  shower <timer | alert>                toggle either timer or alert on/off
*  send XX...                            send raw telegram data in hex to EMS bus
*  thermostat read <hex type ID>         send read request to thermostat
*  thermostat temp <degrees>             set current thermostat temperature
*  thermostat mode <mode>                set mode (0=low/night, 1=manual/day, 2=auto)
*  thermostat scan <hex type ID>         do a force read on all type IDs starting at n
*  boiler read <hex type ID>             send read request to boiler
*  boiler wwtemp <degrees>               set warm water temperature
*  boiler tapwater <on | off>            set warm tap water on or off

autodetect
Unknown command. Use ? for help.
autodetect
Unknown command. Use ? for help.
Unknown command. Use ? for help.
?
Connection closed by foreign host.`

m1588, 2019-02-03:
@dutchrazor, download new version, it is already fixed couple minutes ago.

dutchrazor, 2019-02-03:
Ah it sees the thermostat! I can also read/write the setpoint temperature. Only the current value is slightly too low. 

```
Thermostat stats:
  Thermostat type: RC10/Nefit Moduline 100) [Type ID: 0x17] Product ID:79 Version:01.02
  Setpoint room temperature: 22.0 C
  Current room temperature: -870.40 C
  Thermostat time is 176:00:00 0/0/2152
  Mode is set to ?
```

proddy, 2019-02-03:
I've made the change, and also fixed "thermostat temp" to except floats

dutchrazor, 2019-02-08:
It works, thanks!

dutchrazor, 2020-04-16:
Hello,

It's been some time and I upgraded now to 1.9.4. Web UI looks great! 

Now i do not get the thermostat recognised anymore however, it thinks its a MM100 mixing module:

```
These device IDs are on the EMS Bus: 0x08 0x09 0x17
and 3 were recognized by EMS-ESP as:
 Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID:0x08 ProductID:64 Version:03.05)
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 MM100 Mixer Module (DeviceID:0x17 ProductID:79 Version:01.02)
```

Could you take a look at that?

proddy, 2020-04-16:
welcome back. A lot has changed since 1.9.4 and I should release 1.9.5 from beta soon. But could you try 1.9.5 and see if it detects the RC10 correctly? You can download it from the UI automatically or manually via https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build

dutchrazor, 2020-04-16:
Fixed in 1.9.5 indeed! Thanks.

I saw that MM100 and RC10 have the same id 79, so that was probably the issue.

closed 2020-04-16 by dutchrazor