emsesp/EMS-ESP #49: Domoticz plugun & MQTT receive issues

Gert05, 2019-02-10:
I'm in the process of writing a Domoticz Python plugin that shows the current room tempeature and allows to change the value of the thermostat. So far, I have succeeded in reading the current room temperature and thermostat setting (through MQTT) and transferring those values to Domoticz. Next step is to modify the thermostat setting. This is where I'm having the problem that EMS-ESP seems not to receive temperature updates through MQTT.

The screenshot has no relevance to the MQTT-problem, but shows what I'm trying to achieve in Domoticz...
[image]

Standard EMS-ESP v1.5.2. Compiled through PlaftformIO. Runing on a Wemos D1 mini pro (device setting in PlatformIO: d1_mini).

For testing purposes, I'm sending messages through MQTTbox. See settings below.
[image]

Messages seem not to be received by EMS-ESP. I'm sure I had this working (using the same MQTTbox setup) a couple of versions ago.
Any suggestions? Any specific test setup that is known to work?

proddy, 2019-02-10:
I just did a quick test and it still works for me. I sent the same package from Home Assistant and in the ems-esp logging (log b) you should see:

```
MQTT topic: thermostat temperature value 18.0
Setting new thermostat temperature
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
Write to 0x17 was successful
```

Try also with QOS 1. https://www.hivemq.com/blog/mqtt-essentials-part-6-mqtt-quality-of-service-levels/

Gert05, 2019-02-10:
Hmmm, I'm seeing none of that. QOS=1 doesn't help. Is it possible for you to check with MQTTbox what the raw payload is when you change temp through HA?
[image]

proddy, 2019-02-10:
I'm not familiar with MQTTbox. Looks like the MQTTbox is serializing the data if the payload from your example. I suggest using mqttspy (as mentioned in my doc) to see what is actually sent from the client to the ems-esp.

Gert05, 2019-02-10:
Sorry - should have said "MQTTbox (or your favourite mqtt analyser)". I think that mqttspy will do the trick as well.

What I'm after is the exact byte sequence that is sent by HA. I know what I'm sending as payload, but since it doesn't work I'm interested in a sequence of bytes that does actually work...

Gert05, 2019-02-12:
Ouch. A simple _sudo systemctl restart _mosquitto__ on my MQTT server did the trick...
Case closed.

closed 2019-02-12 by Gert05

proddy, 2019-02-12:
mosquitto -v would have displayed the packages too