emsesp/EMS-ESP #50: Nefit Proline NxT/Trendline Solarsensor

EinStijn74, 2019-02-17:
EMS-ESP version 1.3.2 with converter circuit from bbqkees.

All is working. I can see the data from the "info" command in the menu. (really, really, nice)

Question:
There is a Nefit Solarsensor (sensor to read water temperature from solar boiler system) connected to my Nefit, Is it possible to read this data from the EMS bus too. 

*Fulfilling this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session with the logging set to Verbose mode.*

**Additional context**
*Add any other context about the problem here.*

proddy, 2019-02-18:
Yes it is possible. I think I've seen the telegram formats posted somewhere so with some code changes you could also read the other devices.

proddy, 2019-04-06:
any more info on this @EinStijn74 ?

EinStijn74, 2019-04-08:
> any more info on this @EinStijn74 [1] ?

Hi Proddy, 

To be honest I don't know how to troubleshoot this issue. 

If there is any manual on how to sent and receive messages over the EMS
bus and how to calculate them, could you sent me this link? 

I am cannot read C+ code, so your code is not very helpful for me. ;-) 

It does work great! 

Best, 

Stijn 

 

Links:
------
[1] https://github.com/EinStijn74
[2] https://github.com/proddy/EMS-ESP/issues/50#issuecomment-480492632
[3]
https://github.com/

proddy, 2019-04-08:
I've added support a few weeks back for a Solar Module. See [issue 77](https://github.com/proddy/EMS-ESP/issues/77). I think it may be compatible with your Nefit device. Have you tried this?

EinStijn74, 2019-04-08:
> I've added support a few weeks back for a Solar Module. See issue 77 [1]. I think it may be compatible with your Nefit device. Have you tried this?

Hi, 

I have another brand Solar Module, only using the temperature sensor
module from Nefit: 

https://www.cvtotaal.nl/nefit-ombouwset-naverwarmen-zonneboiler-proline-trendline.html?gclid=EAIaIQobChMI04f_keTA4QIV0ud3Ch3LrwIjEAQYASABEgKgkvD_BwE

I will upgrade my current of your software this week, to see if it makes
any difference. 

Regards, 

Stijn 

 

Links:
------
[1] https://github.com/proddy/EMS-ESP/issues/77
[2] https://github.com/proddy/EMS-ESP/issues/50#issuecomment-480797440
[3]
https://github.com/

proddy, 2019-04-08:
the new firmware is for an external module so it's probably different to the external sensor you have for your Nefit. You might be lucky and the values are captured in the UBAMonitorSlow/0x19 telegram like the external outside temp. Otherwise it'll be trial and error and a bit of googling to find where the value is stored. If you see this on the boiler or in a thermostat you can trace to the source.

proddy, 2019-04-29:
any luck with 1.7 dev? Did it detect your solar sensor?

proddy, 2019-05-03:
The latest 1.7 is ready to support this. As soon as you have the info we need please re-open

closed 2019-05-03 by proddy