emsesp/EMS-ESP #51: EMS stops sending after a couple of hours

rmarinusse, 2019-02-18:
Dear Proddy,

First of all, thanks for your work. I've ordered a pre-installed board from 'bbqkees' and managed to get the whole thing working with my Nefit Topline Compact and Moduline 300. I'm using MQTT to publish data to my domoticz install. Everything works fine, except that after a couple of hours the data stops flowing. When I power cycle the ems board everything starts working again for a couple of hours. 

Any idea what might be causing this? Could it be due to poor WiFi coverage? Should it reconnect to WiFi automatically when the connection was lost?

Thanks

Ron

proddy, 2019-02-18:
Hi Ron, I'm not sure what could be causing this. The code retries on WiFi and MQTT failures. I would recommend doing 'set led on' and when it stops responding see the LED is still solid and not blinking. Also when you say 'data not flowing' can you still ping the EMS?

rmarinusse, 2019-02-18:
Just checked, led is solid when connection is lost and I can't ping it. When I use a wifi scanner the device is not visible until a power cycle. 

One other thing I noticed is that a telnet session to the device gets disconnected after a minute or so of inactivity. Is that intentional?

FWIW: Here's a dump from the info command output and the status lines at the top
* Connected to: EMS-ESP version 1.5.2
* Hostname: ems-esp   IP: 192.168.1.30   MAC: 68:C6:3A:EA:BA:E8
* Connected to WiFi SSID: myssid
* Boot time: start
* Free RAM:18 KB, Load:1%

EMS-ESP System stats:
  System logging set to None
  LED is on
  # connected Dallas temperature sensors=0
  Thermostat is enabled, Boiler is enabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus Stats:
  Bus Connected=yes, # Rx telegrams=114, # Tx telegrams=0, # Crc Errors=241

Boiler stats:
  Boiler type: Nefit Topline Compact [Type ID: 0x08] Product ID:115 Version:03.06
  Hot tap water is off
  Central Heating is off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water is set to ECO
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 40.29 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 45922 times
  Warm Water active time: 74 days 20 hours 36 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 65 C
  Current flow temperature: 56.50 C
  Return temperature: 29.39 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 30 %
  Burner current power: 29 %
  Flame current: 22.0 uA
  System pressure: 2.29 bar
  Current System Service Code: -H
  Heating temperature setting on the boiler: 65 C
  Boiler circuit pump modulation max. power: 100 %
  Boiler circuit pump modulation min. power: 50 %
  Boiler temperature: 48.50 C
  Pump modulation: 65 %
  Burner # restarts: 76281 times
  Total burner operating time: 609 days 13 hours 13 minutes
  Total heat operating time: 534 days 16 hours 37 minutes
  Total UBA working time: 3011 days 9 hours 1 minutes

Thermostat stats:
  Thermostat type: RC20/Nefit Moduline 300) [Type ID: 0x17] Product ID:77 Version:03.03
  Setpoint room temperature: 21.0 C
  Current room temperature: 21.0 C
  Thermostat time is 21:06:47 18/2/2019
  Mode is set to auto

Shower stats:
  Shower Timer is off

proddy, 2019-02-18:
telnet gets disconnected automatically, yes. You could try running it in serial mode (set serial) and monitor the output to see if anything strange happens. I also noticed the 'Tx telegrams is 0' so perhaps your Tx is not working. You can try this by doing a 'log v' and then 'autodetect' to see if any telegrams are returned to you. 

Also to be safe re-flash. I'm not sure which firmware build kees used.

bbqkees, 2019-02-22:
My guess is that the problem is either bad Wi-Fi reception or another network problem at your home.

rmarinusse, 2019-02-22:
OK, did some checking and indeed, one of my AP's is misbehaving. I've ordered a new one and hopefully that will fix the issues.

proddy, 2019-02-23:
closing the issue. if it happens again, please re-open

closed 2019-02-23 by proddy