emsesp/EMS-ESP #52: Intergas Kombi Kompakt + Bosch Easy Control Adapter problems.

daanv22, 2019-02-19:
Dear Proddy,

I'm currently working on a little bit weird setup with the EMS level shifter and the ESP 8266.
I have drawn a schematic scheme to make it clear.

[image]
Boiler: Intergas Kombi Kompakt HRE 24/18
EzC: Bosch EasyControl adapter
TS: Nefit Easy
EMS: EMS Level Shifter v0.9
ESP: ESP8266
Firmware: 1.5.2

So, in this situation the EzC transforms the OpenTherm Protocol to EMS.
The Nefit Easy works fine.
But the EMS-Bus can't detect the Boiler.

When I do a autodetect I get the following result:

```
autodetect
Scanning EMS bus for devices.
Unrecognized device found. TypeID 0x02, Product ID 205, Version 02.01
Unrecognized device found. TypeID 0x02, Product ID 205, Version 02.01
* Setting Thermostat type to Model TC100/Nefit Easy, TypeID 0x18, Product ID 202                               , Version 02.18
```

I also saw the older issue with the Bosch EasyControl adapter and followed the instructions.
After that i did a _info_ command.
```
EMS-ESP System stats:
  System logging set to None
  LED is on
  # connected Dallas temperature sensors=0
  Thermostat is enabled, Boiler is enabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus Stats:
  Bus Connected=yes, # Rx telegrams=11, # Tx telegrams=0, # Crc Errors=1

Boiler stats:
  Boiler type: Generic Type [Type ID: 0x08] Product ID:0 Version:not set
  Hot tap water is off
  Central Heating is off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water is set to Comfort
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: 2.5 l/min
  Warm Water # starts: 0 times
  Warm Water active time: 3 days 20 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 0 C
  Current flow temperature: 58.0 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.10 uA
  System pressure: 1.90 bar
  Current System Service Code: -H
  Heating temperature setting on the boiler: 90 C
  Boiler circuit pump modulation max. power: 100 %
  Boiler circuit pump modulation min. power: 100 %
  Boiler temperature: 2073.60 C
  Pump modulation: 128 %
  Burner # restarts: 32768 times
  Total burner operating time: 5825 days 14 hours 23 minutes
  Total heat operating time: 3276 days 22 hours 32 minutes
  Total UBA working time: 0 days 0 hours 0 minutes

Thermostat stats:
  Thermostat type: TC100/Nefit Easy [Type ID: 0x18] Product ID:202 Version:02.18
  Setpoint room temperature: 30.20 C
  Current room temperature: 64.33 C

Shower stats:
  Shower Timer is off
```
There are a lot of random values.
My first thought was that the boiler isn't supported yet.
Can I write custom code for my boiler?
My code experience is very low, but a friend of my is pretty good with C/, C++ etc.
And how far is the Nefit Easy write function?

Thank you in advance!

Daan
(Sorry for the bad english)

proddy, 2019-02-19:
you can either set the boiler type to 0x02 or add it ems_devices.h. But shouldn't make a difference as there is only one format for EMS boilers. Looking at the data I see the Boiler temperature is wrong but the rest seems ok. Does writing work, for example setting the boiler temp?

Writing to the Nefit Easy is not supported as the Easy doesn't listen to EMS requests. I put some ideas in my README on how to solve this for anyone feeling adventurous.

daanv22, 2019-02-20:
When I set the boiler to type 2. I get the following result:
```
Shower stats:
  Shower Timer is off
info
EMS-ESP System stats:
  System logging set to None
  LED is on
  # connected Dallas temperature sensors=0
  Thermostat is enabled, Boiler is enabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus Stats:
  Bus Connected=yes, # Rx telegrams=49, # Tx telegrams=1, # Crc Errors=3

Boiler stats:
  Boiler type: Generic Type [Type ID: 0x02] Product ID:0 Version:not set
  Hot tap water is off
  Central Heating is off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water is set to Comfort
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: 2.5 l/min
  Warm Water # starts: 0 times
  Warm Water active time: 3 days 20 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 24 C
  Current flow temperature: -1510.40 C
  Return temperature: 0.0 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 0.0 bar
  Current System Service Code:
  Heating temperature setting on the boiler: 90 C
  Boiler circuit pump modulation max. power: 100 %
  Boiler circuit pump modulation min. power: 100 %
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # restarts: 63791 times
  Total burner operating time: 193 days 12 hours 0 minutes
  Total heat operating time: 0 days 0 hours 0 minutes

Thermostat stats:
  Thermostat type: TC100/Nefit Easy [Type ID: 0x18] Product ID:202 Version:02.18
  Setpoint room temperature: 31.35 C
  Current room temperature: 65.48 C

Shower stats:
  Shower Timer is off

```
After I do `boiler wwtemp 40` nothing changes.
Also the thermostat stats are a little bit off. (Real temperatures are: Setpoint: 21,0 Current: 22,1)

I will let my friend look at the README file and see what he can do.

proddy, 2019-02-20:
best way to debug is turn on verbose logging, look for the telegrams and
then check the code to see if the values are in the right place. Its quite
straightforward and I put a lot of effort into making the code readable. If
you find anything strange feel free to put in a pull request.

closed 2019-04-06 by proddy