emsesp/EMS-ESP #55: ESP8266 crashing continously when compiling from source

nomis, 2019-02-20:
I can't get 1.5.2 (or any version) to work on a Wemos D1 Mini V2 if I compile it myself, it only works if I use the precompiled version:
```
$ git rev-parse HEAD
8d5757c00b2638a54d2927064002685e7876ec42
$ platformio upgrade
You're up-to-date!
PlatformIO 3.6.4 is currently the newest version available.
$ platformio update
Updating tool-scons                      @ 2.20501.7      [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.12.5         [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.0      [Up-to-date]
Updating framework-arduinoavr            @ 2.10623.190209 [Up-to-date]
Updating tool-avrdude                    @ 1.60300.2      [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.0.0          [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.3          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20500.0      [Up-to-date]
Updating toolchain-xtensa                @ 2.40802.190218 [Up-to-date]

Library Manager
===============
$ platformio run

Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting ems-esp.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CRC32> 2.0.0
|-- <CircularBuffer> 1.3.0
|-- <JustWifi> 2.0.2
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 5.13.4
|-- <OneWire> 2.3.4
|-- <Ticker> 1.0
|-- <myESP>
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266mDNS> 1.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 5.13.4
|   |-- <JustWifi> 2.0.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <TelnetSpy>
|   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/d1_mini/src/ems-esp.ino.cpp.o
Retrieving maximum program size .pioenvs/d1_mini/firmware.elf
Checking size .pioenvs/d1_mini/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  56.6% (used 46400 bytes from 81920 bytes)
PROGRAM: [=====     ]  45.7% (used 477724 bytes from 1044464 bytes)
============================================================= [SUCCESS] Took 2.01 seconds =============================================================

====================================================================== [SUMMARY] ======================================================================
Environment nodemcuv2	[SKIP]
Environment d1_mini  	[SUCCESS]
============================================================= [SUCCESS] Took 2.01 seconds =============================================================
```

When I read the serial console it's just continuously crashing:
```
   [000107] [MQTT] disabled

Exception (28):
               epc1=0x4022639c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

                                                                                                  >>>stack>>>

                                                                                                             ctx: cont
                                                                                                                      sp: 3ffffd20 end: 3fffffc0 offset: 01a0
     3ffffec0:  3ffe8ffb 3ffe8ff3 3ffffef0 3fff2950  
                                                     3ffffed0:  3fff4f1c 3ffea8ad 3fff3298 40219594  
                                                                                                     3ffffee0:  3ffea8a5 3fff3298 3ffea8a1 4020d6a9  
                                                                                                                                                     3ffffef0:  00000000 00000000 00000000 00000000  
                                             3fffff00:  00000000 00000000 3fff2950 40221370  
                                                                                             3fffff10:  3ffe8ffb 3ffe8ff3 3fff2950 4020d51c  
                                                                                                                                             3fffff20:  3fff41fc 3ffe8ff3 4020cf4c 4022123c  
                                     3fffff30:  3ffe8ffb 3ffe8ff3 3fff2950 4020fa1f  
                                                                                     3fffff40:  40221250 3fff2950 3fffff70 40208d1a  
                                                                                                                                     3fffff50:  00000001 00000000 3ffe8fe4 3ffeabae  
                             3fffff60:  3fffff70 feefeffe feefeffe feefeffe  
                                                                             3fffff70:  4020779c feefeffe 4022132c 40221310  
                                                                                                                             3fffff80:  402090b8 feefeffe 402212fc 402212e4  
                     3fffff90:  feefeffe feefeffe feefeffe 3fff2d5c  
                                                                     3fffffa0:  3fffdad0 00000000 3fff2d28 402116fc  
                                                                                                                     3fffffb0:  feefeffe feefeffe 3ffe855c 40100951  
             <<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
```

proddy, 2019-02-21:
did you have a previous version on the ESP? If so perhaps the SPIFFS could be corrupted after the upgrade. Its hard to tell without seeing any changes you may have made to the platformio.ini.  In these types of situations I would enable the serial (set serial on), plug the Wemos into the PC via USB bare with nothing connected and look at the serial monitor to see the boot up process. If that doesn't work format the SPIFFS partition and flash again.

nomis, 2019-02-21:
Do you know why the binary I get is much larger than the one you've built?

So far I'm getting nothing useful on the serial port even after forcing `_use_serial = true`.

proddy, 2019-02-21:
need more info. what is your platformio. what do you see in the serial monitor. it should be full with information

nomis, 2019-02-21:
I've already provided all the information about platformio above. The resulting binary is 477724 bytes whereas your version is 384240 bytes.

proddy, 2019-02-22:
the code uses many external libraries, perhaps they have been updated. Also the esp framework went through a major update to[ 2.0.0 ](https://github.com/platformio/platform-espressif8266/releases/tag/v2.0.0)in the last days which is my first guess. I wouldn't worry about it.

nomis, 2019-02-22:
Could you provide the dependency graph of the libraries that you compiled it with?

I know your build of the code works so I want to be able to reproduce it.

proddy, 2019-02-22:
```
Processing d1_mini (framework: arduino; platform: espressif8266; board: d1_mini)
----------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting ems-esp.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CRC32> 2.0.0
|-- <CircularBuffer> 1.3.0
|-- <JustWifi> 2.0.2
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 5.13.4
|-- <OneWire> 2.3.4
|-- <myESP>
|   |-- <TelnetSpy>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 5.13.4
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS> 1.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <JustWifi> 2.0.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
```

proddy, 2019-02-22:
as I expected the problem is the esp8266 core update to 2.5. I need to debug what is breaking - perhaps something in the wifi code. For now the fix is to keep using 1.8.0 so use `platform = espressif8266@1.8.0` in your platformio.ini and it should work.

proddy, 2019-02-23:
This should now be fixed with 1.5.3

closed 2019-02-27 by nomis