emsesp/EMS-ESP #56: WiFi AP Mode runs in parallel with Client mode

S-Przybylski, 2019-02-21:
**Bug description**
After some days running the ems esp V. 1.5.2 (and of course a restart of my Fritzbox Wlan) i wonder that i see a running access point in my house named ems-esp. Its interresting, that i can normally access the device using my prefered (private) Wlan (not the ems-esp)!

**Steps to reproduce**
Not quite sure: Restart of my internal WLAN access point (Fritzbox)

**Expected behavior**
I assume that only one access way is currently activated

**Screenshots**
Telnet using my private wlan (not ems-esp): see SSID!
...
* Connected to: EMS-ESP version 1.5.2
* ESP8266 is in AP mode with SSID ems-esp
* Free RAM:17 KB, Load:25%
...

Wlan check at the same time with my mobile phone:
[image]

proddy, 2019-02-21:
if the ems-esp can't connect to the wifi it will go into AP mode. Go check your settings (hint telnet to 192.168.1.4 and do a 'set')

S-Przybylski, 2019-02-21:
The device i connected to my private wlan network and its sending currently data via mqtt. In parallel the device act as an access point at the same time! That is the strange thing. When i restart the device (ems-esp) then the device shows that is connected to the provate wlan!

Is it possible to deactivate the access point in general?

S-Przybylski, 2019-02-21:
Dear @proddy can you please recheck?

proddy, 2019-02-21:
That's not possible in the code. Can you take out the wemos, connect it up to USB and do a serial monitor to see what is happening. Make sure 'set serial on' is enabled before you do this and don't forget to disabled it when you reconnect it to the EMS bus.

proddy, 2019-03-03:
closing. I think you received a new board?

closed 2019-03-03 by proddy