emsesp/EMS-ESP #57: better MQTT handling for testament / last will

S-Przybylski, 2019-02-21:
**Describe the solution you'd like**
In my home automation (home assistant) i commonly use the mqtt lwl/testament to identify if the device is online or offline. Other ESP-solution like the Tasmota firmware or Node Red support this behaviour per default: Besides the last will statement also "online" statements are set up accordingly. Example of an MQTT binary sensor Home Assistant configuration: ...
  availability_topic: "NodeRed/LWT"
  payload_available: "online"
  payload_not_available: "offline"
From a technical point of view it should (i belive) be easy to implement: when the device establish an mqtt connection, publish a second message with the same LWT-topic (but without the last will flag set) and the body "online" (example) using the retain flag.

Benefit: Direct detection if the device crashed or was gone offline! Then Home Assistant set the states as unavailable.

proddy, 2019-02-21:
last will is already setup (see my_config.h) and myESP line 718: 

```
// last will
if (_mqtt_will_topic) {
     myDebug_P(PSTR("[MQTT] Setting last will topic %s"), _mqttTopic(_mqtt_will_topic));
     mqttClient.setWill(_mqttTopic(_mqtt_will_topic), _mqtt_qos, _mqtt_retain, _mqtt_will_payload);
}
```

With your examples like Tasmota, how often do the devices publish to the LWT topic? Is it like a heartbeat?

S-Przybylski, 2019-02-21:
When i understand it correctly: You have to send at least two messages. Both with retain=true
1. LWT like line 718 does
2. Additional "normal" statement with the same lwt-topic but not using the .setWill function and the payload="online" (for example)

The effect of the first line will be that the payload "offline" will only be published to the subscribers if the broker identifies that the client has disconnected. 
The second statement payload "online" is directly visible for all subscribers, also for those which connected later (due to retain=true). 
Therefore the second statement needs only to be send once; there is no need to republish it on a scheduled basis.

In your configuration a new parameter should be included:
#define MQTT_AVAILABILITY_PAYLOAD "Online"
and the additional line like this: 
... mqttClient.publish(_mqttTopic(_mqtt_will_topic), _mqtt_qos, _mqtt_retain, _mqtt_availability_payload); 
Note: The retain=1 should alway be set for both subscriptions

proddy, 2019-02-22:
that's correct. I'll add the changes and test it. Thanks.

proddy, 2019-02-23:
In 1.5.3. Please test and let me know if it works as proposed!

closed 2019-03-03 by proddy