emsesp/EMS-ESP #59: Change flow temperature via telnet and MQTT

egrekov, 2019-02-22:
Hi!
I would like to be able to change the flow temperature.
I understand that it is necessary to implement a function similar to ems_setFlowTemp, but not enough knowledge in the EMS protocol, I could not help in the implementation of this functionality.

Examlpe in telnet 
`* boiler flowtemp <degrees>               set flow temperature`

lobocobra, 2019-03-04:
+1
is there a mqtt SEND command, which returns the value? like this, we could do such things as described by egrekov with node-red

proddy, 2019-03-05:
It’s not hard to add further commands. I tried to make the code as readable and flexible as possible. Try it yourself and can review the pull requests remotely and make changes.

lobocobra, 2019-04-06:
You can check here how I added MQTT commands. Its not hard. Just search for "lobocobra" in the code and you will see what to change where....
https://github.com/lobocobra/EMS-ESP-HC2

proddy, 2019-04-06:
@egrekov I can quickly add this for you. But not sure if its possible. Looking at https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#wwbetriebsart is the flow temperature the 3rd byte of the data?

ReneHonig, 2019-04-06:
@proddy 
Paul, I would also love to try to set the flow temperature (and burner power) by MQTT. Is my understanding that that is the function of the 'UBASollwerte' telegram? And what is the difference between 'Leistungsanforderung HK' and 'Leistungsanforderung WW'?

Let me know if i can help testing in any way.

proddy, 2019-04-06:
@ReneHonig I really have no idea! It's all been trial and error. One thing to try is using this 1A telegram to set the flow temperature, This can be done with a normal 'send' command.

ReneHonig, 2019-04-06:
@proddy , ok, let me try that. In order to be sure I am at least sending the intended telegram, can you give me a hint of what it would look like? Is the 'normal' send command through the telnet interface or is this the code lobocobra just created (@lobocobra , is the new mqtt command documented somewhere?)

Thanks!!

Thanks!!

proddy, 2019-04-07:
ok, say it is UBASollwerte then the command to set the temperature to 20 degrees would be

send 0B 08 1A 00 14

the 0B is us, the 08 is the typeID of the boiler, 1A is UBASollwerte, 00 is the offset for the value you want changed and 14 is the value (20 degrees).

I'm not at home so can't test if this works

ReneHonig, 2019-04-07:
thanks, I can confirm this works!
I will do a couple more  tests later today, also with setting the burner power.

proddy, 2019-04-07:
Ok, I've added to the latest dev. `set flowtemp <n>`

ReneHonig, 2019-04-07:
I have done a bit more investigation. The boiler regularly received 0x1a updates from the thermostat (in my case Evohome+OT converter). The 1A telegram in my case sets the flow temperature and the next byte ('Leistungsanforderung HK') which I have only seen at 100 (the EMS wiki says: 0 or 100).

Here is an example from the log:

`Thermostat -> Boiler, type 0x1A telegram: 18 08 1A 00 1F 64 (CRC=21), ...`

This sets the flow temperature at 31 deg C.

The boiler responds with:
```
<--- UBASetPoints(0x1A) received
 Boiler flow temp 15.5 C, Warm Water power 33 %
```
which I do not understand (seems 50% of the set temperature, also not sure where the warm water power comes from).

@proddy , maybe you know how the boiler response is obtainer, otherwise I will need to dive a bit into the code.

To be continued...

proddy, 2019-04-08:
the mistake is the code then. I assumed, like all temperatures it was multiplied by 2. So if 1F really does set the flow temp to 31 degrees then there is no division needed in the calculation. It also means half-steps are not supported. I'll modify the code and leave the old logic in. See ems.cpp line 1267

ReneHonig, 2019-04-08:
Thanks. My source for the flow temperature is the telnet info, which is obviously the same code...

I guess I can only conclude then that there is an inconsistency between the number on the info screen and the log. 

I will check whether I can look up the action number on the boiler display.

proddy, 2019-04-18:
can we close this?

ReneHonig, 2019-04-18:
apologies for not reverting sooner. I can confirm there is no need to divide the temperature by 2 (by looking at the boiler display). 

I have not heart back from @lobocobra on how to use the MQTT implementation, but agree we can close the issue. Thanks again for all the help.

proddy, 2019-04-18:
great. thanks,

sadrov, 2019-04-25:
> @proddy
> Paul, I would also love to try to set the flow temperature (and burner power) by MQTT. Is my understanding that that is the function of the 'UBASollwerte' telegram? And what is the [...]

Hi ReneHonig, did you manage in any way to change the burner power with telnet or mqqt, if so, I'd like to know how you pulled it off :)

ReneHonig, 2019-04-25:
I have been able to set the flow temperature writing a raw telegram as per Paul instructions earlier in this conversation. Have not yet managed to use MQTT as I am not sure how sending raw telegrams using MQTT has been implemented. 

I have not tried to change the burner power - are you sure that that is even possible? It might be set by the boiler internally only.

proddy, 2019-04-25:
You can change the flow temperature via telnet using the `boiler flowtemp` command. I can also add this to MQTT if needed. As for the burner power we'll need to work out how to change this.

sadrov, 2019-04-25:
boiler flowtemp tells me 'unknown command'. Is this command new since 1.7? Because I did downgrade from dev build 1.7 back to 1.6 though (1.7 wouldn't take my long wifi password full of weird characters, only the simple wifi password of my guest network, but that's another issue)
@ReneHonig I have no clue if that is possible or not. But I thought maybe you had found out :)
The reason I want this, is to have my boiler burn either at its lowest capacity as soon as the set room temperature is reached or not burn at all. Because by default it acts more like an on/off boiler than a modulating one, going from off to high power

proddy, 2019-04-26:
yes 'bolier flowtemp' is in 1.7 and was added 18 days ago based on a request in this issue (scroll up). I thought I also fixed the SSID and Password issue to accept any strange character too in 1.7 so please also check if that is working for you too.

So what's next? Change flow temp via MQTT or set the boiler burn?

sadrov, 2019-04-27:
Hey @proddy I reflashed 1.7 but the password issue remains, even when i hardcode it in the platformio.ini file. So now I'm using it on my guest network that has a simple password.
Now I can do the boiler flowtemp command and I get a response but none of the settings actually change when I look at the info right after (not even when I disconnect the thermostat).
"
boiler flowtemp 65
Setting boiler flow temperature to 65 C
"
The reason I started trying this project was actually to limit the power of my boiler so it would burn for longer times once the room temperature is more or less reached instead of burning short intervals at maximum power (pendelgedrag). So for me ideally an implementation where i can set the "Burner selected max power" would be awesome, than I'd let just domoticz change that once the room temperature is close to the setpoint room temperature.
But since i can't even change the boiler flowtemp, maybe i shouldn't dream about that :P
Have a nice weekend!

sadrov, 2019-04-27:
oh yeah, when I query info, I also get this:
EMS Bus stats:
  Bus is connected
  Rx: Poll=37 ms, # Rx telegrams read=0, # CRC errors=2
  Tx: no signal
Has the no signal thing maybe something to do with the fact that I'm having issues with sending out commands and raw telegrams?

proddy, 2019-04-28:
@sadrov the 'Tx: no signal' means that we're not receiving a Poll request to our device which is the trigger to transmit any Tx packages. If you do a `queue` you probably see all the commands in a wait state. Did sending work for you before? Are you using a Junkers boiler?

Also I'm interested in the wifi password issue since I'm pretty sure I nailed that problem. Can you give me some hints on the special characters you use? And have you been able to get other ESP8266 code to connect?

sadrov, 2019-04-28:
Yes, there's a whole bunch of read actions in the queue and when i do a send command it shows up in the queue as well:
 [19] action=read dest=0x08 type=0x14 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:21:00)
 [20] action=write dest=0x08 type=0x1a offset=0 length=6 dataValue=50 comparisonValue=50 type_validate=0x1a comparisonPostRead=0x1a @ (00:21:12)
 [21] action=read dest=0x10 type=0x06 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:22:00)

It is indeed a junkers cerapur boiler, you think this has something to do with it? That somehow the protocol works differently? Sending never worked before (but I've been experimenting witht his only a few weeks). How could I figure out wether these poll requests are even there?

About the wifi: it worked on 1.5 and 1.6 but not on de 1.7 dev build. I set the ssid and password in AP mode and when I reboot the wemos, the ssid and password field are empty. The special characters I use are: ^ l ( ] * ~ ` ; , " \ ) ' 
It worked on a sonoff device and also on this very wemos, but on earlier firmware.

proddy, 2019-04-28:
same Tx issue as https://github.com/proddy/EMS-ESP/issues/103 then. If you could join that thread we can work it on it collectively.

As for the password I'll do some more checking here. 

Closing this issue as it relates to the flowtemp that was implemented in 1.7

proddy, 2019-04-28:
@sadrov just checked the password. Created a new wifi ssid using the password with the special characters you listed **(^ l ( ] * ~ ` ; , " \ ) ')** and it works fine. Just to be sure make sure all your libraries are updated as it could be a bug in the older 2.5.0 ardunio core

ReneHonig, 2019-04-29:
> yes 'bolier flowtemp' is in 1.7 and was added 18 days ago based on a request in this issue (scroll up). I thought I also fixed the SSID and Password issue to accept any strange character too in 1.7 [...]

Hi Paul,
Setting the flow temp through MQTT would be highest priority for me, I will see whether I can figure out whether/how burning power can be set by a telegram.

proddy, 2019-04-29:
Wim implemented this in his fork. Perhaps ask him to merge into into dev.
https://github.com/curantil/EMS-ESP/commit/7d407fa992e41aaad38be087d3aa92a69d4fbc86

ReneHonig, 2019-04-29:
Thanks and done

https://github.com/curantil/EMS-ESP/commit/7d407fa992e41aaad38be087d3aa92a69d4fbc86#commitcomment-33343320

ReneHonig, 2019-04-29:
loos like Wim's implementation is also addressing the flow temperature and not burner power.

egrekov, 2019-12-04:
Figured out to change boiler flowtemp, you change byte 1A, with an offset of 00, and you need 16 with an offset of 01.

ReneHonig, 2019-12-04:
@egrekov , what exactly do you mean by çhange the boiler flowtemp'? Is this different from setting the boiler flow temperature via the telnet interface `boiler flowtemp xx` or the equivalent MQTT command?

egrekov, 2019-12-04:
@proddy `boiler flowtemp xx` it doesn’t work for us at all.
Command `send 0B 08 16 01 3C` change to 60C `Selected flow temperature: 60 C` and `Heating temperature setting on the boiler: 60 C`

egrekov, 2019-12-04:
[image]

ReneHonig, 2019-12-04:
@egrekov , so `send 0B 08 16 01 3C ` changes the heating temperature setting on the boiler' as per your picture?

egrekov, 2019-12-04:
@proddy, yes changes the temperature as in the picture by 60 degrees)

ReneHonig, 2019-12-04:
I tried it and it works indeed. perfect!
@proddy, can we have this in MQTT as well?

egrekov, 2019-12-04:
[emswiki.thefischer.net](https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme), MC10Parameter

egrekov, 2019-12-04:
Сan also be changed maximum and minimum boiler output:
`send 0B 08 16 02 5A` maximum to 90%
`send 0B 08 16 03 1E` minimum to 30%

proddy, 2019-12-04:
ok, so what is wrong actually? Are you saying to change the boiler flowtemp you first need to use 0x1A (UBASetPoints) followed by 0x16 (MC10Parameter)? 

Did the original implementation every work?

ReneHonig, 2019-12-04:
@proddy , nothing wrong from my perspective. 
These are two different functionalities, the UBASetpoints sets the 'currently required flow temperature' and MC10 sets the upper boundary on the flow temparature. The latter is especially useful when the thermostat/control system sets the currently required flow temperature, but you want to put an upper limit on maximum temperature.

proddy, 2019-12-04:
ok, I can add that functionality. How should it be configured in Telnet and MQTT?

egrekov, 2019-12-04:
@proddy It is better that there is the possibility of both telnet and mqtt

ReneHonig, 2019-12-04:
great, I think exactly the same as the `boiler flowtemp xx`, so something like `boiler maxflowtemp xx`, likewise for MQTT (both R/W).

egrekov, 2019-12-04:
`boiler minflowtemp xx` would be nice too

proddy, 2020-07-31:
@egrekov @ReneHonig do you still want this feature implemented?

ReneHonig, 2020-07-31:
Yes, would be great, especially MQTT.

proddy, 2020-07-31:
Added to v2, but needs further testing by users.

Console commands are `temp <n>`, `maxpower <%>`, `minpower <%>` and same commands via MQTT using the topic `command_cmd` as the payload with value (as described in the wiki). I haven't tested any of this. It'll be interesting to know if changing the flow temp now works. If not I need to add the extra call. It was a little unclear from the previous dialog what works and what doesn't. Let me know how you get on.

ReneHonig, 2020-08-01:
Super, Ii uploaded the new firmware but cannot fully test it for a while as I am remote with limited access at the moment.

closed 2020-08-26 by proddy