emsesp/EMS-ESP #60: Cannot seem to compile a working version for Wemos Mini

higgers, 2019-02-22:
Hi,

I received an interface board and Wemos Mini from bbqkees today so I immediately set about building the project and flashing it to the Wemos.

I'm using Visual Studio Code v1.31.1 on Windows 10 and installed the PlatformIO extension v1.60.  I cloned the latest version of EMS-ESP (v1.5.2), copied the example platformio file to platformio.ini and changed the upload port in the [env:d1_mini] section to COM3.  COM3 is listed in device manager; under the Ports section there's a device called "USB-SERIAL CH340 (COM3)".

The output of the build task is:

```
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting ems-esp.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CRC32> 2.0.0
|-- <CircularBuffer> 1.3.0
|-- <JustWifi> 2.0.2
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 5.13.4
|-- <OneWire> 2.3.4
|-- <Ticker> 1.0
|-- <myESP>
|   |-- <TelnetSpy>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 5.13.4
|   |-- <JustWifi> 2.0.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266mDNS> 1.2
|   |   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs\d1_mini\src\ds18.cpp.o
Compiling .pioenvs\d1_mini\src\ems-esp.ino.cpp.o
Compiling .pioenvs\d1_mini\src\ems.cpp.o
Compiling .pioenvs\d1_mini\src\emsuart.cpp.o
Generating LD script .pioenvs\d1_mini\ld\local.eagle.app.v6.common.ld
Compiling .pioenvs\d1_mini\lib25d\CRC32_ID1202\CRC32.cpp.o
Archiving .pioenvs\d1_mini\libe35\libCircularBuffer_ID1796.a
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\BearSSLHelpers.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\CertStoreBearSSL.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Archiving .pioenvs\d1_mini\lib25d\libCRC32_ID1202.a
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\ESP8266WiFiSTA-WPS.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\WiFiClient.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\WiFiClientSecureAxTLS.cpp.o
Compiling .pioenvs\d1_mini\liba6d\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o
[... 180 lines cut ...]
Use manually specified: COM3
Uploading .pioenvs\d1_mini\firmware.bin
Uploading 481968 bytes from .pioenvs\d1_mini\firmware.bin to flash at 0x00000000
................................................................................ [ 16% ]
................................................................................ [ 33% ]
................................................................................ [ 50% ]
................................................................................ [ 67% ]
................................................................................ [ 84% ]
.......................................................................          [ 100% ]
========================================================================================================================================================================== [SUCCESS] Took 20.61 seconds ==========================================================================================================================================================================

=================================================================================================================================================================================== [SUMMARY] ===================================================================================================================================================================================
Environment nodemcuv2   [SKIP]
Environment d1_mini     [SUCCESS]
========================================================================================================================================================================== [SUCCESS] Took 20.62 seconds ==========================================================================================================================================================================

```

At this point the blue LED on the Wemos was not lit up.  I pressed the reset button on the side of the Wemos after which the blue LED starts flashing, approximately 4 times a second.  However, there is no EMS-ESP AP to be found by a wifi scan.  Next, I tried to connect to COM3 via putty.  Putty connects but then prints a string of garbage in the putty window.

At this point I'm a bit stuck.  I've tried building and flashing on my laptop and my workstation (both Windwos 10), I've tried two different USB cables.  I've tried flashing the firmware manually using esptool.py, the flashing process worked but the result was the same, no AP and garbage in the putty window.

The only thing I've not been able to test is using a different Wemos though hopefully two new ones will arrive in the post tomorrow.

Can anyone suggest anything else I can try?

proddy, 2019-02-22:
Try using the serial monitor that’s built into platformio on windows. You
should see something. If using putty make sure the baud is 11520. Some esp
libraries changed during the week so I’ll try building on my system to see
if it all still works. Also I would do a ‘pio run -t erase’ to clear the
esp’s memory as I suspect you have Kees’ old settings installed when he
tested the board.

proddy, 2019-02-22:
ok, I didn't work on my test system either. The problem is the new esp8266 core which was recently upgraded [2.5.0](https://github.com/esp8266/Arduino/releases/tag/2.5.0) and platformio from 1.8.0 to 2.0.0.  I need to debug why. For now the workaround is to replace the 'platform' in your platformio.ini with `platform = espressif8266@1.8.0`

proddy, 2019-02-23:
This is now fixed in the upcoming 1.5.3. Please test.

higgers, 2019-02-23:
Success!  Thanks for such a speedy reply!

Compiling and flashing the 1.5.3 firmware didn't work until I issued the 'pio run -t erase' command you mentioned yesterday.  When I say it didn't work I could see that the blue LED on the Wemos was flashing once every two seconds instead of the very rapid flashing I saw yesterday but my laptop couldn't see an EMS-ESP access point.  After running 'pio run -t erase' and flashing again I could connect to EMS-ESP.

What changes did you have to make to get the code to work with the newer libraries?

Now to try and crack open my Worcester-Bosch and try and get at those EMS pins...

proddy, 2019-02-23:
I had to remove some of the legacy SPIFFs code that used the old SD libraries. Not sure why this kept failing. But don't get too excited, I'm running some tests now and the connection to the UART rx/tx is problematic too.

higgers, 2019-02-23:
Would you recommend sticking with 1.5.2 + `platform = espressif8266@1.8.0`?

higgers, 2019-02-23:
I've reflashed with 1.5.2 but have yet to hook up the Wemos to my boiler.  It's getting late here and I'd rather not run the risk of somehow taking my boiler out of action overnight. :)

Reflashing 1.5.2 over 1.5.3 didn't work until I'd run ‘pio run -t erase’ but I can only run ‘pio run -t erase’ from a Visual Studio Code terminal within the 1.5.3 folder, is this what you'd expect?

proddy, 2019-03-03:
the 1.5.3 and 1.5.4 should work. You can run the pio commands from VS Code or just standalone. See the documentation. But I don't think you need to wipe the flash, it should just work.

higgers, 2019-03-03:
A-ha!  v1.5.4 works!!!  Thanks!

I couldn't get the Wemos to connect to the EMS bus with 1.5.1, 1.5.2 or 1.5.3 but after flashing 1.5.4 it worked.  Whilst the Wemos was at v1.5.3 I had an oscilloscope hooked up and could see the serial data on the EMS boiler pins at 12v low and 16v high and then on D7 and D8 on the Wemos at 0v and 3.3v.  Anyway, the Wemos says it's connected to the bus now.  However, the output of info is:

The output of the info command is:

`EMS-ESP System stats:
  System logging set to Verbose
  LED is on
  # connected Dallas temperature sensors=0
  Thermostat is disabled, Boiler is disabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus Stats:
  Bus Connected=yes, # Rx telegrams=0, # Tx telegrams=0, # Crc Errors=53

Boiler stats:
  Boiler type: <not enabled>
  Hot tap water is running
  Central Heating is active
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water is set to Comfort
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: ? l/min
  Warm Water # starts: ? times
  Warm Water 3-way valve: ?
  Selected flow temperature: ? C
  Current flow temperature: ? C
  Return temperature: ? C
  Gas: ?
  Boiler pump: ?
  Fan: ?
  Ignition: ?
  Circulation pump: ?
  Burner selected max power: ? %
  Burner current power: ? %
  Flame current: ? uA
  System pressure: ? bar
  Current System Service Code: ?
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max. power: ? %
  Boiler circuit pump modulation min. power: ? %
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # restarts: ? times

Shower stats:
  Shower Timer is active`

A few lines of the verbose log:

`System Logging set to Verbose
(01:50:33) 0x08 -> all, type 0x18 telegram: 88 00 18 00 2F 01 85 4C 00 01 02 20 C0 01 E4 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=9A), #data=25
(01:50:34) 0x08 -> all, type 0x34 telegram: 88 00 34 00 33 01 E4 80 00 80 00 00 01 00 01 B2 9A 00 B0 2B 00 (CRC=51), #data=17
(01:50:43) 0x08 -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(01:50:43) 0x08 -> all, type 0x18 telegram: 88 00 18 00 2F 01 83 4C 00 01 02 20 C0 01 E4 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=6A), #data=25
(01:50:44) 0x08 -> all, type 0x34 telegram: 88 00 34 00 33 01 E5 80 00 80 00 00 01 00 01 B2 9A 00 B0 2B 00 (CRC=47), #data=17
(01:50:53) 0x08 -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(01:50:53) 0x08 -> all, type 0x18 telegram: 88 00 18 00 2F 01 82 4C 00 01 02 20 C0 01 E5 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=A3), #data=25
(01:50:54) 0x08 -> all, type 0x34 telegram: 88 00 34 00 33 01 E5 80 00 80 00 00 01 00 01 B2 9A 00 B0 2B 00 (CRC=47), #data=17
Calling scheduled data refresh from EMS devices..
Requesting type UBAMonitorFast(0x18) from dest 0x00
Requesting type UBAMonitorSlow(0x19) from dest 0x00
Requesting type UBAParameterWW(0x33) from dest 0x00
Requesting type UBAParametersMessage(0x16) from dest 0x00
Requesting type UBATotalUptimeMessage(0x14) from dest 0x00
(01:51:03) 0x08 -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(01:51:03) 0x08 -> all, type 0x18 telegram: 88 00 18 00 2F 01 81 4C 00 01 02 20 C0 01 E4 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=CD), #data=25
[... 9 lines cut ...]
(01:51:33) 0x08 -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(01:51:33) 0x08 -> all, type 0x18 telegram: 88 00 18 00 2F 01 7E 4C 00 01 02 20 C0 01 E3 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=F6), #data=25
(01:51:34) 0x08 -> all, type 0x34 telegram: 88 00 34 00 33 01 E3 80 00 80 00 00 01 00 01 B2 9A 00 B0 2B 00 (CRC=33), #data=17
(01:51:43) 0x08 -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(01:51:43) 0x08 -> all, type 0x18 telegram: 88 00 18 00 2F 01 7E 4C 00 01 02 20 C0 01 E2 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=E0), #data=25
(01:51:44) 0x08 -> all, type 0x34 telegram: 88 00 34 00 33 01 E2 80 00 80 00 00 01 00 01 B2 9A 00 B0 2B 00 (CRC=25), #data=17
(01:51:53) 0x08 -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(01:51:53) 0x08 -> all, type 0x18 telegram: 88 00 18 00 2F 01 7E 4C 00 01 02 20 C0 01 E1 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=DA), #data=25
(01:51:54) 0x08 -> all, type 0x34 telegram: 88 00 34 00 33 01 E1 80 00 80 00 00 01 00 01 B2 9A 00 B0 2B 00 (CRC=1F), #data=17
Publishing hot water and heating states via MQTT
Calling scheduled data refresh from EMS devices..
Requesting type UBAMonitorFast(0x18) from dest 0x00
Requesting type UBAMonitorSlow(0x19) from dest 0x00
Requesting type UBAParameterWW(0x33) from dest 0x00
Requesting type UBAParametersMessage(0x16) from dest 0x00
Requesting type UBATotalUptimeMessage(0x14) from dest 0x00
(01:52:03) 0x08 -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(01:52:03) 0x08 -> all, type 0x18 telegram: 88 00 18 00 2F 01 7E 4C 00 01 02 20 C0 01 E0 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=CC), #data=25
(01:52:04) 0x08 -> all, type 0x34 telegram: 88 00 34 00 33 01 E0 80 00 80 00 00 01 00 01 B2 9A 00 B0 2B 00 (CRC=09), #data=17`

The boiler is a Worcester-Bosch Greenstar 550CDi.

I spent a few minutes decoding some UBAMonitorSlow datagrams and see the water temperature go up after I'd turned a hot water tap on.  I could also see the flow rate go up and down.

What should I do to get the info command to show the values from the boiler?

proddy, 2019-03-03:
if you do an `autodetect` what does it tell you about the type ID and product Id for your 550CDi? Because its not in the recognized list (see ems_devices.h) it doesn't know the type. Do a `set boiler_type 8` and see if that helps.

higgers, 2019-03-03:
Output of autodetect with log set to none:

```
autodetect
Scanning EMS bus for devices.
```

with log set to verbose:

```
Scanning EMS bus for devices.
Requesting type Version(0x02) from dest 0x02
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
Requesting type Version(0x02) from dest 0x10
Requesting type Version(0x02) from dest 0x11
Requesting type Version(0x02) from dest 0x17
Requesting type Version(0x02) from dest 0x18
Requesting type Version(0x02) from dest 0x21
```

After set boiler_type 8:

set boiler_type 8(00:18:40) 0x08 -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1

```
boiler_type changed.
(00:18:40) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 BE 4C 00 01 02 20 C0 01 D6 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=24), #data=25
<--- UBAMonitorFast(0x18) received
(00:18:40) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 D6 80 00 80 00 00 01 00 01 B2 9E 00 B0 32 00 (CRC=84), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:18:46) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 B2 4C 00 01 02 20 C0 01 D6 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=DD), #data=25
<--- UBAMonitorFast(0x18) received
(00:18:50) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:18:50) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 AE 4C 00 01 02 20 C0 01 D6 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=61), #data=25
<--- UBAMonitorFast(0x18) received
(00:18:50) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 D6 80 00 80 00 00 01 00 01 B2 9E 00 B0 32 00 (CRC=84), #data=17
<--- UBAMonitorWWMessage(0x34) received
```

That looks like it's read the messages now.  Output of info:

```
info
EMS-ESP System stats:
  System logging set to Verbose
  LED is on
  # connected Dallas temperature sensors=0
  Thermostat is disabled, Boiler is enabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus Stats:
  Bus Connected=yes, # Rx telegrams=0, # Tx telegrams=0, # Crc Errors=0

Boiler stats:
  Boiler type: Generic Type [Type ID: 0x08] Product ID:0 Version:not set
  Hot tap water is off
  Central Heating is off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water is set to Comfort
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 47.9 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 45106 times
  Warm Water active time: 77 days 6 hours 22 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 47 C
  Current flow temperature: 40.0 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 76 %
  Burner current power: 0 %
  Flame current: 0.10 uA
  System pressure: 0.0 bar
  Current System Service Code:
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max. power: ? %
  Boiler circuit pump modulation min. power: ? %
  Boiler temperature: ? C
  Pump modulation: 100 %
  Burner # restarts: 441417 times
  Total burner operating time: 404 days 14 hours 51 minutes
  Total heat operating time: 327 days 8 hours 29 minutes

Shower stats:
  Shower Timer is off
```

Awesome!!!  Thanks!!!

proddy, 2019-03-03:
strange that nothing gets displayed from the autodetect. what does `boiler read 2` tell you?

higgers, 2019-03-03:
I'm not sure there's any response from boiler read 2:

```
log v
System Logging set to Verbose
(00:56:08) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:56:08) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 DE 4C 00 01 02 24 C0 01 5E 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=02), #data=25
<--- UBAMonitorFast(0x18) received
(00:56:08) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 5E 80 00 80 00 00 01 00 01 B2 9F 00 B0 36 00 (CRC=CF), #data=17
<--- UBAMonitorWWMessage(0x34) received
boiler read 2
Requesting type Version(0x02) from dest 0x08
(00:56:13) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 D0 4C 00 01 02 24 C0 01 5E 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=5C), #data=25
<--- UBAMonitorFast(0x18) received
(00:56:18) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:56:18) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 C2 4C 00 01 02 24 C0 01 5D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=84), #data=25
<--- UBAMonitorFast(0x18) received
(00:56:18) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 5E 80 00 80 00 00 01 00 01 B2 9F 00 B0 36 00 (CRC=CF), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:56:19) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 BF 4C 00 01 02 24 C0 01 5E 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=5A), #data=25
<--- UBAMonitorFast(0x18) received
(00:56:20) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 BC 4C 00 01 02 20 C0 01 5D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=EA), #data=25
<--- UBAMonitorFast(0x18) received
(00:56:24) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 B2 4C 00 01 02 20 C0 01 5E 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=8E), #data=25
<--- UBAMonitorFast(0x18) received
boiler (00:56:28) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
re(00:56:28) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 A8 4C 00 01 02 20 C0 01 5E 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=C2), #data=25
<--- UBAMonitorFast(0x18) received
ad(00:56:28) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 5D 80 00 80 00 00 01 00 01 B2 9F 00 B0 36 00 (CRC=F5), #data=17
<--- UBAMonitorWWMessage(0x34) received
 2(00:56:30) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 A3 4C 00 01 02 20 C0 01 5D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=2E), #data=25
<--- UBAMonitorFast(0x18) received

Requesting type Version(0x02) from dest 0x08
(00:56:38) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:56:38) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 93 4C 00 01 02 20 C0 01 5D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=E1), #data=25
<--- UBAMonitorFast(0x18) received
(00:56:38) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 5D 80 00 80 00 00 01 00 01 B2 9F 00 B0 36 00 (CRC=F5), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:56:48) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:56:48) Boiler -> all, type 0x07 telegram: 88 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=E6), #data=15
(00:56:48) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 88 4C 00 01 02 20 C0 01 5D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=72), #data=25
<--- UBAMonitorFast(0x18) received
(00:56:49) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 5D 80 00 80 00 00 01 00 01 B2 9F 00 B0 36 00 (CRC=F5), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:56:49) Boiler -> all, type 0x19 telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 03 64 06 BC 5E 08 E4 08 00 00 00 07 31 68 06 0C 28 80 00 (CRC=75), #data=27
<--- UBAMonitorSlow(0x19) received
(00:56:50) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 85 4C 00 01 02 20 C0 01 5C 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=42), #data=25
<--- UBAMonitorFast(0x18) received
(00:56:58) Boiler -> 0x13, type 0x05 telegram: 88 13 05 22 00 (CRC=80), #data=1
(00:56:58) Boiler -> all, type 0x18 telegram: 88 00 18 00 2F 01 80 4C 00 01 02 20 C0 01 5C 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=CA), #data=25
<--- UBAMonitorFast(0x18) received
(00:56:58) Boiler -> all, type 0x34 telegram: 88 00 34 00 33 01 5C 80 00 80 00 00 01 00 01 B2 9F 00 B0 36 00 (CRC=E3), #data=17
<--- UBAMonitorWWMessage(0x34) received
Calling scheduled data refresh from EMS devices..
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
```

proddy, 2019-03-03:
ah, damn. You have the same issue as reported by others. The Tx is not working, i.e. you're not able to send commands to the EMS devices. I haven't been able to fix this, unfortunately. It happens on certain systems (look at the open github issues) and very difficult to debug remotely. You seem to know about electronics seeing you own an oscilloscope - how good are your C++ debugging skills?

higgers, 2019-03-03:
I'm not that great with electronics but I know a man who is an absolute god with anything electrical.  The last time I did an C programming was many years ago but I'm sure I can work my way through the code.  Did you have a plan in mind for how to approach this?  We'd benefit from having some hardware that can talk to the boiler via the EMS bus so we could monitor the comms.  I don't have any hardware like that at the moment but if any exists for a reasonable price then I could buy it.

higgers, 2019-03-03:
I've just read through this issue #23 which seems like the same thing I'm seeing here.  I'll try and hook up the oscilloscope on the TX line and see if data is making its way to the boiler.

higgers, 2019-03-04:
Is it possible to flash a Wemos with new firmware via the network rather than with it locally connected over USB?  The boiler is in the cellar so it would be handy if I could update the code without have to go venturing across the house every time I want to make a change.

proddy, 2019-03-04:
yes, via OTA. It's always been in the code and is the default way for
flashing. The platormio.ini is setup this way, just use the upload_port
'ems-esp.local'

higgers, 2019-03-05:
I'd tried changing the upload port to the IP I'd assigned to the Wemos, didn't work.  Then I tried turning off the windows firewall on my laptop.  Success!

proddy, 2019-03-06:
great!

closed 2019-03-06 by proddy

proddy, 2019-10-08:
@higgers you're using a Worcester-Bosch Greenstar 550CDi correct? Can you tell me the Product ID and version of this. Also did you get Tx to work and if so which tx_mode are you using?