emsesp/EMS-ESP #61: Thermostat is overruled

henderikus, 2019-02-23:
Hi Proddy,

I have two questions regarding the temperature control and it is not a real issue..

When I use the "thermostat temp" command on the ESP, the set(ingestelde) thermostat temperature on the thermostat itself is not updated/synchronized with the ESP set temperature. Is this supposed to happen? I am wondering because it is synchronizing the mode control, when used on the ESP..

Another thing: once I set the the thermostat temp with the ESP, the thermostat temperature control is ignored. It looks like the ESP overrules the thermostat. Is this also supposed to happen? Only when I switch between auto/manual mode on the thermostat, the temperature control is working again.

Best regards,

Rick

proddy, 2019-02-23:
This may actually be a bug. You're supposed to be able to set the thermostat temps just like using a real thermostat - as that was the original point of my project to begin with! Can you try and reproduce the steps so I can recreate it her in my test lab? Also tell me which model thermostat you have. So set the logging to thermostat only (`set log t`) and do a test both in auto and manual mode (`thermostat temp <val>`). After each operation look at the info/stats to see if the values are correct. It may take a few seconds for the EMS to respond.

henderikus, 2019-02-23:
>You're supposed to be able to set the thermostat temps just like using a real thermostat
To be sure you understand me. Setting the temperature via the ESP is working. It updates the "current temperature" in the ESP, which is shown with "info" and the flame sign is appearing on the real thermostat, but the current temperature on the the real thermostat isn't updated when altering the current temperature on the ESP...

proddy, 2019-03-03:
not sure I unuderstand. You can't set the current temperature is this is coming from a sensor built into the thermostat. You can set the setpoint temp and if this is showing correctly in the ESP code then its reading it correctly from the thermostat. Can you explain how to reproduce?

henderikus, 2019-03-09:
First thing:
-Current thermostat temperature 18.
-Login with telnet into ESP
-Run "thermostat temp 20"
-Wait 30 seconds.
-Turn the wheel a little on the thermostat, so the  set temperature is shown on the display, which is still 18 and not the expected value 20.

Second thing:
-Current thermostat temperature 18.
-Login with telnet into ESP
-Run "thermostat temp 20"
-Wait 30 seconds.
-Turn the wheel of thermostat to 16.
-Wait 30 seconds.
-run "show info" in ESP.
-Set temperature is still 20.

proddy, 2019-03-09:
Rick - it may be because Tx is not working for you. Try the same with logging enabled and see if there is a response back from the Send. Also helpful to know which model thermostat you're using

henderikus, 2019-03-10:
Second thing with log:

Set temperature from ESP:

log t
thermostat temp 23
Setting new thermostat temperature
Write to 0x17 was successful
(00:08:56) Thermostat -> Boiler, type 0x1A telegram: 17 08 1A 00 5A 28 00 00 (CRC=78), #data=4
(00:08:56) Thermostat -> me, type 0x91 telegram: 17 0B 91 00 80 2E 00 C6 51 00 00 02 64 5A 07 00 C7 00 (CRC=DD), #data=14
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
(00:08:57) Thermostat -> Boiler, type 0x1A telegram: 17 08 1A 00 5A 64 00 00 (CRC=51), #data=4
Calling scheduled data refresh from EMS devices..
(00:09:02) Thermostat -> me, type 0x91 telegram: 17 0B 91 00 80 2E 00 C6 51 00 00 02 64 5A 07 00 C7 00 (CRC=DD), #data=14
Publishing boiler data via MQTT
(00:09:02) Thermostat -> me, type 0xA8 telegram: 17 0B A8 00 01 00 FF F6 01 06 00 01 0D 03 00 FF FF 01 02 02 02 00 00 05 23 05 23 01 0E 00 FF (CRC=D2), #data=27
(00:09:02) Thermostat -> me, type 0x06 telegram: 17 0B 06 00 13 03 0B 0A 3A 38 06 00 00 00 00 00 00 (CRC=00), #data=13
Publishing boiler data via MQTT
(00:09:06) Thermostat -> all, type 0xA8 telegram: 17 00 A8 00 01 00 FF F6 01 06 00 01 0D 03 00 FF FF 01 02 02 02 00 00 05 23 05 23 01 0E 00 FF (CRC=2C), #data=27
(00:09:06) Thermostat -> all, type 0xA8 telegram: 17 00 A8 1B 00 2E 29 (CRC=6A), #data=3
(00:09:27) Thermostat -> all, type 0x91 telegram: 17 00 91 00 80 2E 00 C7 51 00 00 04 64 5A 07 00 C7 00 (CRC=3A), #data=14
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
(00:09:27) Thermostat -> all, type 0x06 telegram: 17 00 06 00 13 03 0B 0A 3B 14 06 00 00 00 00 00 00 (CRC=CB), #data=13
(00:09:29) Thermostat -> all, type 0xA3 telegram: 17 00 A3 00 00 00 00 (CRC=77), #data=3
(00:09:29) Thermostat -> Boiler, type 0x35 telegram: 17 08 35 00 11 00 (CRC=C1), #data=2
(00:09:57) Thermostat -> Boiler, type 0x1A telegram: 17 08 1A 00 5A 64 00 00 (CRC=51), #data=4
Publishing boiler data via MQTT
Calling scheduled data refresh from EMS devices..
(00:10:01) Thermostat -> me, type 0x91 telegram: 17 0B 91 00 80 2E 00 C7 51 00 00 04 64 5A 07 00 C7 00 (CRC=20), #data=14
[601924] [MQTT] Setting last will topic home/ems-esp/status
[601924] [MQTT] Connecting to MQTT using user mqtt_user...
[601929] [MQTT] TCP Disconnected. Check mqtt logs.
(00:10:02) Thermostat -> me, type 0xA8 telegram: 17 0B A8 00 01 00 FF F6 01 06 00 01 0D 03 00 FF FF 01 02 02 02 00 00 05 23 05 23 01 0E 00 FF (CRC=D2), #data=27
(00:10:02) Thermostat -> me, type 0x06 telegram: 17 0B 06 00 13 03 0B 0A 3B 37 06 00 00 00 00 00 00 (CRC=D6), #data=13
Publishing boiler data via MQTT
info
EMS-ESP System stats:
  System logging set to Thermostat only
  LED is on
  # connected Dallas temperature sensors=0
  Thermostat is enabled, Boiler is enabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus Stats:
  Bus Connected=yes, # Rx telegrams=98, # Tx telegrams=1, # Crc Errors=0

Boiler stats:
  Boiler type: Nefit Topline Compact [Type ID: 0x08] Product ID:115 Version:03.05
  Hot tap water is off
  Central Heating is active
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water is set to ECO
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 59.50 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 87579 times
  Warm Water active time: 125 days 9 hours 51 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 85 C
  Current flow temperature: 63.0 C
  Return temperature: 39.50 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
[... 37 lines cut ...]
Publishing boiler data via MQTT
(00:12:28) Thermostat -> all, type 0x06 telegram: 17 00 06 00 13 03 0C 0A 02 15 06 00 00 00 00 00 00 (CRC=3D), #data=13
(00:12:30) Thermostat -> all, type 0xA3 telegram: 17 00 A3 00 00 00 00 (CRC=77), #data=3
(00:12:30) Thermostat -> Boiler, type 0x35 telegram: 17 08 35 00 11 00 (CRC=C1), #data=2
[766924] [MQTT] Setting last will topic home/ems-esp/status
[766924] [MQTT] Connecting to MQTT using user mqtt_user...
[766929] [MQTT] TCP Disconnected. Check mqtt logs.
(00:12:58) Thermostat -> Boiler, type 0x1A telegram: 17 08 1A 00 5A 64 00 00 (CRC=51), #data=4
Calling scheduled data refresh from EMS devices..
(00:13:00) Thermostat -> me, type 0x91 telegram: 17 0B 91 00 80 2E 00 C7 51 00 00 02 64 5A 07 00 C8 00 (CRC=A7), #data=14
Publishing boiler data via MQTT
(00:13:01) Thermostat -> me, type 0xA8 telegram: 17 0B A8 00 01 00 FF F6 01 06 00 01 0D 03 00 FF FF 01 02 02 02 00 00 05 23 05 23 01 0E 00 FF (CRC=D2), #data=27
(00:13:01) Thermostat -> me, type 0x06 telegram: 17 0B 06 00 13 03 0C 0A 02 36 06 00 00 00 00 00 00 (CRC=20), #data=13
Publishing boiler data via MQTT

Expecting Setpoint room temperature to 18, but Setpoint is still 23.
Thermostat stats:
  Thermostat type: RC20/Nefit Moduline 300) [Type ID: 0x17] Product ID:77 Version:02.20
  Setpoint room temperature: 23.0 C
  Current room temperature: 19.89 C
  Thermostat time is 12:03:22 10/3/2019
  Mode is set to manual

Shower stats:
  Shower Timer is off

proddy, 2019-03-10:
You should check the code and try and figure out the logic yourself.

The RC20 receives the temperature values on type 0x91 in byte 5. It's multiplied by 2. You can see when that the thermostat is always broadcasting 23 degrees when you manually turn the wheel:

`Thermostat -> me, type 0x91 telegram: 17 0B 91 00 80 2E 00 C7 51 00 00 02 64 5A 07 00 C8 00`

so its behaving as it should. Why its always sending 23 I don't know. I tested on my RC20/Moduline 300 it and does change. Perhaps something with the auto/manual/clock mode?

proddy, 2019-04-06:
closing due to no activity. If its still an issue please reopen.

closed 2019-04-06 by proddy

henderikus, 2019-04-13:
Ok, I will check the code.

It is in manual mode. When switching the mode on the thermostat from manual to auto to manual it is not overruled anymore and I can use it again.

Hereby a photo of the print of my thermostat.
[image]