emsesp/EMS-ESP #62: Junkers FW100 example + support for Junkers Solar Controller ISM1

Vuego123, 2019-02-24:
**Is your feature request related to a problem? Please describe.**

I would like to have support for the junkers FW100 thermostat and junkers ISM1 solar controller. In attachment I have a working example. The example is based on version 1.1.0 of this project. Is it possible to add this to the latest version and keep it in the project?

**My setup:** 

- Boiler: Junkers Cerapur Smart Top 22-3 Zsb
- Thermostat: Junkers FW100
- Solar: Junkers Solar panels with ISM1 controller and SK300 solar boiler

**The code in attachment supports:**

- Junkers ISM1 solar controller: Temperature Solar boiler, Temperature Solar Panels (on the roof)
- Junkers FW100 thermostat: Current & Selected Room temperature, Current & Selected Flow Temperature, Heating power, Outside temperature, Current & Selected Warm Water Temperature (non solar), ...
- It is running stable without issues for 2 months with a dashboard in home assistant (screenshot below).

**What can be found in the code in attachment:**

- 2 new message types: Both the thermostat FW100 and the ISM1 solar controller broadcast the same message type 0XFF, but with a different length. For the thermostat I was able to parse the current and selected room temperature. For the solar controller I was able to parse the current solar panel temperature and the current solar boiler temperature.
- MQTT message for the solar information

**Open questions**

- Like some other people, I also get the error message that I am "not connected to the EMS bus" and TX is not working. Maybe this is because the boiler is using EMS+? Or maybe this is because I was using an old version (1.1.0) of this project.
- There are more values in the 0xFF messages of the thermostat and solar boiler, but so far I can't figure out how they need to be parsed.
- Next thing I would like to do is set temperatures. I noticed that the current version (1.5.3) is already supporting this. That is the reason why I want to merge my example into this project. Hopefully this will also work for my setup.

**Finally**
Many thanks to everyone who is contributing to this project. (and also BBQKees for providing me a PCB). I hope my contribution can help.

[**Link to example code src.zip**](https://github.com/proddy/EMS-ESP/files/2897828/src.zip)

[image]

proddy, 2019-02-24:
this is awesome! thanks for your contribution. I'll take a look at merging this into master and start the support for EMS+.

proddy, 2019-03-04:
the 0xFF is not the message type, it means its an EMS+ message. I haven't worked on the EMS+ parsing yet mainly because I don't have an EMS plus device at home to test with. There are other users who are also needing this support so if you're daring enough please have a go since you can write code. The logic is to determine if its EMS or EMS+ and then adjust the rest the data parsing.

Vuego123, 2019-03-04:
I will try to code the EMS+ parsing. Testing is not an issue for me, but
understanding EMS+ is less easy. Is there information somewhere about how
EMS+ works?

proddy, 2019-03-04:
I've been collecting snippets which i'll share. Perhaps in the GitHub wiki.

proddy, 2019-04-06:
the latest dev has the basis for supporting EMS+ but probably still doesn't detect your Junkers. If you can provide details of the types/versions and data coming in we can add it. Also support for the Solar Module SM1 has been added.

proddy, 2019-05-03:
1.7 should be easier now to add your Junkers devices. But I need the device data first including some samples of telegram messages to reverse engineer.

moustic999, 2019-05-03:
here are some product Id that could help 

product Id 105 : Junkers FW100
product Id 101 : Junkers ISM1

proddy, 2019-05-03:
@moustic999 what are the device IDs for these? Like 0x08 for UBA/Boilers? Then I can add to the list.

moustic999, 2019-05-04:
@proddy no, this is the ProductId which is returned by 02 Version command
I will check in the HT3 code on which Bus Address you could find them :-)

Vuego123, 2019-06-19:
Update for this issue: 

Since version 1.8.1 there is support for 

**Solar Module: Junkers - ISM1**
- Solar Boiler temperature
- Solar panel temperature
- Solar pump (on/off)
- Energy produced during previous hour (Wh) 

**Thermostat: Junkers - FW100**
- Room Temperature
- Setpoint Temperature
- Time

Still on my requirements list:
- bring the features for solar modules on the same level as for boilers & thermostats
  > work in progress
- set temperature on my thermostat
  > I did not yet detect the telegram type for the "set" telegram of the FW100 thermostat. I tried to trigger such a telegram by changing the temperature on the thermostat, but it does not send a "set" telegram to itself :-) . Any input on how I can find this telegram type is welcome.

closed 2019-07-12 by proddy