emsesp/EMS-ESP #63: TelnetSpy interferes with EMS bus communication on the ESP8266

nomis, 2019-02-27:
After much debugging the UART interrupt handler, I've discovered I need to initialise `usedSer` to `NULL` in `lib/TelnetSpy/TelnetSpy.cpp` otherwise the hardware `Serial` functions will overwrite the RX interrupt handler (as well as output messages over serial) and no messages will ever be received.

proddy, 2019-02-27:
that's correct. the hardware Serial uses the same UART0. This is why I built in the 'set serial' command which disables the EMS when Serial is enabled and had to use Telnet instead. Serial is only really used for debugging as you don't plug one end of the circuit into the EMS line and the other end into a PC. I also describe this in the documentation.

closed 2019-02-27 by proddy

nomis, 2019-02-27:
Serial is set to `off` and it doesn't work properly. There is no interaction between the serial on/off setting and the TelnetSpy code.

proddy, 2019-02-27:
Check if you're using the latest version. This shouldn't happen. Line 146 of MyEsp.cpp should say:

```
        // finally if we don't want Serial anymore, turn it off
        if (!_use_serial) {
            Serial.println("Disabling serial port");
            Serial.flush();
            Serial.end();
            SerialAndTelnet.setSerial(NULL);
        } else {
            Serial.println("Using serial port output");
        }
```

If it still happens, create a case so I can reproduce it. Or try and fix it in the code.

nomis, 2019-02-28:
Immediately above that code is the call to `_wifi_callback` which calls `emsuart_init()`. It's still using the serial output after it has been reconfigured.

proddy, 2019-02-28:
you're using an old build. do a git pull. That problem was solved a while
back.