emsesp/EMS-ESP #64: Setup help needed

lobocobra, 2019-03-03:
**my setup**
I have a buderus GB125 with an RC35. 
I have a bbqkees board with a WEMOS with your firmware on it (loaded by bbqkees)
I connected the EMS bus through the Jack.
I connected the bbqkees board to the WEMOS like described (and even switched rx/tx just to be sure
=> I can connect over telnet, tried autoconnect but nothing happens, the LED blinks.

**Question**
My WEMOS is blinking, which seems to mean, that it did not connect. Probably I did something wrong, 
1) Where do I find a step by step howto, what to connect first
2) I connected my  Buderus through the Jack because both EMS ports have a cable in => Can I simply add the board by adding the cable to the used plug? 
3) How do I use "autoconnect", do I have to initialize first the Buderus, Do I find somewhere the code for the RC35?
=> Please give me a hint where to start. I usually find out how things work alone, but I am stuck here.

**Screenshots**
=> if requested I will add it

**Device information**
*Verbose mode shows SILENCE.... I have no connection to EMS*

**Additional context**
I tried the following:
0) Did the cabling connect to USB
1) changed the WIFI / PW an was able to connect through telnet to the WEMOS
2) I connected the jack of the bbqkees board to my Buderus (like I saw it on the Wiki)
3) The WEMOS is blinking, so I tried to switch RX/TX => it continues to blink
4) So I started autodetect over telnet, but nothing happens => I seem to have no connection to EMS
5) I checked if I have the right jack-cable, it is a stereo cable and works fine.
6) I stopped the heating and restarted it, hoping it had to do with the initialization => Failed
7) I searched for a beginner Howto for 2hours to see if I jumped a step. => Nothing found.
8)  checked the cabeling, seems to work according to WIKI and the instructions of bbqkees.
9) Restarted WEMOS/Buderus Heating around 10x to understand if one needs to be there before the other.
10) now I have no clue what to do else.

proddy, 2019-03-03:
lots of questions! First you need to get the EMS messages flowing in. You said you're able to Telnet to the ESP which is good, so I assume you have set the wifi username and password and have turned serial off. If then you still get the flashing LED, plug the ESP into the PC via the USB, turn the serial on and try and see what error messages show up.

If in doubt download the source code, compile and upload yourself. Also do a 'pio run -t erase' to remove any left over settings.

lobocobra, 2019-03-03:
MANY THANKS @proddy!

The serial OFF did the trick! I will write here a short howto, in the next days for others!

proddy, 2019-03-03:
great, I'll use this feedback to make a special section in the README about
setting up for first time users

lobocobra, 2019-03-03:
**First time users setup guide**

**Hardware pre-requisites**
- bbqkees EMS board
- Wemos D1 mini (from bbqkees with pre-installe firmware) 
- EMS compatible heating/boiler
- a stereo jack cable 

**Hardware wiring**
- connect J2 ground to Wemos "G" 
- connect J2 RX to D7 
- connect J2 TX to D8 
- connect J2 VC to 3V3 (! check this twice)
- set the resistor Jumper to 100k
- connect wemos usb-mini to the USB of a laptop/computer with WIFI
Remarks
- If RX/TX are switched, there is no harm for your board, but transmission will not work.
  => RX/TX are already crossed with this setup, just plug them as described above.
- Make sure you connect only 3volt to the EMS interface board. 
- WEMOS D1 will power the bbqkees board, with this wiring. 

**Connect to WEMOS** (assuming win10, if you have linux, you know what to do)
- in Win10 activate telnet client (if you have linux, install it): 
  dism /online /Enable-Feature /FeatureName:TelnetClient
  or
  use putty
  or check this link
  https://www.tobias-hartmann.net/2016/11/windows-10-telnet-aktivieren/
- connect your computer/laptop to the WEMOS wifi (search for the EMS-ESP wifi)
Remarks
Your computer should be now connected to the WIFI of the WEMOS. If not, you need to use google, to achieve this.
The WEMOS will continue to blink, this means that it is not connected to EMS, we fix that in the next steps.

**Configure WEMOS**
- blue LED is blinking (means not connected)
- open a non-administrative CMD window and insert following commands 
- "telnet 192.168.4.1" (connect to wemos, check that you are connected)
- "set wifi your_ssid your_password"
- "reboot" (now your WEMOS should be connected to your WLAN and the EMS-ESP domain is not visible anymore)
- check the IP of your wemos in your router
- "telnet xx.xx.xx.xx" (the new ip of your WEMOS)
- "set serial off"
Remarks
After those steps your WEMOS is still blinking and you can access the WEMOS over the new IP.

**Connect your setup with your heating/boiler**
- when the WEMOS is blinking, this means its not connected to the EMS bus!
- connect the stereo-jack cable to your heating
  => you do NOT need to restart your heating, if all is correct its PLUG & PLAY
  => at this point the WEMOS needs to stop blinking and the LED is ON

lobocobra, 2019-03-03:
Solution summary:
=> "serial off" did the trick!

Many thanks to proddy for the solution. I hope this guide help some others.

closed 2019-03-03 by lobocobra

proddy, 2019-03-03:
I modified my Getting Started text in the README to hopefully make it more understandable.