emsesp/EMS-ESP #67: Additional warm water comfort mode for Nefit Trendline Aquapower Plus HRC 30 CW6

ReneHonig, 2019-03-05:
Hi Paul,
First of all many thanks for this excellent project!

I have a Nefit Trendline Aquapower Plus HRC 30 CW6, which can the ability to store warm water to meet the CW6 requirements. This boiler has an additional warm water comfort setting, called 'intelligent'. In this mode it will track the warm water usage over time and pre-heat the stored water only at times when it is frequently used.

I managed to extract the setting from the 0x33 telegram (the value for this mode is 'EC':
type 0x33 telegram: 08 0B 33 00 08 FF 3C FB 00 28 00 00 46 EC FF FF 00 (CRC=7B), #data=13

I have looked into implementing it in your code, but it is not that straightforward, as I am not sure whether we can distinguish this boiler from types that do not have this option (for write purposes), and the current implementation uses a boolean (so bit more work to change that). 

Here are the boiler details: 
Boiler type: Buderus GB172/Nefit Trendline [Type ID: 0x08] Product ID:123 Version:06.07

I thought I start with reporting it, so we can build up the EMS catalogue.

Best, Rene

proddy, 2019-03-06:
Hi Rene, and welcome. I checked on my boiler and it seems that data value (your EC value) is set to zero (0x00) so I think it should be safe to assume it was a reserved byte and only for those boilers that support it. How would you like to handle this? Just display the value and send it MQTT? I can help with the coding as its pretty straight forward (but I do always say that!).

thanks for the boiler details. Added to 1.5.5.

ReneHonig, 2019-03-07:
Hi Paul,
Thanks for the quick response! 
I think as a first step displaying the value and sending it through MQTT as you suggested would be great. 
Once that is done, I am happy to look into creating the ability to change the setting (probably easier to test for me with the specific boiler).

Thanks for adding the OT converter.

proddy, 2019-03-16:
@ReneHonig I modified version 1.6.0 (still in testing in the 'dev' branch) to handle the 3 comfort modes (Hot, Eco, Intelligent)

closed 2019-03-16 by proddy

ReneHonig, 2019-03-16:
Great! Have installed it and it reports the 'intelligent' mode fine.

Thanks!

proddy, 2019-03-17:
Thanks for testing. I will add mqtt support next