emsesp/EMS-ESP #68: Opentherm converter as thermostat

ReneHonig, 2019-03-05:
Hi Paul,
Another one:

I use a Nefit Opentherm to EMS Converter for my system (with a Honeywell Modulating Thermostat).
I have added it in my ems_devices.h, and it is recognised, but does not seem to be readable (I am not sure whether it is dependent on the actual OT thermostat used).

Here is the line I added (probably quicker than a pull request...):
{EMS_MODEL_OT, 171, 0x02, "Opentherm converter", EMS_THERMOSTAT_READ_NO, EMS_THERMOSTAT_WRITE_NO}

Best, Rene

proddy, 2019-03-06:
There was someone else with a similar problem using the OpenTherm EMS converter. It doesn't seem to be able to handle all the message types. It could be the code or the circuit but difficult to determine without further testing.

ReneHonig, 2019-03-07:
Agree, ultimately I would like to get rid of it and control the heating directly through the EMS bus. Need to look at setting flow temperatures etc, through MQTT, but that is off-topic...

closed 2019-03-14 by ReneHonig