emsesp/EMS-ESP #69: Compilation issue

lobocobra, 2019-03-07:
Hey all

The JSON error is now gone. Thanks for updating it. 
=> by the  way I did hit the "rror: expected unqualified-id before '(' token" error also when I added ttps://github.com/bblanchon/ArduinoJson#v5.13.5 in the platform.ini

What could I possibly have missed?

`------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WEMOS D1 mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting ems-esp.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CRC32> 2.0.0
|-- <CircularBuffer> 1.3.0
|-- <JustWifi> 2.0.2
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 6.9.1
|-- <OneWire> 2.3.4
|-- <myESP>
|   |-- <TelnetSpy>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 6.9.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <JustWifi> 2.0.2
|   |   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
Compiling .pioenvs\d1_mini\src\ems-esp.ino.cpp.o
Compiling .pioenvs\d1_mini\libb98\myESP\MyESP.cpp.o
In file included from C:\users\rrs\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiClient.h:25:0,
from C:\users\rrs\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,
from C:\users\rrs\.platformio\packages\framework-arduinoespressif8266\libraries\ArduinoOTA/ArduinoOTA.h:4,
from lib\myESP\MyESP.h:15,
from lib\myESP\MyESP.cpp:9:
c:\users\rrs\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
C:\users\rrs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
#define max(a,b) ((a)>(b)?(a):(b))
^
c:\users\rrs\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_bvector.h: In member function 'std::vector<bool, _Alloc>::size_type std::vector<bool,
_Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const':
C:\users\rrs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
#define max(a,b) ((a)>(b)?(a):(b))
^
*** [.pioenvs\d1_mini\libb98\myESP\MyESP.cpp.o] Error 1
============================================================================ [ERROR] Took 3.01 seconds ============================================================================
Der Terminalprozess wurde mit folgendem Exitcode beendet: 1`

proddy, 2019-03-07:
I don't get that error. Can you try updating everything using `pio lib update` and `pio update`. And then `pio run -t clean` just to be 100% safe

lobocobra, 2019-03-07:
I seem to have a basic issue, because I get this:

`C:\Arduino` Git\EMS-ESP\src>pio lib update
Error: The `C:\Arduino Git\EMS-ESP\src` is not a PlatformIO project.

To manage libraries in global storage `C:\Users\rrs\.platformio\lib`,
please use `platformio lib --global update` or specify custom storage `platformio lib --storage-dir /path/to/storage/ update`.
Check `platformio lib --help` for details.

lobocobra, 2019-03-07:
Ahh I guess I know in which direction I have to search....
ems-esp.in shows 8 errors... saying....

cannot open source file "OneWire.h ... bla bla
=> it does not find the libs, I will start from scratch and delete all.

lobocobra, 2019-03-07:
**Hmm... it compiles now.... for others... I did....**
- un-installed VSC
- deleted the VSC folder in documents
- re-installed VSC
- installed platformio
- opened the project over OPEN PROJECT in platformio
- tried pio lib update => FAILED
- got pissed
- tried to compile the firmware => WORKED

So probably the "I got pissed" part did the difference

@proddy
thanks your hint pointed me to the right path... there must have been an problem with my VSC installation.

closed 2019-03-07 by lobocobra