emsesp/EMS-ESP #70: OTA upload issue since 1.5.5

SpaceTeddy, 2019-03-07:
hi proddy,

since the new 1.5.5 I can not update the FW via OTA.
any idea?
```
Checking size .pioenvs/nodemcuv2/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  57.4% (used 47040 bytes from 81920 bytes)
PROGRAM: [====      ]  39.4% (used 411056 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = espota
Uploading .pioenvs/nodemcuv2/firmware.bin
19:32:26 [DEBUG]: Options: {'timeout': 10, 'esp_ip': '192.168.0.100', 'host_port': 39495, 'image': '.pioenvs/nodemcuv2/
firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
19:32:26 [INFO]: Starting on 0.0.0.0:39495
19:32:26 [INFO]: Upload size: 415200
Sending invitation to 192.168.0.100
19:32:26 [INFO]: Waiting for device...
Uploading: [=                                                           ] 1% 
19:32:36 [ERROR]: Error Uploading
*** [upload] Error 1```

gl3nni3, 2019-03-07:
did you clean the spiffs? you can do that with
pio run -t erase

proddy, 2019-03-07:
I made an error and fixed it this morning. Make sure you pull the latest code.

SpaceTeddy, 2019-03-07:
well, since this update I got a lot of trouble. TCP connection is really slow/unstable. I get many many timeout if I ping the esp. Unfortunatly I updated the libs and the esp8266 sdk, therefore I can not recompile the older versions. my last binary (nodemcuv2) is 1.4.x and after reflashing this one, all works fine again
do you know how to downgrade the libs in PlatformIO?

lobocobra, 2019-03-07:
Does this help you?
https://community.platformio.org/t/show-or-get-older-versions-of-libraries/569

SpaceTeddy, 2019-03-08:
OK, thanks. I will try this weekend :)

SpaceTeddy, 2019-03-08:
Ok, it is more easy to downgrade the libs in PlatformIO IDE. It is even possible to install an dedicated lib version for each Project. Perfectly easy :)

lobocobra, 2019-03-08:
I was about to install 1.5.5.
=> Would you recommend not to do it currently? A
=> above it was stated that one error was fixed few hours before you pasted the bug.
=> Did you try with the latest code? I want to avoid to jeopardize my running system.
:)

SpaceTeddy, 2019-03-08:
well, hard to say. Usually proddy tests all versions very well and it seems, that only I have trouble with that version. Therefore I think there is something with my setup. My recommendation is to backup you working firmware.bin file. If the new version fails, you can easily reflash it to your old version.

proddy, 2019-03-08:
I've noticed a few wifi instability issues since the latest esp8266 2.0.4 update too but I reckoned it was because I kept on fiddling with my own WiFi network. Thats the reason I print the wifi strength in the telnet session. If more people noticed this then it's worth looking into and fixing. I just did a complete rebuild of VSC and PlatformIO with the latest firmware and it seems to work find. But...if you don't want to downgrade just put `platform=espressif8266@1.8.0` in your platformio.ini file.

lobocobra, 2019-03-09:
@SpaceTeddy 
Thanks, all worked out, its running stable on me now.

Would you mind to pass me your plaform.ini file, be cause my OTA problems are like 20cm in front of the screen. :-)
=> I would need a working example, I removed the semicolon but OTA does not work (I never did it with Visual Code)

```
; for OTA comment out these sections
upload_protocol = espota
upload_port = ems-esp.local

```

proddy, 2019-03-09:
@lobocobra those setting looks fine. Can you ping to 'ems-esp.local'? Perhaps the mdns is not resolving. It is hard to tell without seeing the exact errors platformio gives you. In any case you can also substitute the `ems-esp.local` for the IP address and that will always work.

lobocobra, 2019-03-09:
Many thanks!

**For the records**
=> all works great and stable. OTA works perfectly with 1.5.5.

For those having issues.  Uncomment platform.ini and set ur IP. For example:
`; for OTA comment out these sections
upload_protocol = espota
upload_port = 192.168.20.35
;upload_port = ems-esp.local`

closed 2019-03-10 by proddy

SpaceTeddy, 2019-03-17:
@proddy , I updated to 1.6.0_b dev and it seems to work fine now. I did not made an erase. It worked out of the box again.

```
[223892] [APP] EMS-ESP version: 1.6.0b
[223893] [APP] MyESP version: 1.1.6
[223894] [APP] Build timestamp: 2019-03-17 09:03:53
[223894] [APP] Boot time: start
[223894] [APP] Uptime: 223 seconds
[223895] [APP] System Load: 4%
[223895] [WIFI] WiFi Hostname: ems-esp
[223896] [WIFI] WiFi IP: 192.168.XX.XX
[223896] [WIFI] WiFi signal strength: 100%
[223897] [WIFI] WiFi MAC: 60:01:94:XX:XX:XX
[223897] [EEPROM] EEPROM size: 4096
[223898] [EEPROM] EEPROM Sector pool size is 4, and in use are: 1019 1018 1017 1016 
[223898] [SYSTEM] Board: PLATFORMIO_NODEMCUV2
[223899] [SYSTEM] CPU chip ID: 0x0EA3E6
[223899] [SYSTEM] CPU frequency: 80 MHz
[223899] [SYSTEM] SDK version: 3.0.0-dev(c0f7b44)
[223900] [SYSTEM] Core version: 2_5_0
[223900] [SYSTEM] Boot version: 31
[223900] [SYSTEM] Boot mode: 1
[223901] [FLASH] Flash chip ID: 0x1640EF
[223901] [FLASH] Flash speed: 40000000 Hz
[223902] [FLASH] Flash mode: DOUT
[223902] [FLASH] Flash size (CHIP): 4194304
[223903] [FLASH] Flash size (SDK): 4194304
[223903] [FLASH] Flash Reserved: 4096
[223904] [MEM] Firmware size: 424096
[223904] [MEM] Max OTA size: 2715648
[223904] [MEM] OTA Reserved: 16384
[223905] [MEM] Free Heap: 17520
[223905] [MEM] Stack: 1924
```