emsesp/EMS-ESP #71: Read failed. Giving up message

lobocobra, 2019-03-10:
**My setup**
I have a Buderus125GB with a floor heating. => in consequence it uses heating ciruit 2 on my RC35.
=> I adapted the code and can now read / write to Circut 2 over telnet/mqtt, all works perfectly
=> https://github.com/lobocobra/EMS-ESP-HC2   // still early testing, will do a pull request once debugged
I have no boiler and my "boiler warm water temperature" is shown as -3200° (which I fixed in my code).

Summa summarum all work now except 2 error messages that pop up in the console all 2 min.

**Error message**
I get following error, with my adapted-code AND with the 1.5.5 code

`<--- RC35StatusMessage(0x3E) received
(00:45:00) Sending read of type 0x47 to 0x10: telegram: 0B 90 47 00 63 (CRC=32), #data=1
...Retrying read. Attempt 1/2...
(00:45:00) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=C1), #
data=16
<--- RC35StatusMessage(0x3E) received
`
**If I send the RAW Telegram `0B 90 47 00 63 `  I get no error.**

and the second
```
(00:39:02) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 63 (CRC=AF), #data=1
Read failed. Giving up, removing from queue

```
Same issue, if I send it as raw telegram, there is no error.

**Any ideas if it has to do with my missing boiler?**

proddy, 2019-03-10:
does `thermostat read 47` work?

Also the -3200° problem, I sent you an earlier message to capture the data output so I can fix it. Checking for negative values is not the right fix, and I want to be able to detect it properly.

lobocobra, 2019-03-10:
Yes thermostat read 47 works and shows:
> thermostat read 47
> Requesting type (0x47) from dest 0x10
> (00:26:31) Sending read of type 0x47 to 0x10: telegram: 0B 90 47 00 63 (CRC=32), #data=1
> (00:26:31) Thermostat -> me, type 0x47 telegram: 10 0B 47 00 03 26 32 20 00 28 00 01 00 05 05 2D 01 01 04 4B 0A 4B 01
>  00 3C FF 10 0A 00 01 02 (CRC=23), #data=27
> (00:26:31) Boiler -> all, type 0x18 telegram: 08 00 18 00 25 01 CC 64 64 09 01 35 01 83 00 7D 00 80 00 02 0B FF 2D 48
>  00 00 FF 00 10 83 00 (CRC=67), #data=27
> <--- UBAMonitorFast(0x18) received

regarding the negative celsiusinn
=> I will check how I can capture data output on windows
=> I will do it after I fixed the issue that "thermostat temp" always and only changes the night temp of the heating (found the code area in ems.epp and am working on it)

proddy, 2019-03-10:
which of the following 3 is showing up as -3200 ?

    _renderIntValue("Warm Water selected temperature", "C", EMS_Boiler.wWSelTemp);
    _renderIntValue("Warm Water desired temperature", "C", EMS_Boiler.wWDesiredTemp);
    _renderFloatValue("Warm Water current temperature", "C", EMS_Boiler.wWCurTmp);

lobocobra, 2019-03-10:
"Warm Water current temperature" shows a negative value of -3200°

My quick and dirty fix I did here in ems.cpp:
> void _process_UBAMonitorWWMessage(uint8_t type, uint8_t * data, uint8_t length) {
>     EMS_Boiler.wWCurTmp  = _toFloat(1, data);
>     //lobocobra start fix negative celius if you have no boiler
>     if (EMS_Boiler.wWCurTmp <0) EMS_Boiler.wWCurTmp =0;
>     //lobocobra end
>     EMS_Boiler.wWStarts  = _toLong(13, data);
>     EMS_Boiler.wWWorkM   = _toLong(10, data);
>     EMS_Boiler.wWOneTime = bitRead(data[5], 1);
>     EMS_Boiler.wWCurFlow = data[9];
> }

proddy, 2019-03-10:
ok thanks. This value is coming from the 0x34 type. Could you do a 'boiler read 34' in verbose logging and copy the telegram here? The temperature is the 6th and 7th byte and my code assumes 0x8000 represents a null value.

lobocobra, 2019-03-11:
Here we are :-)

I uploaded firmware 1.5.6 (not modified) and executed `boiler read 34`:
>(00:02:25) Sending read of type 0x34 to 0x08: telegram: 0B 88 34 00 63 (CRC=27), #data=1
>(00:02:25) Boiler -> me, type 0x34 telegram: 08 0B 34 00 00 83 00 7D 00 00 00 00 03 00 00 00 00 00 00 00 (CRC=1B), #data=16

As I have no boiler attached 0x8300 might mean that there is no usefull data available?

proddy, 2019-03-11:
@lobocobra ok, I made the fix in the latest dev branch. _toFloat() in ems.cpp. Thanks for pointing this out.

lobocobra, 2019-03-13:
@proddy 
1) The read errors are gone with version 1.5.6 :-)
    => Many thanks!
2) The negative Temperatur is still there, but if you ask me, it's not a bug, it's a feature, saying there is no boiler.

For the records `boiler read 34 shows`
```
 Boiler -> me, type 0x34 telegram: 08 0B 34 00 00 83 00 7D 00 00 00 00 03 00 00 00 00 00 00 00 (CRC=1B), #data
=16
```

lobocobra, 2019-03-13:
Marked issue as closed as the read errors are now fixed by Proddy.
=> The temperature reading of -3200° is due to my missing boiler, if someone wants to quickfix it use above fix from me.

closed 2019-03-13 by lobocobra

proddy, 2019-03-13:
yes I fixed the -3200 in the dev branch which is 1.5.7