emsesp/EMS-ESP #72: error code of boiler

IGastel, 2019-03-12:
question: is there a way to add the error code of the boiler to the MQTT data?

the error code should be the 2 bytes after the service code.
the service code ASCII data provides interesting data, but the error code adds more detail to the service code. the installation manual of my Nefit trendline explains the meaning of the error code. some service codes have multiple error codes. others have none.

on a personal note: i am really impressed by your work and that of BBQKees and i am exited to start working with this project. since i use the Honeywell evohome, i miss being able to check my boiler status. adding it to my home automation (domoticz) would be very helpfull.

proddy, 2019-03-12:
Welcome @IGastel - yes, its easy to add the service error code. Are you able to modify the code yourself or do you need some help?

IGastel, 2019-03-12:
Hi Proddy, I am a total beginner with respect to programming. i started learning arduino (by myself) about a year ago and only wrote 2 programs so far. simple 1 file programs. So if you can help me, i would  appreciate it.
I can understand mostly what bbqkees wrote. but your program has a lot more functionality, so i get lost.

proddy, 2019-03-12:
no problem. I just added it to my dev branch so you can take a look how it was done.

closed 2019-03-16 by proddy