emsesp/EMS-ESP #73: V1.5.6 constantly reboots

Gert05, 2019-03-12:
Problems in paradise.  I tried to upgrade from 1.5.4 to 1.5.6:
compile ok
burn ok
startup in AP mode ok
set wifi credentials ok
reboot ok
LED stops blinking

So far so good & as expected, But then after < 1min:
a reboot
all settings are lost
device seems to keep rebooting every half minute
I've tried multiple new installs, but result stays the same.
Have reverted back to 1.5.4, which is now running fine.

proddy, 2019-03-12:
I've seen this behavior before. My dev branch contains some extra code to capture the stack dump so I can find why it keeps resetting. Its something to do with the new JSON library (or my implementation more like). Thanks for pointing this out. Still investigating...

lobocobra, 2019-03-13:
I just loaded code 1.5.6 and I do not have any reboot issues. 
=> Did you try a SET ERASE one time?

proddy, 2019-03-13:
I'm getting constant restarts too, both with 1.5.6 and 1.5.7 and still debugging. I'm comparing the platform libraries to see differences between espressif8266 2.0.4 and 1.8.0

lobocobra, 2019-03-14:
hmm after I restarted pio home/visual code and uploaded a firmware, I have the same issue....
=> so it might have really to do with libraries

proddy, 2019-03-14:
@lobocobra I found a workaround for the wifi restarts. The issue with losing the config, can you describe the steps to reproduce?

Please try with 1.5.7 from the dev branch for both issues, thanks.

lobocobra, 2019-03-14:
Thanks! All works now with version 1.5.7. 
=> also the -3200° message is gone and all looks stable now, including the config. 
=> What might be an interesting observation (in case this comes up again), is that my additional variables were not affected by the data loss. Means only the config of WIFI,etc was gone, but not my newly saved variables. So in case this is popping up again, we might want to check if the code overwrites spiff with empty data?

Anyhow, I would say that the code could go life.

Gert05, 2019-03-15:
Tried 1.5.7 dev; could not get it to run - mainly because the ESP wouldn't reliably connect to wifi after credentials were set. 
Ping with lots of losses; no telnet connection possible over wifi.
Back to 1.5.4 which runs normal.

proddy, 2019-03-15:
Did you change the platform to use the 2.4.x core?

Gert05, 2019-03-15:
You mean the Espressif 8266 v2.0.4 platform? Well, PlatformIO was automatically upgraded to that version, but the platformio.ini for 157dev still specifies 1.8.0. All I did was adapt the .ini for the d1_mini_pro.

```
Processing d1_mini_pro (platform: espressif8266@1.8.0; board: d1_mini_pro; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_pro.html
PLATFORM: Espressif 8266 > WeMos D1 mini Pro
HARDWARE: ESP8266 80MHz 80KB RAM (16MB Flash)
```

proddy, 2019-03-15:
Yes, there are a few known issues with the latest 2.5.0 arduino core which is why the WiFi keeps dropping out. There are not crashes or WDTs as first expected. Something related to the signal power (e.g. [here](https://github.com/esp8266/Arduino/issues/5825)). My dev version reverts back to the old 2.4.2 libraries in the `platformio.ini-example` which you probably missed, but now OTA is not stable. So for now your platformio.ini file it should say platform=espressif8266@1.8.0 until I've fixed it.

Gert05, 2019-03-15:
Well, I did a compile with ..@1.8.0 (see previous comment for the copy/past of the compile output...) and still have the unstable wifi. As far as I can see I run both 154 and 157dev on ..@1.8.0 but only 157dev has unstable wifi.

proddy, 2019-03-15:
ok, thanks for the feedback

proddy, 2019-03-15:
@geezerrr fancy guinea-pigging that latest 1.5.7 dev build? I think I found a problem for the wifi dropouts

proddy, 2019-03-16:
bumped to 1.6.0. Been running for >24hrs with no resets or timeouts so think its ok now. Using the latest 2.5.0 core libraries.

Gert05, 2019-03-16:
This is turning messy... Tried the dev build but still had wifi problems. Tried going back to 154 but even that is no longer possible. Must have thoroughly messed up the d1_mini_pro. Back to the soldering iron now for a new d1_mini_pro.
Any suggestions how to completely erase a board via PlatformIO? Cannot connect to it over wifi/telnet any more.

proddy, 2019-03-16:
plug it into your PC, then `pio run -t erase`. Then upload the firmware again.

Gert05, 2019-03-16:
OK. Apparently bricked the old d1_mini_pro.
On a new one still having wifi problems with 160-dev. Able to connect to AP,set wifi, reboot, but not connecting to wifi; resetting; loss of settings.
Now happily back on 154.

SpaceTeddy, 2019-03-17:
@Gert05 ,
I had also massive trouble with V1.5.6. I saw also, reconnects, bad pings and very slow telnet performance. I went back to 1.4.1 (my latest stable version) and today straight forward to 1.6.0b_dev without any setting erase. (https://github.com/proddy/EMS-ESP/issues/70)
It runs good and fast again. 
hope that helps

proddy, 2019-03-17:
@Gert05 I suspect its the EEPROM code since it has [16MB](https://docs.platformio.org/en/latest/boards/espressif8266/d1_mini_pro.html) of flash compared to 4MB on the D1 Mini. Make sure in your `platformio.ini` you are **not** bulding with `-DCRASH` and that **you are** using the latest core libs with `platform=espressif8266`. If it still doesn't work I'll get a Pro and run some tests. Thanks for hanging in there!

Gert05, 2019-03-19:
Strange... I seem to have brought the dead D1 mini pro back to life... :-)
Now running 1.6.0b3 without (wifi or other) problems on this previously dead wemos.

closed 2019-03-19 by Gert05

proddy, 2019-03-19:
LOL. Was it the platformio.ini file that needed adjusting?

Gert05, 2019-03-19:
No, only mods to the platform.ini I make are replacing d1_mini by d1_mini_pro.
In my opinion nothing has changed. I know that sounds stupid, but I have no explanation...