emsesp/EMS-ESP #75: receive multiple WW requests 08 0B 33 00

SpaceTeddy, 2019-03-17:
Dear proddy,

not sure if it is normal, but I saw a probably untypical behaviour for WW message requests. 
It seems that I receive four messages for only one WW requests.
Is this normal?

Version 1.6.0_b 

```
log v:
(00:27:07) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(00:27:07) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF (CRC=FA), #data=11
<--- UBAParameterWW(0x33) received
Publishing boiler data via MQTT
(00:27:07) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF (CRC=FA), #data=11
<--- UBAParameterWW(0x33) received
(00:27:07) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF (CRC=FA), #data=11
<--- UBAParameterWW(0x33) received

log r:
08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF FA 
08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF FA 
08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF FA 
08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF FA 
```

proddy, 2019-03-17:
I've reproduced this, its a bug. I'm looking into it - thanks for reporting

proddy, 2019-03-17:
I had it once, but can't reproduce it anymore. Have you been able too?

SpaceTeddy, 2019-03-17:
yes, after a boiler and ESP restart still there. I get also some "read retries"...

```
log v
System Logging set to Verbose
(00:01:49) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 02 27 00 C4 00 00 2B 2B 2B 00 00 64 11 2B (CRC=22), #data=15
<--- RC35StatusMessage(0x3E) received
Publishing boiler data via MQTT
(00:01:51) 0x18 -> all, type 0xAF telegram: 18 00 AF 00 00 C4 00 06 09 (CRC=1A), #data=5
(00:01:54) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 02 02 02 02 21 00 05 03 00 02 42 AC 00 21 7A 00 (CRC=C7), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:01:55) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 2B 64 64 00 (CRC=91), #data=4
<--- UBASetPoints(0x1A) received
(00:01:55) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 53 64 3D 09 01 A5 60 80 00 02 02 00 FC 00 D2 0C 2D 48 00 C8 00 02 00 (CRC=2B), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:55) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 54 64 3D 09 01 A5 60 80 00 02 02 00 FC 00 D2 0C 2D 48 00 C8 00 02 00 (CRC=04), #data=25
<--- UBAMonitorFast(0x18) received
(00:01:58) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 54 64 3E 09 01 A5 60 80 00 02 02 00 FC 00 D6 0B 2D 48 00 C8 00 02 00 (CRC=D5), #data=25
<--- UBAMonitorFast(0x18) received
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:02:00) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(00:02:00) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 27 00 C4 00 00 2B 2B 2B 00 00 64 11 2B (CRC=16), #data=15
<--- RC35StatusMessage(0x3E) received
Publishing boiler data via MQTT
(00:02:00) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 01 00 00 00 00 00 00 00 00 00 00 (CRC=92), #data=13
(00:02:00) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
(00:02:01) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 04 23 27 22 00 28 00 02 00 05 05 2D 01 01 04 41 05 4B 01 00 3C FF 11 02 05 01 01 (CRC=04), #data=27
<--- RC35Set(0x3D) received
(00:02:01) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 63 (CRC=2F), #data=1
(00:02:01) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 03 10 11 08 19 06 00 (CRC=A5), #data=8
<--- RCTime(0x06) received
(00:02:01) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(00:02:01) Boiler -> me, type 0x18 telegram: 08 0B 18 00 2B 01 5B 64 3E 09 01 A5 60 80 00 02 02 00 FC 00 DA 0B 2D 48 00 C8 00 02 00 (CRC=4B), #data=25
<--- UBAMonitorFast(0x18) received
(00:02:02) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(00:02:02) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 64 01 66 80 00 00 00 00 4D 01 61 C5 1C 3D 11 00 00 00 19 FA 65 01 40 4B 00 00 (CRC=75), #data=27
<--- UBAMonitorSlow(0x19) received
(00:02:02) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(00:02:02) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF (CRC=FA), #data=11
<--- UBAParameterWW(0x33) received
Publishing boiler data via MQTT
(00:02:02) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF (CRC=FA), #data=11
<--- UBAParameterWW(0x33) received
(00:02:02) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF (CRC=FA), #data=11
<--- UBAParameterWW(0x33) received
(00:02:03) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 63 (CRC=AF), #data=1
(00:02:03) Boiler -> me, type 0x16 telegram: 08 0B 16 00 FF 5A 64 00 06 FA 0A 01 05 64 37 02 (CRC=D4), #data=12
<--- UBAParametersMessage(0x16) received
(00:02:03) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 63 (CRC=A7), #data=1
(00:02:03) Boiler -> me, type 0x14 telegram: 08 0B 14 00 43 6E 32 (CRC=E9), #data=3
<--- UBATotalUptimeMessage(0x14) received
(00:02:03) Boiler -> me, type 0x14 telegram: 08 0B 14 00 43 6E 32 (CRC=E9), #data=3
<--- UBATotalUptimeMessage(0x14) received
(00:02:03) Boiler -> me, type 0x14 telegram: 08 0B 14 00 43 6E 32 (CRC=E9), #data=3
<--- UBATotalUptimeMessage(0x14) received
(00:02:03) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 5E 64 3F 09 01 A5 60 80 00 02 02 00 FC 00 DC 0C 2D 48 00 C8 00 02 00 (CRC=70), #data=25
<--- UBAMonitorFast(0x18) received
(00:02:04) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 01 00 00 00 00 00 00 00 00 00 00 (CRC=BE), #data=13
(00:02:04) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 61 64 3F 09 01 A5 60 80 00 02 02 00 FC 00 DE 0B 2D 48 00 C8 00 02 00 (CRC=A7), #data=25
<--- UBAMonitorFast(0x18) received
(00:02:05) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 02 02 02 02 21 00 05 03 00 02 42 AC 00 21 7A 00 (CRC=C7), #data=17
<--- UBAMonitorWWMessage(0x34) received
[... 107 lines cut ...]
(00:03:14) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 02 02 02 02 21 00 01 03 00 02 42 AC 00 21 7A 00 (CRC=0F), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:03:22) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 03 10 11 09 2D 06 00 (CRC=DA), #data=8
<--- RCTime(0x06) received
[... cut ...]
o(00:03:34) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 02 02 02 02 21 00 01 03 00 02 42 AC 00 21 7A 00 (CRC=0F), #data=17
<--- UBAMonitorWWMessage(0x34) received
g(00:03:35) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 01 00 00 00 00 00 00 00 00 00 00 (CRC=BE), #data=13
(00:03:35) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 86 64 4B 09 01 25 60 80 00 02 02 00 FF 01 1B 0C 2D 48 00 C8 00 02 00 (CRC=53), #data=25
<--- UBAMonitorFast(0x18) received
 n(00:03:36) 0x18 -> Boiler, type 0x35 telegram: 18 08 35 00 00 00 (CRC=1A), #data=2

System Logging set to None

```

SpaceTeddy, 2019-03-17:
```
(00:14:28) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2B 09 01 25 60 80 00 02 01 01 50 00 A6 0D 2D 48 00 C8 00 02 00 (CRC=12), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:31) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 60 01 B4 80 00 00 00 00 41 01 61 C5 1C 3D 1D 00 00 00 19 FA 71 01 40 4B 00 00 (CRC=52), #data=27
<--- UBAMonitorSlow(0x19) received
(00:14:31) Boiler -> all, type 0x1C telegram: 08 00 1C 00 93 01 04 18 27 00 00 00 00 00 00 (CRC=33), #data=11
(00:14:31) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 02 01 02 01 21 00 01 03 00 02 42 AC 00 21 7A 00 (CRC=B7), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:14:32) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 01 00 00 00 00 00 00 00 00 00 00 (CRC=BE), #data=13
(00:14:32) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2A 09 01 25 60 80 00 02 01 01 52 00 A5 0C 2D 48 00 C8 00 02 00 (CRC=8C), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:34) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2A 09 01 25 60 80 00 02 01 01 52 00 A5 0C 2D 48 00 C8 00 02 00 (CRC=8C), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:35) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2A 09 01 25 60 80 00 02 01 01 52 00 A4 0D 2D 48 00 C8 00 02 00 (CRC=15), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:36) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2A 09 01 25 60 80 00 02 01 01 52 00 A3 0C 2D 48 00 C8 00 02 00 (CRC=DA), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:36) 0x18 -> Boiler, type 0x35 telegram: 18 08 35 00 00 00 (CRC=1A), #data=2
(00:14:37) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2A 09 01 25 60 80 00 02 01 01 53 00 A2 0D 2D 48 00 C8 00 02 00 (CRC=27), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:38) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2A 09 01 25 60 80 00 02 01 01 52 00 A3 0C 2D 48 00 C8 00 02 00 (CRC=DA), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:39) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2A 09 01 25 60 80 00 02 01 01 53 00 A2 0D 2D 48 00 C8 00 02 00 (CRC=27), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:41) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 02 01 02 01 21 00 01 03 00 02 42 AC 00 21 7A 00 (CRC=B7), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:14:41) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2A 09 01 25 60 80 00 02 01 01 52 00 A3 0D 2D 48 00 C8 00 02 00 (CRC=5A), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:42) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 2A 09 01 25 60 80 00 02 01 01 52 00 A0 0D 2D 48 00 C8 00 02 00 (CRC=71), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:43) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 29 09 01 25 60 80 00 02 01 01 53 00 A1 0D 2D 48 00 C8 00 02 00 (CRC=12), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:49) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AE 64 29 09 01 25 60 80 00 02 01 01 53 00 9F 0D 2D 48 00 C8 00 02 00 (CRC=D1), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:50) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AE 64 29 09 01 25 60 80 00 02 01 01 53 00 9D 0D 2D 48 00 C8 00 02 00 (CRC=E3), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:51) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 02 01 02 01 21 00 01 03 00 02 42 AC 00 21 7A 00 (CRC=B7), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:14:51) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AE 64 29 09 01 25 60 80 00 02 01 01 53 00 9D 0D 2D 48 00 C8 00 02 00 (CRC=E3), #data=25
<--- UBAMonitorFast(0x18) received
(00:14:53) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 02 27 00 C4 00 00 2B 2B 2B 00 00 64 11 2B (CRC=22), #data=15
<--- RC35StatusMessage(0x3E) received
```

proddy, 2019-03-17:
if I look at the CRC I don't see any duplicates so those messages coming in are all uniquely different in one way or form. It is however strange that so many messages are sent.

SpaceTeddy, 2019-03-17:
I downgraded my setup to 1.5.4 and with this version, there are no dupes.
Log:

```
* Connected to: EMS-ESP version 1.5.4
* Hostname: ems-esp  IP: 192.168.0.100  MAC: 60:01:94:xx:xx:xx  Board: PLATFORMIO_NODEMCUV2
* Connected to WiFi SSID: UPC7160713 (signal 100%)
* Boot time: start
* Free RAM:21 KB, Load:0%
```

```
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:03:10) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(00:03:10) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 27 00 C4 00 00 23 23 23 00 00 64 11 23 (CRC=51), #data=15
<--- RC35StatusMessage(0x3E) received
Publishing boiler data via MQTT
(00:03:10) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 01 00 00 00 00 00 00 00 00 00 00 (CRC=92), #data=13
(00:03:10) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
(00:03:10) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 04 23 27 22 00 28 00 02 00 05 05 2D 01 01 04 41 05 4B 01 00 3C FF 11 02 05 01 01 (CRC=04), #data=27
<--- RC35Set(0x3D) received
(00:03:10) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 02 27 00 C4 00 00 23 23 23 00 00 64 11 23 (CRC=65), #data=15
<--- RC35StatusMessage(0x3E) received
(00:03:11) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 11 (CRC=30), #data=2
(00:03:11) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 63 (CRC=2F), #data=1
(00:03:11) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 03 13 11 28 26 06 00 (CRC=20), #data=8
<--- RCTime(0x06) received
(00:03:11) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(00:03:11) Boiler -> me, type 0x18 telegram: 08 0B 18 00 23 01 73 64 19 09 01 25 60 80 00 01 F6 01 3B 00 34 0C 2D 48 00 C8 00 02 00 (CRC=B4), #data=25
<--- UBAMonitorFast(0x18) received
(00:03:12) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(00:03:12) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 50 01 76 80 00 00 00 00 37 01 61 C7 1C 3D E4 00 00 00 19 FB 13 01 40 4D 00 00 (CRC=10), #data=27
<--- UBAMonitorSlow(0x19) received
(00:03:12) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(00:03:12) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF (CRC=FA), #data=11
<--- UBAParameterWW(0x33) received
Publishing boiler data via MQTT
(00:03:12) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 63 (CRC=AF), #data=1
(00:03:12) Boiler -> me, type 0x16 telegram: 08 0B 16 00 FF 5A 64 00 06 FA 0A 01 05 64 37 02 (CRC=D4), #data=12
<--- UBAParametersMessage(0x16) received
(00:03:13) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 63 (CRC=A7), #data=1
(00:03:13) Boiler -> me, type 0x14 telegram: 08 0B 14 00 43 6F 08 (CRC=D1), #data=3
<--- UBATotalUptimeMessage(0x14) received
(00:03:13) Boiler -> all, type 0x18 telegram: 08 00 18 00 23 01 73 64 19 09 01 25 60 80 00 01 F6 01 3B 00 34 0C 2D 48 00 C8 00 02 00 (CRC=07), #data=25
<--- UBAMonitorFast(0x18) received
(00:03:13) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 01 F6 01 F6 21 00 01 03 00 02 42 D1 00 21 7A 00 (CRC=41), #data=17
```

proddy, 2019-03-17:
I still can't reproduce. Would you mind testing again with the latest dev build, then `set test_mode on` then `log r` and see if you have duplicate messages. The CRC should be same.

SpaceTeddy, 2019-03-17:
```
ESP8266 System stats:

 [APP] EMS-ESP version: 1.6.0b3
 [APP] MyESP version: 1.1.6b1
 [APP] Build timestamp: 2019-03-17 21:53:29
 [APP] Boot time: start
 [APP] Uptime: 55 seconds
 [APP] System Load: 0%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 192.168.xx.xx
 [WIFI] WiFi signal strength: 100%
 [WIFI] WiFi MAC: 60:01:94:0E:xx:xx
 [SYSTEM] Board: PLATFORMIO_NODEMCUV2
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 3.0.0-dev(c0f7b44)
 [SYSTEM] CPU chip ID: 0x0EA3E6
 [SYSTEM] Core version: 2_5_0
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [FLASH] Flash chip ID: 0x1640EF
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DOUT
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 423120
 [MEM] Max OTA size: 2715648
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 22288
 [MEM] Stack: 2496
```
log r
```
log r
System Logging set to Raw mode
18 08 35 00 00 00 1A 
08 00 07 00 03 01 01 00 00 00 00 00 00 00 00 00 00 BE 
08 00 18 00 25 01 77 64 19 09 01 25 60 80 00 02 22 01 3F 00 36 0C 2D 48 00 C8 00 02 00 D8 
08 00 19 00 00 40 01 7A 80 00 00 00 00 37 01 61 C9 1C 3E 48 00 00 00 19 FB 6D 01 40 4E 00 00 2E 
08 00 1C 00 93 01 04 18 27 00 00 00 00 00 00 33 
08 00 34 00 0A 02 22 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 F7 
10 00 3E 00 04 02 27 00 C2 00 00 25 25 25 00 00 64 11 25 50 
10 08 35 00 01 11 10 
08 00 18 00 25 01 77 64 19 09 01 25 60 80 00 02 22 01 3F 00 36 0C 2D 48 00 C8 00 02 00 D8 
08 00 34 00 0A 02 22 02 22 20 00 01 03 00 02 42 DB 00 21 7B 00 75 
10 88 14 00 03 6E 
08 10 14 00 43 6F 90 02 
10 00 A2 00 00 00 00 00 00 00 00 00 00 00 18 
08 00 18 00 25 01 76 64 19 09 01 25 60 80 00 02 22 01 3F 00 36 0C 2D 48 00 C8 00 02 00 07 
08 00 34 00 0A 02 22 02 22 20 00 01 03 00 02 42 DB 00 21 7B 00 75 
10 00 06 00 13 03 15 11 37 2E 06 00 9F 
08 00 18 00 25 01 77 64 19 09 01 25 60 80 00 02 22 01 3F 00 38 0C 2D 48 00 C8 00 02 00 46 
08 00 34 00 0A 02 22 02 22 20 00 01 03 00 02 42 DB 00 21 7B 00 75 
18 00 AF 00 00 C2 00 06 08 2B 
10 00 A3 00 08 00 01 8F 
10 08 1A 00 25 64 64 00 E1 
08 00 18 00 25 01 77 64 19 09 01 25 60 80 00 02 22 01 3F 00 35 0C 2D 48 00 C8 00 02 00 F3 
08 00 34 00 0A 02 22 02 22 20 00 01 03 00 02 42 DB 00 21 7B 00 75 
10 89 29 00 01 90 
09 10 29 00 6B DF 
08 00 18 00 25 01 77 64 19 09 01 25 60 80 00 02 21 01 40 00 39 0C 2D 48 00 C8 00 02 00 91 
08 00 34 00 0A 02 22 02 22 20 00 01 03 00 02 42 DB 00 21 7B 00 75 
18 08 35 00 00 00 1A 
08 00 07 00 03 01 01 00 00 00 00 00 00 00 00 00 00 BE 
08 00 18 00 25 01 77 64 19 09 01 25 60 80 00 02 21 01 3F 00 35 0C 2D 48 00 C8 00 02 00 71 
08 00 19 00 00 40 01 7B 80 00 00 00 00 37 01 61 C9 1C 3E 49 00 00 00 19 FB 6E 01 40 4E 00 00 21 
08 00 1C 00 93 01 04 18 27 00 00 00 00 00 00 33 
08 00 34 00 0A 02 21 02 22 20 00 01 03 00 02 42 DB 00 21 7B 00 4F 
10 00 3E 00 04 02 27 00 C2 00 00 25 25 25 00 00 64 11 25 50 
10 08 35 00 01 11 10 
08 00 18 00 25 01 76 64 19 09 01 25 60 80 00 02 21 01 3F 00 34 0C 2D 48 00 C8 00 02 00 B7 
08 00 34 00 0A 02 21 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 CD 
08 00 18 00 25 01 76 64 19 09 01 25 60 80 00 02 21 01 3F 00 35 0C 2D 48 00 C8 00 02 00 AE 
08 00 34 00 0A 02 22 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 F7 
10 00 06 00 13 03 15 11 38 2E 06 00 E7 
08 00 18 00 25 01 76 64 19 09 01 25 60 80 00 02 22 01 3F 00 34 0C 2D 48 00 C8 00 02 00 35 
08 00 34 00 0A 02 22 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 F7 
18 00 AF 00 00 C2 00 06 08 2B 
10 00 A3 00 08 00 01 8F 
10 08 1A 00 25 64 64 00 E1 
08 00 18 00 25 01 76 64 19 09 01 25 60 80 00 02 21 01 3F 00 3C 0C 2D 48 00 C8 00 02 00 7F 
08 00 34 00 0A 02 21 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 CD 
```

set test_mode off
```
log r
System Logging set to Raw mode
18 08 35 00 00 00 1A 
08 00 07 00 03 01 01 00 00 00 00 00 00 00 00 00 00 BE 
08 00 18 00 25 01 7F 64 19 09 01 25 60 80 00 02 21 01 46 00 3A 0C 2D 48 00 C8 00 02 00 55 
08 00 19 00 00 3F 01 82 80 00 00 00 00 37 01 61 C9 1C 3E 4D 00 00 00 19 FB 72 01 40 4E 00 00 E0 
08 00 1C 00 93 01 04 18 27 00 00 00 00 00 00 33 
08 00 34 00 0A 02 21 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 CD 
10 00 3E 00 04 02 27 00 C2 00 00 25 25 25 00 00 64 11 25 50 
10 08 35 00 01 11 10 
08 00 18 00 25 01 81 64 19 09 01 25 60 80 00 02 21 01 46 00 3B 0C 2D 48 00 C8 00 02 00 CA 
08 00 34 00 0A 02 21 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 CD 
08 00 18 00 25 01 81 64 19 09 01 25 60 80 00 02 21 01 47 00 35 0C 2D 48 00 C8 00 02 00 30 
08 00 34 00 0A 02 21 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 CD 
10 00 06 00 13 03 16 11 00 2F 06 00 5A 
08 00 18 00 25 01 81 64 19 09 01 25 60 80 00 02 21 01 47 00 3F 0C 2D 48 00 C8 00 02 00 CA 
08 00 34 00 0A 02 21 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 CD 
10 0B 3E 00 04 02 27 00 C2 00 00 25 25 25 00 00 64 11 25 64 
08 00 07 00 0B 01 01 00 00 00 00 00 00 00 00 00 00 92 
10 0B 3E 00 04 02 27 00 C2 00 00 25 25 25 00 00 64 11 25 64 
10 0B 3D 00 04 23 27 22 00 28 00 02 00 05 05 2D 01 01 04 41 05 4B 01 00 3C FF 11 02 05 01 01 04 
10 0B 06 00 13 03 16 11 00 36 06 00 99 
08 0B 18 00 25 01 82 64 19 09 01 25 60 80 00 02 21 01 47 00 3E 0C 2D 48 00 C8 00 02 00 18 
08 0B 19 00 00 3F 01 87 80 00 00 00 00 37 01 61 C9 1C 3E 4E 00 00 00 19 FB 73 01 40 4E 00 00 76 
08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF FA 
08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF FA 
08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF FA 
08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF FA 
18 00 AF 00 00 C2 00 06 08 2B 
10 08 1A 00 25 64 64 00 E1 
10 00 A3 00 08 00 01 8F 
08 00 07 00 03 01 01 00 00 00 00 00 00 00 00 00 00 BE 
08 00 18 00 25 01 82 64 19 09 01 25 60 80 00 02 21 01 47 00 3A 0C 2D 48 00 C8 00 02 00 CF 
08 00 34 00 0A 02 21 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 CD 
08 00 18 00 25 01 82 64 19 09 01 25 60 80 00 02 21 01 47 00 39 0C 2D 48 00 C8 00 02 00 E4 
08 00 34 00 0A 02 21 02 21 20 00 01 03 00 02 42 DB 00 21 7B 00 CD 
```

proddy, 2019-03-17:
wow, that's clear as day. I'll have a think of what is happening. Other than adding a delay to the loop nothing much has changed in the UART code.

SpaceTeddy, 2019-03-18:
hi,
verbose log with 1.6.0_b3
less type 0x33 dupes, but still other repeated messages there. mhhhh.
```
ESP8266 System stats:

 [APP] EMS-ESP version: 1.6.0b3
 [APP] MyESP version: 1.1.6b1
 [APP] Build timestamp: 2019-03-18 21:06:40
 [APP] Boot time: start
 [APP] Uptime: 33 seconds
 [APP] System Load: 0%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 192.168.xx.xx
 [WIFI] WiFi signal strength: 100%
 [WIFI] WiFi MAC: 60:01:94:0E:xx:xx
 [SYSTEM] Board: PLATFORMIO_NODEMCUV2
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 3.0.0-dev(c0f7b44)
 [SYSTEM] CPU chip ID: 0x0EA3E6
 [SYSTEM] Core version: 2_5_0
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [FLASH] Flash chip ID: 0x1640EF
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DOUT
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 423136
 [MEM] Max OTA size: 2715648
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 25272
 [MEM] Stack: 1936
```

log v
```
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:04:13.284) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 28 (CRC=84), #data=1
(00:04:13.381) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 27 00 BA 00 00 41 41 41 00 01 64 11 41 (CRC=A5), #data=15
(00:04:13.530) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 01 00 00 00 00 00 00 00 00 00 00 (CRC=92), #data=13
(00:04:13.657) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 28 (CRC=88), #data=1
(00:04:13.792) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 04 23 27 22 00 28 00 02 00 05 05 2D 01 01 04 41 05 4B 01 00 3C FF 11 02 05 01 01 (CRC=04), #data=27
(00:04:14.055) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 28 (CRC=64), #data=1
(00:04:14.116) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 03 15 12 0D 39 00 00 (CRC=91), #data=8
(00:04:14.329) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 28 (CRC=DC), #data=1
(00:04:14.385) Boiler -> me, type 0x18 telegram: 08 0B 18 00 41 02 8A 64 35 09 01 25 60 80 00 02 0D 02 1B 00 D2 0F 2D 48 00 C8 00 02 00 (CRC=F8), #data=25
(00:04:14.702) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 28 (CRC=D8), #data=1
(00:04:14.761) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 39 02 93 80 00 00 00 00 48 01 61 CE 1C 41 34 00 00 00 19 FE 20 01 40 53 00 00 (CRC=9E), #data=27
(00:04:15.050) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 28 (CRC=70), #data=1
(00:04:15.091) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF (CRC=FA), #data=11
Publishing boiler data via MQTT
(00:04:15.246) Boiler -> all, type 0x18 telegram: 08 00 18 00 41 02 8A 64 35 09 01 25 60 80 00 02 0D 02 1B 00 D3 0F 2D 48 00 C8 00 02 00 (CRC=52), #data=25
(00:04:15.433) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 39 02 93 80 00 00 00 00 48 01 61 CE 1C 41 34 00 00 00 19 FE 20 01 40 53 00 00 (CRC=60), #data=27
(00:04:15.638) Boiler -> all, type 0x1C telegram: 08 00 1C 00 93 01 04 18 27 00 00 00 00 00 00 (CRC=33), #data=11
(00:04:15.869) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 0D 02 0D 20 00 01 03 00 02 43 14 00 21 7B 00 (CRC=21), #data=17
(00:04:16.120) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 28 (CRC=E4), #data=1
(00:04:16.162) Boiler -> me, type 0x16 telegram: 08 0B 16 00 FF 5A 64 00 06 FA 0A 01 05 64 37 02 (CRC=D4), #data=12
(00:04:16.187) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 01 00 00 00 00 00 00 00 00 00 00 (CRC=92), #data=13
(00:04:16.567) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 28 (CRC=EC), #data=1
(00:04:16.600) Boiler -> me, type 0x14 telegram: 08 0B 14 00 43 75 14 (CRC=F9), #data=3
[... 14 lines cut ...]
(00:04:34.974) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 0D 02 0D 20 00 01 03 00 02 43 14 00 21 7B 00 (CRC=21), #data=17
(00:04:35.207) Boiler -> all, type 0x18 telegram: 08 00 18 00 41 02 8C 64 35 09 01 25 60 80 00 02 0D 02 1C 00 D4 10 2D 48 00 C8 00 02 00 (CRC=CF), #data=25
(00:04:36.998) Boiler -> all, type 0x18 telegram: 08 00 18 00 41 02 8C 64 35 09 01 25 60 80 00 02 0D 02 1C 00 D3 0F 2D 48 00 C8 00 02 00 (CRC=87), #data=25
(00:04:37.993) Boiler -> all, type 0x18 telegram: 08 00 18 00 41 02 8C 64 35 09 01 25 60 80 00 02 0D 02 1C 00 D2 10 2D 48 00 C8 00 02 00 (CRC=99), #data=25
(00:04:38.988) Boiler -> all, type 0x18 telegram: 08 00 18 00 41 02 8C 64 35 09 01 25 60 80 00 02 0D 02 1D 00 D2 0F 2D 48 00 C8 00 02 00 (CRC=FA), #data=25
(00:04:39.983) Boiler -> all, type 0x18 telegram: 08 00 18 00 41 02 8D 64 35 09 01 25 60 80 00 02 0D 02 1D 00 D2 10 2D 48 00 C8 00 02 00 (CRC=22), #data=25
(00:04:40.978) Boiler -> all, type 0x18 telegram: 08 00 18 00 41 02 8D 64 34 09 01 25 60 80 00 02 0D 02 1D 00 D0 0F 2D 48 00 C8 00 02 00 (CRC=EA), #data=25
(00:04:41.973) Boiler -> all, type 0x18 telegram: 08 00 18 00 41 02 8D 64 34 09 01 25 60 80 00 02 0D 02 1D 00 D0 10 2D 48 00 C8 00 02 00 (CRC=ED), #data=25
(00:04:42.969) Boiler -> all, type 0x18 telegram: 08 00 18 00 41 02 8D 64 34 09 01 25 60 80 00 02 0D 02 1E 00 D0 0F 2D 48 00 C8 00 02 00 (CRC=46), #data=25
[... cut ...]
(00:05:04.900) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 0D 02 0D 20 00 01 03 00 02 43 14 00 21 7B 00 (CRC=21), #data=17
(00:05:05.108) Boiler -> all, type 0x18 telegram: 08 00 18 00 3F 02 88 64 21 09 01 25 60 80 00 02 0D 02 21 00 77 0F 2D 48 00 C8 00 02 00 (CRC=85), #data=25
(00:05:10.855) Boiler -> all, type 0x18 telegram: 08 00 18 00 3F 02 7F 64 1D 09 01 25 60 80 00 02 0D 02 21 00 53 0F 2D 48 00 C8 00 02 00 (CRC=9C), #data=25
Calling scheduled data refresh from EMS devices....
```

proddy, 2019-03-18:
I've seen it happen once, but can't reproduce it anymore. I think I'll hook up an Arduino and use bbqkees original code to see if I'm going mad or not.

SpaceTeddy, 2019-03-23:
so, I went back each version and I get the doubled messages since Ver. 1.5.6.
Till 1.5.5 everything is fine. I compiled V1.5.6 with esp 1.8.0 and 2.5.0 but there is no influence.
May the new ArduinoJson lib cause this? -> just an idea.

proddy, 2019-03-23:
so since 1.5.6 you're seeing double messages? That really helps in narrowing down the root cause.

SpaceTeddy, 2019-03-24:
i verified it and since *V1.5.5* I see the doubled messages.

proddy, 2019-03-24:
ok, so it worked in 1.5.4 but 1.5.5 shows the double messages. I'll compare the sources in a while. 

Did you compile both from source using the latest Arduino core libraries or take the firmware? I'm trying to rule out any external factors like library changes.

SpaceTeddy, 2019-03-24:
I compiled all versions new with the latest libs or sdk. There is one exception, because V1.5.4 must be compiled with Arduinojson 5. I also compiled each version with 1.8.0 and 2.5.0 but there was no difference ... I guess :)
I had also some trouble to compile the dev branch with 1.8.0. Basically I can compile, flash and see the data via mitt, but I can not login via telnet.

proddy, 2019-03-24:
I saw one small thing which I changed just now in dev - to do with the uart pointer, so worth trying with the latest 1.6. if you still get double try changing

line 27 in emsuart.cpp to static uint8_t length;
line 28 in emsaurt.h to     uint8_t writePtr;

proddy, 2019-03-24:
can you test for duplicates with the last dev? I added a line to remove the contents of the last Rx package just in case its overwritting

SpaceTeddy, 2019-03-24:
wow, seems to be fixed with latest dev version (2019-03-24:1800).
verbose out put log looks different, but there are no dupes anymore .)

good job, proddy! 👍

proddy, 2019-03-24:
phew, glad that worked! Yes I change verbose logging to also show the milliseconds for more accurate timings. I also made a change to remove the strange 0x00 telegrams. If you have time could you do one last check? Then I think I'll push this version 1.6.0 to the master.

SpaceTeddy, 2019-03-25:
... last check done and the 0x00 messages are gone! 👍 

```
Calling scheduled data refresh from EMS devices...
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:01:12.462) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 20 (CRC=8C), #data=1
(00:01:12.564) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 27 00 BB 00 00 3D 3D 3D 00 04 64 11 3D (CRC=80), #data=15
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
(00:01:12.718) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 01 00 00 00 00 00 00 00 00 00 00 (CRC=92), #data=13
(00:01:12.984) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 20 (CRC=80), #data=1
(00:01:13.100) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 04 23 27 22 00 28 00 02 00 05 05 2D 01 01 04 41 05 4B 01 00 3C FF 11 02 05 01 01 (CRC=04), #data=27
(00:01:13.358) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 20 (CRC=6C), #data=1
(00:01:13.424) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 03 06 19 29 16 00 00 (CRC=F6), #data=8
(00:01:13.632) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 20 (CRC=D4), #data=1
(00:01:13.687) Boiler -> me, type 0x18 telegram: 08 0B 18 00 3F 02 70 64 19 09 01 A5 60 80 00 02 24 02 2E 00 3D 0E 2D 48 00 C8 00 02 00 (CRC=1E), #data=25
(00:01:14.004) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 20 (CRC=D0), #data=1
(00:01:14.063) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 36 02 77 80 00 00 00 00 37 01 62 1C 1C 53 47 00 00 00 1A 0E 53 01 40 95 00 00 (CRC=B1), #data=27
(00:01:14.353) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 20 (CRC=78), #data=1
(00:01:14.393) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 37 FB 00 0F FF 02 41 00 FF (CRC=FA), #data=11
Publishing boiler data via MQTT
(00:01:14.676) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 20 (CRC=EC), #data=1
(00:01:14.718) Boiler -> me, type 0x16 telegram: 08 0B 16 00 FF 5A 64 00 06 FA 0A 01 05 64 37 02 (CRC=D4), #data=12
(00:01:14.950) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 20 (CRC=E4), #data=1
(00:01:14.981) Boiler -> me, type 0x14 telegram: 08 0B 14 00 43 99 5F (CRC=73), #data=3
(00:01:15.839) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 01 00 00 00 00 00 00 00 00 00 00 (CRC=BE), #data=13
(00:01:17.538) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 3D 64 64 00 (CRC=21), #data=4
(00:01:17.665) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 11 (CRC=30), #data=2
(00:01:17.842) Boiler -> all, type 0x18 telegram: 08 00 18 00 3F 02 6E 64 19 09 01 A5 60 80 00 02 24 02 2E 00 3C 0E 2D 48 00 C8 00 02 00 (CRC=AF), #data=25
(00:01:18.058) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 02 24 02 24 21 00 05 03 00 02 44 F4 00 21 87 00 (CRC=32), #data=17
(00:01:18.473) 0x18 -> Boiler, type 0x35 telegram: 18 08 35 00 00 00 (CRC=1A), #data=2
(00:01:27.034) Thermostat -> 0x09, type 0x29 telegram: 10 89 29 00 01 (CRC=90), #data=1
(00:01:27.054) 0x09 -> Thermostat, type 0x29 telegram: 09 10 29 00 6B (CRC=DF), #data=1
```

closed 2019-03-25 by SpaceTeddy

proddy, 2019-03-25:
phew!