emsesp/EMS-ESP #77: Buderus SM10 Solar Module Support

mediaDS, 2019-03-19:
Hi

Software (1.55 if I remember corectly) is working fine with  my Logatrend boiler. Except for the SM10 solar module, which is quite common here in Germany. Is it possible to add it to the MQTT data? 

It is described technically in the Wiki, unfortunately I am new to programming an ESP8266, so I can't do it myself without help.  The SM10 is just a pump and some sensors, it would be nice to have the pump modulation, if there is solar energy at all and the temperatures received from the solar panels.

If you need additional information please don't hesitate to ask.

Dieter

proddy, 2019-03-19:
First, could you post back the version and product ID for the SM10 so I can add it to the known devices list. An `autodetect` will show you these. I think that function was there in 1.55.

Then do a `log v` and  `send 0B 90 97 00 FF` so I can see the telegrams to decode according to https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#sm10monitor 

(I'm assuming 0x10 is the type ID for the SM10 here)

mediaDS, 2019-03-19:
I got the following after autodetect (did it a few times, always the same response):

`Scanning EMS bus for devices.
Boiler type device found. Model Buderus GB172/Nefit Trendline with TypeID 0x08, Product ID 123, Version 04.05
* Setting Boiler type to Model Buderus GB172/Nefit Trendline, TypeID 0x08, Product ID 123, Version 04.05
Boiler type device found. Model BC25 Base Controller with TypeID 0x09, Product ID 125, Version 01.05
* Setting Thermostat type to Model RC35, TypeID 0x10, Product ID 86, Version 01.15
`

The logging after sending those commands is the following:

`send 0B 90 97 00 FF
(17:16:28) Sending raw telegram: 0B 90 97 00 FF (CRC=C5), #data=1
(17:16:28) Thermostat -> me, type 0x97 telegram: 10 0B 97 00 (CRC=9B)
(17:16:28) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(17:16:29) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(17:16:31) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 42 64 00 01 01 20 40 80 00 01 EE 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=B7), #data=25
<--- UBAMonitorFast(0x18) received
(17:16:31) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(17:16:31) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 EF 01 EE 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=29), #data=19
<--- UBAMonitorWWMessage(0x34) received
(17:16:40) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 41 64 00 01 01 20 40 80 00 01 EE 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=CF), #data=25
<--- UBAMonitorFast(0x18) received
(17:16:41) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(17:16:41) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 EE 01 EE 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=71), #data=19
<--- UBAMonitorWWMessage(0x34) received
(17:16:42) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 02 26 00 CC 00 00 00 00 00 00 00 64 11 2B 00 (CRC=FF), #data=16
<--- RC35StatusMessage(0x3E) received
Publishing boiler data via MQTT
(17:16:43) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 08 01 00 00 CC 00 D2 00 D2 7D 00 7D 00 (CRC=6E), #data=13
<--- RCOutdoorTempMessage(0xA3) received
(17:16:50) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 2B 64 64 00 (CRC=91), #data=4
<--- UBASetPoints(0x1A) received
(17:16:50) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(17:16:51) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 41 64 00 01 01 20 40 80 00 01 EE 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=CF), #data=25
<--- UBAMonitorFast(0x18) received
(17:16:51) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 6A 80 00 80 00 00 00 00 40 00 2C 9F 00 F2 CE 00 00 00 00 E4 D1 00 2B 3E 80 00 (CRC=24), #data=27
<--- UBAMonitorSlow(0x19) received
(17:16:51) Boiler -> all, type 0x1C telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=D2), #data=11
(17:16:51) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(17:16:52) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 EE 01 EE 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=71), #data=19
<--- UBAMonitorWWMessage(0x34) received
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(17:17:00) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(17:17:00) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 26 00 CC 00 00 00 00 00 00 00 64 11 2B 00 (CRC=97), #data=16
<--- RC35StatusMessage(0x3E) received
Publishing boiler data via MQTT
(17:17:00) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(17:17:00) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
(17:17:00) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 01 20 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=50), #data=27
<--- RC35Set(0x3D) received
(17:17:01) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 40 64 00 01 01 20 40 80 00 01 EE 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=10), #data=25
<--- UBAMonitorFast(0x18) received
(17:17:01) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(17:17:01) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 EE 01 EE 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=71), #data=19
<--- UBAMonitorWWMessage(0x34) received
(17:17:01) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 63 (CRC=2F), #data=1
(17:17:02) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 03 0F 13 36 19 01 00 (CRC=A9), #data=8
<--- RCTime(0x06) received
(17:17:02) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(17:17:02) Boiler -> me, type 0x18 telegram: 08 0B 18 00 2B 01 40 64 00 01 01 20 40 80 00 01 EE 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=A3), #data=25
[... cut ...]
(17:17:11) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 3F 64 3B 09 01 25 40 80 00 01 EE 80 00 00 B3 FF 2D 48 00 C8 00 02 00 (CRC=29), #data=25
<--- UBAMonitorFast(0x18) received
(17:17:11) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 3E 64 3B 09 01 25 40 80 00 01 EE 80 00 00 B2 FF 2D 48 00 C8 00 02 00 (CRC=EF), #data=25
<--- UBAMonitorFast(0x18) received
(17:17:13) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 40 64 3A 09 01 25 40 80 00 01 EE 80 00 00 B1 FF 2D 48 00 C8 00 02 00 (CRC=A5), #data=25
<--- UBAMonitorFast(0x18) received
`

Hope this helps.

proddy, 2019-03-19:
I made a mistake, the SM10's ID is 0x30 so try this instead:

`send 0B B0 97 00 16` and also `send 0B B0 02 00 03`

mediaDS, 2019-03-19:
No prob, thanks for your help.

Here is the logs, let me know if you need more:

send 0B B0 97 00 16

(17:36:38) Sending raw telegram: 0B B0 97 00 16 (CRC=35), #data=1
(17:36:38) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 BD 00 00 FB 01 00 00 CF 00 00 (CRC=36), #data=13
(17:36:38) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(17:36:39) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(17:36:46) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 33 64 00 01 01 20 40 80 00 01 EB 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=62), #data=25
<--- UBAMonitorFast(0x18) received
(17:36:46) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 5F 80 00 80 00 00 00 00 3C 00 2C A1 00 F2 D3 00 00 00 00 E4 D6 00 2B 40 80 00 (CRC=0D), #data=27
<--- UBAMonitorSlow(0x19) received
(17:36:46) Boiler -> all, type 0x1C telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=D2), #data=11
(17:36:46) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(17:36:47) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 EA 01 EA 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=50), #data=19
<--- UBAMonitorWWMessage(0x34) received
(17:36:47) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(17:36:47) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 33 64 00 01 01 20 40 80 00 01 EA 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=EB), #data=25
<--- UBAMonitorFast(0x18) received
(17:36:49) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 02 26 00 CC 00 00 00 00 00 00 00 64 11 2A 00 (CRC=FD), #data=16
<--- RC35StatusMessage(0x3E) received
Publishing boiler data via MQTT
(17:36:50) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 08 01 00 00 CC 00 D2 00 D2 7D 00 7D 00 (CRC=6E), #data=13
<--- RCOutdoorTempMessage(0xA3) received
(17:36:55) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 2A 64 64 00 (CRC=99), #data=4
<--- UBASetPoints(0x1A) received
(17:36:56) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(17:36:56) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 EA 01 EA 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=50), #data=19
<--- UBAMonitorWWMessage(0x34) received
(17:36:56) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 33 64 00 01 01 20 40 80 00 01 EA 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=EB), #data=25
<--- UBAMonitorFast(0x18) received
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(17:37:00) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 32 64 00 01 01 20 40 80 00 01 EA 80 00 00 00 FF 30 41 00 CA 00 02 00 (CRC=30), #data=25
<--- UBAMonitorFast(0x18) received
(17:37:01) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
(17:37:01) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 26 00 CC 00 00 00 00 00 00 00 64 11 2A 00 (CRC=95), #data=16
<--- RC35StatusMessage(0x3E) received
Publishing boiler data via MQTT
(17:37:01) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(17:37:02) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 32 64 00 01 01 24 40 80 00 01 EA 80 00 00 00 FF 30 43 01 1B 00 02 00 (CRC=4C), #data=25
<--- UBAMonitorFast(0x18) received
(17:37:02) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 63 (CRC=C3), #data=1
(17:37:02) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 01 20 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=50), #data=27
<--- RC35Set(0x3D) received
(17:37:02) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 63 (CRC=2F), #data=1
(17:37:02) Corrupt telegram: telegram: 90 06 00 63 (CRC=2F)
(17:37:02) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 03 10 13 0E 1A 01 00 (CRC=B7), #data=8
<--- RCTime(0x06) received
(17:37:03) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 63 (CRC=97), #data=1
(17:37:03) Corrupt telegram: telegram: 88 18 00 63 (CRC=97)
(17:37:03) Boiler -> me, type 0x18 telegram: 08 0B 18 00 2A 01 32 64 00 01 01 25 40 80 00 01 EB 80 00 00 00 FF 30 4C 01 1C 00 02 00 (CRC=07), #data=25
<--- UBAMonitorFast(0x18) received
(17:37:03) Corrupt telegram: telegram: 0B 08 00 18 00 2A 01 32 64 00 01 01 2D 40 80 00 01 EA 80 00 00 00 FF 30 4C 01 1C 00 02 00 (CRC=C0), #data=26
(17:37:03) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 63 (CRC=93), #data=1
(17:37:03) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 5F 80 00 80 00 00 00 00 3C 00 2C A2 00 F2 D3 00 00 00 00 E4 D6 00 2B 41 80 00 (CRC=CD), #data=27
<--- UBAMonitorSlow(0x19) received
(17:37:04) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 63 (CRC=3B), #data=1
(17:37:04) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 32 FB 00 28 00 02 46 00 FF FF 00 (CRC=5E), #data=13
<--- UBAParameterWW(0x33) received
Publishing boiler data via MQTT
[... 126 lines cut ...]
[... cut ...]
(17:40:04) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 32 FB 00 28 00 02 46 00 FF FF 00 (CRC=5E), #data=13
<--- UBAParameterWW(0x33) received
Publishing boiler data via MQTT
(17:40:04) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 63 (CRC=AF), #data=1
(17:40:04) Corrupt telegram: telegram: 88 16 00 63 (CRC=AF)
(17:40:04) Boiler -> me, type 0x16 telegram: 08 0B 16 00 FF 50 64 00 06 FA 0A 01 03 64 0A 02 00 00 00 00 00 00 00 00 00 00 00 00 50 00 1E (CRC=B3), #data=27
<--- UBAParametersMessage(0x16) received

proddy, 2019-03-19:
```
send 0B B0 02 00 03

(17:39:01) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 63 (CRC=CF), #data=1
```

looks funny. Can you retry `send 0B B0 02 00 03` please

So for the returned SM10 data its sayings
 Pump modulation = 0% (byte 4)
 Pump is on (byte 6, bit 1)
 Collected Temperature = 18.9% (bytes 2&3)

does that look correct?

mediaDS, 2019-03-19:
> looks funny. Can you retry `send 0B B0 02 00 03` please

Sure, see below.

> So for the returned SM10 data its sayings
> Pump modulation = 0% (byte 4)
> Pump is on (byte 6, bit 1)
> Collected Temperature = 18.9% (bytes 2&3)
> 
> does that look correct?

Temperature may be right, pump modulation too but the pump is off. Right after logging I had a look at the RC35, collector temp was 16°C (overclouded, a few minutes ago it was sunny), pump modulation was 0, boiler temp middle was 47°C.

send 0B B0 02 00 03

(18:39:09) Sending raw telegram: 0B B0 02 00 03 (CRC=46), #data=1
(18:39:09) 0x30 -> me, type 0x02 telegram: 30 0B 02 00 49 02 00 (CRC=CC), #data=3
<--- Version(0x02) received
Unrecognized device found. TypeID 0x02, Product ID 73, Version 02.00
(18:39:09) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(18:39:10) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(18:39:11) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 CC 64 00 01 01 24 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=16), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:11) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 CF 80 00 00 80 00 80 00 80 00 00 (CRC=55), #data=21
(18:39:11) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 D3 01 D3 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=C9), #data=19
<--- UBAMonitorWWMessage(0x34) received
(18:39:12) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 CA 64 00 01 01 24 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=E6), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:12) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 2A 64 64 00 (CRC=99), #data=4
<--- UBASetPoints(0x1A) received
(18:39:15) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 C5 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=95), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:15) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 02 26 00 CD 00 00 00 00 00 00 00 64 11 2A 00 (CRC=35), #data=16
<--- RC35StatusMessage(0x3E) received
Publishing boiler data via MQTT
(18:39:16) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 08 01 00 00 CD 00 D3 00 D3 7D 00 7D 00 (CRC=27), #data=13
<--- RCOutdoorTempMessage(0xA3) received
(18:39:17) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 C1 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=C2), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:21) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 0E 80 00 00 80 00 80 00 80 00 00 (CRC=D9), #data=21
(18:39:21) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 D3 01 D3 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=C9), #data=19
<--- UBAMonitorWWMessage(0x34) received
(18:39:21) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 BA 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=D6), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:23) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 B6 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=2F), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:30) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 AC 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=63), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:31) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 4F 80 00 80 00 00 00 07 3C 00 2C A8 00 F2 EB 00 00 00 00 E4 EE 00 2B 47 80 00 (CRC=F6), #data=27
<--- UBAMonitorSlow(0x19) received
(18:39:31) Boiler -> all, type 0x1C telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=D2), #data=11
(18:39:31) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(18:39:32) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 D3 01 D3 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=C9), #data=19
<--- UBAMonitorWWMessage(0x34) received
(18:39:32) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(18:39:32) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 A9 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=EB), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:36) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 A3 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=E2), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:41) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(18:39:41) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 D3 01 D3 21 00 00 03 00 00 0D FD 00 01 61 00 80 00 (CRC=C9), #data=19
<--- UBAMonitorWWMessage(0x34) received
(18:39:41) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 9E 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=0B), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:46) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 99 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=24), #data=25
<--- UBAMonitorFast(0x18) received
(18:39:51) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 03 11 13 11 0E 01 00 (CRC=98), #data=8
<--- RCTime(0x06) received
(18:39:51) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 A2 00 00 EF 01 00 00 CF 00 00 (CRC=5A), #data=13
[... cut ...]
(18:41:04) Boiler -> me, type 0x16 telegram: 08 0B 16 00 FF 50 64 00 06 FA 0A 01 03 64 0A 02 00 00 00 00 00 00 00 00 00 00 00 00 50 00 1E (CRC=B3), #data=27
<--- UBAParametersMessage(0x16) received
(18:41:04) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 63 (CRC=A7), #data=1
(18:41:04) Boiler -> me, type 0x14 telegram: 08 0B 14 00 02 44 97 (CRC=05), #data=3
<--- UBATotalUptimeMessage(0x14) received
(18:41:05) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13

proddy, 2019-03-19:
I added SM10 support. It should automatically detect the Solar Module and refresh results along with the boiler values. The information should be displayed in the `info` section but obviously, I can't test any of this so please can you try it out? Take the dev branch. If you need a firmware file let me know.

mediaDS, 2019-03-19:
Wow, thanks so much, you are very quick :)

I will try the dev branch first and do it myself (I gotta learn it anyway). If I run into problems then I will report back. I'll keep you posted.

Again, thanks for the quick support.

mediaDS, 2019-03-19:
Little problem, the dev version does not store settings so I can't connect it to my Wifi.

After setting my MQTT broker with:

set mqtt_host 10.42.0.3
mqtt_host changed.

[FS] Failed to open config file for writing
[FS] Failed to write to file

I get an error. Setting my WiFi info will store them:

Stored settings:

  wifi_ssid=mediaDS 
  wifi_password=*************
  mqtt_host=10.42.0.3
  mqtt_username=<not set>
  mqtt_password=<not set>
  serial=on
  led=on
  led_gpio=2
  dallas_gpio=14
  dallas_parasite=off
  thermostat_type=<not set>
  boiler_type=<not set>
  silent_mode=off
  shower_timer=off
  shower_alert=off
  publish_time=120

After reboot everything is erased again. Any ideas?

mediaDS, 2019-03-19:
Okay, I hardcodec SSID, pass and MQTT host so at least I can test it. Here is what I get via Telnet:

autodetect
Started scan of EMS bus for known devices
Boiler found. Model Buderus GB172/Nefit Trendline with TypeID 0x08, Product ID 123, Version 04.05
* Setting Boiler type to Model Buderus GB172/Nefit Trendline, TypeID 0x08, Product ID 123, Version 04.05
[FS] Failed to open config file for writing
[FS] Failed to write to file
Device found. Model BC25 Base Controller with TypeID 0x09, Product ID 125, Version 01.05
* Setting Thermostat type to Model RC35, TypeID 0x10, Product ID 86, Version 01.15
[FS] Failed to open config file for writing
[FS] Failed to write to file

And here is what I get via MQTT:

{"wWSelTemp":"50","selFlowTemp":"43.0","outdoorTemp":"0.40","wWActivated":"on","wWComfort":"Hot","wWCurTmp":"45.90","wWCurFlow":"0.0","wWHeat":"off","curFlowTemp":"46.70","retTemp":"?","burnGas":"off","heatPmp":"on","fanWork":"on","ignWork":"off","wWCirc":"off","selBurnPow":"100","curBurnPow":"0","sysPress":"0.0","boilTemp":"?","pumpMod":"60","ServiceCode":"0Y","ServiceCodeNumber":204}

I can't find any reference to SM10. Here is a log file for a few seconds:

(00:10:11.270) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 2B 64 64 00 (CRC=91), #data=4
(00:10:11.394) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 01 01 (CRC=00), #data=2
(00:10:13.166) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 03 16 13 3B 12 01 00 (CRC=41), #data=8
(00:10:13.502) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 A4 64 17 09 01 25 40 80 00 01 CB 80 00 00 55 FF 2D 48 00 C8 00 02 00 (CRC=3E), #data=25
(00:10:17.544) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 C6 00 C6 80 00 00 80 00 80 00 80 00 00 (CRC=9F), #data=21
(00:10:17.760) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 01 CA 01 CA 20 00 00 03 00 00 0E 06 00 01 62 00 80 00 (CRC=8C), #data=19
(00:10:18.022) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 A5 64 17 09 01 25 40 80 00 01 CB 80 00 00 4E FF 2D 48 00 C8 00 02 00 (CRC=8B), #data=25
(00:10:27.443) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 06 80 00 80 00 00 00 00 40 00 2C C7 00 F3 7E 00 00 00 00 E5 78 00 2B 65 80 00 (CRC=AE), #data=27
(00:10:27.753) Boiler -> all, type 0x1C telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=D2), #data=11
(00:10:28.063) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 C6 00 C6 80 00 00 80 00 80 00 80 00 00 (CRC=9F), #data=21
(00:10:28.288) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 01 CA 01 CB 20 00 00 03 00 00 0E 06 00 01 62 00 80 00 (CRC=9A), #data=19
(00:10:28.538) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(00:10:28.817) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 A9 64 17 09 01 25 40 80 00 01 CA 80 00 00 51 FF 2D 48 00 C8 00 02 00 (CRC=F5), #data=25
(00:10:36.426) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AE 64 17 09 01 25 40 80 00 01 CB 80 00 00 51 FF 2D 48 00 C8 00 02 00 (CRC=53), #data=25
(00:10:37.416) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 C6 00 C6 80 00 00 80 00 80 00 80 00 00 (CRC=9F), #data=21
(00:10:37.675) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 01 CB 01 CA 20 00 00 03 00 00 0E 06 00 01 62 00 80 00 (CRC=D4), #data=19
(00:10:37.894) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 AF 64 17 09 01 25 40 80 00 01 CA 80 00 00 53 FF 2D 48 00 C8 00 02 00 (CRC=37), #data=25
(00:10:41.446) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 00 20 00 CD 00 00 00 00 00 00 00 64 00 2B 00 (CRC=65), #data=16
Publishing boiler data via MQTT
(00:10:47.378) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 C6 00 C6 80 00 00 80 00 80 00 80 00 00 (CRC=9F), #data=21
(00:10:47.692) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 01 CA 01 CA 20 00 00 03 00 00 0E 06 00 01 62 00 80 00 (CRC=8C), #data=19
(00:10:47.908) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 B8 64 17 09 01 25 40 80 00 01 CA 80 00 00 53 FF 2D 48 00 C8 00 02 00 (CRC=5D), #data=25
(00:10:48.473) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 B8 64 17 09 01 25 40 80 00 01 CB 80 00 00 50 FF 2D 48 00 C8 00 02 00 (CRC=FF), #data=25
(00:10:48.940) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 04 00 00 00 CD 00 D3 00 D3 7D 00 7D 00 (CRC=D5), #data=13
(00:10:49.960) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 FF F8 00 00 CE 01 00 00 CF 00 00 (CRC=E4), #data=13
(00:10:50.172) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3
(00:10:57.371) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 C6 00 C6 80 00 00 80 00 80 00 80 00 00 (CRC=9F), #data=21
(00:10:57.586) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 01 CB 01 CA 20 00 00 03 00 00 0E 06 00 01 62 00 80 00 (CRC=D4), #data=19
(00:10:57.873) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 BE 64 17 09 01 25 40 80 00 01 CA 80 00 00 52 FF 2D 48 00 C8 00 02 00 (CRC=B4), #data=25
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
[... cut ...]
(00:11:47.353) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 01 CA 01 CA 20 00 00 03 00 00 0E 06 00 01 62 00 80 00 (CRC=8C), #data=19
(00:11:47.666) Boiler -> all, type 0x18 telegram: 08 00 18 00 2B 01 CF 64 17 09 01 25 40 80 00 01 CA 80 00 00 56 FF 2D 48 00 C8 00 02 00 (CRC=CD), #data=25
(00:11:49.298) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 04 00 00 00 CD 00 D3 00 D3 7D 00 7D 00 (CRC=D5), #data=13
(00:11:50.122) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 FF F8 00 00 CE 01 00 00 CF 00 00 (CRC=E4), #data=13
(00:11:50.348) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

proddy, 2019-03-19:
the "failed to write" error is odd. I've seen this once before. To be sure could you update everything and wipe the ESP with `platformio update` `platformio lib update` `pio run -t erase` and then re-flash.

The autodetect should discover the SM10. I'll check the code again. I haven't added the MQTT yet - I want to make sure the detection and reading values works first.

mediaDS, 2019-03-19:
I don't use platformio and don't even know what it is but I will have a look. I'll report back after a good night sleep.

Have a good night,
Dieter

proddy, 2019-03-20:
well then you're uploading it incorrectly. If you're not using platformio or looked at my README best familiarize yourself on how to build using SPIFFS and memory partitioning & sectors.

mediaDS, 2019-03-20:
Don't worry, I did read your Readme - and I did every step explained in section "Building Using Arduino IDE". It compiles fine. But I noticed I flashed it witout SPIFFS - my bad, mea culpa :)

Anyway, will this have anything to do with the SM10 not listed with autodetect? What else can I help you with?

proddy, 2019-03-20:
do a `log v` and then `autodetect` then I can see if your SM10 is sent a version request

mediaDS, 2019-03-20:
Ok, now everything is working as intended. As a side note - on first reboot after setting the wifi data it crashes with:

Exception (28):
epc1=0x402244a8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffc0 offset: 01a0
3fffff30:  00000000 3fff25ac 3fff25ac 40221878  
3fffff40:  00000000 00000000 3fff25ac 3fff2630  
3fffff50:  0000000d 3fff25ac 3fff2358 4021108d  
3fffff60:  0000000d 00000000 4bc6a7f0 00000000  
3fffff70:  3fff2358 3fff2044 3fff2348 3fff2a44  
3fffff80:  3fffdad0 3fff2044 3fff2358 40211164  
3fffff90:  3fffdad0 00000000 3fff2044 4020d4e8  
3fffffa0:  3fffdad0 00000000 3fff2a14 4021f3a0  
3fffffb0:  feefeffe feefeffe 3ffe8560 40100a51  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa

Here is the log (I did autodetect twice):

Started scan of EMS bus for known devices
Requesting type Version(0x02) from dest 0x02
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
Requesting type Version(0x02) from dest 0x10
Requesting type Version(0x02) from dest 0x11
Requesting type Version(0x02) from dest 0x17
Requesting type Version(0x02) from dest 0x18
Requesting type Version(0x02) from dest 0x21
(00:03:06.005) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(00:03:06.300) Boiler -> all, type 0x18 telegram: 08 00 18 00 2E 01 66 64 00 01 01 20 40 80 00 01 E8 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=65), #data=25
(00:03:06.553) Sending read of type 0x02 to 0x02: telegram: 0B 82 02 00 28 (CRC=E4), #data=1
(00:03:06.586) Corrupt telegram: telegram: 82 02 00 28 (CRC=E4)
(00:03:07.723) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(00:03:09.714) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 28 (CRC=B4), #data=1
(00:03:09.764) Corrupt telegram: telegram: 02 00 28 B4 08 0B 02 00 7B 04 05 00 00 00 00 00 00 03 (CRC=44), #data=14
(00:03:10.030) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(00:03:10.212) Sending read of type 0x02 to 0x09: telegram: 0B 89 02 00 28 (CRC=BC), #data=1
(00:03:10.285) echo: telegram: 0B 89 02 00 28 (CRC=BC), #data=1
(00:03:10.303) 0x09 -> me, type 0x02 telegram: 09 0B 02 00 7D 01 05 (CRC=C1), #data=3
Device found. Model BC25 Base Controller with TypeID 0x09, Product ID 125, Version 01.05
(00:03:10.560) Sending read of type 0x02 to 0x10: telegram: 0B 90 02 00 28 (CRC=74), #data=1
(00:03:10.622) Thermostat -> me, type 0x02 telegram: 10 0B 02 00 56 01 0F (CRC=71), #data=3
Thermostat found. Model RC35 with TypeID 0x10, Product ID 86, Version 01.15
* Setting Thermostat type to Model RC35, TypeID 0x10, Product ID 86, Version 01.15
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
(00:03:10.914) Sending read of type 0x02 to 0x11: telegram: 0B 91 02 00 28 (CRC=7C), #data=1
(00:03:12.104) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(00:03:13.179) Sending read of type 0x02 to 0x17: telegram: 0B 97 02 00 28 (CRC=4C), #data=1
(00:03:13.440) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(00:03:13.622) Sending read of type 0x02 to 0x18: telegram: 0B 98 02 00 28 (CRC=34), #data=1
(00:03:13.695) echo: telegram: 0B 98 02 00 28 (CRC=34), #data=1
(00:03:14.724) Boiler -> all, type 0x18 telegram: 08 00 18 00 2E 01 66 64 00 01 01 20 40 80 00 01 E7 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=DD), #data=25
(00:03:14.940) Sending read of type 0x02 to 0x21: telegram: 0B A1 02 00 28 (CRC=E5), #data=1
(00:03:15.250) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(00:03:15.521) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 E7 01 E8 21 00 00 03 00 00 0E 13 00 01 63 00 80 00 (CRC=C7), #data=19
(00:03:15.887) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 28 (CRC=84), #data=1
(00:03:15.919) Corrupt telegram: telegram: 90 3E 00 28 (CRC=84)
[... 106 lines cut ...]
(00:03:54.959) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(00:03:55.174) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 E7 01 E7 21 00 00 03 00 00 0E 13 00 01 63 00 80 00 (CRC=15), #data=19
(00:03:56.081) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:04:02.057) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 28 (CRC=84), #data=1
(00:04:02.170) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 26 00 CF 00 00 00 00 00 00 00 64 11 2E 00 (CRC=DC), #data=16
[... cut ...]
g (00:04:03.427) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 28 (CRC=DC), #data=1
(00:04:03.459) Corrupt telegram: telegram: 88 18 00 28 (CRC=DC)
(00:04:03.511) Boiler -> me, type 0x18 telegram: 08 0B 18 00 2E 01 64 64 00 01 01 20 40 80 00 01 E8 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=71), #data=25
n(00:04:03.755) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 28 (CRC=D8), #data=1
(00:04:03.808) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 7A 80 00 80 00 00 00 02 40 00 2D 05 00 F5 0C 00 00 00 00 E6 F9 00 2B A2 80 00 (CRC=73), #data=27

proddy, 2019-03-20:
I made a typo with the ID (again!). Now fixed in the latest dev branch.

mediaDS, 2019-03-20:
No big deal, I could have seen it myself, unfortunately I am busy with 3D design at the moment. I'll check the latest dev in a few and report back to you.

mediaDS, 2019-03-20:
Well done! It identifies the SM10 (but no real data). Her is the logs:

Started scan of EMS bus for known devices
Boiler found. Model Buderus GB172/Nefit Trendline with TypeID 0x08, Product ID 123, Version 04.05
* Setting Boiler type to Model Buderus GB172/Nefit Trendline, TypeID 0x08, Product ID 123, Version 04.05
Device found. Model BC25 Base Controller with TypeID 0x09, Product ID 125, Version 01.05
* Setting Thermostat type to Model RC35, TypeID 0x10, Product ID 86, Version 01.15
Device found. Model SM10 Solar Module with TypeID 0x30, Product ID 73, Version 02.00
SM10 Solar Module support enabled.

Here is the info:

EMS-ESP system stats:
  System logging set to None
  LED is on, Silent mode is off
  # connected Dallas temperature sensors=0
  Thermostat is enabled, Boiler is enabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus Connected=yes, Tx is active, # Rx telegrams=46, # Tx telegrams=0, # Crc Errors=15

Boiler stats:
  Boiler type: Buderus GB172/Nefit Trendline [Type ID: 0x08] Product ID:123 Version:04.05
  Hot tap water: off
  Central Heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Eco
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 48.0 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 355 times
  Warm Water active time: 2 days 12 hours 3 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 44 C
  Current flow temperature: 45.20 C
  Return temperature: ? C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 23 %
  Flame current: 8.30 uA
  System pressure: 0.0 bar
  System service code: -H (200)
  Heating temperature setting on the boiler: 80 C
  Boiler circuit pump modulation max. power: 100 %
  Boiler circuit pump modulation min. power: 10 %
  Outside temperature: 13.0 C
  Boiler temperature: ? C
  Pump modulation: 60 %
  Burner # starts: 11531 times
  Total burner operating time: 43 days 13 hours 45 minutes
  Total heat operating time: 41 days 1 hours 42 minutes
  Total UBA working time: 104 days 0 hours 40 minutes

  SM10 modulation pump: ? %
  SM10 collector temperature: ? C
  SM10 pump: ?
  SM10 uptime: 11650 days 20 hours 15 minutes

Thermostat stats:
  Thermostat type: RC35 [Type ID: 0x10] Product ID:86 Version:01.15
  Setpoint room temperature: 19.0 C
  Current room temperature: 20.60 C
  Thermostat time is 12:42:20 20/3/2019
  Mode is set to auto
[... 93 lines cut ...]
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
Requesting type SM10Monitor(0x97) from dest 0x30
(00:04:05.242) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 28 (CRC=84), #data=1
(00:04:05.357) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 26 00 CE 00 00 00 00 00 00 00 64 11 2B 00 (CRC=1E), #data=16
(00:04:05.496) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(00:04:05.759) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 28 (CRC=88), #data=1
(00:04:06.017) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 01 20 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=50), #data=27
(00:04:06.283) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 28 (CRC=64), #data=1
(00:04:06.315) Corrupt telegram: telegram: 90 06 00 28 (CRC=64)
(00:04:06.445) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 03 0C 14 2B 37 02 00 (CRC=EF), #data=8
(00:04:06.656) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 28 (CRC=DC), #data=1
(00:04:06.688) Corrupt telegram: telegram: 88 18 00 28 (CRC=DC)
(00:04:06.744) Boiler -> me, type 0x18 telegram: 08 0B 18 00 2C 01 F5 64 00 01 01 24 40 80 00 01 E0 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=9A), #data=25
(00:04:08.250) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 28 (CRC=D8), #data=1
(00:04:08.284) Corrupt telegram: telegram: 88 19 00 28 (CRC=D8)
(00:04:08.338) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 83 80 00 80 00 00 00 07 3C 00 2D 0B 00 F5 1A 00 00 00 00 E7 07 00 2B A8 80 00 (CRC=9A), #data=27
(00:04:08.400) Boiler -> all, type 0x18 telegram: 08 00 18 00 2C 01 F2 64 00 01 01 24 40 80 00 01 E0 80 00 00 00 FF 30 59 00 CC 00 00 00 (CRC=06), #data=25
(00:04:08.845) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 28 (CRC=70), #data=1
(00:04:08.896) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 32 FB 00 28 00 02 46 D8 FF FF 00 (CRC=C8), #data=13
Publishing boiler data via MQTT
(00:04:09.037) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 28 (CRC=E4), #data=1
...Retrying read. Attempt 1/2...
(00:04:09.045) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 32 FB 00 28 00 02 46 D8 FF FF 00 (CRC=C8), #data=13

mediaDS, 2019-03-20:
Ok, here is some more information:

It is sunny outside, no clouds, I expect the temperature to go up from here. Here is what my RC35 displays:

```
Collector Temp. 36°C
WW middle 47°C
WW bottom 28°C
Solar Pump 0% modulation
```

Here is a rather long log file, maybe you'll find the above data somewhere in it:

```
(00:36:41.073) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 2A 64 64 00 (CRC=99), #data=4
(00:36:44.965) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 03 0D 14 10 22 02 00 (CRC=FD), #data=8
(00:36:49.533) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(00:36:49.792) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 D2 01 D2 21 00 00 03 00 00 0E 13 00 01 63 00 80 00 (CRC=EF), #data=19
(00:36:50.012) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 46 64 00 01 01 20 40 80 00 01 D2 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=65), #data=25
(00:36:59.458) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(00:36:59.736) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 D2 01 D2 21 00 00 03 00 00 0E 13 00 01 63 00 80 00 (CRC=EF), #data=19
(00:36:59.968) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 45 64 00 01 01 20 40 80 00 01 D3 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=94), #data=25
Calling scheduled data refresh from EMS devices..
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
Requesting type SM10Monitor(0x97) from dest 0x30
(00:37:06.014) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 28 (CRC=84), #data=1
(00:37:06.106) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 26 00 CE 00 00 00 00 00 00 00 64 11 2A 00 (CRC=1C), #data=16
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
(00:37:06.249) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(00:37:06.251) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 28 (CRC=88), #data=1
...Retrying read. Attempt 1/2...
(00:37:06.260) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 02 26 00 CE 00 00 00 00 00 00 00 64 11 2A 00 (CRC=1C), #data=16
(00:37:07.478) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 28 (CRC=88), #data=1
(00:37:07.643) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 01 20 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=50), #data=27
(00:37:07.851) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 28 (CRC=64), #data=1
(00:37:07.883) Corrupt telegram: telegram: 90 06 00 28 (CRC=64)
(00:37:07.956) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 03 0D 14 10 39 02 00 (CRC=36), #data=8
(00:37:08.274) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 28 (CRC=DC), #data=1
(00:37:08.306) Corrupt telegram: telegram: 88 18 00 28 (CRC=DC)
(00:37:08.362) Boiler -> me, type 0x18 telegram: 08 0B 18 00 2A 01 45 64 00 01 01 20 40 80 00 01 D2 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=AE), #data=25
(00:37:09.849) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 D2 01 D2 21 00 00 03 00 00 0E 13 00 01 63 00 80 00 (CRC=EF), #data=19
(00:37:10.216) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 28 (CRC=D8), #data=1
(00:37:10.248) Corrupt telegram: telegram: 88 19 00 28 (CRC=D8)
(00:37:10.303) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 89 80 00 80 00 00 00 04 3C 00 2D 0E 00 F5 26 00 00 00 00 E7 13 00 2B AB 80 00 (CRC=C0), #data=27
(00:37:10.590) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 28 (CRC=70), #data=1
(00:37:10.621) Corrupt telegram: telegram: 88 33 00 28 (CRC=70)
(00:37:10.658) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 32 FB 00 28 00 02 46 D8 FF FF 00 (CRC=C8), #data=13
(00:37:10.801) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 28 (CRC=E4), #data=1
(00:37:10.809) Corrupt telegram: telegram: 88 33 00 28 (CRC=70)
(00:37:10.825) Corrupt telegram: telegram: 0B 88 00 28 (CRC=E4)
(00:37:12.845) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(00:37:15.367) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 28 (CRC=EC), #data=1
(00:37:15.404) Corrupt telegram: telegram: 14 00 28 EC 08 0B 14 00 02 49 4B (CRC=C3), #data=7
(00:37:15.671) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(00:37:15.889) Sending read of type 0x97 to 0x30: telegram: 0B B0 97 00 28 (CRC=0B), #data=1
(00:37:15.959) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 01 79 00 01 19 01 00 00 DF 00 00 (CRC=E4), #data=13
(00:37:16.059) Thermostat -> Boiler, type 0x14 telegram: 10 88 14 00 03 (CRC=6E), #data=1
(00:37:16.198) Boiler -> Thermostat, type 0x14 telegram: 08 10 14 00 02 49 4B (CRC=88), #data=3
(00:37:16.336) Thermostat -> all, type 0xA2 telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 (CRC=9B), #data=14
(00:37:17.284) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
[... cut ...]
(00:40:38.663) Boiler -> all, type 0x18 telegram: 08 00 18 00 2A 01 3C 64 00 01 01 20 40 80 00 01 D2 80 00 00 00 FF 30 41 00 CA 00 00 00 (CRC=AE), #data=25
(00:40:38.892) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(00:40:39.113) Boiler -> all, type 0x34 telegram: 08 00 34 00 32 01 D2 01 D2 21 00 00 03 00 00 0E 13 00 01 63 00 80 00 (CRC=EF), #data=19
log n(00:40:41.697) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 2A 64 64 00 (CRC=99), #data=4
```

proddy, 2019-03-20:
ok, back to the drawing board then!

mediaDS, 2019-03-20:
Yep, take your time as I will continue my 3D work now. As I always say: have fun :)

proddy, 2019-03-20:
I can get the collector temp and bottom temp but not sure where % modulation and the middle temps are. I need more data. Could you do a few spot checks between settings shown on the RC35 and the contents of a `send 0B B0 97 00 28`?

 I just need the line that looks like:

(00:39:10.886) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 01 7F 50 01 19 03 00 00 E1 00 00 (CRC=6F), #data=13

mediaDS, 2019-03-20:
Okay, I did some spot checks and still looking through it. The RC35 displayed steady values during logging. They are:

```
Collector 17°C
WW middle 44°C
WW bottom 30°C
Solar pump 0%

```
None of the values changed while logging.

Here is the filtered output containing only data packets from the SM10 to all, the boiler or the EMS-ESP. The value that's bothering me is "AD" which then counts down and jumps between "A6" and "A5". I have no idea what that might be.

```
(03:35:12.463) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AD 00 01 29 01 00 00 E6 00 00 (CRC=7F), #data=13

(03:35:29.228) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 AD 00 01 29 01 00 00 E6 00 00 (CRC=72), #data=13
(03:35:29.445) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:36:13.866) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AD 00 01 29 01 00 00 E6 00 00 (CRC=7F), #data=13

(03:36:29.519) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 AC 00 01 29 01 00 00 E6 00 00 (CRC=40), #data=13
(03:36:29.735) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:36:41.922) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AC 00 01 29 01 00 00 E6 00 00 (CRC=4D), #data=13
(03:37:11.872) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AB 00 01 29 01 00 00 E6 00 00 (CRC=D3), #data=13
(03:37:12.233) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AB 00 01 29 01 00 00 E6 00 00 (CRC=D3), #data=13

(03:37:29.628) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 AB 00 01 29 01 00 00 E6 00 00 (CRC=DE), #data=13
(03:37:29.846) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:38:14.295) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AB 00 01 29 01 00 00 E6 00 00 (CRC=D3), #data=13

(03:38:29.821) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 AB 00 01 29 01 00 00 E6 00 00 (CRC=DE), #data=13
(03:38:30.086) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:38:39.174) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AB 00 01 29 01 00 00 E6 00 00 (CRC=D3), #data=13
(03:39:06.841) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AA 00 01 29 01 00 00 E6 00 00 (CRC=E1), #data=13
(03:39:07.621) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AA 00 01 29 01 00 00 E6 00 00 (CRC=E1), #data=13
(03:39:15.775) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AA 00 01 29 01 00 00 E6 00 00 (CRC=E1), #data=13
(03:39:28.981) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 AA 00 01 29 01 00 00 E6 00 00 (CRC=E1), #data=13

(03:39:30.171) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 AA 00 01 29 01 00 00 E6 00 00 (CRC=EC), #data=13
(03:39:30.382) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:39:43.942) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A9 00 01 29 01 00 00 E6 00 00 (CRC=B7), #data=13
(03:39:52.247) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A9 00 01 29 01 00 00 E6 00 00 (CRC=B7), #data=13
(03:40:11.586) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A8 00 01 29 01 00 00 E6 00 00 (CRC=85), #data=13
(03:40:12.014) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A8 00 01 29 01 00 00 E6 00 00 (CRC=85), #data=13
(03:40:17.305) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A8 00 01 29 01 00 00 E6 00 00 (CRC=85), #data=13

(03:40:30.522) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 A8 00 01 29 01 00 00 E6 00 00 (CRC=88), #data=13
(03:40:30.743) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:40:37.579) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A7 00 01 29 01 00 00 E6 00 00 (CRC=92), #data=13
(03:40:52.241) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A7 00 01 29 01 00 00 E6 00 00 (CRC=92), #data=13
(03:41:11.938) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=A0), #data=13
(03:41:17.862) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=A0), #data=13

(03:41:30.692) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=AD), #data=13
(03:41:30.917) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:43:26.120) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=A0), #data=13

(03:43:31.106) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 A5 00 01 29 01 00 00 E6 00 00 (CRC=FB), #data=13
(03:43:31.328) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:43:51.763) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=A0), #data=13
(03:44:11.837) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A5 00 01 29 01 00 00 E6 00 00 (CRC=F6), #data=13

(03:44:12.287) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=A0), #data=13
(03:44:17.693) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=A0), #data=13

(03:44:31.355) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=AD), #data=13
(03:44:31.564) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:44:37.778) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A5 00 01 29 01 00 00 E6 00 00 (CRC=F6), #data=13
[... cut ...]
(03:45:13.755) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=A0), #data=13
(03:45:18.992) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=A0), #data=13

(03:45:31.575) 0x30 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 A6 00 01 29 01 00 00 E6 00 00 (CRC=AD), #data=13
(03:45:31.801) 0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3

(03:45:41.142) 0x30 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 A5 00 01 29 01 00 00 E6 00 00 (CRC=F6), #data=13

```

proddy, 2019-03-20:
thanks - we're getting closer. It would still help to get the % modulation as its always zero and there are so many 0x00s in the telegram :)

mediaDS, 2019-03-20:
> thanks - we're getting closer. It would still help to get the % modulation as its always zero and there are so many 0x00s in the telegram :)

You're welocme and yes, I agree - too many zero's. However, I will not climp up my roof to imitate the sun with my flashlight :)

At the moment it is very cloudy and dark, looks like it will rain soon. Maybe we will see some better data tomorow - weather forecast says it should be a warm and sunny day. Let's hope for the best.

Thanks so much for all the help.

proddy, 2019-03-20:
LOL Ok! I'll commit some changes for the collective and bottom temps and we'll guess the % modulation. We can always fix that later when the sun comes out. That 44C middle temp - I have no idea where that comes from! I guess its not that important. You always do a 'log t' and just see how your RC35 is sending requests to the SM10.

mediaDS, 2019-03-20:
Hmm, ok, Byte 7 and 8 are the collector temp (*10) which makes sense - 17.3°C * 10 = 00AD, Byte 9 is the pump modulation in % (00 in our case) followed by 2 bytes with the value of the temp sensor of the warm water (0129 in our case, if divided by 10 you'll get 29,7 - which is the 30°C dsiplayed by the RC25). The next byte contains in bit 1 if the pump is on or off - in our case it is on which is weird - the pump should be off, maybe it is inverse?. The next 3 bytes are the uptime of the SM10 - in our case 00 00 E6(?). The last byte in the payload is of unknown functionality.

mediaDS, 2019-03-20:
> That 44C middle temp - I have no idea where that comes from! I guess its not that important. You always do a 'log t' and just see how your RC35 is sending requests to the SM10.

I maybe able to shine some light on this (no pun intended) - there is only one NTC wired to the SM10 which is actiually in the hot water tank. 

I assume that's the bottom temperatur reported by the SM10. The sensor of the middle temperatur is not wired to the SM10 but to the boiler directly.. So I assume the RC35 is collecting the date from the SM10 and the boiler and displays it's values.

So, forget about the middle temperature - that's not from the SM10.

mediaDS, 2019-03-21:
Little update on the weather: no sun, no fun. 

The solar collector does not generate enough heat for the pump to be activated. Beside that we (my girlfriend and I) did a real good job with the insulation, so the hot water tank won't loose much temperature. That's good for me and my wallet but bad for testing software. 

I assume we will not have a real life testing of the SM10 today.

proddy, 2019-03-21:
No worries. I’m pretty sure the last change is accurate and if it isn’t
it’s only one line to change. It’s sunny here in NL btw

mediaDS, 2019-03-21:
Then have fun in the sun :)

Anyway, I made a mistake above - the pump seems to be Bit 1, not Bit 0, since Bit 0 is always "1" it might be another status (maybe "pump is present"). Please let me know whenever you have something to test.

proddy, 2019-03-21:
I made the changes yesterday so you can test that. I ignored the pump
on/off and just show the % modulation

mediaDS, 2019-03-21:
Well, I'm blind as a bat sometimes. Will check and report back later today.

mediaDS, 2019-03-21:
Did you change anything related to WiFi? It cannot connect to my WiFi anymore. I reset all settings and re-entered it again - nothing, just "Cant connect to SSID mediaDS".

proddy, 2019-03-21:
no changes to that logic. Platform did release a new version of the platformio libraries and the new arduino core libraries so if you're using that make sure you clean your environment and then upgrade the core and the libs. Then try again. If it still fails then perhaps your SPIFFS has been erased. You'll need to hook it up to a serial monitor and see if its booting in AP mode

mediaDS, 2019-03-21:
I found the problem - it was my WiFi router (Fritzbox). The WiFi was unresponsible. I resetted it and everything is working now. Phew.  Sorry to wake you up with my crap :)

I'll begin testing now.

mediaDS, 2019-03-21:
I did a quick check - result is similar to yesterday. It detects the SM10 but no data is presented:

```
autodetect
Started scan of EMS bus for known devices
Device found. Model BC25 Base Controller with TypeID 0x09, Product ID 125, Version 01.05
* Setting Thermostat type to Model RC35, TypeID 0x10, Product ID 86, Version 01.15
Device found. Model SM10 Solar Module with TypeID 0x30, Product ID 73, Version 02.00
SM10 Solar Module support enabled.

```
```
info
EMS-ESP system stats:
  System logging set to None
  LED is on, Silent mode is off
  # connected Dallas temperature sensors=0
  Thermostat is enabled, Boiler is enabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus Connected=yes, Tx is active, # Rx telegrams=20, # Tx telegrams=0, # Crc Errors=3

Boiler stats:
  Boiler type: Buderus GB172/Nefit Trendline [Type ID: 0x08] Product ID:123 Version:04.05
  Hot tap water: off
  Central Heating: active
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Eco
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 43.70 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 358 times
  Warm Water active time: 2 days 12 hours 38 minutes
  Warm Water 3-way valve: on
  Selected flow temperature: 85 C
  Current flow temperature: 76.50 C
  Return temperature: ? C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 108 %
  Burner current power: 108 %
  Flame current: 28.79 uA
  System pressure: 0.0 bar
  System service code: =H (201)
  Heating temperature setting on the boiler: 80 C
  Boiler circuit pump modulation max. power: 100 %
  Boiler circuit pump modulation min. power: 10 %
  Outside temperature: 6.19 C
  Boiler temperature: ? C
  Pump modulation: 100 %
  Burner # starts: 11681 times
  Total burner operating time: 43 days 20 hours 30 minutes
  Total heat operating time: 41 days 7 hours 51 minutes
  Total UBA working time: 156 days 2 hours 8 minutes

  SM10 collector temperature: ? C
  SM10 bottom temperature: ? C
  SM10 pump: ? %

Thermostat stats:
  Thermostat type: RC35 [Type ID: 0x10] Product ID:86 Version:01.15
  Setpoint room temperature: 19.0 C
  Current room temperature: 20.60 C
  Thermostat time is 21:08:23 21/3/2019
  Mode is set to auto

```
logs are similar to the ones we know already.

proddy, 2019-03-21:
ok, think I fixed that. I missed a call to fetch the values after detection.

mediaDS, 2019-03-21:
Alright, I'll check it right now and will let you know.

mediaDS, 2019-03-21:
Nope, the same output as before. Detected but only "?" as data.

proddy, 2019-03-21:
oh. can you `log v` and do another `autodetect`. look for a send from the SM10, something like `0B B0 97 00 16` and that a response is received. 

Also - do you want these values in a single JSON object sent to an MQTT topic? Handy if using Home Assistant or Gafana

mediaDS, 2019-03-21:
Done, see the log below. I marked the two lines for your convenience. I am actually not a big fan of gigantic JSON blobs, because I have to write so much Javascript to extract all the data for display (and other stuff) but for now it would be nice to have all SM10 data in a single JSON object.

I have to be careful because of data lenght restrictions. My MQTT broker is already choking on the boiler JSON :)

Btw, I am using ioBroker in a Multihost system with vis as visualisation (Android tablets inside the walls) and Influxdb and Grafana as a data logging/visualisation graph system.

Here is the log:

```
Started scan of EMS bus for known devices
Requesting type Version(0x02) from dest 0x02
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
Requesting type Version(0x02) from dest 0x10
Requesting type Version(0x02) from dest 0x11
Requesting type Version(0x02) from dest 0x17
Requesting type Version(0x02) from dest 0x18
Requesting type Version(0x02) from dest 0x21
Requesting type Version(0x02) from dest 0x30
(00:35:37.177) Sending read of type 0x02 to 0x02: telegram: 0B 82 02 00 20 (CRC=EC), #data=1
(00:35:37.493) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(00:35:37.799) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 20 (CRC=BC), #data=1
(00:35:37.849) Corrupt telegram: telegram: 02 00 20 BC 08 0B 02 00 7B 04 05 00 00 00 00 00 00 03 (CRC=44), #data=14
(00:35:38.171) Thermostat -> Boiler, type 0x14 telegram: 10 88 14 00 03 (CRC=6E), #data=1
(00:35:38.296) Boiler -> Thermostat, type 0x14 telegram: 08 10 14 00 02 51 31 (CRC=C2), #data=3
(00:35:38.426) Thermostat -> all, type 0xA2 telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 (CRC=9B), #data=14
(00:35:38.769) Sending read of type 0x02 to 0x09: telegram: 0B 89 02 00 20 (CRC=B4), #data=1
(00:35:38.791) 0x09 -> me, type 0x02 telegram: 09 0B 02 00 7D 01 05 (CRC=C1), #data=3
Device found. Model BC25 Base Controller with TypeID 0x09, Product ID 125, Version 01.05
Requesting type SM10Monitor(0x97) from dest 0x30
(00:35:40.185) SM10 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 24 00 00 F3 01 00 00 FC 00 00 (CRC=2D), #data=13
(00:35:40.398) SM10 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3
(00:35:40.542) Sending read of type 0x02 to 0x10: telegram: 0B 90 02 00 20 (CRC=7C), #data=1
(00:35:40.680) Thermostat -> me, type 0x02 telegram: 10 0B 02 00 56 01 0F (CRC=71), #data=3
Thermostat found. Model RC35 with TypeID 0x10, Product ID 86, Version 01.15
* Setting Thermostat type to Model RC35, TypeID 0x10, Product ID 86, Version 01.15
Requesting type RC35StatusMessage(0x3E) from dest 0x10
Requesting type RC35Set(0x3D) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
(00:35:41.060) Sending read of type 0x02 to 0x11: telegram: 0B 91 02 00 20 (CRC=74), #data=1
(00:35:41.092) Corrupt telegram: telegram: 91 02 00 20 (CRC=74)
(00:35:42.111) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 44 00 00 00 CC 00 00 00 00 00 00 00 00 00 05 00 (CRC=60), #data=16
(00:35:42.478) Sending read of type 0x02 to 0x17: telegram: 0B 97 02 00 20 (CRC=44), #data=1
(00:35:42.838) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 08 00 00 00 CC 00 D2 00 D2 7D 00 7D 00 (CRC=A6), #data=13
(00:35:43.324) Sending read of type 0x02 to 0x18: telegram: 0B 98 02 00 20 (CRC=3C), #data=1
(00:35:44.442) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(00:35:44.717) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 00 E4 00 00 00 00 20 40 80 00 02 22 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=75), #data=25
(00:35:45.000) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(00:35:45.215) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 22 02 22 20 00 00 03 00 00 0E 3C 00 01 66 00 80 00 (CRC=3A), #data=19
(00:35:45.683) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 64 00 (CRC=59), #data=4
(00:35:47.008) Sending read of type 0x02 to 0x21: telegram: 0B A1 02 00 20 (CRC=ED), #data=1
(00:35:47.357) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76), #data=13
(00:35:47.636) Sending read of type 0x02 to 0x30: telegram: 0B B0 02 00 20 (CRC=65), #data=1
(00:35:47.698) SM10 -> me, type 0x02 telegram: 30 0B 02 00 49 02 00 (CRC=CC), #data=3
Device found. Model SM10 Solar Module with TypeID 0x30, Product ID 73, Version 02.00
SM10 Solar Module support enabled.
Requesting type SM10Monitor(0x97) from dest 0x30

--->  (00:35:48.009) Sending read of type 0x97 to 0x30: telegram: 0B B0 97 00 20 (CRC=03), #data=1

(00:35:48.106) SM10 -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 24 00 00 F3 01 00 00 FC 00 00 (CRC=20), #data=13
(00:35:48.352) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 20 (CRC=8C), #data=1
(00:35:48.471) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 44 00 00 00 CC 00 00 00 00 00 00 00 00 00 05 00 (CRC=08), #data=16
(00:35:48.600) Sending read of type 0x3D to 0x10: telegram: 0B 90 3D 00 20 (CRC=80), #data=1
(00:35:50.593) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(00:35:51.588) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 20 (CRC=6C), #data=1
(00:35:51.688) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 03 16 15 32 20 03 00 (CRC=02), #data=8
[... cut ...]
(00:35:53.082) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A), #data=13
(00:35:54.688) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 00 E4 00 00 00 00 20 40 80 00 02 22 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=75), #data=25
(00:35:54.965) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=93), #data=21
(00:35:55.196) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 22 02 22 20 00 00 03 00 00 0E 3C 00 01 66 00 80 00 (CRC=3A), #data=19
Calling scheduled data refresh from EMS devices..

```

mediaDS, 2019-03-21:
Take your time, please. I am tired and go to sleep now. Have a good night and I'm looking forward to more testing of your software tomorrow.

proddy, 2019-03-21:
I found the error and fixed it, I think. I'm getting sloppy. Good point on the MQTT - it's on my list to make the JSON configurable. So what format would you like the SM10 messages in? Same as the thermostat with one topic and 3 key/values?

mediaDS, 2019-03-22:
Good morning, I'd like that, just one topic and three objects would be fine. I'll do testing once I finish my first coffee (I'm useless before that).

mediaDS, 2019-03-22:
Well done, now I get valid data from the SM10. So, goal achieved from your side. What I don't understand is the differenz between the "bottom" temp and the current WW temp (middle). The NTC for WW sits just above the SM10 NTC but the difference is more than 20°C (bottom 30.5°C vs. middle 56.7°C.). This is really weird.

The current output of "info":

```
EMS-ESP system stats:
  System logging set to None
  LED is on, Silent mode is off
  # connected Dallas temperature sensors=0
  Thermostat is enabled, Boiler is enabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus Connected=yes, Tx is active, # Rx telegrams=101, # Tx telegrams=0, # Crc Errors=22

Boiler stats:
  Boiler type: Buderus GB172/Nefit Trendline [Type ID: 0x08] Product ID:123 Version:04.05
  Hot tap water: off
  Central Heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Eco
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 55.79 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 359 times
  Warm Water active time: 2 days 12 hours 57 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 37 C
  Current flow temperature: 34.20 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 0.0 bar
  System service code: 0Y (204)
  Heating temperature setting on the boiler: 80 C
  Boiler circuit pump modulation max. power: 100 %
  Boiler circuit pump modulation min. power: 10 %
  Outside temperature: 15.10 C
  Boiler temperature: ? C
  Pump modulation: 60 %
  Burner # starts: 11738 times
  Total burner operating time: 44 days 2 hours 7 minutes
  Total heat operating time: 41 days 13 hours 10 minutes
  Total UBA working time: 106 days 2 hours 18 minutes

Solar Module stats:
    Collector temperature: 39.90 C
    Bottom temperature: 30.29 C
    Pump modulation: 0 %

Thermostat stats:
  Thermostat type: RC35 [Type ID: 0x10] Product ID:86 Version:01.15
  Setpoint room temperature: 19.0 C
  Current room temperature: 20.79 C
  Thermostat time is 14:07:36 22/3/2019
  Mode is set to auto

```

mediaDS, 2019-03-22:
Update: :)

The pump modulation also seems to be valid:

```
Solar Module stats:
    Collector temperature: 40.20 C
    Bottom temperature: 30.29 C
    Pump modulation: 82 %

Thermostat stats:
  Thermostat type: RC35 [Type ID: 0x10] Product ID:86 Version:01.15
  Setpoint room temperature: 19.0 C
  Current room temperature: 20.70 C
  Thermostat time is 14:11:14 22/3/2019
  Mode is set to auto

```

mediaDS, 2019-03-22:
Addition info:

Would be nice to add the pump active/not active to be able to identify a pump failure. And yes, it is bit 1 of byte 11. If the pump modulates, the byte read is 3. If modulation is 0% **and** the pump is activated means the pump is probably "kaputt" and needs to be serviced.

proddy, 2019-03-22:
pump active added. Also MQTT

mediaDS, 2019-03-22:
All working except pump active - it says pump "on" but the data says it is off. My fault - it is not byte 11 but byte 10, bit 1 (bit 0 is always 1). Sorry for the typo.

data telegram:

SM10 -> all, type 0x97 telegram: 30 00 97 00 00 00 00 5B 00 01 32 01 00 01 21 00 00

mediaDS, 2019-03-22:
MQTT works fine, very well done! If the pump activation is fixed it is perfect. Thanks a lot for all your work, now it is my turn to make something useful out of all the data. I'll stay in touch with you - if you need anything, just let me know.

Have a nice weekend

Dieter

mediaDS, 2019-03-23:
you nailed it, proddy. Have it running for a few hours. Works fine. Thanks so much!

proddy, 2019-03-23:
great, it was fun working together. Thanks for helping out. If you notice anything strange please just re-open this issue and report back.

closed 2019-03-23 by proddy

mediaDS, 2019-03-24:
Sorry, there seems to be something wrong regarding the "Pump activation flag". The Wiki has a mistake:

Byte 7-8 are the temperature
Byte 9 is the pump modulation
Byte 10 and 11 is the bottom temperature of the hot tank
**Byte 12 is the pump active flag, Bit 1 is "1" if the pump is on and "0" if the pump is off**
Byte 13 -15 is the uptime of the SM10

Example of a data packet:

SM10 -> all, type 0x97 telegram: 30 00 97 00 00 00 01 95 1E 01 29 02 00 01 38 00 00 (CRC=AE), #data=13

proddy, 2019-03-24:
no problem. Changed line #1132 in `ems.cpp` to 

`EMS_Other.SM10pump           = _bitRead(7, 1); // active if bit 1 is set (to 1)`

The 7 is the offset in the data package after the header, so position 12 (+5) when comparing to the German ems wiki.

mediaDS, 2019-03-24:
Yep, did  that already - I just sat here studying your code :) 

All good now, enjoy the rest of your weekend.

mediaDS, 2019-03-24:
FYI, the ems-esp.ino is missing from the dev branch :)

proddy, 2019-03-24:
It’s renamed to .cpp which makes the build slightly faster and debugging work. It should still build fine with platformio

mediaDS, 2019-03-24:
Ah, ok. That makes sense.

[1 more reply]