emsesp/EMS-ESP #79: Rounding / Number conversion

NFis, 2019-03-22:
I watched a roundig problems with temperatures:  at certain temperatures there are steps up to the next integer. This behavior started when precision was changed to "2"  with release 1.4.0

Histograms from Homeassistant:

[image]

[image]

As a quick fix I changed "precision" to "1" in function "_float_to_char" in ems-esp.ino. The smoother result is seen on the right side of the graphics. 
Perhaps it would be better solution to avoid floating point calculations and use fixed point arithmetics. Values calculated from two bytes devided by 10 cannot have a higher precision then 0.1

My configuration: Buderus GB162 V3, UBA 3.5, BC10, RC35, RC20 RF

proddy, 2019-03-22:
that's a really good idea. I'll think about removing floats completely. They're costly

proddy, 2019-03-23:
I changed this in 1.6.0 (in dev branch) and removed all float pre-processing

SpaceTeddy, 2019-03-23:
@proddy ,
may you double check the rounding / floating algorithm please, because since dev 1.6.0.b5 I get via telnet "info" and mqtt a wrong "Setpoint room temperature" value. Target is 19.5 but the ESP says 19.1

thx

proddy, 2019-03-23:
@SpaceTeddy yup, its a bug. well spotted. The way I calculate a / 2 is just balls. I need to fix that later tonight. thanks for reporting.

proddy, 2019-03-24:
think its fixed now

SpaceTeddy, 2019-03-24:
fixed!

closed 2019-03-24 by proddy